fwupd/firmware-lenovo #2: ThinkPad L380 Yoga recognized as X280

andreasnuesslein, 2019-09-14:
Hi all,

I own a ThinkPad L380 Yoga and `fwupdmgr` is offering me 3 `ThinkPad X280 Consumer ME Update` updates, which is wrong. (I'm also on the newest correct L380 BIOS.)

I'd like to help so let me know how/if I can give you some IDs you might require.

Cheers

MichalM1, 2020-01-29:
In my case is **Lenovo ThinkPad L380** recognized as X280 too.

kmauleon, 2020-06-23:
> In my case is **Lenovo ThinkPad L380** recognized as X280 too.

Hi MichalM1
I am not handling L380 but I am the owner of X280 ME FW...
may I know the result of the below command...thank you very much.

fwupdmgr get-devices

diego-treitos, 2022-06-17:
Same here my L380 is recognized as X280. Also the Intel ME fw update fails (even with secure boot disabled) so I guess this is the reason?

mrhpearson, 2022-06-17:
Hi @diego-treitos - please see Katherene's note above. Can you provide the output from `fwupdmgr get-devices` please

diego-treitos, 2022-06-18:
Hi @mrhpearson thank you for your response. I am sorry but I do not see any Katherene in this thread.

My `fwupdmgr get-devices` is

```
20M5CTO1WW
│
├─Embedded Controller:
│     Device ID:          ca368aebcf7da847029e9f2520ec55fb7a036b31
│     Current version:    0.0.1
│     Minimum Version:    0.0.1
│     Vendor:             DMI:LENOVO
│     GUIDs:              3f4a527b-6588-45b8-b2d3-dc61189b63cb
│                         4a3e39e1-1d69-5ae4-bb2c-b60efa74daf7 ← UEFI\RES_{3F4A527B-6588-45B8-B2D3-DC61189B63CB}
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│
├─Intel Management Engine:
│     Device ID:          093ef0be8328a2c4ed2fe55cd36aae3171b92ade
│     Current version:    184.60.3561
│     Minimum Version:    0.0.1
│     Vendor:             DMI:LENOVO
│     GUIDs:              6d28cd9f-7bcd-4fb9-9f10-0372e2962fc4
│                         0318543c-9b6e-5191-8368-2bcfad7ffdff ← UEFI\RES_{6D28CD9F-7BCD-4FB9-9F10-0372E2962FC4}
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│
├─SanDisk SD9SN8W128G1001:
│     Device ID:          e421b2fc248391f6fe3e55ddbb3c9043be068bd0
│     Summary:            ATA Drive
│     Current version:    X6106001
│     Vendor:             SanDisk (ATA:0x15B7|OUI:001b44)
│     Serial Number:      190145802569
│     GUIDs:              a872c7b5-fe7b-55d6-9f69-86e5b04623cf ← IDE\SanDisk_SD9SN8W128G1001_________________X6106001
│                         e098e88b-d820-5299-8d4a-b3512b2cb287 ← IDE\0SanDisk_SD9SN8W128G1001_________________
│                         bcfacd2e-c424-5166-933c-9e85ba79d74a ← SanDisk SD9SN8W128G1001
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│
└─System Firmware:
  │   Device ID:          c249035ac01959a5d3a8f9b38ee7d66c166b7ddd
  │   Current version:    65562
  │   Minimum Version:    1
  │   Vendor:             LENOVO (DMI:LENOVO)
  │   GUIDs:              74edf5c5-58bb-46aa-aa7d-d655dd77284c
  │                       230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
  │                       d0dd759e-406e-5eee-9c85-e6f51f783f78 ← UEFI\RES_{74EDF5C5-58BB-46AA-AA7D-D655DD77284C}
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Needs a reboot after installation
  │                       • Cryptographic hash verification is available
  │                       • Device is usable for the duration of the update
  │
  └─UEFI dbx:
        Device ID:        362301da643102b9f38477387e2193e57abaa590
        Summary:          UEFI Revocation Database
        Current version:  77
        Minimum Version:  77
        Vendor:           UEFI:Linux Foundation
[... 124 lines cut ...]
        Vendor:           Lenovo
        Flags:            is-upgrade
        Description:
        • 8 Quarterly Security Release

        Version 11.8.65.3590 (LVFS: 184.65.3590)

        Problem Fixes

        • Intel CSME PSIRT-TA-201901-002 Q1 2019 QSR:
```

And this is from `dmidecode`

```
System Information
        Manufacturer: LENOVO
        Product Name: 20M5CTO1WW
        Version: ThinkPad L380
        Serial Number: **DELETED_FOR_GITHUB**
        UUID: **DELETED_FOR_GITHUB**
        Wake-up Type: Power Switch
        SKU Number: LENOVO_MT_20M5_BU_Think_FM_ThinkPad L380
        Family: ThinkPad L380
```

kmauleon, 2022-06-19:
hi @diego-treitos @mrhpearson it seems L380 yoga is using X280's GUID.  not sure who's handling L380 yoga though... I am still checking with ODMs.... will feedback once I have an update.  Thank you very much

kmauleon, 2022-06-21:
hi @diego-treitos will forward this information to ODM... thank you very much

kmauleon, 2022-07-25:
ODM is testing fixed BIOS to be release hopefully August mid... thank you

diego-treitos, 2022-07-27:
> ODM is testing fixed BIOS to be release hopefully August mid... thank you

That is fantastic! Thank you very much @kmauleon !

diego-treitos, 2022-09-02:
Thank you very much @kmauleon , although I am not sure how to get that new version. I tried with `fwupdmgr refresh` but after that `fwupdmgr get-updates` still returns the `Thinkpad X380` version. I also tried to purge the packages and remove `/var/lib/fwupd`, but still nothing. Also tried the update itself and still fails.

Maybe I need to try in a few days?

hughsie, 2022-09-02:
Try `fwupdmgr refresh --force`

diego-treitos, 2022-09-02:
> Try `fwupdmgr refresh --force`

Thanks for the suggestion but it didn't work either. As I said already removed anything related to fwupd and reinstalled so I guess that the X280 information is the one that is getting.

My guess is that either this is not fixed or that there is some intermediate cache that I need to wait to expire so I get the new information.

kmauleon, 2022-09-04:
hi @diego-treitos can you please try to download the BIOS and install it manually as below... thank you very much
BIOS link >> https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadR0RET.firmware
**fwupdmgr install xxx.cab**

diego-treitos, 2022-09-05:
hi @kmauleon thanks for the follow up. I downloaded the firmwar and run the command but nothing happened after reboot.

Then I run `fwupdmgr get-updates` to get the report. Here it is:

```json
{
   "ReportVersion" : 2,
   "MachineId" : "7481283370ab95fc303805a3b062ae5afe0c6a6037914daa9547fc9d1a72d89d",
   "Metadata" : {
     "DistroId" : "debian",
     "DistroVersion" : "11"
   },
   "Reports" : [
     {
       "Checksum" : "8ba19dcf27a6588a302cc374374d6b7117d249d7",
       "UpdateState" : 3,
       "UpdateError" : "failed to run update on reboot",
       "Guid" : [
         "74edf5c5-58bb-46aa-aa7d-d655dd77284c"
       ],
       "Plugin" : "uefi_capsule",
       "VersionOld" : "0.1.26",
       "VersionNew" : "0.1.29",
       "Flags" : 574619947,
       "Created" : 1662396297,
       "Modified" : 1662396298,
       "Metadata" : {
         "RuntimeVersion(org.freedesktop.gusb)" : "0.3.5",
         "HostFamily" : "ThinkPad L380",
         "BootMgrDesc" : "legacy",
         "CompileVersion(com.redhat.fwupdate)" : "12",
         "RuntimeVersion(com.redhat.fwupdate)" : "12",
         "SecureBoot" : "Enabled",
         "LinuxLockdown" : "integrity",
         "EspPath" : "/boot/efi",
         "UEFIUXCapsule" : "Enabled",
         "LastAttemptVersion" : "0x0",
         "CompileVersion(org.freedesktop.fwupd)" : "1.5.7",
         "RuntimeVersion(org.kernel)" : "5.10.0-17-amd64",
         "MissingCapsuleHeader" : "False",
         "RuntimeVersion(com.dell.libsmbios)" : "2.4",
         "CompileVersion(org.freedesktop.gusb)" : "0.3.5",
         "DistroId" : "debian",
         "CpuArchitecture" : "x86_64",
         "BootTime" : "1662395453",
         "HostVendor" : "LENOVO",
         "EfivarNvramUsed" : "40726",
         "RuntimeVersion(org.freedesktop.appstream-glib)" : "0.7.14",
         "KernelVersion" : "5.10.0-17-amd64",
         "HostProduct" : "20M5CTO1WW",
         "DistroVersion" : "11",
         "KernelCmdline" : "vt.global_cursor_default=0 workqueue.power_efficient=1 i915.modeset=1 intel_pstate=disable",
         "HostSku" : "LENOVO_MT_20M5_BU_Think_FM_ThinkPad L380",
         "RuntimeVersion(org.freedesktop.fwupd)" : "1.5.7",
         "LastAttemptStatus" : "0x0"
       }
     }
   ]
 }
```

kmauleon, 2022-09-05:
thanks for the feedback @diego-treitos ...will forward your results to BIOS dev for checking...

kmauleon, 2022-09-07:
hi @diego-treitos BIOS dev did not encounter the issue... still asking them for any suggestion.  while waiting... could you please try below commands and after system restart, rerun the system update...  thank you very much...
sudo apt-get update && sudo apt install && sudo apt dist-upgrade && sudo apt full-upgrade && sudo apt autoremove && sudo reboot

diego-treitos, 2022-09-14:
@kmauleon I was finally able to update. I found that I had to disable *secure boot* for the update to work. After disabling it, the firmware update was triggered and it updated. Oddly enough, after the reboot, `fwupdatemgr get-updates` reported that there was an error with the update BUT, `fwupdmgr get-history` reported the update and the "BIOS" interface shows version `1.29`. Also next executions of `fwupdmgr get-updates` did not show any available updates anymore.

My wander is if from now on the updates will appear to be fore L380 or for X280 again...

In any case THANK YOU VERY MUCH!!

kmauleon, 2022-09-16:
hi @diego-treitos thank you very much for the update.  it should always be for L380.  if in any case you received any update offer again for X280, kindly report it here.  Again thank you very much.  will close this issue for now.  please reopen if i missed anything....

closed 2022-09-16 by kmauleon