fwupd/firmware-lenovo #4: Missing Latest firmware for Lenovo Thinkpad P50

ajes, 2019-09-14:
For missing firmware please provide the following items:

* Link to firmware update at OEM website

Bios 1.56: https://pcsupport.lenovo.com/pl/pl/products/LAPTOPS-AND-NETBOOKS/THINKPAD-P-SERIES-LAPTOPS/THINKPAD-P50/downloads/DS106108

Intel Management Engine 11.8: https://pcsupport.lenovo.com/pl/pl/products/LAPTOPS-AND-NETBOOKS/THINKPAD-P-SERIES-LAPTOPS/THINKPAD-P50/downloads/DS105420

* Firmware release date (if available on OEM website)

Bios: March 18th, 2019
Intel ME: Jan 17th, 2019

* Link to the device page on LVFS.

Bios: https://fwupd.org/lvfs/device/ddc0ee61-e7f0-4e7d-acc5-c070a398838e
Intel ME: https://fwupd.org/lvfs/device/671d19d0-d43c-4852-98d9-1ce16f9967e4

kmauleon, 2020-10-22:
target to release new versions by november...please wait...thank you very much

kmauleon, 2020-11-16:
hi @ajes please try latest MEFW >> https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN1DRMP50.firmware
thank you very much...

kmauleon, 2020-12-23:
hi @ajes latest 11.8.80.3746 is available already.  please check with this.  thank you very much.
https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN1DRMP50.firmware

kmauleon, 2020-12-23:
latest MEFW 11.8.80.3746 has been released already.  closing this but please reopen if any concern.  thank you very much...

closed 2020-12-23 by kmauleon