fwupd/firmware-lenovo #5: [missing firmware] ThinkPad Thunderbolt 3 Dock

kat-co, 2019-09-14:
I'm hoping that with the recent support from Lenovo, they can also provide firmware for this. Currently the only updater is a Windows .exe :(

For missing firmware please provide the following items:

* Link to firmware update at OEM website
https://support.lenovo.com/us/en/accessories/ACC100356

* Firmware release date (if available on OEM website)
2017-10-31

* Link to the device page on LVFS.
(I don't believe this exists)

joakim-tjernlund, 2019-09-25:
> I can't say anything official. It could be a long wait, but things are moving.

It has been a long wait now :)

hughsie, 2019-09-25:
This stuff takes time. We have to talk with the OEM, the ODM and then the silicon provider. Each might be coming from a Windows-centric world and quite a lot of discussion has to be done around licencing and support. In better news, check out the 100% unrelated (ish) blog from today: https://blogs.gnome.org/hughsie/2019/09/25/synaptics-cx-audio-support/

I.e. patience :)

joakim-tjernlund, 2019-09-25:
Trying to be patient but its hard when the only updater is through Windows.

CX Audio FW support, great! I see you already have some ThinkPads in there :)

alisianoi, 2019-12-27:
My condolences for having to remove "macros-of-macros-of-macros" abstractions as described in the blog from above.

Just another [ThinkPad USB-C Dockstation](https://support.lenovo.com/us/en/solutions/acc100348) user checking in (different from topic starter's Thunderbolt Dock). I cannot update the firmware on my dockstation because the installer is an `.exe` :(

hughsie, 2019-12-28:
Docking station stuff is coming on now, but everyone is on holiday and hence a lot of the work has paused for a few weeks. You can see some of the most recent work here: https://github.com/fwupd/fwupd/pull/1663

joakim-tjernlund, 2020-01-05:
@alisianoi , curious w.r.t Wine
What version did you use? I see that stable Wine is fairly old.
Also, did you pull in default USE flags? These rebuilds alot.
Wonder if somthing simpler would work:
```
USE="-* abi_x86_64 run-exes ncurses threads" emerge -p =app-emulation/wine-vanilla-4.21

NickeZ, 2020-01-13:
Will it work for the usb-c gen 2 dock?

edoiks, 2020-01-21:
Is there any estimation when it will be available?

joakim-tjernlund, 2020-02-13:
Is fwdup 1.3.8 ready for Docks now? Just waiting for Lenovo to release dock FW?

hughsie, 2020-02-17:
> Is fwdup 1.3.8 ready for Docks now?

There is one plugin waiting for a silicon supplier, but I don't think that's blocking things.

> Just waiting for Lenovo to release dock FW?

Yes, it's under QA now. I can't share progress, sorry.

joakim-tjernlund, 2020-02-17:
> > Is fwdup 1.3.8 ready for Docks now?
> 
> There is one plugin waiting for a silicon supplier, but I don't think that's blocking things.
> 
> > Just waiting for Lenovo to release dock FW?
> 
> Yes, it's under QA now. I can't share progress, sorry.

Great, should not be long now then :)

nazco, 2020-04-04:
Is there any update on this? I guess that it is delayed because of the covid19 thing :( Thanks and stay healthy!

hughsie, 2020-04-04:
Not sure how much I can say, but the sheet looks green with the exception of one orange. I think COV19 really pushed things back a bit. Lots of people QA testing this internally now.

joakim-tjernlund, 2020-04-08:
Seems like a bunch of USB-C dock FW just hit the LVFS, happy testing :)

joakim-tjernlund, 2020-04-09:
Any idea if other Lenovo docks are coming soon?
We have a bunch of ThinkPad Basic/Pro/Ultra Docking Stations

hughsie, 2020-04-09:
> Any idea if other Lenovo docks are coming soon?

I have 22 docks on my desk right now. I can't share what is being worked on, but I have all this hardware for a reason :)

joakim-tjernlund, 2020-04-09:
That is a big desk to fit all those :)
Good to hear that it is coming along, thanks.

joakim-tjernlund, 2020-05-13:
I figured Lenovo  Basic/Pro/Ultra dock would be ready by now?

hughsie, 2020-05-13:
It turns out a global pandemic slows things down. My apologies.

alisianoi, 2020-05-16:
@joakim-tjernlund I mean, let's face it: kvm/qemu + an unactivated copy of Windows 10 is enough to get you started. I suggest you try `virt-manager` as your GUI, then forwarding a USB device is a point-and-click process: `Virtual Machine -> Redirect USB device -> Lenovo Dockstation`. Then you can just run the `.exe` updater and see how far it goes.

https://www.microsoft.com/en-gb/software-download/windows10ISO

Of course, when you pay ~200 euros for a box, you would expect it to just work (it looks like mine has DisplayPort issues, so I just use HDMI), but at least the developer here responds :)

hughsie, 2020-05-18:
> I suggest you try virt-manager as your GUI, then forwarding a USB device is a point-and-click process

joakim-tjernlund, 2020-05-18:
> > I suggest you try virt-manager as your GUI, then forwarding a USB device is a point-and-click process
> 
> I would advise against this. Source: Bricked a USB3 hub doing exactly this.

I will wait for the real fwupd pkgs :)
Hopefully they will show u any day now...

joakim-tjernlund, 2020-05-29:
Considering how many other FW updates Lenovo has managed to put out, I doubt that the
pandemic is the reason. Something else is holding back the dock updates.

joakim-tjernlund, 2020-06-08:
Yet another batch of non dock FW uploaded ...

hughsie, 2020-06-08:
> Yet another batch of non dock FW uploaded

Dude the peripherals team is different from the UEFI team. Patience please.

joakim-tjernlund, 2020-08-27:
Dare I ask if ThinkPad Basic/Pro/Ultra Docking FW will be out soon? Perhaps with 1.5.0 ?

hughsie, 2020-08-27:
All the code landed in 1.4.5 which was released some time ago. Now it's on Lenovo :)

thesilas, 2020-10-26:
Any update from Lenovo?

albertogarciacz, 2020-11-10:
I'm also interested. I've got 40AN0 and I'd like to update it's firmware. Last time I run  `fwupdmgr refresh ; fwupdmgr get-update` I got `ThinkPad Thunderbolt 3 Dock has no available firmware updates`. Any update from Lenovo so far?

bin101, 2020-11-10:
Not that I know. For linux only users I can highly recommend to install a portable unregistered windows on a flash drive and install the firmware update from windows. Some older versions have some great issues (at least under my system).

hughsie, 2020-11-10:
> Any update from Lenovo so far?

I'm working with the ODM on an issue they discovered during QA: https://github.com/fwupd/fwupd/pull/2570

joakim-tjernlund, 2020-11-10:
> > Any update from Lenovo so far?
> 
> I'm working with the ODM on an issue they discovered during QA: [fwupd/fwupd#2570](https://github.com/fwupd/fwupd/pull/2570)

For which Dock is this ?

hughsie, 2020-11-10:
> For which Dock is this

It was discovered on the TB3 Gen2, but probably affects anything with an MST device and no EC, which includes the Ultra etc.

joakim-tjernlund, 2020-11-10:
> > For which Dock is this
> 
> It was discovered on the TB3 Gen2, but probably affects anything with an MST device and no EC, which includes the Ultra etc.

I see this was merged, will there be a new release then soon?

hughsie, 2020-11-10:
> will there be a new release then soon

If the ODM confirms it fixes the problem, yes. Patience dude.

joakim-tjernlund, 2020-11-10:
> > will there be a new release then soon
> 
> If the ODM confirms it fixes the problem, yes. Patience dude.

Thanks, trying to muster some more patience then ... :)

okapies, 2021-01-18:
Hi, any updates on this?

mrhpearson, 2021-01-18:
They're still testing (I confirmed just last week)....and I'm frustrated too at this point :(

LordChaos73, 2021-01-25:
I tried to create a bootable USB stick with a portable version of Windows on it as I have some issues with my TB3 gen 2 dock (https://pcsupport.lenovo.com/in/en/solutions/pd500265). Unfortunately that didn't work. Any news on when we might be able to flash the dock using fwupdmgr? I have quite a few issues with the dock at the moment which seems to have been fixed with the latest firmware.

wllhf, 2021-01-29:
Is it possible to do the update from a non-thunderbolt Windows device via USB?

LordChaos73, 2021-02-17:
> Is it possible to do the update from a non-thunderbolt Windows device via USB?

I have no clue to be honest, but that's good questions actually. I will try this this evening.
Do you have an update for us @mrhpearson ?

leahoswald, 2021-02-17:
> Is it possible to do the update from a non-thunderbolt Windows device via USB?
It isn't. 

The only way is to create a Windows on an USB stick and than try to boot from it. You can do that by downloading a developer VM image from MS. Boot that image and enable pass through for your USB stick. Than download a Win10 ISO and use a (shady) software like Win2USB to create a Windows on a stick (use a fast stick!). Than boot from the image (sometimes it needs multiple tries, oh and disable secure boot and switch to bios mode if you've selected MBR in Win2USB). Configure the windows, install the update tool from Lenovo and run it. That's the way I did it multiple times now.

wllhf, 2021-02-17:
Thanks for your answer, but it seems easier to send the hardware back and buy another dock. 

As the dock leads to unbearable instability of the system it is connected to and there are no firmware updates available, Lenovo is basically selling 300$ bricks. This thread started in August 2018! :)

LordChaos73, 2021-02-17:
> As the dock leads to unbearable instability of the system it is connected to and there are no firmware updates available, Lenovo is basically selling 300$ bricks. This thread started in August 2018! :)

I agree this is taking way too long and the only real annoying issue I have is the unstable DP & HDMI connection, which is a major PITA. However, Linux is not officially supported for this dock as far as I'm aware.

wllhf, 2021-02-17:
Sure, officially, it's a platform independent device though and the Thinkpads with which they are supposed to used are sold with Fedora. I didn't expect this, next time I'll look it up before I buy Lenovo.

LordChaos73, 2021-03-01:
Can somebody provide us with an update please?

mrhpearson, 2021-03-01:
There was an issue with the DisplayPort component - if it's not connected and active then it doesn't get updated.

It's not confirmed if there will be a workaround for this, or if it will just be a known limitation - I think the latter but the docking firmware and test team need to confirm.

I'll push for an ETA - apart from the DP issue my understanding is it was all ready to go.

nkalkhof, 2021-03-01:
So if we do the update we'll need a Monitor connected to the DP outlet in active mode in order to get the DP component updated successfully, yes?
Will there be an update availble for the 1st gen TB3 dock (40AC) as well?

mrhpearson, 2021-03-01:
Correct on the DP component.

The 1st gen isn't getting an update, but my understanding is that if it has the same components they'll get updated....I'm not sure how that actually works. 
Once we have the G2 up I'll see if I can figure out more details but I don't want to distract the docking team until we've got the G2 delivered. I don't have a G1 dock to test myself unfortunately.

LordChaos73, 2021-03-03:
Thanks for the update @mrhpearson

nkalkhof, 2021-03-15:
Apparently the Gen2 dock update is now availble [https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadThunderbolt3DockGen2.Audio.firmware](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadThunderbolt3DockGen2.Audio.firmware)
Hopefully Gen1 owners will get an update too before the end of this decade ;)

LordChaos73, 2021-03-17:
Has anybody tested the firmware on the TB3G2 dock by any chance?

mrhpearson, 2021-03-17:
Yes - I did succesfully. 
There is an issue I'm looking into where the audio component sometimes disappears and it reappears after hotplugging the dock. I don't know yet if it's a firmware issue, or a firmware update issue (my guess is on the former).
For me the update was clean with no issues. A colleague hit an error around the audio component update which is making us hesitant to move the firmware to enabled state until we've understood it better. The error had limited impact - they hotplugged the dock, found that the audio component hadn't been updated, and then re-ran the update succesfully.
There is also the known issue with the display port component not showing up or getting updated - digging into that one as well, that's less likely to be a blocker as it's more understood and we do want to get all the other pieces out to you guys.
Mark

LordChaos73, 2021-03-17:
Thank you for all your effort @mrhpearson & @hughsie and hopefully the firmware updates will land in stable soon.

okapies, 2021-03-21:
I also got the same issue as @mrhpearson written. The audio component is visible in my ThinkPad, but audio was not outputted from the connected speaker.

thescouser89, 2021-03-22:
I've had the audio issue using the old firmware and was hoping the new firmware would fix this :/

Fedora 33 on a Thinkpad T490s

LordChaos73, 2021-03-30:
Any news on when this will be moved to the stable channel?

mrhpearson, 2021-03-31:
They're discussing it today....I'm not promising anything but it should be enabled really soon (just waiting for a couple more folk to give the thumbs up - I gave mine).

On that note though, has anybody on this thread had issues with using the testing version? We hit a couple of items and wouldn't mind getting update from other users. The issues we saw were:

1) audio device sometimes missing the update. This has been fixed by hughsie with an update to fwupd, but that won't necessarily be rolled out everywhere. If you see this unplug/replug the dock and confirm the audio component is seen with fwupd get-devices and retry

2) I had a strange issue where the main dock component wasn't updating. It didn't have any GUIDs show under get-devices unless I used sudo (which shouldn't be required). I did an update succesfully using sudo, but I think I just needed to reboot my system. If anybody else is seeing this let me know - I've not been able to reproduce it. 

3) The DP component doesn't get updated unless it is connected (On my system it doesn't actually show up at all which I'm trying to figure out). If you're not using DP then it shouldn't be a concern, if you are using DP then it should work so we decided not to delay the release for this issue.

Mark

maikzumstrull, 2021-03-31:
Ran the testing version. Screens went dark for quite some time but did come back. One USB device (Logitech wireless receiver for a mouse) didn't come back until I pulled and replugged it.

Of the three USB hubs in the dock, it seems to have left one on an older firmware. The first USB 3.1 hub and the USB 2.0 hub report `51.63`. The chained USB 3.1 hub reports `51.33`.

I don't use the built-in audio so I couldn't tell you if it worked before or after.

mrhpearson, 2021-04-13:
Yeah - it's literally down to just get a couple of people's approvals who haven't given it yet....I nagged last night. 
Mark

[62 more replies]