fwupd/firmware-lenovo #10: fwupd proposing incorrect firmware for Thinkpad T460

jgottlander, 2019-09-23:
Fwupd is proposing an update for the Thinkpad T460s, but I have a T460.
I guess it's because they have the same GUID.

```shell
$ fwupdmgr --version
client version:	1.0.9
daemon version:	1.0.9
compile-time dependency versions
	appstream-glib:	0.7.7
	gusb:	0.2.11
	fwupdate:	10
	efivar:	34
```
```shell
$ fwupdmgr get-updates
UEFI Device Firmware has firmware updates:
GUID:                    ffaf3e12-a6ea-4352-851b-76c6455913c6
ID:                      com.lenovo.ThinkPadN1CHT.firmware
Update Version:          0.1.14
Update Name:             ThinkPad T460s Embedded Controller Update
Update Summary:          Lenovo ThinkPad T460s Embedded Controller Firmware
Update Remote ID:        lvfs
Update Checksum:         SHA1(ecb06ffd30f958f35ce9d7d9a8add8c53f445c91)
Update Location:         https://fwupd.org/downloads/b28bf8f2c28cf63bff1faec6b89aec0a263152b3-Lenovo-ThinkPad-T460s-ECFirmware-1.14.cab
Update Description:      Lenovo ThinkPad T460s Embedded Controller Firmware Version 1.14
                         
                         The computer will be restarted automatically after updating BIOS completely . Do NOT turn off your computer or remove the AC adaptor while update is in progress.
                         
                         New functions or enhancements
                          • Initial version on LVFS.
No upgrades for 20FMS5BT00 System Firmware, current is 0.1.41: 0.1.41=same, 0.1.40=older, 0.1.39=older, 0.1.38=older, 0.1.37=older
cannot handle firmware requirement not-child

```

```shell
$ fwupdmgr get-devices
UEFI Device Firmware
  DeviceId:             33bc73bf0f48eb6177a007e9351e22d5bce2c04d
  Guid:                 ffaf3e12-a6ea-4352-851b-76c6455913c6
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.9
  VersionLowest:        0.0.1
  Icon:                 audio-card
  Created:              2019-07-11

UEFI Device Firmware
  DeviceId:             e47aa2e9c7ed86bbb41cbf7c7e9ca72504497207
  Guid:                 304a041a-be63-4f29-b429-168def233a23
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              184.50.3425
  VersionLowest:        0.0.1
  Icon:                 audio-card
  Created:              2019-07-11

UEFI Device Firmware
  DeviceId:             1e27bf0e569658d727f2d126752b270ed048ea45
  Guid:                 dad450ea-85da-4121-ae37-ce857cfa2b8e
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              0.1.11
  VersionLowest:        0.0.1
  Icon:                 audio-card
  Created:              2019-07-11

20FMS5BT00 System Firmware
  DeviceId:             218dfc72dd94339f3c3fe47b0df642273d049170
  Guid:                 c9a0acef-aa63-4b85-a52d-a9384163c0e8
[... 346 lines cut ...]
│       ├── xnu.mod
│       ├── xnu_uuid.mod
│       ├── xnu_uuid_test.mod
│       ├── xzio.mod
│       ├── zfscrypt.mod
│       ├── zfsinfo.mod
│       └── zfs.mod
├── initrd.img-4.18.0-15-generic
├── initrd.img-4.18.0-25-generic
├── memtest86+.bin
├── memtest86+.elf
├── memtest86+_multiboot.bin
├── System.map-4.18.0-15-generic
├── System.map-4.18.0-25-generic
├── vmlinuz-4.18.0-15-generic
└── vmlinuz-4.18.0-25-generic
```

Please answer the following questions:

- Operating system and version: Ubuntu 18.04.2 LTS
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): Installed from start
- Have you tried rebooting? Yes
- Are you using an NVMe disk? no
- Is secure boot enabled (only for the UEFI plugin)?

PhilippWendler, 2020-11-16:
This is probably the same as #19 and might be solved by doing a firmware upgrade under Windows once?

kmauleon, 2020-11-16:
@PhilippWendler is correct!  Thank you very much