fwupd/firmware-lenovo #11: nothing happens after reboot

emilas44, 2019-09-23:
```
fwupdmgr --version
client version:	1.2.9
checkout info:	1.2.9
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.2.9
```

```
sudo fwupdmgr install 31c5f61ceb395c6a7379b3481f1560fe7dabe274-Lenovo-ThinkPad-P1X1Extreme-ConsumerMEFirmware-12.0.35.1427.cab
[sudo] password for elvis: 
Decompressing?           [***************************************]
Authenticating?          [***************************************]
Installing on UEFI Device Firmware?                              ]
Scheduling?              [***************************************]

An update requires a reboot to complete. Restart now? [Y|n]: 

```

I have a X1 Extreme and fwupdate does nothing after reboot.... if I run `fwupdmgr update` it downloads the update file and it asks if I want to reboot and If I type in Y and Enter it will reboot and comes normally up to login screen and does nothing. I even made a video so you guys can see it. Maybe someone can suggest something what can I try or do to make it work?

https://www.youtube.com/watch?v=awlPrj4YX0w

- Operating system and version: POP-OS 18.04 
- How did you install fwupd: from snap master branch
- Have you tried rebooting? of course
- Are you using an NVMe disk? yes
- Is secure boot enabled (only for the UEFI plugin)? secure boot is disabled

cdock1029, 2020-02-06:
Same issue X1 Extreme Gen 2 and Pop!_OS 19.10

hughsie, 2020-02-06:
Is "Boot Order Lock" turned on in the bios?

cdock1029, 2020-02-06:
No sir

ChaotiX66, 2020-02-19:
I have the same issue. I run Pop_OS! 19.10 on an internal SSD on my ThinkPad X1 Extreme (Gen1). However, someone suggested to enter the Grub Boot Menu and choose fwupd boot environment from there over at this thread:
https://github.com/fwupd/fwupd/issues/1685#issuecomment-587329349
Have yet to confirm that it works though. Maybe someone following this thread can?

debalance, 2021-07-28:
Same problem here with Debian 10 on several Lenovo T490.

debalance, 2021-07-28:
Solved it using the latest fwupd flatpak from flathub.