fwupd/firmware-lenovo #6: Lenovo P50: After firmware update, grub is gone from UEFI boot; only Windows left

RalfJung, 2019-09-23:
I just installed a UEFI update on my Lenovo ThinkPad P50. I have a dual-boot system. Usually, the firmware loads grub which then offers to boot Linux or Windows. The firmware update itself seems to have worked fine. However, after the firmware update, it instead Windows immediately instead of loading grub. Going into the firmware setup shows that grub entirely disappeared from the list of boot options.  Lucky enough I had a boot stick handy that I could use to chroot into my system and setup grub again.

```shell
$ fwupdmgr --version
client version:	1.1.1
daemon version:	1.1.1
compile-time dependency versions
	appstream-glib:	0.7.10
	gusb:	0.2.11
	efivar:	34
```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

```shell
$ fwupdmgr get-devices
20ENCTO1WW System Firmware
  DeviceId:             2c1302f31806a0e0d57c377d99e18dae56351413
  Guid:                 ddc0ee61-e7f0-4e7d-acc5-c070a398838e
  Guid:                 230c8b18-8d9b-53ec-838b-6cfc0383493a
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.52
  VersionLowest:        0.1.46
  Icon:                 computer
  Created:              2018-09-02

UEFI Device Firmware
  DeviceId:             1f9ed21ce218ebcd0e1c803b7da08dd4e33d01a3
  Guid:                 671d19d0-d43c-4852-98d9-1ce16f9967e4
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              184.50.3425
  VersionLowest:        0.0.1
  Icon:                 audio-card
  Created:              2018-09-02

UEFI Device Firmware
  DeviceId:             6c8326bcda55dc53571921c60da36f320d3efa3f
  Guid:                 a9971959-9246-4a5b-b2f2-ba6fdcb19349
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              0.1.17
  VersionLowest:        0.0.1
  Icon:                 audio-card
  Created:              2018-09-02
```

```shell
$ efibootmgr -v
BootCurrent: 0001
Timeout: 0 seconds
BootOrder: 0001,0000,0017,0018,0019,001A,001B,001C,001D,001E,001F,0020
Boot0000  Windows Boot Manager	HD(1,GPT,28c13da1-69ee-4c7a-a7f4-6852aeac394e,0x800,0x100000)/File(\EFI\Microsoft\Boot\bootmgfw.efi)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}...a................
Boot0001* debian	HD(1,GPT,28c13da1-69ee-4c7a-a7f4-6852aeac394e,0x800,0x100000)/File(\EFI\debian\grubx64.efi)
Boot0010  Setup	FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
Boot0011  Boot Menu	FvFile(126a762d-5758-4fca-8531-201a7f57f850)
Boot0012  Diagnostic Splash Screen	FvFile(a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380)
Boot0013  Lenovo Diagnostics	FvFile(3f7e615b-0d45-4f80-88dc-26b234958560)
Boot0014  Startup Interrupt Menu	FvFile(f46ee6f4-4785-43a3-923d-7f786c3c8479)
Boot0015  Rescue and Recovery	FvFile(665d3f60-ad3e-4cad-8e26-db46eee9f1b5)
Boot0016  MEBx Hot Key	FvFile(ac6fd56a-3d41-4efd-a1b9-870293811a28)
Boot0017* USB CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,86701296aa5a7848b66cd49dd3ba6a55)
Boot0018* USB FDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,6ff015a28830b543a8b8641009461e49)
Boot0019* NVMe0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a400)
Boot001A* NVMe1	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a401)
Boot001B* ATA HDD2	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f600)
Boot001C* ATA HDD3	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f604)
Boot001D* ATA HDD0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f603)
Boot001E* ATA HDD1	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f602)
Boot001F* USB HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,33e821aaaf33bc4789bd419f88c50803)
Boot0020* PCI LAN	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
Boot0021* IDER BOOT CDROM	PciRoot(0x0)/Pci(0x14,0x0)/USB(15,1)
Boot0022* IDER BOOT Floppy	PciRoot(0x0)/Pci(0x14,0x0)/USB(15,0)
Boot0023* ATA HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f6)
Boot0024* ATAPI CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a354)
```

```shell
$ efivar -l | grep fw
# no output
```

```shell
$ tree /boot
/boot
├── config-4.16.0-2-amd64
├── config-4.17.0-1-amd64
├── config-4.17.0-3-amd64
├── efi
│   ├── EFI
│   │   ├── Boot
│   │   │   └── bootx64.efi
│   │   ├── debian
│   │   │   ├── fw
│   │   │   ├── fwupdx64.efi
│   │   │   └── grubx64.efi
│   │   └── Microsoft
│   │       └── # snip
│   └── System Volume Information
│       └── IndexerVolumeGuid
├── grml
│   └── grml64-full_2017.05.iso
├── grub
│   ├── fonts
│   │   └── unicode.pf2
│   ├── grub.cfg
│   ├── grubenv
│   ├── locale
│   │   ├── ast.mo
│   │   ├── ca.mo
│   │   ├── da.mo
│   │   ├── de_CH.mo
│   │   ├── de@hebrew.mo
│   │   ├── de.mo
│   │   ├── en@arabic.mo
│   │   ├── en@cyrillic.mo
│   │   ├── en@greek.mo
│   │   ├── en@hebrew.mo
│   │   ├── en@piglatin.mo
│   │   ├── en@quot.mo
│   │   ├── eo.mo
│   │   ├── es.mo
│   │   ├── fi.mo
│   │   ├── fr.mo
│   │   ├── gl.mo
│   │   ├── hr.mo
│   │   ├── hu.mo
│   │   ├── id.mo
│   │   ├── it.mo
│   │   ├── ja.mo
│   │   ├── ko.mo
│   │   ├── lt.mo
│   │   ├── nb.mo
│   │   ├── nl.mo
│   │   ├── pa.mo
│   │   ├── pl.mo
│   │   ├── pt_BR.mo
│   │   ├── ru.mo
│   │   ├── sl.mo
│   │   ├── sr.mo
│   │   ├── sv.mo
│   │   ├── tr.mo
│   │   ├── uk.mo
│   │   ├── vi.mo
│   │   ├── zh_CN.mo
│   │   └── zh_TW.mo
│   ├── unicode.pf2
│   └── x86_64-efi
│       └── # snip
├── initrd.img-4.16.0-2-amd64
├── initrd.img-4.17.0-1-amd64
├── initrd.img-4.17.0-3-amd64
├── lost+found
├── mk-chroot
├── System.map-4.16.0-2-amd64
├── System.map-4.17.0-1-amd64
├── System.map-4.17.0-3-amd64
├── vmlinuz-4.16.0-2-amd64
├── vmlinuz-4.17.0-1-amd64
└── vmlinuz-4.17.0-3-amd64

```

Please answer the following questions:

- Operating system and version: Debian testing
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): `apt install`
- Have you tried rebooting? Well it didn't boot into Linux any more after the update
- Are you using an NVMe disk? No (as far as I know -- the SSD has the M.2 firm but I think it uses normal SATA to connect)
- Is secure boot enabled (only for the UEFI plugin)? No

the8472, 2020-01-19:
Also seen this happening on a P53s.

nmoreaud, 2020-01-20:
Each time I update Lenovo firmware, it boots on Windows at next startup.
Here is a solution to bring back grub : 
In windows, open "cmd" with elevated rights and run `bcdedit /set {bootmgr} path \EFI\ubuntu\grubx64.efi` (https://doc.ubuntu-fr.org/uefi#lenovo)

However, I would like to upgrade firmware without having to run a command manually each time

paolomainardi, 2020-03-07:
It happens as well to me on thinkpad x1 Carbon and Ubuntu 18.04.

Is there a technical reason for this ? What should be the standard behavior ? What I had to do was also to move “Linux firmware updater” manually as the first item to make it works and then grub gone.

houdini68, 2020-10-09:
It happened on a T590 and Ubuntu 20.04.1LTS.

mrhpearson, 2020-10-11:
Hi @houdini68 - do you happen to know what version of firmware you were updating from and to? And just to confirm you running a dual boot system?
I hadn't sen this thread before and we do test the firmware updates but I don't think we've tested in a dual-boot environment - have to see if we can reproduce the issue.
Thanks
Mark

greyltc, 2020-11-14:
I had this today with my ThinkCentre M920 Tower. It's really quite terrible since there seems to be no way in the firmware user interface to point to a new UEFI boot stub.

guillemglez, 2020-12-06:
I have this same issue on a Lenovo X1 Carbon 6th gen. Happened several times, got used at reinstalling grub every time I update, but still very annoying...

kmauleon, 2020-12-11:
hi all... tried to consult Dev BIOS and here's their confirmation... thank you very much...
----------------------------------------------------------------------
Grub is still displayed after BIOS update. Since BIOS code is similar on SKL/KBL/WHL/CML products, this symptom is customer’s environment, such as Linux distribution, dual-boot setting, etc.
----------------------------------------------------------------------
Below is the result.
1.	Install grub
2.	Confirm grub menu appears at POST.
-	GRUB menu was not displayed, so I had to modify the grub configuration ( /etc/default/grub )
                  GRUB_STYLE_MENU=menu
                  GRUB_TIMEOUT=10
3.	Update BIOS by ShellFlash.
4.	Confirm grub menu appears or not at POST.
-	GRUB is still displayed

guillemglez, 2020-12-11:
@kmauleon You mean that by adding `GRUB_STYLE_MENU=menu` to our `/etc/default/grub` will fix the issue? Why does this happen?

kmauleon, 2020-12-11:
hi @guillemglez ... this is not actually a fix.  Dev BIOS Team needed to set these parameters in order to confirm if GRUB menu will disappear after BIOS update since initially his environment is not displaying GRUB at POST.  Thus, procedure 2.
     3.  Update BIOS by ShellFlash.
     4.  Confirm grub menu appears or not at POST.
               GRUB is still displayed >>> **PASS** (**GRUB menu is still displayed after BIOS update**)
I hope this clarifies your question... thank you very much.

joff13, 2021-02-11:
Hi for those that will get across this issue...
Same for me with lenovo legion..

solution

enter boot at startup

1 in bios menu check boot sequence (should not be changed but make sure 1# is the right harddrive)
2 change EFI sequence to have Ubuntu/Linux or whatever as grub
3 save and reboot 

4 if starting Linux distro you get black screen ( it appends to me) enter uefi bios again and restore factory default. In boot menu, boot sequence has changed with legacy

5 reboot
6 have a nice day

danneboom, 2021-07-08:
> Each time I update Lenovo firmware, it boots on Windows at next startup.
> Here is a solution to bring back grub :
> In windows, open "cmd" with elevated rights and run `bcdedit /set {bootmgr} path [...]

Maybe worth mentioning that it was a slightly different command for me (Thinkpad P52, secure boot enabled):
`bcdedit /set {bootmgr} path \EFI\ubuntu\shimx64.efi`

Maltimore, 2022-09-16:
Could someone tell me in concrete steps what I need to do to "get grub back"? I do not have windows installed, so for me it just boots into an error screen. I do have a live USB stick available. I followed guides to reinstall grub like here: http://subinsb.com/how-to-repairrestorereinstall-grub-2-using-a-ubuntu-live-cd/ but it doesn't seem to solve the problem.
I don't know what some abstract step like "change EFI sequence to have Ubuntu/Linux or whatever has grub" means. I don't know how to change the EFI sequence. In the UEFI startup screen I don't see an option to do that.

Maltimore, 2022-09-16:
Turns out, that one newer systems there's one more directory that needs to be bind-mounted, see this answer here: https://unix.stackexchange.com/a/418913
All the bind-mounted directories can be mounted like so:
`for i in /dev /dev/pts /proc /sys /sys/firmware/efi/efivars /run; do sudo mount -B $i /mnt$i; done`
Apart from that, follow any guide like http://subinsb.com/how-to-repairrestorereinstall-grub-2-using-a-ubuntu-live-cd/ and it'll hopefully work.