fwupd/firmware-lenovo #8: Lenovo X1C6 update is not applied

danielmellado, 2019-09-23:
While trying to get an update for a component on X1C6 as follows:

```shell
UEFI Device Firmware
  DeviceId:             e563ad307df81c99f0de8c26292afd71cf409673
  Guid:                 42a0a96e-c9f3-438f-9687-7826be33e4ce
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              184.60.3561
  VersionLowest:        184.60.3561
  VersionFormat:        triplet
  Icon:                 audio-card
  Created:              2019-06-12
  Modified:             2019-06-12
  UpdateState:          needs-reboot

```
this always stays in version 184.60.3561, even if it should update to 11.8.65.3590. After rebooting it shows the updating screen but then get-updates asks to update the same update again and shows that it has been succesfully update from 180.50 to 180.60.

I've tried also to download the cap file directly and apply it from cli with same results.

```shell
$ fwupdmgr --version
client version:	1.2.9
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.2.9
```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

```shell
$ fwupdmgr get-devices
X1 Carbon Thunderbolt Controller
  DeviceId:             4631827429fe709891845868ba9887ddae5b020f
  Guid:                 5ff9544e-fce2-5fd9-9cd2-671c95ac91dc <- TBT-01091706
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|require-ac|registered
  Vendor:               Lenovo
  VendorId:             TBT:0x0109
  Version:              33.00
  VersionFormat:        pair
  Icon:                 computer
  Created:              2019-06-12

ThinkPad Thunderbolt 3 Dock
  DeviceId:             5db4f3ee776387a10c57f4fe7fcb06197da2f637
  Guid:                 df675f10-53ac-59ca-bddd-0a86ee492920 <- TBT-01081630
  Plugin:               thunderbolt
  Flags:                updatable|require-ac|registered
  Vendor:               Lenovo
  VendorId:             TBT:0x0108
  Version:              15.00
  VersionFormat:        pair
  Icon:                 audio-card
  Created:              2019-06-12

20KGS23S06 System Firmware
  DeviceId:             1c53551e7da69d896138fac1ae131c83ad46d923
  Guid:                 a4b51dca-8f97-4310-8821-3330f83c9135
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Checksum:             SHA1(0fef62b744fa13d59b6947518fc79d13f0c038b4)
  Checksum:             SHA256(6b7f41b2505f44be4943f603a6c98d76ac243dfd6e702d364c008da85791c430)
  Version:              0.1.38
  VersionLowest:        0.0.1
[... 209 lines cut ...]
├── loader
│   └── entries
│       ├── 2628eeaba54e4585b96f0c3c122e8039-0-rescue.conf
│       ├── 2628eeaba54e4585b96f0c3c122e8039-5.1.5-300.fc30.x86_64.conf
│       ├── 2628eeaba54e4585b96f0c3c122e8039-5.1.6-300.fc30.x86_64.conf
│       └── 2628eeaba54e4585b96f0c3c122e8039-5.1.7-300.fc30.x86_64.conf
├── lost+found
├── memtest86+-5.01
├── System.map-5.1.5-300.fc30.x86_64
├── System.map-5.1.6-300.fc30.x86_64
├── System.map-5.1.7-300.fc30.x86_64
├── vmlinuz-0-rescue-2628eeaba54e4585b96f0c3c122e8039
├── vmlinuz-5.1.5-300.fc30.x86_64
├── vmlinuz-5.1.6-300.fc30.x86_64
└── vmlinuz-5.1.7-300.fc30.x86_64

```

Please answer the following questions:

- Operating system and version: fedora30
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): dnf
- Have you tried rebooting? yes
- Are you using an NVMe disk? yes
- Is secure boot enabled (only for the UEFI plugin)? no

kmauleon, 2020-10-23:
hi.. can you please try this >> https://github.com/fwupd/firmware-lenovo-thinkpad/issues/70#issuecomment-714491086
downgrade / update seem to fix the issue as stated... thank you very much

kmauleon, 2022-11-16:
Closing this as BIOS ESRT fix has been rolled out already.  Please reopen if you still see the problem.  Thank you very much.

closed 2022-11-16 by kmauleon