fwupd/firmware-lenovo #16: Thinkpad T460s Management Engine Update fails to update

atrueba, 2019-09-24:
The machine is a 2017 Thinkpad P40 Yoga (20GQ). A fresh installation of fwupdate and gnome-software shown an available update for "Thinkpad T460s Management Engine", which was in its factory version (170.xxx or 171.xxx, I don't recall). The latest available firmware was v184.60.3561

The update seemed to work: download OK, reboot, BIOS messages about finding and installing a new capsule and reboot again into the OS, but a new run of gnome-software and package refresh still showed the same situation as before: installed version 17x and new available version v184. I repeated the process a couple of times during a few days, but the firmware was never updated. Tried also with the command line (fwupmgr refresh & install), with the same results

I noticed there was also an older update available (v184.55.3510) and figured maybe the updates had to be accumulative, so I downloaded the cab and installed it manually with "fwupdmagr install xxx.cab". This time the process seemed to work and gnome-software now reports that  version as the currently installed one and wants to update to the latest one, but this time neither the automated install nor the manual download/install seem to work: as at the begining, everything looks like it's updating but after the last reboot the firmware is still at v184.55.3510.

I'm guessing the problem here resides in Lenovo's firmware rather that in fwupd. Notice that there are two variants of this firmware: Consumer (that's the one I'm having trouble with) and Corporate.

```
$ fwupdmgr --version
client version:	1.2.6
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.2.6
```

```
$ fwupdmgr get-devices
20GQCTO1WW System Firmware
  DeviceId:             8be4cbd5f33a57bd5ac69de9004c4e5ecc91325f
  Guid:                 b8d73eb5-feba-42e1-a106-04540b6ae7d6
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Checksum:             SHA1(de773864c70add85c32c36c8ee79e985c47c4837)
  Version:              0.1.55
  VersionLowest:        0.1.55
  Icon:                 computer
  Created:              2019-05-04

UEFI Device Firmware
  DeviceId:             aff24b50c0a8c292c6c2a81d3827c36d7a86f62a
  Guid:                 03d3297b-3851-4379-95ad-12e21a96c80a
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              184.55.3510
  VersionLowest:        184.55.3510
  Icon:                 audio-card
  Created:              2019-05-04
  UpdateState:          success

SanDisk SD8TB8U256G1001
  DeviceId:             5dbeb140337f610d54913c0f43d15fdaa0eafb51
  Guid:                 9bbc7ebe-9b6f-5f43-a56e-e5037c19ec56
  Guid:                 818c75b7-74ff-5960-85e0-cdce1cb7b55a
  Guid:                 9a64ac8d-a549-58b3-afb0-277bdd6ac025
  Summary:              ATA Drive
  Plugin:               ata
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              X4133101
  Icon:                 drive-harddisk
  Created:              2019-05-04
```

```
$ efibootmgr -v
BootCurrent: 0001
Timeout: 0 seconds
BootOrder: 0001,0000,0017,0018,0019,001A,001B,001C,0002
Boot0000  Windows Boot Manager	HD(1,GPT,e15f6a59-b630-4bb2-aef7-7700d5b24ca7,0x800,0x100000)/File(\EFI\Microsoft\Boot\bootmgfw.efi)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}....................
Boot0001* debian	HD(1,GPT,e15f6a59-b630-4bb2-aef7-7700d5b24ca7,0x800,0x100000)/File(\EFI\debian\grubx64.efi)
Boot0002* Linux-Firmware-Updater	HD(1,GPT,e15f6a59-b630-4bb2-aef7-7700d5b24ca7,0x800,0x100000)/File(\EFI\debian\fwupdx64.efi)
Boot0010  Setup	FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
Boot0011  Boot Menu	FvFile(126a762d-5758-4fca-8531-201a7f57f850)
Boot0012  Diagnostic Splash Screen	FvFile(a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380)
Boot0013  Lenovo Diagnostics	FvFile(3f7e615b-0d45-4f80-88dc-26b234958560)
Boot0014  Startup Interrupt Menu	FvFile(f46ee6f4-4785-43a3-923d-7f786c3c8479)
Boot0015  Rescue and Recovery	FvFile(665d3f60-ad3e-4cad-8e26-db46eee9f1b5)
Boot0016  MEBx Hot Key	FvFile(ac6fd56a-3d41-4efd-a1b9-870293811a28)
Boot0017* USB CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,86701296aa5a7848b66cd49dd3ba6a55)
Boot0018* USB FDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,6ff015a28830b543a8b8641009461e49)
Boot0019* ATA HDD0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f600)
Boot001A* ATA HDD1	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f601)
Boot001B* USB HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,33e821aaaf33bc4789bd419f88c50803)
Boot001C* PCI LAN	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
Boot001D* IDER BOOT CDROM	PciRoot(0x0)/Pci(0x16,0x2)/Ata(0,1,0)
Boot001E* IDER BOOT Floppy	PciRoot(0x0)/Pci(0x16,0x2)/Ata(0,0,0)
Boot001F* ATA HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f6)
Boot0020* ATAPI CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a354)
```

```
$ efivar -l | grep fw
$ tree /boot

****** efivar and tree are not installed
****** fwupdx64 resides in /boot/efi/EFI/debian
****** the .cab files prepared to install where available inside  /boot/efi/EFI/debian/fw. At the moment that folder is empty
```

Please answer the following questions:

- Operating system and version: Debian 10.0. A few packages from testing & experimental
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): apt-get
- Have you tried rebooting? Yes
- Are you using an NVMe disk? No
- Is secure boot enabled (only for the UEFI plugin)? No

kmauleon, 2020-06-23:
> This is still a problem with the latest 184.65.3590 firmware. Please reopen.

Hi PhilippWendler
Please show results of the below command...thank you very much.

fwupdmgr get-devices

PhilippWendler, 2020-06-23:
For my T460p, I currently do not get any updates for the  offered, I am stuck at `176.12.1008`. When will updates for this be available on LVFS?

Note that for the System Firmware, on the same system it seems that it still thinks

PhilippWendler, 2020-06-23:
For the Intel Management Engine, I currently do not get any updates offered on my T460p, I am stuck at `176.12.1008`. When will updates for this be available on LVFS?

Note that for the System Firmware, on the same system it seems that there is still a confusion with updates for the T460s as I explained in https://github.com/fwupd/firmware-lenovo-thinkpad/issues/19#issuecomment-630034660. So currently I cannot upgrade any firmware on this device, it is still one the same version as bought. Not sure whether this is related.

The ME info from `fwupdmgr get-devices -v` (version 1.4.4) is this:
```
Intel Management Engine                                                               
  DeviceId:             25127f5d4d5dfd6e903cda1a61a67cb092e4bb0f                                                                   
  Guid:                 63451986-0c42-42db-9a18-342a8455ac68                                                                       
  Plugin:               uefi                                                                                                       
  Protocol:             org.uefi.capsule                                                                                           
  Flags:                internal|updatable|require-ac|registered|needs-reboot|usable-during-update|md-set-name-category|md-set-verf
mt                                                                                                                                 
  VendorId:             DMI:LENOVO                                                                                                 
  Version:              176.12.1008                                                                                                
  VersionLowest:        176.12.1008                                                                                                
  VersionFormat:        triplet                                                                                                    
  VersionRaw:           0xb00c03f0                                                                                                 
  VersionLowestRaw:     0xb00c03f0                                                                                                 
  Icon:                 audio-card                                                                                                 
  Created:              2020-06-23                                                                                                 
```

In the comment linked above I also provided full output of `get-updates -v` and `get-devices -v` (expand them at the bottom).

If you need any further information I will be happy to provide it.

kmauleon, 2020-06-23:
> For the Intel Management Engine, I currently do not get any updates offered on my T460p, I am stuck at `176.12.1008`. When will updates for this be available on LVFS?
> 
> Note that for the System Firmware, on the same system it seems that there is still a confusion with updates for the T460s as I explained in [#19 (comment)](https://github.com/fwupd/firmware-lenovo-thinkpad/issues/19#issuecomment-630034660). So currently I cannot upgrade any firmware on this device, it is still one the same version as bought. Not sure whether this is related.
> 
> The ME info from `fwupdmgr get-devices -v` (version 1.4.4) is this:
> 
> ```
> Intel Management Engine                                                               
>   DeviceId:             25127f5d4d5dfd6e903cda1a61a67cb092e4bb0f                                                                   
>   Guid:                 63451986-0c42-42db-9a18-342a8455ac68                                                                       
>   Plugin:               uefi                                                                                                       
>   Protocol:             org.uefi.capsule                                                                                           
>   Flags:                internal|updatable|require-ac|registered|needs-reboot|usable-during-update|md-set-name-category|md-set-verf
> mt                                                                                                                                 
>   VendorId:             DMI:LENOVO                                                                                                 
>   Version:              176.12.1008                                                                                                
>   VersionLowest:        176.12.1008                                                                                                
>   VersionFormat:        triplet                                                                                                    
>   VersionRaw:           0xb00c03f0                                                                                                 
>   VersionLowestRaw:     0xb00c03f0                                                                                                 
>   Icon:                 audio-card                                                                                                 
>   Created:              2020-06-23                                                                                                 
> ```
> 
> In the comment linked above I also provided full output of `get-updates -v` and `get-devices -v` (expand them at the bottom).
> 
> If you need any further information I will be happy to provide it.

Hi PhilippWendler... it seems T460p is using T460s GUID... but I believe they have fixed this issue last october 2019 with their BIOS update.  can you please try to run the following commands...

fwupdmgr refresh
fwupdmgr update

PhilippWendler, 2020-06-23:
I try to run these commands regularly, but there is no update available:
```
• Lenovo ThinkPad Dock has no available firmware updates
• Intel Management Engine has no available firmware updates
• Samsung SSD 850 EVO 1TB has no available firmware updates
• System Firmware has the latest available firmware version
• VMM2322 has no available firmware updates
```

kmauleon, 2020-06-23:
Hi Philipp...latest version for system firmware is as below...

ThinkPad T460p System Update
Vendor: Lenovo Ltd.
Latest Version: 0.2.30

But I can't find any update for MEFW for T460p.... T460s MEFW updates will not be executed since I restrict the update to T460s only.... let me try to call the attention of T460p owner....

PhilippWendler, 2020-06-23:
I know that it is available, but how can I install the 0.2.30 system firmware?

It is not offered as upgrade, and if I download it manually from [LVFS](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadR07ET.firmware) and execute `fwupdmgr install 217bbab44027f49628cff3074e8c334bec40663d-Lenovo-ThinkPad-T460p-SystemFirmware-2.30.cab` it says `No supported devices found`.

kmauleon, 2020-06-23:
I already called the attention of the owner... let me get back to you once I have some updates from them..... thank you very much...

PhilippWendler, 2020-06-23:
Thank you very much for taking care of this!

kmauleon, 2020-07-06:
Hi @PhilippWendler ,
A new MEFW capsule for T460p has been uploaded to LVFS...please check... Thank you very much

lijunwang1990, 2020-07-06:
> I know that it is available, but how can I install the 0.2.30 system firmware?
>
> It is not offered as upgrade, and if I download it manually from [...]

pleasre use below command have a try:
fwupdmgr --allow-older --allow-reinstall install  217bbab44027f49628cff3074e8c334bec40663d-Lenovo-ThinkPad-T460p-SystemFirmware-2.30.cab

lijunwang1990, 2020-07-06:
> pleasre use below command have a try:
> fwupdmgr --allow-older --allow-reinstall install 217bbab44027f49628cff3074e8c334bec40663d-Lenovo-ThinkPad-T460p-SystemFirmware-2.30.cab

Hi @PhilippWendler 
I see your GUID is 6345.., but T460p system firmware GUID should be 90EF2839-18D5-4BCA-8F7C-9F846031D025, it is different, could you please help to check your BIOS Build ID on setup? if it begain with R07ET..

PhilippWendler, 2020-07-06:
I tried `fwupdmgr refresh --force; fwupdmgr get-updates`, still no updates offered.

I also tried to install it manually with `fwupdmgr --allow-older --allow-reinstall install 217bbab44027f49628cff3074e8c334bec40663d-Lenovo-ThinkPad-T460p-SystemFirmware-2.30.cab`, but it says `No supported devices found`.

I downloaded the 2.31 firmware manually from [LVFS](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadR07ET.firmware) and tried to install it with the same command, also `No supported devices found`.

The output of `dmidecode` shows the following part:
```
BIOS Information
        Vendor: LENOVO
        Version: R07ET71W (2.11 )
        Release Date: 09/26/2016
        Address: 0xE0000
        Runtime Size: 128 kB
        ROM Size: 16 MB
        Characteristics:
                PCI is supported
                PNP is supported
                BIOS is upgradeable
                BIOS shadowing is allowed
                Boot from CD is supported
                Selectable boot is supported
                EDD is supported
                3.5"/720 kB floppy services are supported (int 13h)
                Print screen service is supported (int 5h)
                8042 keyboard services are supported (int 9h)
                Serial services are supported (int 14h)
                Printer services are supported (int 17h)
                CGA/mono video services are supported (int 10h)
                ACPI is supported
                USB legacy is supported
                BIOS boot specification is supported
                Targeted content distribution is supported
                UEFI is supported
        BIOS Revision: 2.11
        Firmware Revision: 1.4
```
So the BIOS version is as expected (`R07ET71W`).

kmauleon, 2020-10-22:
new versions target to be released within november... please wait.. thank you very much

lijunwang1990, 2020-11-13:
Hi @PhilippWendler : I find the system GUID was changed in 2.18,  the new GUID is {90EF2839-18D5-4BCA-8F7C-9F846031D025},please update BIOS to 2.18 or later version use USB device or CD ,the latest BIOS version is 2.32, you can not update view fwupd as it will check GUID, after that I think you can update ME. any other question, please let me known, thank you.

lijunwang1990, 2020-11-13:
HOW TO CREATE A BOOTABLE BIOS FLASH USB MEMORY KEY
(please refer to lenovo readme https://download.lenovo.com/pccbbs/mobiles/r07uj29ww.txt)
1. Refer to the "HOW TO INSTALL" Section 1 to 7.
2. Attach a USB memory key to your Windows system, then format the USB memory key.
   Note: A medium formatted with NTFS or exFAT is not supported by this Flash Update process.
         Use FAT32 or FAT16 as a file system of the medium.
3. Open command prompt with administrator right, then move to a directory of this
   package.
4. Run mkusbkey.bat in the directory with following command format, then finish 
   creating BIOS flash USB memory key.

        Syntax : mkusbkey.bat [Drive]

        [Drive]
                D:,E:,F:... : Drive letter assigned to the USB memory key

  Example : mkusbkey D:

PhilippWendler, 2020-11-13:
Ok thanks for the response. I will upgrade manually. Is there no way to do this first upgrade without Windows?

lijunwang1990, 2020-11-16:
you can also download ISO file and use CD device update.

PhilippWendler, 2020-11-16:
Thanks, upgrading under Windows solved this for me. I upgraded my T460p's UEFI firmware to 2.30, and after that fwupd correctly offered me version 2.31 and finally also the correct "ThinkPad T460p Corporate ME Update". Upgrading the ME firmware with fwupd from 176.12.1008 to 184.77.3664 also worked fine.

Thank you for your help!

My issue is solved (although the original issue reported here was about the same problem on a different model).

kmauleon, 2020-11-16:
@PhilippWendler thank you very much!
@atrueba sorry for the delay in response ...unfortunately, P40 Yoga is not LVFS supported. thank you very much
@Forbo sorry for the delay in response ... please update BIOS/EC to the latest first then MEFW... after update of MEFW, please check version in BIOS Setup Menu... thank you very much

Forbo, 2021-02-16:
Eventually I just updated under Windows, which has since allowed me to install subsequent updates from Ubuntu.

kmauleon, 2021-02-17:
thank you very much for the information @Forbo ... closing the ticket...

closed 2021-02-17 by kmauleon