fwupd/firmware-lenovo #17: Can't update UEFI and System firmware on my Thinkpad P52s

FFY00, 2019-09-25:
To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version:	1.3.1
checkout info:	1.3.1
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.3.1

```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

```shell
$ fwupdmgr get-devices
20LBCTO1WW
│
├─P52s Thunderbolt Controller:
│     Device ID:           685059db1d7bba3ac49b1e7810636e1e64f7938b
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     18.00
│     Vendor:              Lenovo (TBT:0x0109)
│     Flags:               internal|updatable|require-ac|supported|registered
│     GUID:                eb8bf568-00a2-51e9-aa09-894358506c50
│
├─INTEL SSDPEKKF256G8L:
│     Device ID:           be29aadda98b637d33b6b740ff7b367f61886a31
│     Summary:             NVM Express Solid State Drive
│     Current version:     L08P
│     Vendor:              Intel Corporation (NVME:0x8086)
│     Flags:               internal|updatable|require-ac|registered|needs-reboot
│     GUIDs:               79517f86-8df8-5d6e-a18b-33f0b36a78e9
│                          68db11e5-b0cf-5bc9-a94e-17e28496e505
│                          cdc79848-b022-5a05-a77a-862f21df0c29
│
├─System Firmware:
│     Device ID:           facae3f8c9e996531c93d00358a7d5b40d3260f6
│     Current version:     0.1.18
│     Minimum Version:     0.0.1
│     Vendor:              LENOVO
│     Update State:        needs-reboot
│     Flags:               internal|updatable|require-ac|supported|registered|needs-reboot
│     GUID:                52dd29cd-24b9-4d60-8a45-e72670901924
│
├─UEFI Device Firmware:
│     Device ID:           7bd9d67d86feb8000c1a8b147d33ff4198bafefc
│     Current version:     184.55.3510
│     Minimum Version:     0.0.1
│     Update State:        needs-reboot
│     Flags:               internal|updatable|require-ac|supported|registered|needs-reboot
│     GUID:                72e9217b-717e-4908-b428-fe4df390177b
│
└─UEFI Device Firmware:
      Device ID:           66021d7392bed66cc5f15d3e6802f755ce0f63ad
      Current version:     0.1.5
      Minimum Version:     0.1.5
      Flags:               internal|updatable|require-ac|registered|needs-reboot
      GUID:                61b58b80-a21a-4185-8913-4bddfc68d6c9
```

```shell
$ efibootmgr -v
BootNext: 0004
BootCurrent: 0002
Timeout: 0 seconds
[... 507 lines cut ...]
Scheduling…              [    \                                  ](fwupdmgr:3954): Fwupd-DEBUG: 12:54:05.455: Emitting ::status-changed() [idle]
Scheduling…              [***************************************]
(fwupdmgr:3954): Fwupd-DEBUG: 12:54:05.455: Emitting ::changed()
(fwupdmgr:3954): Fwupd-DEBUG: 12:54:05.457: Emitting ::status-changed() [unknown]
(fwupdmgr:3954): Fwupd-DEBUG: 12:54:05.457: Emitting ::device-changed(facae3f8c9e996531c93d00358a7d5b40d3260f6)
Downloading 184.65.3590 for UEFI Device Firmware...
(fwupdmgr:3954): FuCommon-DEBUG: 12:54:05.460: creating path /home/anubis/.cache/fwupd
(fwupdmgr:3954): FuMain-DEBUG: 12:54:05.467: skpping download as file already exists
(fwupdmgr:3954): Fwupd-DEBUG: 12:54:05.471: Emitting ::status-changed() [decompressing]
Decompressing…           [    \                                  ](fwupdmgr:3954): Fwupd-DEBUG: 12:54:05.481: Emitting ::status-changed() [idle]
Decompressing…           [***************************************]
(fwupdmgr:3954): Fwupd-DEBUG: 12:54:05.481: Emitting ::status-changed() [waiting-for-auth]
Authenticating…          [    \                                  ](fwupdmgr:3954): Fwupd-DEBUG: 12:54:05.487: Emitting ::status-changed() [idle]
Authenticating…          [***************************************]
(fwupdmgr:3954): Fwupd-DEBUG: 12:54:05.492: Emitting ::status-changed() [scheduling]
Updating UEFI Device Firmware… \                                 ]
Scheduling…              [     \                                 ](fwupdmgr:3954): Fwupd-DEBUG: 12:54:05.493: Emitting ::device-changed(7bd9d67d86feb8000c1a8b147d33ff4198bafefc)
Scheduling…              [      \                                ](fwupdmgr:3954): Fwupd-DEBUG: 12:54:05.572: Emitting ::status-changed() [idle]
Scheduling…              [***************************************]
(fwupdmgr:3954): Fwupd-DEBUG: 12:54:05.572: Emitting ::changed()
(fwupdmgr:3954): Fwupd-DEBUG: 12:54:05.574: Emitting ::status-changed() [unknown]
(fwupdmgr:3954): Fwupd-DEBUG: 12:54:05.574: Emitting ::device-changed(7bd9d67d86feb8000c1a8b147d33ff4198bafefc)

An update requires a reboot to complete. Restart now? [Y|n]:
```

asheshambasta, 2019-11-19:
Hello, I've got a P52s and I'm looking to update my fw. Have you made any progress on this @FFY00?

asheshambasta, 2019-11-20:
FYI, I just upgraded my firmware without issues.