fwupd/firmware-lenovo #19: fwupd recommends Thinkpad T460s UEFI firmware for T460p

lzaoral, 2019-09-30:
Fwupd confuses Thinkpad T460p with T460s and offers wrong updates, even though the T460p firmware is available in LVFS (currently 2.28).

```shell
$ fwupdmgr get-updates
No upgrades for 20FXS0BB03 System Firmware, current is 0.2.12: 0.1.45=older, 0.1.44=older, 0.1.43=older, 0.1.42=older, 0.1.41=older
UEFI Device Firmware has firmware updates:
Device ID:               25127f5d4d5dfd6e903cda1a61a67cb092e4bb0f
GUID:                    63451986-0c42-42db-9a18-342a8455ac68
ID:                      com.lenovo.ThinkPadN1CRM.firmware
Update Version:          184.65.3590
Update Name:             ThinkPad T460s Corporate ME Update Update
Update Summary:          Lenovo ThinkPad T460s Corporate ME Firmware
Update Remote ID:        lvfs
Update Checksum:         SHA1(6182571ac60db540c18f87418195cb7f20c639ff)
Update Location:         https://fwupd.org/downloads/9f021b7b43bd68535390c0c8b7310ca7f88bd974-Lenovo-ThinkPad-T460s-CorporateMEFirmware-11.8.65.3590.cab
Update Description:       • 8 Quarterly Security Release
                         
                         Version 11.8.65.3590 (LVFS: 184.65.3590)
...
```

To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version:	1.2.10
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.2.10

```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

```shell
$ fwupdmgr get-devices
20FXS0BB03 System Firmware
  DeviceId:             444a229da2b21bc0d24faf814d1c58c5bff3315f
  Guid:                 42f29620-3d63-4f09-950b-9b7055570f28
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Checksum:             SHA1(224857caffd00074c1da8df9440bf8193c42a644)
  Version:              0.2.12
  VersionLowest:        0.2.12
  VersionFormat:        triplet
  Icon:                 computer
  Created:              2019-09-29

UEFI Device Firmware
  DeviceId:             25127f5d4d5dfd6e903cda1a61a67cb092e4bb0f
  Guid:                 63451986-0c42-42db-9a18-342a8455ac68
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              176.0.1205
  VersionLowest:        176.0.1205
  VersionFormat:        triplet
  Icon:                 audio-card
  Created:              2019-09-29

SAMSUNG MZ7LN512HMJP-000L7
  DeviceId:             5dbeb140337f610d54913c0f43d15fdaa0eafb51
  Guid:                 b987c66f-bb70-54a1-b880-e1c41ba2c222
  Guid:                 c9706c50-0f93-5b57-af6e-4e3fd77ea10e
  Guid:                 9aa5c587-9906-564a-b5d6-baffbf019028
  Summary:              ATA Drive
  Plugin:               ata
[... 499 lines cut ...]
│       ├── xnu_uuid.mod
│       ├── xnu_uuid_test.mod
│       ├── xzio.mod
│       ├── zfscrypt.mod
│       ├── zfsinfo.mod
│       ├── zfs.mod
│       └── zstd.mod
├── initramfs-5.2-x86_64-fallback.img
├── initramfs-5.2-x86_64.img
├── intel-ucode.img
├── linux52-x86_64.kver
├── memtest86+
│   └── memtest.bin
└── vmlinuz-5.2-x86_64

10 directories, 321 files
```

Please answer the following questions:

- Operating system and version: Manjaro 18.1.0 (Juhraya)
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): pacman
- Have you tried rebooting? yes
- Are you using an NVMe disk? no
- Is secure boot enabled (only for the UEFI plugin)? no

hughsie, 2019-09-30:
I've pinged Lenovo about this.

PhilippWendler, 2019-10-29:
Today I noticed that the suggested wrong T460s update for a T460p is gone, but it does not show any real updates for a T460p either:

```
$ fwupd.fwupdmgr --version
client version: 1.3.3
checkout info:  1.3.2-5-g506ef63a
compile-time dependency versions
        gusb:   0.2.11
        efivar: 34
daemon version: 1.3.3
```

```
$ fwupd.fwupdmgr get-devices
20FXS1B700
│
├─Samsung SSD 850 EVO 1TB:
│     Device ID:           5dbeb140337f610d54913c0f43d15fdaa0eafb51
│     Summary:             ATA Drive
│     Current version:     EMT01B6Q
│     Flags:               internal|updatable|require-ac|registered|needs-reboot
│     GUIDs:               e7307cf4-8f2f-5162-b7d2-7942747a765b
│                          bab8801e-be63-581d-a10e-14074d3d6b6d
│                          7faa6c7f-3fb9-5e4f-864e-4bf771d91769
│   
├─System Firmware:
│     Device ID:           444a229da2b21bc0d24faf814d1c58c5bff3315f
│     Current version:     0.2.11
│     Minimum Version:     0.2.11
│     Vendor:              LENOVO
│     Flags:               internal|updatable|require-ac|supported|registered|needs-reboot
│     GUID:                42f29620-3d63-4f09-950b-9b7055570f28
│   
├─UEFI Device Firmware:
│     Device ID:           25127f5d4d5dfd6e903cda1a61a67cb092e4bb0f
│     Current version:     176.12.1008
│     Minimum Version:     176.12.1008
│     Update State:        success
│     Flags:               internal|updatable|require-ac|supported|registered|needs-reboot
│     GUID:                63451986-0c42-42db-9a18-342a8455ac68
│   
└─VMM2322 inside ThinkPad Workstation Dock:
      Device ID:           2d9bd19a73af0e1160733d42c55f1d9de90ed451
      Summary:             Multi-Stream Transport Device
      Current version:     2.30.000
      Vendor:              Synaptics
      Flags:               updatable|registered
      GUIDs:               49ec4eb4-c02b-58fc-8935-b1ee182405c7
                           8eba3da0-2998-58bf-a7d7-210c15b3a3a8
                           75f1e01e-356f-5534-a8be-211d879838f1
```

```
$ fwupd.fwupdmgr get-updates
No upgrades for System Firmware, current is 0.2.11: 0.1.45=older, 0.1.44=older, 0.1.43=older, 0.1.42=older, 0.1.41=older
No releases found for device: no HWIDs matched 90706264-e399-575b-a9fb-077ead03b9b4
```

Of course I did a `refresh`. The firmware 176.12.1008 is several years old, I would expect to get a more recent firmware such as the 184.65.3590 for the T460s that was wrongly offered.

PhilippWendler, 2020-05-18:
Is there any update on this?

`fwupd` still says `System Firmware has the latest available firmware version`, although I have 0.2.11 installed and 0.2.30 is [available on LVFS](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadR07ET.firmware).

When I run `get-updates -v`, it mentions firmware versions 0.1.43 to 0.1.47, so it seems that the cause is that it still detects my T460p as a T460s and considers only [firmware updates for the T460s](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN1CET.firmware).

I tried to install 0.2.30 manually from LVFS with `install-blob`, and during reboot got some messages about firmware updating, but after reboot I am still stuck with 0.2.11.

```
# fwupd.fwupdtool --version
client version: 1.4.1
compile-time dependency versions
        gusb:   0.2.11
        efivar: 34
```
(installed from Snap)

Hardware: ThinkPad T460p 20FXS1B700

Device Info from `get-devices`:
```
├─System Firmware:
│     Device ID:           444a229da2b21bc0d24faf814d1c58c5bff3315f
│     Current version:     0.2.11
│     Minimum Version:     0.2.11
│     Vendor:              LENOVO (DMI:LENOVO)
│     GUID:                42f29620-3d63-4f09-950b-9b7055570f28
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
```
BIOS Info from `dmidecode`:
```
Handle 0x000B, DMI type 0, 24 bytes
BIOS Information
	Vendor: LENOVO
	Version: R07ET71W (2.11 )
	Release Date: 09/26/2016
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 16 MB
	Characteristics:
		PCI is supported
		PNP is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		3.5"/720 kB floppy services are supported (int 13h)
		Print screen service is supported (int 5h)
		8042 keyboard services are supported (int 9h)
		Serial services are supported (int 14h)
		Printer services are supported (int 17h)
		CGA/mono video services are supported (int 10h)
		ACPI is supported
		USB legacy is supported
		BIOS boot specification is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 2.11
	Firmware Revision: 1.4
```

For the Intel Management Engine, it says `no available firmware updates`, which sadly matches what LVFS shows, although there have been several updates according to the Lenovo website and ME updates are available for the T460s on LVFS, but not for the T460p.

[... 2005 lines cut ...]
08:35:11:0020 FuEngine             requirement 1.1.0 ge 1.4.1 on org.freedesktop.fwupd passed
08:35:11:0020 XbSilo               0,01ms      query on component with `releases/release` limit=0 -> 5 results
08:35:11:0020 XbSilo               0,01ms      query on component with `id` limit=1 -> 1 results
08:35:11:0020 XbSilo               0,03ms      query on component with `url[@type='homepage']` limit=1 -> 1 results
08:35:11:0020 XbSilo               0,01ms      query on component with `project_license` limit=1 -> 1 results
08:35:11:0020 XbSilo               0,01ms      query on component with `name` limit=1 -> 1 results
08:35:11:0021 XbSilo               0,01ms      query on component with `summary` limit=1 -> 1 results
08:35:11:0021 XbSilo               0,01ms      query on component with `developer_name` limit=1 -> 1 results
08:35:11:0021 XbSilo               0,03ms      query on component with `custom/value[@key='LVFS::VersionFormat']` limit=0 -> 1 results
08:35:11:0021 XbSilo               0,06ms      query on component with `../custom/value[@key='fwupd::RemoteId']` limit=1 -> 1 results
08:35:11:0021 XbSilo               0,01ms      query on release with `description` limit=1 -> 1 results
08:35:11:0021 XbSilo               0,01ms      query on release with `location` limit=1 -> 1 results
08:35:11:0021 XbSilo               0,03ms      query on release with `checksum[@target='content']` limit=1 -> 1 results
08:35:11:0021 XbSilo               0,03ms      query on release with `url[@type='details']` limit=1 -> 0 results
08:35:11:0021 XbSilo               0,02ms      query on release with `url[@type='source']` limit=1 -> 0 results
08:35:11:0021 XbSilo               0,03ms      query on release with `checksum[@target='container']` limit=1 -> 1 results
08:35:11:0021 XbSilo               0,03ms      query on release with `size[@type='installed']` limit=1 -> 1 results
08:35:11:0021 XbSilo               0,01ms      query on component with `screenshots/screenshot/caption` limit=1 -> 0 results
08:35:11:0021 XbSilo               0,01ms      query on component with `screenshots/screenshot/image` limit=1 -> 0 results
08:35:11:0021 XbSilo               0,03ms      query on component with `custom/value[@key='LVFS::UpdateProtocol']` limit=1 -> 1 results
08:35:11:0021 XbSilo               0,03ms      query on component with `custom/value[@key='LVFS::UpdateMessage']` limit=1 -> 0 results
08:35:11:0021 XbSilo               0,01ms      query on component with `id` limit=1 -> 1 results
08:35:11:0021 XbSilo               0,03ms      query on component with `url[@type='homepage']` limit=1 -> 1 results
08:35:11:0021 XbSilo               0,01ms      query on component with `project_license` limit=1 -> 1 results
08:35:11:0021 XbSilo               0

kmauleon, 2020-10-20:
hi T460p firmware.. thank you very much >> https://fwupd.org/lvfs/search?value=T460p

kmauleon, 2020-11-16:
same as #16 https://github.com/fwupd/firmware-lenovo-thinkpad/issues/16#issuecomment-726637384

PhilippWendler, 2020-11-16:
Indeed, and this solved the problem. I upgraded the UEFI firmware of my T460p from version 2.11 to 2.30 under Windows, and then fwupd correctly offered me version 2.31. Installing this with fwupd worked fine. Now I guess I just have to wait until 2.32 (already available on the Lenovo website) appears on LVFS.

Thank you for your help!

I think this issue can now be closed.

kmauleon, 2020-11-16:
thank you very much @PhilippWendler ... closing the issue ^_^

closed 2020-11-16 by kmauleon