fwupd/firmware-lenovo #20: No supported devices found

MountainX, 2019-10-03:
To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
fwupdmgr --version
client version: 1.3.2
compile-time dependency versions
        gusb:   0.3.0
        efivar: 37
daemon version: 1.3.2
```

```shell
$ fwupdmgr get-devices
20KHCTO1WW
│
├─X1 Carbon Thunderbolt Controller:
│     Device ID:           af0931ae5a019de856a5c4fc815e6928d5616dc7
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     33.00
│     Vendor:              Lenovo (TBT:0x0109)
│     Flags:               internal|updatable|require-ac|registered
│     GUID:               xxxx  ← TBT-01091706
│   
└─Samsung SSD 970 EVO 2TB:
      Device ID:           xxxx
      Summary:             NVM Express Solid State Drive
      Current version:     1B2QEXE7
      Vendor:              Samsung Electronics Co Ltd (NVME:0x144D)
      Serial Number:       xxxxxxx
      Flags:               internal|updatable|require-ac|registered|needs-reboot
      GUIDs:               xxx ← NVME\VEN_144D&DEV_A808
                           xxxx ← Samsung SSD 970 EVO 2TB

```

```shell
$ efibootmgr -v
BootCurrent: 001A
Timeout: 0 seconds
BootOrder: 001C,0018,0019,001A,0000,001B,001D,001E,001F,0024
Boot0000* Windows Boot Manager  HD(xxxxxxx)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}...B................
Boot0010  Setup FvFile(xxxxxxx)
Boot0011  Boot Menu     FvFile(xxxxxxx)
Boot0012  Diagnostic Splash Screen      FvFile(xxxxxxx)
Boot0013  Lenovo Diagnostics    FvFile(xxxxxxx)
Boot0014  Regulatory Information        FvFile(xxxxxxx)
Boot0015  Startup Interrupt Menu        FvFile(xxxxxxx)
Boot0016  Rescue and Recovery   FvFile(xxxxxxx)
Boot0017  MEBx Hot Key  FvFile(xxxxxxx)
Boot0018* USB CD        VenMsg(xxxxxxx)
Boot0019* USB FDD       VenMsg(xxxxxxx)
Boot001A* NVMe0 VenMsg(xxxxxxx)
Boot001B* ATA HDD0      VenMsg(xxxxxxx)
Boot001C* USB HDD       VenMsg(xxxxxxx)
Boot001D  PCI LAN       VenMsg(xxxxxxx)
Boot001E  Other CD      VenMsg(xxxxxxx)
Boot001F  Other HDD     VenMsg(xxxxxxx)
Boot0020* USBR BOOT CDROM       PciRoot(xxxxxxx)
Boot0021* USBR BOOT Floppy      PciRoot(xxxxxxx)
Boot0022* ATA HDD       VenMsg(xxxxxxx)
Boot0023* ATAPI CD      VenMsg(xxxxxxx)
Boot0024* PCI LAN       VenMsg(xxxxxxx)

```

```shell
$ efivar -l | grep fw
EMPTY (no output)
echo $?
1
```

```shell
$ tree /boot
/boot
├── EFI
│   ├── BOOT
│   │   └── BOOTX64.EFI
│   └── systemd
│       └── systemd-bootx64.efi
├── initramfs-linux-fallback.img
├── initramfs-linux.img
├── initramfs-linux-lts-fallback.img
├── initramfs-linux-lts.img
├── intel-ucode.img
├── loader
│   ├── entries
│   │   ├── arch.conf
│   │   ├── arch-fallback.conf
│   │   └── arch-lts.conf
│   └── loader.conf
├── vmlinuz-linux
└── vmlinuz-linux-lts

```

Please answer the following questions:

- Operating system and version: Arch Linux
- How did you install fwupd: `pacman`
- Have you tried rebooting? YES
- Are you using an NVMe disk? YES
- Is secure boot enabled (only for the UEFI plugin)? NO

Here are my problems:

```shell
fwupdmgr refresh
Fetching metadata https://cdn.fwupd.org/downloads/firmware.xml.gz
Downloading…             [***************************************]
Fetching signature https://cdn.fwupd.org/downloads/firmware.xml.gz.asc

fwupdmgr get-updates
No updatable devices
```
Since that did not work, I next tried this method:

```shell
wget https://fwupd.org/downloads/c04107f51e405d4921b2c933d5305c0adbca7116-Lenovo-ThinkPad-X1Carbon6th-SystemFirmware-1.41.cab
fwupdmgr install c04107f51e405d4921b2c933d5305c0adbca7116-Lenovo-ThinkPad-X1Carbon6th-SystemFirmware-1.41.cab
Decompressing…           [***************************************]
No supported devices found
```
I have the following BIOS settings:
Flash BIOS Updating by End-Users [Enabled]
Secure RollBack Prevention [Enabled]
Windows UEFI Firmware Update [Enabled]

exploide, 2019-10-03:
You were right.

No need to reset to defaults, but I found the respective option. It is called "**Windows** UEFI Firmware Update"... I wonder why I missed that -.-

No, to be honest, I remember I turned it off initially, since I was not sure if this is necessary for fwupd and thought I will get a warning if so. What I did not expect is that the "devices" don't show up at all (because ESRT isn't populated I guess).

Now updating the UEFI works, thanks! :smiley: 

But I also needed to turn off "Boot Order Lock". But I can imagine why :smile: Though, I'm wondering if this is only necessary for the first time to create the boot entry, or also for subsequent updates to set BootNext?

LIttleAncientForestKami, 2020-03-03:
@exploide was right with the *Windows* UEFI Firmware Update. Thanks to seeing this comment few days earlier I had a "hmm!" moment when `fwudpmgr` didn't offer updates for my BIOS and for the embedded controller. The other setting wasn't needed in my case (or perhaps it was already set? didn't check).

acrellin, 2020-04-19:
I'm having this issue, too, and no combination of BIOS settings seems to help. Lenovo X1 Carbon 7th Gen. Running Ubuntu 20.04. `fwupdmgr` was already installed (via `apt`). Secure boot not enabled; NVMe - yes; definitely tried rebooting; 

```
fwupdmgr --version
client version:	1.3.9
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.3.9
```

```
fwupdmgr get-devices
20R1CTO1WW
│
├─Thunderbolt Controller:
│     Device ID:           50341c1d669896bbe22ae85ee576dad79ac85c1e
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     44.00
│     Vendor:              Lenovo (TBT:0x0109)
│     GUIDs:               c7920601-0cda-507f-851e-92229eb1d370
│                          96c6512b-0138-51da-b02f-f44f2c4c7744
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Device stages updates
│   
├─Prometheus:
│ │   Device ID:           d432baa2162a32c1554ef24bd8281953b9d07c11
│ │   Summary:             Fingerprint reader
│ │   Current version:     10.01.3077709
│ │   Vendor:              Synaptics (USB:0x06CB)
│ │   Install Duration:    2 seconds
│ │   GUIDs:               09e8bf16-3e69-50f5-bb66-c7a040248352
│ │                        8088f861-6318-5b1e-9ce4-fbddbedb09ac
│ │   Device Flags:        • Updatable
│ │                        • Cryptographic hash verification is available
│ │ 
│ └─Prometheus IOTA Config:
│       Device ID:         8dcffb5d059857368caa2b7b22371ebd831b0c0c
│       Current version:   0021
│       Minimum Version:   0021
│       Vendor:            Synaptics (USB:0x06CB)
│       GUID:              7c5a1e62-38fa-5859-9337-09dbac6377e4
│       Device Flags:      • Updatable
│     
├─System Firmware:
│     Device ID:           09647b198775f1868cc695ec2685477e0964c323
│     Current version:     0.1.16
│     Vendor:              LENOVO (DMI:LENOVO)
│     GUID:                765f81e8-cc44-4d09-8fc4-9a47b167166e
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
│   
├─Touchpad:
│     Device ID:           2bd025a21741529fac39d6f088b9debfee7d4c9a
│     Current version:     1.3.3013337
│     Bootloader Version:  8.0
│     Vendor:              Synaptics (HIDRAW:0x06CB)
│     GUIDs:               4150912a-f249-5f7e-bfa7-2d49e9c9de8c
│                          9c27eba7-52a4-5e6f-b28c-b5d9a3991e35
│                          19326b41-711b-5b64-89a0-6c86f3139bf1
│                          54ed5b90-c4f7-5551-a4cc-e6ac9957a4a8
│     Device Flags:        • Internal device
[... 290 lines cut ...]
fwupdmgr get-updates                   
• Thunderbolt Controller has the latest available firmware version
• Prometheus has no available firmware updates
• Prometheus IOTA Config has no available firmware updates
• System Firmware has the latest available firmware version
• Touchpad has the latest available firmware version
• UEFI Device Firmware has no available firmware updates
• UEFI Device Firmware has the latest available firmware version
• UEFI Device Firmware has no available firmware updates
• UEFI Device Firmware has no available firmware updates
• UEFI Device Firmware has no available firmware updates
• WDC PC SN730 SDBQNTY-1T00-1001 has the latest available firmware version
```

When I try to install the BIOS update file from the Lenovo website (https://pcsupport.lenovo.com/us/en/products/laptops-and-netbooks/thinkpad-x-series-laptops/thinkpad-x1-carbon-7th-gen-type-20qd-20qe/downloads/ds540232), I get:

```
fwupdmgr install N2HET47W.cab          
Decompressing…           [***************************************]
No supported devices found
```

I have tried both enabling and disabling the "Windows UEFI Firmware Update" option in BIOS, to no avail. Secure boot is not enabled. Boot order lock is off. 

Any help would be greatly appreciated!

rikvanderkemp, 2020-06-09:
@acrellin did you manage to solve this? I just ran into the same issue.
Running version 1.3.9 of fwupd

SoumenDass, 2020-07-19:
Same error for Thinkpad X1 Extreme Gen 2. Output from "fwupdmgr get-devices --show-all-devices" here: https://pastebin.com/AvKVvyxd

jketcham, 2020-08-28:
Don't suppose anyone has an update on this? I'm running into this on debian testing on a T580. Just getting 'No supported devices found', and verbose output doesn't reveal any other clues. Enabling/disabling "Windows UEFI Firmware Update" didn't work. Boot order lock is turned off.

hughsie, 2020-08-28:
> I'm running into this on debian testing on a T580

Can you try a Fedora 32 LIveUSB and see if the devices appear? Then we know if it's distro or machine specific.

jketcham, 2020-08-28:
> > I'm running into this on debian testing on a T580
> 
> Can you try a Fedora 32 LIveUSB and see if the devices appear? Then we know if it's distro or machine specific.

@hughsie Just tested on Fedora 32 and unfortunately got the same 'No supported devices found' message.

ipstone, 2020-10-29:
reporting the same issue on thinkpad carbon x1 gen6

dandv, 2021-01-20:
I managed to work around this by writing [the ISO](https://pcsupport.lenovo.com/us/en/products/laptops-and-netbooks/thinkpad-x-series-laptops/thinkpad-x1-carbon-7th-gen-type-20qd-20qe/downloads/ds540232) to a USB disk (dd if=xxx.iso of=/dev/xxx bs=1MB), then booting from it.

allmazz, 2021-01-27:
Hello. Fwupd does not see the Synaptics fingerprint sensor. Compiled fwupd 1.3.5. Tried to disable Windows UEFI Update. frpintd and libfprint are compiled by me. I am using Manjaro. Lenovo Thinkpad E14 Gen 2 laptop. The sensor works in Windows. BIOS updated. Secure Boot is also disabled. Please tell me what I can try.

dariusk, 2021-11-29:
I'm on a Thinkpad X1 Carbon Gen 9 running Ubuntu 20.04 and I was able to get my firmware updated from 1.27 to 1.49 simply by doing:

```bash
$ sudo fwupdmgr refresh
$ sudo fwupdmgr update
```

and following the instructions and rebooting. I still can't get it to update past 1.49 but this was still a big, and easy, improvement for me.

msureshsns, 2022-05-31:
Hello, I am also facing a similar issue on Ubuntu 20.04 with Fwupd 1.7.5. When I run the "fwupdmgr get-devices --show-all-devices" it does not list my usb devices which are connected to the machine. It shows only 

[image]

When I checked "lsusb", it shows connected USB devices.
Any idea why Fwupd is not enumerating connected USB devices?

hughsie, 2022-05-31:
@msureshsns you're running in a VM.

msureshsns, 2022-05-31:
> @msureshsns you're running in a VM.

Yes, I am running in a VM and tried on a separate Linux machine as well, which is running Ubuntu 18.04 with Fwupd 1.2.14. In that machine, it also does not list connected devices.

msureshsns, 2022-06-01:
> Firmware updates won't work inside a VM, and 1.2.14 is ancient.

When I run the "fwupdmgr get-devices --show-all-devices"

It shows only the "synaptics_cape" plugin device (EPOS ADAPT 160 ANC).

[image]

But the other device (usb dongle) is Qualcomm chipset device and I have added that device VID and PID in the "dfu-csr.quirk" file.

But its not showing in the connected device list.

[image]

In the above dfu-csr.quirk file, a new device instance ([[USBVID_1395 & PID_002d]]) has been added.

But when I connect this device, it's not showing in the list.

Is the above dfu-csr.quirk file changes enough to enumerate the device or does any other place need to be changed?

hughsie, 2022-06-01:
> Is the above dfu-csr.quirk file changes enough to enumerate the device

The instance ID is case sensitive, so you want PID_002D

msureshsns, 2022-06-01:
> > Is the above dfu-csr.quirk file changes enough to enumerate the device
> 
> The instance ID is case sensitive, so you want PID_002D

Yeah, I have tried with both PID_002d and PID_002D also, but it is still not enumerating.
I added one more new device instance as well, but even that device was not enumerated.
[image]

hughsie, 2022-06-01:
I assume you did "ninja install"? I think we should move this to a new issue, but the output of `sudo FWUPD_PROBE_VERBOSE=1 ./src/fwupdtool --plugins dfu-csr get-devices --verbose` should help.

msureshsns, 2022-06-01:
> I assume you did "ninja install"? I think we should move this to a new issue, but the output of `sudo FWUPD_PROBE_VERBOSE=1 ./src/fwupdtool --plugins dfu-csr get-devices --verbose` should help.

Thank you for your time.
Do you want me to create another issue for this discussion? 

Not exactly. I have just done "sudo apt-get install fwupd"

Yeah, I have tried this command "sudo fwupdtool --plugins dfu-csr get-devices --verbose". 

It gives the following error: I am new to this tool and am not sure why this error occurs only for these devices.
Currently, two devices are connected, and each device is failing for a different reason. Please see the attached image.
 
[image]