fwupd/firmware-lenovo #24: Thinkpad Carbon X1 Gen6 : Thunderbold upgrade issue with version numbers

beavis69, 2019-11-02:
Hi,

I got an issue on my X1 Gen6 with the following firmware upgrade (thunderbold) : 

```
fwupdmgr get-updates 
No upgrades for System Firmware, current is 0.1.41: 0.1.41=same, 0.1.40=older, 0.1.38=older, 0.1.37=older, 0.1.36=older
No upgrades for UEFI Device Firmware, current is 184.65.3590: 184.65.3590=same, 184.60.3561=older, 184.55.3510=older
No upgrades for UEFI Device Firmware, current is 0.1.17: 0.1.17=same, 0.1.15=older, 0.1.13=older, 0.1.12=older
No releases found for device: Not compatible with bootloader version: failed predicate [BOT01.0[0-3]_* regex BOT01.04_B0016]
No upgrades for Unifying Receiver, current is RQR24.10_B0036: RQR24.10_B0036=same, RQR24.06_B0030=older, RQR24.05_B0029=older
20KHCT01WW
│
└─X1 Carbon Thunderbolt Controller:
  │   Device ID:           0651e78b29431a1275d4742fdb591590bbbf6a69
  │   Summary:             Unmatched performance for high-speed I/O
  │   Current version:     43.00
  │   Vendor:              Lenovo (TBT:0x0109)
  │   Update State:        failed
  │   Update Error:        device version not updated on success, 43.0.0 != 43.00
  │   Flags:               internal|updatable|require-ac|supported|registered
  │   GUID:                2da42a33-cd30-5ef5-a8fb-2c800a4b760f ← TBT-01091706-native
  │ 
  └─ThinkPad X1 Carbon Gen6 Thunderbolt Controller:
        New version:       43.0.0
        Remote ID:         lvfs
        Summary:           Lenovo ThinkPad X1 Carbon Gen6 Thunderbolt Firmware
        License:           proprietary
        Size:              262.1 kB
        Vendor:            Lenovo Ltd.
        Flags:             is-upgrade
        Description:       Lenovo ThinkPad X1 Carbon Gen6 Thunderbolt Firmware
        
         • TBT SPI ROM Wear out issue.
        
         • Before update, set the Thunderbolt BIOS Assist Mode to "Disable" in BIOS setup under Config > Thunderbolt 3. * DO NOT FORCE UPDATE Thunderbolt Controller. This may damage the firmware.
```

Upgrade goes well, (it doesnt need a reboot) : 
```
fwupdmgr update
Downloading 43.0.0 for X1 Carbon Thunderbolt Controller...
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Updating X1 Carbon Thunderbolt Controller…                       ]
Restarting device…       [***************************************]
```

The issue is version_old vs version_new :  43.00 vs 43.0.0
```
        "Reports" : [
                             {
                               "Checksum" : "1e75dfb3e3b2b69c48c601101a4552033b4a775f",
                               "UpdateState" : 3,
                               "UpdateError" : "device version not updated on success, 43.0.0 != 43.00",
                               "Guid" : [
                                 "2da42a33-cd30-5ef5-a8fb-2c800a4b760f"
                               ],
                               "Plugin" : "thunderbolt",
                               "VersionOld" : "43.00",
                               "VersionNew" : "43.0.0",
                               "Flags" : 4194347,
                               "Created" : 1572685038,
                               "Modified" : 0,
                               "Metadata" : {
                                 "CpuArchitecture" : "x86_64",
                                 "RuntimeVersion(org.freedesktop.fwupd)" : "1.3.2",
                                 "DistroId" : "debian",
                                 "CompileVersion(org.freedesktop.fwupd)" : "1.3.2",
                                 "RuntimeVersion(org.kernel)" : "4.19.0-6-amd64",
                                 "CompileVersion(com.redhat.fwupdate)" : "12",
                                 "ThunderboltNative" : "True",
                                 "CompileVersion(org.freedesktop.gusb)" : "0.3.0",
                                 "RuntimeVersion(org.freedesktop.appstream-glib)" : "0.7.14",
                                 "DistroVersion" : "10",
                                 "CompileVersion(com.redhat.efivar)" : "37",
                                 "RuntimeVersion(com.dell.libsmbios)" : "2.4",
                                 "BootTime" : "1572684769",
                                 "RuntimeVersion(com.redhat.fwupdate)" : "12"
                               }
                             }
                           ]
```
Tried with : Debian 10 with fwupd 1.2.5 (debian stable), but same happened with Debian 10 with fwupd 1.3.2 (testing)

gkelly, 2019-11-02:
I'm having the same issue.  I also force-upgraded the controller without disable BIOS assist because nothing told me not to, oops.  I'm adding a warning to https://github.com/fwupd/fwupd/wiki/Thunderbolt%3A-Validation-failed-or-unknown-device/ so others aren't hit by this.

hughsie, 2019-11-05:
Lets continue in #1534

closed 2019-11-05 by hughsie