fwupd/firmware-lenovo #25: Thunderbolt controller critical update for ThinkPad T480/T580

xvitaly, 2019-11-06:
Missing Thunderbolt controller firmware critical update for ThinkPad T480/T580 models: https://pcsupport.lenovo.com/nl/en/solutions/ht508988

Currently users of such hardware need to install Windows in order to apply this update and this is too bad.

xvitaly, 2019-11-06:
Found [T580 update](https://fwupd.org/lvfs/device/com.lenovo.ThinkPadN27TF.firmware), but still no updates for T480.

xvitaly, 2019-11-06:
```T480 Thunderbolt Controller
  DeviceId:             c994c96d4262316c36825e4480dc79e618baa1e7
  Guid:                 7d131734-64ed-5865-a46e-a67cfb9b5a46
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|require-ac|registered
  Vendor:               Lenovo
  VendorId:             TBT:0x0109
  Version:              14.00
  VersionFormat:        pair
  Icon:                 thunderbolt
  Created:              2019-11-06```

joakim-tjernlund, 2019-11-12:
Same for T480s, strange Lenovo has released other T480s FW but this one.

joakim-tjernlund, 2020-01-10:
I think this is resolved, my T480s has:
Thunderbolt Controller: current version is 20.00: 20.00=same

xvitaly, 2020-01-10:
T480s != T480. T480s, T580 updated, but T480 - don't.

peculiarity, 2020-01-23:
What's the status of this issue?FW doesn't see any new versions and I'm still on 14.00 ( T480 here )?

xvitaly, 2020-01-23:
@hughsie Why you closed this issue? fwupd still has 14.00 version for T480.

hughsie, 2020-01-23:
> Why you closed this issue

Because @joakim-tjernlund said it was fixed...

xvitaly, 2020-01-23:
> Because @joakim-tjernlund said it was fixed

He has T480s, not T480. T480 still has buggy 14.00 version.

varnie, 2020-01-23:
T480 user here. I have 12.02 version and no updates available.

xvitaly, 2020-01-24:
> T480 user here. I have 12.02 version and no updates available.

No updates available because v20.00 still not uploaded to fwupd.

markun, 2020-02-01:
@hughsie can you please reopen as this ticket is not resolved for T480? (@joakim-tjernlund was talking about T480**S** as @xvitaly pointed out).

The new thunderbolt firmware fixes a very serious issue:

See: https://www.notebookcheck.net/Lenovo-statement-Thunderbolt-firmware-responsible-for-ThinkPad-USB-C-failures.451307.0.html

"Fix Thunderbolt SPI ROM Wear out issue"
(T480**S** link: https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN22TF.firmware)

I'm wondering if the thunderbolt controllers are the same, otherwise we can maybe manually edit the cab file and manually install it with `fwupdmgr`.

hughsie, 2020-02-02:
> otherwise we can maybe manually edit the cab

I would advise against not doing this.

kflak, 2020-02-15:
Any news on this one? I've got a T480s and my Thunderbolt controller is stuck on v 14.00 and I'm not so keen on going down the Windows to Go hole...

forestEagle, 2020-02-16:
Been holding off on this one since I don't have any TB appliances, but I just read that the chip itself wears down to dust and breaks no matter what - so I figured it'd be a good idea.
I investigated a little regarding the firmware.bin-file; the T480s one is not identical to the .bin-file from the "n24th08w" fw update available to T480.
Creating a cab with the firmware-packager might be possible with the latter I suppose - but I didn't dare follow through and went with a temporary Windows install instead.
 Upgrading on Windows was a HUGE hassle and I almost wish I'd just taken the risk with fwupdmgr on my Arch setup...

kflak, 2020-02-16:
Yeah, I finally caved and did the update in Windows. Luckily I kept the original ssd lying around, so it was a simple matter of swapping drives, going through a few hours of system updates, manually downloading and installing the tb fix, as it wasn’t included in Lenovo vantage.  This is why I use Linux...

markun, 2020-02-16:
I compared the two firmware images (the T480s one from lvfs and the T480 one from the windows tool extracted with wine). With hexcompare I see they are largely identical. Obvious changes are some strings like "Thinkpad T480s" -> "T480" and "N22TF16W" -> "N24TF16W" but there are also a few differences in the binary data.

I was stubborn and created a .cab file based on the T480s cab file with the image from the windows update tool and editing firmware.metainfo.xml. I tried to flash it, but I get:

```
# sudo fwupdmgr install ./t480update.cab
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Installing on Thunderbolt Controller…                            ]
Restarting device…       [***************************************]
could not validate firmware: FW image image not compatible to this controller (Native). See https://github.com/fwupd/fwupd/wiki/Thunderbolt:-Validation-failed-or-unknown-device for more information.
```

I don't want to force the update, so giving up for now :)

hughsie, 2020-02-16:
> I don't want to force the update

Really don't do that please!

markun, 2020-02-26:
@hughsie I manged to do the update!

I had to switch off bios assist mode (and then update the device ID in firmware.metainfo.xml)

```
# sudo fwupdmgr install ./t480update.cab
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Installing on Thunderbolt Controller…                            ]
Restarting device…       [***************************************]
Successfully installed firmware
```

```
fwupdmgr get-devices                  
20L5CTO1WW
│
├─Thunderbolt Controller:
│     Device ID:           c23d991ae57a8cff5b5cc242a8674002c181613a
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     20.00
│     Vendor:              Lenovo (TBT:0x0109)
│     GUIDs:               fe328c2d-5300-597b-a21d-991116db8ab6
│                          88848ff6-38a2-52d3-8aa4-d4f1a3491d00
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Device stages updates
...
```

If you want I can send you the cab file I created.

basicinside, 2020-02-26:
@markun Can you share the cab with me?

crbble, 2020-02-26:
It seems that Lenovo released a new N24TF18W firmware for the T480. That contains a AssistTBT.bin and a TBT.bin with the difference that AssistTBT.bin contains the string N24TF19W and TBT containing N24TF18W, did anyone try to flash one of those?

basicinside, 2020-02-26:
The AssistTBT.bin always has a higher release than the TBT.bin.
n24th08w contains N24TF16W (TBT.bin) and N24TF17W (AssistTBT.bin)
n24th09w contains N24TF18W (TBT.bin) and N24TF19W (AssistTBT.bin)

I flashed N24TF16W and N24TF18W successfully. One after the other.
Both are exposed with Version 20.00.

crbble, 2020-02-26:
OK, I managed to update, too. Thanks for the previous hints.

```
─Thunderbolt Controller:
│     Device ID:           5d58c012bdcb14c78eaa255b8702249f393b11d7
│     Zusammenfassung:     Unmatched performance for high-speed I/O
│     Current version:     20.00
│     Vendor:              Lenovo (TBT:0x0109)
│     GUIDs:               fe328c2d-5300-597b-a21d-991116db8ab6
│                          88848ff6-38a2-52d3-8aa4-d4f1a3491d00
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Device stages updates
```
Functionality checked with eGPU. Any side effects?

**HERE BE DRAGONS, YOU ARE ON YOUR OWN AND THERE IS NO WARRANTY:**

Some more notes during the update:
- disable TB Assist mode in BIOS (though AssistTBT.bin _may_ be the right firmware for you)
- download n24th09w.exe
- you can extract this file with _innoextract_
- as mentioned you are looking for TBT.bin
- download the LVFS Update file for T480s as mentioned before as a template
- this is a cab file, use something like _cabextract_ to extract it
- extract the update file, rename TBT.bin to firmware.bin, remove the signature firmware.bin.asc
- change firmware.metainfo.xml line 6 to contain the GUID of the TB3 controller (Should be fe328c2d-5300-597b-a21d-991116db8ab6)
- repack the cab file with _lcab_
- install the resulting file with _fwupdmgr_
- reboot for good measure

xvitaly, 2020-02-27:
You don't even need to create CAB files. You can flash firmware bin file directly into controller on your own risk:

```sudo /usr/libexec/fwupd/fwupdate -a XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXX firmware.bin

xvitaly, 2020-02-27:
You don't even need to create CAB files. You can flash firmware bin file directly into controller on your own risk:

```bash
sudo /usr/libexec/fwupd/fwupdate -a XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXX TBT.bin
```

joakim-tjernlund, 2020-02-27:
Does innoextract work for Lenovo Dock FW too?

xvitaly, 2020-03-03:
Thunderbolt update for T480 is now available on LVFS: https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN24TF.firmware (in testing).

yann-soubeyrand, 2020-04-19:
Hi @xvitaly,

> Thunderbolt update for T480 is now available on LVFS: https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN24TF.firmware (in testing).

How can I apply it? I activated the testing LVFS repository, did a `fwupdmgr refresh` but `fwupdmgr update` still says that `Thunderbolt Controller has no available firmware updates`. I'm however still on an old version:

```
├─Thunderbolt Controller:
│     Device ID:           a8a720ab1893ac015af9e4220022c4853fca5539
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     14.00
│     Vendor:              Lenovo (TBT:0x0109)
│     GUIDs:               7d131734-64ed-5865-a46e-a67cfb9b5a46
│                          7c99fbc4-39cb-5eb2-a0b4-274db5472295
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Device stages updates
```

yann-soubeyrand, 2020-04-19:
Answering to myself: disabling Bios assist as advised above is the solution ;-)

closed 2020-04-20 by xvitaly

alex-ie, 2021-12-16:
@basicinside 
> The AssistTBT.bin always has a higher release than the TBT.bin. n24th08w contains N24TF16W (TBT.bin) and N24TF17W (AssistTBT.bin) n24th09w contains N24TF18W (TBT.bin) and N24TF19W (AssistTBT.bin)

What is the purpose of AssistTBT.bin? Is it not for assist mode of thunderbolt in UEFI setup? Why for Linux I see everywhere to use only TBT.bin?

Thank you.