fwupd/firmware-lenovo #26: lost thunderbolt controller after upgrade

Macbet, 2019-11-06:
To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
**client version:	1.3.3
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.3.3
**
```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

```shell
$ fwupdmgr get-devices
**20KH006MRT
│
├─SAMSUNG MZVLB1T0HALR-000L7:
│     Device ID:           04e17fcf7d3de91da49a163ffe4907855c3648be
│     Summary:             NVM Express Solid State Drive
│     Current version:     5L2QEXA7
│     Vendor:              Samsung Electronics Co Ltd (NVME:0x144D)
│     GUIDs:               47335265-a509-51f7-841e-1c94911af66b
│                          08575ca8-c22c-540a-8ac9-e784c04f6e5a
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─System Firmware:
│     Device ID:           1c53551e7da69d896138fac1ae131c83ad46d923
│     Current version:     0.1.41
│     Minimum Version:     0.0.1
│     Vendor:              LENOVO
│     GUID:                a4b51dca-8f97-4310-8821-3330f83c9135
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           29749712e93b58730ecd1b8537a5503ff5b7fb27
│     Current version:     184.65.3590
│     Minimum Version:     0.0.1
│     GUID:                9c9d9769-32fa-4841-b550-ea998e754e99
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           9698faabddf0d7b18925cfbbda95f8b0d0dacc53
│     Current version:     0.1.17
│     Minimum Version:     0.1.17
│     GUID:                3babca5f-b2bf-4f4b-a72e-2bdc84eb4019
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
└─Unifying Receiver:
      Device ID:           fb1a5fd6e7d8cbf5d5a2ba7df68aee106ce41027
      Summary:             A miniaturised USB wireless receiver
      Current version:     RQR12.10_B0032
      Bootloader Version:  BOT01.02_B0015
      Vendor:              Logitech, Inc. (USB:0x046D)
      Install Duration:    30 seconds
      GUIDs:               9d131a0c-a606-580f-8eda-80587250b8d6
                           279ed287-3607-549e-bacc-f873bb9838c4
      Device Flags:        • Updatable
                           • Supported on remote server
**
```

```shell
$ efibootmgr -v
**INSERT OUTPUT HERE**
**BootCurrent: 0000
Timeout: 2 seconds
BootOrder: 0000,0003,0034,002E,002F,0030,0031,0032,0033,0027,0028,002D,0004
Boot0000* archlinux	HD(1,GPT,99b65f67-7006-c147-bcf4-b28a8f040837,0x1000,0x96000)/File(\EFI\archlinux\grubx64.efi)
Boot0003* EndeavourOS	HD(1,GPT,99b65f67-7006-c147-bcf4-b28a8f040837,0x1000,0x96000)/File(\EFI\EndeavourOS\grubx64.efi)
Boot0004* Linux-Firmware-Updater	HD(1,GPT,99b65f67-7006-c147-bcf4-b28a8f040837,0x1000,0x96000)/File(\EFI\endeavouros\fwupdx64.efi)
Boot000A  Setup	FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
Boot000B  Boot Menu	FvFile(126a762d-5758-4fca-8531-201a7f57f850)
Boot000C  Diagnostic Splash Screen	FvFile(a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380)
Boot000D  Lenovo Diagnostics	FvFile(3f7e615b-0d45-4f80-88dc-26b234958560)
Boot000E  Regulatory Information	FvFile(478c92a0-2622-42b7-a65d-5894169e4d24)
Boot000F  Startup Interrupt Menu	FvFile(f46ee6f4-4785-43a3-923d-7f786c3c8479)
Boot0025  Rescue and Recovery	FvFile(665d3f60-ad3e-4cad-8e26-db46eee9f1b5)
Boot0026  MEBx Hot Key	FvFile(ac6fd56a-3d41-4efd-a1b9-870293811a28)
Boot0027  Other CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a35406)
Boot0028  Other HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f606)
Boot0029* USBR BOOT CDROM	PciRoot(0x0)/Pci(0x14,0x0)/USB(11,1)
Boot002A* USBR BOOT Floppy	PciRoot(0x0)/Pci(0x14,0x0)/USB(11,0)
Boot002B* ATA HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f6)
Boot002C* ATAPI CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a354)
Boot002D* PCI LAN	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
Boot002E* USB CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,86701296aa5a7848b66cd49dd3ba6a55)
Boot002F* USB FDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,6ff015a28830b543a8b8641009461e49)
Boot0030* NVMe0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a400)
Boot0031* ATA HDD0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f601)
Boot0032* USB HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,33e821aaaf33bc4789bd419f88c50803)
Boot0033* PCI LAN	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
Boot0034* Linux Boot Manager	HD(1,GPT,5133c104-112f-b443-b52f-5f58071c44bb,0x800,0x1dd000)/File(\EFI\systemd\systemd-bootx64.efi)
**
```

```shell
$ efivar -l | grep fw
**INSERT OUTPUT HERE**
**This is only required if you use the UEFI plugin**
```

```shell
$ tree /boot
**INSERT OUTPUT HERE**
**This is only required if you use the UEFI plugin**
**We're looking for any `.cap` files and the location of `fwupx64.efi`**
```

Please answer the following questions:

- Operating system and version: archlinux
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): pacman 
- Have you tried rebooting? - yes 
- Are you using an NVMe disk? - yes
- Is secure boot enabled (only for the UEFI plugin)?

yesterday came the update for thunderbolt controller, before installing it turned off Thunderbolt BIOS Assist Mode as indicated in the description, after the update was gone thunderbolt controller from the list of devices, tell me it immediately contact lenovo support or have a chance to return the controller to work without them ?

fwupdmgr get-history - 
├─Thunderbolt Controller:
│ │   Device ID:           9de9f2423b63f31e5a968d3781ecffb4b4ba6bca
│ │   Previous version:    43.00
│ │   Update State:        failed
│ │   Update Error:        device version not updated on success, 43.0.0 != 43.00
│ │   GUID:                2da42a33-cd30-5ef5-a8fb-2c800a4b760f
│ │   Device Flags:        • Internal device
│ │                        • Updatable
│ │                        • Requires AC power
│ │                        • Supported on remote server
│ │                        • Reported to remote server
│ │                        • Device stages updates
│ │ 
│ └─  New version:       43.0.0
│       Remote ID:         lvfs
│       License:           Unknown

LinuxOnTheDesktop, 2019-11-08:
I too get the same update - or is it updates? (I find it hard to tell from I am seeing) over and over again. What to do?

I attach the fwupdmgr output and details of my system.

[My system.txt](https://github.com/fwupd/missing-firmware-lenovo-thinkpad/files/3824922/My.system.txt)
[fwupdmgr output.txt](https://github.com/fwupd/missing-firmware-lenovo-thinkpad/files/3824924/fwupdmgr.output.txt)

LinuxOnTheDesktop, 2020-01-26:
Is this bug fixed, please? I'm seeing a lot of confusing and indeed sometimes downright contradictory information?

I can find [this LVFS](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN2HTF.firmware) page for the update and that page pertains to the 7th Generation X1. I can't find a corresponding page for the 6th Generation model.

tazjin, 2021-03-24:
> If they only support the native GUID, only include the native GUID. You're going to potentially brick controllers left and right if you allow people to flash on the non-native GUID.

I suppose unbricking these isn't possible? For me this controller completely disappeared, presumably after the Thunderbolt update which happened months (but I only just noticed it). BIOS Assist was enabled, so it might have flashed to the non-native one as you suggested ...