fwupd/firmware-lenovo #29: Missing Thunderbolt firmware for T480 (Type 20L5, 20L6) Laptop (ThinkPad)

dash17291, 2020-01-02:
For missing firmware please provide the following items:

* Link to firmware update at OEM website:
https://download.lenovo.com/pccbbs/mobiles/n24th08w.exe

* Firmware release date (if available on OEM website):
25 Jul 2019

* Link to the device page on LVFS:

ohazi, 2020-01-23:
Link to firmware update at OEM website:
https://support.lenovo.com/ca/en/downloads/ds502613
https://download.lenovo.com/pccbbs/mobiles/n24th08w.exe

Firmware release date:
24 Jul 2019

Link to the device page on LVFS:
There aren't any Thunderbolt firmware images for the T480 on LVFS.

ohazi, 2020-01-23:
There was another issue for this that probably shouldn't have been closed: https://github.com/fwupd/missing-firmware-lenovo-thinkpad/issues/25

ohazi, 2020-01-23:
Aside:

How does getting firmware images added to LVFS actually work? Does somebody at Lenovo have to do it, or is this something that users can help with? The fwupd.org login page seems to suggest that this is really meant to be done by the hardware vendors. If that's the case, are these issues just for tracking? Is there *anything* that the concerned public can help with?

xvitaly, 2020-01-24:
Please post message in [this thread](https://forums.lenovo.com/t5/Other-Linux-Discussions/Thunderbolt-Software-and-Firmware-updates-and-Linux/td-p/4552148/). I already did it 3 times ([1](https://forums.lenovo.com/t5/Other-Linux-Discussions/Thunderbolt-Software-and-Firmware-updates-and-Linux/m-p/4578798/highlight/true#M13982), [2](https://forums.lenovo.com/t5/Other-Linux-Discussions/Thunderbolt-Software-and-Firmware-updates-and-Linux/m-p/4626711/highlight/true#M14252), [3](https://forums.lenovo.com/t5/Other-Linux-Discussions/Thunderbolt-Software-and-Firmware-updates-and-Linux/m-p/4629609/highlight/true#M14259)), but got generic answers:
> I don't have permission to share the planned date.
> No ETA, sorry.

bgermann, 2020-01-24:
Does anyone know how to generate a UEFI capsule from the .exe content?