fwupd/firmware-lenovo #31: X390 fails to install ThinkPad T490s Corporate ME Update

foshaug, 2020-01-24:
My system is unable to install the "ThinkPad T490s Corporate ME Update". 
Every time I try to install it the system offers the same update again. 

```shell
$ fwupdmgr get-updates
UEFI Device Firmware has firmware updates:
GUID:                    aeac19da-0c78-4b86-a718-1a043947d83b
ID:                      com.lenovo.ThinkPadN2JRM.firmware
Update Version:          192.47.1524
Update Name:             ThinkPad T490s Corporate ME Update
Update Summary:          Lenovo ThinkPad T490s Corporate ME Firmware
Update Remote ID:        lvfs
Update Checksum:         SHA1(afa2196c27376214d475213bfec45570aa260ab3)
Update Location:         https://fwupd.org/downloads/b2c442db905872d9d4725a038b29450edaea8361-Lenovo-ThinkPad-T490s-CorporateMEFirmware-12.0.47.1524.cab
Update Description:       • 0 Q2'19 Intel Platform Update (Hot Fix Release)
                         
                         Version 12.0.47.1524 (LVFS: 192.47.1524)
                         
                         Problem Fixes
                          • Intel CSME PSIRT-TA-201905-011 Q2 2019 IPU:
                          • Mitigated the following security vulnerabilities under issues:
                         
                         Security issues fixed:
                          • CVE-2019-0086
                          • CVE-2019-0090
                          • CVE-2019-0092
                          • CVE-2019-0093
                          • CVE-2019-0094
                          • CVE-2019-0096
                          • CVE-2019-0097
                          • CVE-2019-0098
                          • CVE-2019-0153
                          • CVE-2019-0170
No upgrades for UEFI Device Firmware, current is 0.1.15: 0.1.15=same, 0.1.12=older
No upgrades for 20Q0CTO1WW System Firmware, current is 0.1.51: 0.1.51=same
```

[image]
[image]

To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version:	1.0.9
daemon version:	1.0.9
compile-time dependency versions
	appstream-glib:	0.7.7
	gusb:	0.2.11
	fwupdate:	10
	efivar:	34
```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

```shell
$ fwupdmgr get-devices
X390/T490s Thunderbolt Controller
  DeviceId:             d13dc094e8febb10211a5395d0961b529c340495
  Guid:                 e773c51e-a20c-5b29-9f09-6bb0e0ef7560
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|registered
  Vendor:               Lenovo
  VendorId:             TBT:0x0109
  Version:              18.00
  Icon:                 computer
[... 381 lines cut ...]
│       ├── xzio.mod
│       ├── zfscrypt.mod
│       ├── zfsinfo.mod
│       └── zfs.mod
├── initrd.img-5.0.0-37-generic
├── initrd.img-5.3.0-26-generic
├── lost+found
├── memtest86+.bin
├── memtest86+.elf
├── memtest86+_multiboot.bin
├── System.map-5.0.0-37-generic
├── System.map-5.3.0-26-generic
├── vmlinuz-5.0.0-37-generic
└── vmlinuz-5.3.0-26-generic

10 directories, 297 files
```

Please answer the following questions:

- Operating system and version: Ubuntu 18.04.3 LTS
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): Came installed
- Have you tried rebooting? Yes
- Are you using an NVMe disk? Not sure - Yes
- Is secure boot enabled (only for the UEFI plugin)? Yes

neuged, 2020-01-25:
@superm1 Could you maybe explain why this was transferred to "missing-firmware-..."? The firmware doesn't seem to be missing in this case or am I missing something?

superm1, 2020-01-25:
This repo tracks issues with the firmware not the open source software. Lenovo should be monitoring it too.

gustafla, 2020-01-27:
I also have this problem on the T490 running Arch Linux.

`client version:	1.3.6
compile-time dependency versions
	gusb:	0.3.1
	efivar:	37
daemon version:	1.3.6`

The installer runs successfully from a reboot, the machine reboots again and the Lenovo logo flashes with "firmware upgrade" or something like that, but after it's done and back to Linux, fwupdmgr says I'm still on 192.35.1427.

`20N3S2NJ00
│
└─UEFI Device Firmware:
  │   Device ID:           b1c32d855d85a2124e865f0a03f8c2384dcca167
  │   Current version:     192.35.1427
  │   Minimum Version:     0.0.1
  │   Vendor:              DMI:LENOVO
  │   Update State:        success
  │   GUID:                bf2d0f8b-f9a9-400c-8914-36c225d16eb4
  │   Device Flags:        • Internal device
  │                        • Updatable
  │                        • Requires AC power
  │                        • Supported on remote server
  │                        • Needs a reboot after installation
  │                        • Device is usable for the duration of the update
  │
  └─ThinkPad T490/ThinkPad P43s/ThinkPad T590/ThinkPad P53s Consumer ME Update:
        New version:       192.47.1524
        Remote ID:         lvfs
        Summary:           Lenovo ThinkPad T490/P43s/T590/P53s Consumer ME Firmware
        License:           Proprietary
        Size:              2.7 MB
        Vendor:            Lenovo Ltd.
        Flags:             is-upgrade
        Description:       • 0 Q2'19 Intel Platform Update (Hot Fix Release)

        Version 12.0.47.1524 (LVFS: 192.47.1524)

        Problem Fixes

         • Intel CSME PSIRT-TA-201905-011 Q2 2019 IPU:
         • Mitigated the following security vulnerabilities under issues:

        Security issues fixed:

         • CVE-2019-0086
         • CVE-2019-0090
         • CVE-2019-0092
         • CVE-2019-0093
         • CVE-2019-0094
         • CVE-2019-0096
         • CVE-2019-0097
         • CVE-2019-0098
         • CVE-2019-0153
         • CVE-2019-0170
`

alextrosta, 2020-02-06:
Same issue here on Ubuntu 18.04.4 LTS using a 20Q0 ThinkPad X390

gregorriegler, 2020-02-10:
Same. Ubuntu 19.10 T490s

tompreston, 2020-02-15:
Same issue here. Fedora 31, x390.

I had two firmware updates to begin with, but I was able to install the first with the [BIOS update (Utility & Bootable CD)](https://pcsupport.lenovo.com/gb/en/products/laptops-and-netbooks/thinkpad-x-series-laptops/thinkpad-x390/20q0/20q0003vuk/pc1a0yg4/downloads/driver-list/component?name=BIOS%2FUEFI).

When I reboot to install the update I see the following messages and it hangs for 5+ minutes.
```
Found update fwupd-ux-capsule
Found update fwupd-ef5cdc85-9cf6-469d-9cb7-920b7dd6672b-0
Adding new capsule
Adding new capsule
```

I can turn off by holding the power button for two seconds, and it boots into Fedora without errors. `fwupdmgr` still reports an update is available for "ThinkPad T490s Consumer ME Update":
```
$ fwupdmgr get-updates
No upgrades for System Firmware: current version is 0.1.62: 0.1.51=older
No upgrades for UEFI Device Firmware: current version is 0.1.17: 0.1.15=older, 0.1.12=older
20Q0003VUK
│
└─UEFI Device Firmware:
  │   Device ID:           07edea1cf12d20814225b44de22b33758db1a1c8
  │   Current version:     192.40.1433
  │   Minimum Version:     0.0.1
  │   Vendor:              DMI:LENOVO
  │   Update State:        success
  │   GUID:                ef5cdc85-9cf6-469d-9cb7-920b7dd6672b
  │   Device Flags:        • Internal device
  │                        • Updatable
  │                        • Requires AC power
  │                        • Supported on remote server
  │                        • Needs a reboot after installation
  │                        • Device is usable for the duration of the update
  │
  └─ThinkPad T490s Consumer ME Update:
        New version:       192.47.1524
        Remote ID:         lvfs
        Summary:           Lenovo ThinkPad T490s Consumer ME Firmware
        Licence:           Proprietary
        Size:              2.7 MB
        Vendor:            Lenovo Ltd.
        Flags:             is-upgrade
        Description:       • 0 Q2'19 Intel Platform Update (Hot Fix Release)

        Version 12.0.47.1524 (LVFS: 192.47.1524)

        Problem Fixes

         • Intel CSME PSIRT-TA-201905-011 Q2 2019 IPU:
         • Mitigated the following security vulnerabilities under issues:

        Security issues fixed:

         • CVE-2019-0086
         • CVE-2019-0090
         • CVE-2019-0092
         • CVE-2019-0093
         • CVE-2019-0094
         • CVE-2019-0096
         • CVE-2019-0097
         • CVE-2019-0098
         • CVE-2019-0153
         • CVE-2019-0170
```

sdague, 2020-02-17:
Same issue with Fedora 31.

stefanb2, 2020-02-20:
Same issue with T490s, Fedora 31, SecureBoot disabled in BIOS:

````
20NX000EMX
...
  │   Current version:     192.40.1433
...
  └─ThinkPad T490s Consumer ME Update:
        New version:       192.47.1524
        Remote ID:         lvfs
        Summary:           Lenovo ThinkPad T490s Consumer ME Firmware
...
        Description:       • 0 Q2'19 Intel Platform Update (Hot Fix Release)
        
        Version 12.0.47.1524 (LVFS: 192.47.1524)
````

I've tried several times to install this update over the past few months and always the same result:
* `fwupdmgr` downloads & prepares the update
* reboot
* update package is detected & starts up
* Lenovo Logo with `Installing Firmware Update` text shows up on the screen (for a fraction of a second), no progress bar is shown
* laptop immediately resets
* laptop starts normal boot procedure into OS

IMHO this is not a `fwupdmgr` issue but a faulty/corrupted upgrade package, i.e. the upgrade code itself crashes. As this seems to affect several different Lenovo update packages that try to install the same `Q2'19 Intel Platform Update (Hot Fix Release)`, the problem may originate with the Intel ME update itself.

wienerneustadt, 2020-03-01:
X390: exactly the same problem. OS: Ubuntu 18.04.4 LTS:
download -- update -- reboot -- update package detected -- Firmware update is announced to be starting -- "Press any key to continue" -- reset -- boot procedure

I called Lenovo Tech Support, hardware dept (still have warranty). They did not seem to understand the problem at all and claimed there was no Linux support offered. Still, I believe it is a hardware problem and, consequently an issue they should deal with.

Sorry for my non-tech approach. I am just an interested user and overjoyed to find someone who reports the same problems. At Lenovo they acted as if I had been the first one.

dannypaz, 2020-03-07:
Just wanted to chime in and say Ubuntu 18.04.4 LTS on T490s

53RT, 2020-03-10:
> 
> 
> Just wanted to chime in and say Ubuntu 18.04.4 LTS on T490s

Same setup and problem here

mayk78, 2020-03-11:
I think i suffer the same issue.
When receiving the laptop i managed to install a fw update, but the next one i just can't get it to install (stick). 
When i install it on my t490s , i can see the capsule files in /boot/efi/EFI/ubuntu/fw . Before pressing update the directory is empty.  efibootmgr -v tells me boot next is linux-firmware-updater , but i can't get it to install anything..   Read somewhere to turn off secureboot , that did't help.  Booting with F12 to select bootdevice i select linux-firmware-updater, nothing happens.  Every boot the updatemanager nags me about the same update..  
Anyone any clue ?

PrinceWang2018, 2020-03-17:
Just wanted to chime in and say Ubuntu 18.04.4 LTS on T490s

haeringer, 2020-03-19:
I think the problem is solved by updating the system BIOS.

I have updated the T490s to Package **1.62** [[1.62 (N2JET84W) + 1.17 (N2JHT33W)]](https://support.lenovo.com/de/en/downloads/DS539062) and the problematic FW update 192.47.1524 seems to have been installed fine afterwards:

```
~$ fwupdmgr get-updates
No upgrades for 20NX003CGE System Firmware, current is 0.1.62: 0.1.51=older
No upgrades for UEFI Device Firmware, current is 192.47.1524: 192.47.1524=same, 192.35.1427=older
No upgrades for UEFI Device Firmware, current is 0.1.17: 0.1.15=older, 0.1.12=older
________________________________________________

Devices that have been updated successfully:

 • UEFI Device Firmware (192.40.1433 → 192.47.1524)
```

I can not guarantee that this is the solution, but I haven't done any other changes to the system and the correlation seems obvious. Probably the package has some undocumented dependencies to the BIOS version.

53RT, 2020-03-19:
I have updated the BIOS from 'N2JET77W (1.55)' to `N2JET85W (1.63)` but still have the same problems to install the `192.47.1524` FW update. 

PS: After updating the BIOS the my T490s restarted ~ 5 times for 3 seconds and then continued with a Self-Healing BIOS Backup. Was a bit scared if the update bricked my laptop

haeringer, 2020-03-19:
> I have updated the BIOS from 'N2JET77W (1.55)' to N2JET85W (1.63) but still have the same problems to install the 192.47.1524 FW update.

Did you also install the Embedded Controller (1.17, N2JHT33W) part of the package?

metze-samba, 2020-03-19:
I also did the upgrade using n2jur23w.iso. That updated both on my x390 20Q0
from 'N2JET73W (1.51)' to N2JET85W (1.63) and N2JHT31W (1.15) to 
N2JHT33W (1.17). The Bios says ME Firmware Version is still 12.0.40.1433
(before and after).

After 'fwupdmgr update' and a reboot (without any update happening) I'm still getting:
~# fwupdmgr get-updates
No upgrades for X390/T490s Thunderbolt Controller, current is 20.00: 20.00=same
No upgrades for 20Q0S0HM00 System Firmware, current is 0.1.63: 0.1.51=older
UEFI Device Firmware has firmware updates:
Device ID:               943ac8e39266b524856ff8e415f05b6f2bcbb148
GUID:                    aeac19da-0c78-4b86-a718-1a043947d83b
ID:                      com.lenovo.ThinkPadN2JRM.firmware
Update Version:          192.47.1524
Update Name:             ThinkPad T490s Corporate ME Update Update
Update Summary:          Lenovo ThinkPad T490s Corporate ME Firmware
Update Remote ID:        lvfs
Update Checksum:         SHA1(afa2196c27376214d475213bfec45570aa260ab3)
Update Location:         https://fwupd.org/downloads/b2c442db905872d9d4725a038b29450edaea8361-Lenovo-ThinkPad-T490s-CorporateMEFirmware-12.0.47.1524.cab
Update Description:       • 0 Q2'19 Intel Platform Update (Hot Fix Release)
                         
                         Version 12.0.47.1524 (LVFS: 192.47.1524)
                         
                         Problem Fixes
                         
                          • Intel CSME PSIRT-TA-201905-011 Q2 2019 IPU:
                          • Mitigated the following security vulnerabilities under issues:
                         
                         Security issues fixed:
                         
                          • CVE-2019-0086
                          • CVE-2019-0090
                          • CVE-2019-0092
                          • CVE-2019-0093
                          • CVE-2019-0094
                          • CVE-2019-0096
                          • CVE-2019-0097
                          • CVE-2019-0098
                          • CVE-2019-0153
                          • CVE-2019-0170
                         
No upgrades for UEFI Device Firmware, current is 0.1.17: 0.1.15=older, 0.1.12=older

metze-samba, 2020-03-19:
https://pcsupport.lenovo.com/de/de/products/laptops-and-netbooks/thinkpad-x-series-laptops/thinkpad-x390/20q0/downloads/DS539850
says there's already 12.0.49.1534 (07 Jan 2020), maybe that would work, but there's only
n2jrg24w.exe and no .iso.

53RT, 2020-03-19:
> > I have updated the BIOS from 'N2JET77W (1.55)' to N2JET85W (1.63) but still have the same problems to install the 192.47.1524 FW update.
> 
> Did you also install the Embedded Controller (1.17, N2JHT33W) part of the package?

Thanks @haeringer for reminding me of the missing part of the installation.

No I didn't but after updating this part too I experience the same problems as @metze-samba has.

metze-samba, 2020-03-19:
I run n2jrg24w.exe within a Windows VM in order to extract it:

  FwDetect.exe                        A   178200  Fri Feb 19 18:15:12 2016
  FWUpdate.exe                        A  2032600  Thu Feb 28 15:17:08 2019
  FWUpdLcl64.exe                      A   855528  Tue Nov 12 14:26:16 2019
  Idrvdll32e.dll                      A    89576  Tue Nov 12 14:25:06 2019
  MEInfoWin64.exe                     A  1016808  Tue Nov 12 14:23:38 2019
  MEManufWin64.exe                    A  1307112  Tue Nov 12 14:24:20 2019
  MEUpdate.CMD                        A      590  Mon Apr  8 11:12:44 2019
  ME_12.0_Cons.bin                    A  2585504  Tue Dec 10 16:50:22 2019
  ME_12.0_Corp.bin                    A 12039072  Tue Dec 10 16:49:14 2019
  Pmxdll32e.dll                       A   124392  Tue Nov 12 14:25:12 2019
  SLA_TOOLS.PDF                       A    80469  Tue Nov 12 14:53:54 2019

And b2c442db905872d9d4725a038b29450edaea8361-Lenovo-ThinkPad-T490s-CorporateMEFirmware-12.0.47.1524.cab:

12.0.47.1524.cab.d$ ls -l
insgesamt 11924
-rw-rw---- 1 metze metze 12193472 Mär 19 14:08 firmware.bin
-rw-rw---- 1 metze metze      490 Mär 19 14:08 firmware.bin.asc
-rw-rw---- 1 metze metze     1956 Mär 19 14:08 firmware.metainfo.xml
-rw-rw---- 1 metze metze      924 Mär 19 14:08 README.txt

Just looking at the sizes of firmware.bin, ME_12.0_Corp.bin and ME_12.0_Cons.bin.
The first two are similar, but ME_12.0_Cons.bin is much smaller.

Is there a way to find out if I need the consumer firmware of the corporation version?
Maybe that's the problem here...

haeringer, 2020-03-19:
Hm, ok, then I don't have a clue what made the installation of the 192.47.1524 update work on my machine suddenly :/

foshaug, 2020-03-23:
Updated to 1.63 and 1.15 to see if that would install the 192.47.1524 update, but alas, no such luck for me.

slavoroi, 2020-05-27:
The same for me in Lenovo T480, and ubuntu 20.04, pops up on every day:

[image]

sudo efibootmgr
```
BootCurrent: 0001
Timeout: 0 seconds
BootOrder: 0001,0000,0017,0018,0019,001A,001B,001C,001D,001E,001F,0024,0002
Boot0000* Windows Boot Manager
Boot0001* ubuntu
Boot0002* Linux-Firmware-Updater
Boot0010  Setup
Boot0011  Boot Menu
Boot0012  Diagnostic Splash Screen
Boot0013  Lenovo Diagnostics
Boot0014  Startup Interrupt Menu
Boot0015  Rescue and Recovery
Boot0016  MEBx Hot Key
Boot0017* USB CD
Boot0018* USB FDD
Boot0019* NVMe1
Boot001A* NVMe0
Boot001B* ATA HDD0
Boot001C* USB HDD
Boot001D* PCI LAN
Boot001E  Other CD
Boot001F  Other HDD
Boot0020* USBR BOOT CDROM
Boot0021* USBR BOOT Floppy
Boot0022* ATA HDD
Boot0023* ATAPI CD
Boot0024* PCI LAN
```
fwupdmgr --version 
``` 
client version:	1.3.9
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.3.9
```

fwupdmgr get-updates
```
fwupdmgr get-updates
• Thunderbolt Controller has the latest available firmware version
• System Firmware has the latest available firmware version
• UEFI Device Firmware has the latest available firmware version
• UEFI Device Firmware has no available firmware updates
• UEFI Device Firmware has no available firmware updates
• USB2.0 Hub has no available firmware updates
• USB2.0 Hub has no available firmware updates
• USB3.0 Hub has no available firmware updates
• USB3.0 Hub has no available firmware updates
• Unifying Receiver has the latest available firmware version
• VMM5322 has no available firmware updates
• WDC PC SN720 SDAQNTW-512G-1001 has the latest available firmware version
20L5000AIV
│
└─UEFI Device Firmware:
  │   Device ID:           ea0fd60df22477668e549841f6a22a8c31cae587
  │   Current version:     184.70.3626
  │   Minimum Version:     0.0.1
  │   Vendor:              DMI:LENOVO
  │   Update State:        success
  │   GUID:                9e21f98b-fe98-455c-b388-da5450ab6979
  │   Device Flags:        • Internal device
  │                        • Updatable
  │                        • Requires AC power
  │                        • Supported on remote server
  │                        • Needs a reboot after installation
  │                        • Device is usable for the duration of the update
  │ 
  └─ThinkPad T480 Consumer ME Update:
        New version:       184.77.3664
        Remote ID:         lvfs
        Summary:           Lenovo ThinkPad T480 Consumer ME Firmware
        License:           Proprietary
        Size:              2.2 MB
        Vendor:            Lenovo Ltd.
        Flags:             is-upgrade
        Description:       Version 11.8.77.3664 (LVFS: 184.77.3664)
        
        Problem Fixes
        
         • Mitigated security vulnerabilities
        
        Security issues fixed:
        
         • CVE-2020-0531
         • CVE-2020-0532
         • CVE-2020-0533
         • CVE-2020-0535
         • CVE-2020-0536
         • CVE-2020-0537
         • CVE-2020-0538
         • CVE-2020-0539
         • CVE-2020-0540
         • CVE-2020-0545
         • CVE-2020-0594
         • CVE-2020-0595
         • CVE-2020-0596
         • CVE-2020-8674
```

nmoreaud, 2020-05-29:
Same on Ubuntu 20.04

mberionni, 2020-05-29:
same issue here on ubuntu 18.04 and Thinkpad T480 Consumer Update

shayaviv, 2020-05-30:
Same on Fedora 32 and Lenovo T480

Chuckaluphagus, 2020-06-02:
Also have the problem here, Thinkpad X1 (6th Generation) and Ubuntu 18.04.

dal00, 2020-06-02:
Ubuntu 20.04 Thinkpad X1C6, Can't get this update to stick. Have tried a had full of times.
When rebooting something is happening, but can't tell what since these updates have zero debug output. Then it reboots back into the os with no update performed.

`$ fwupdmgr get-updates
• HP Thunderbolt 3 Dock has no available firmware updates
• Thunderbolt Controller has the latest available firmware version
• KXG5AZNV512G TOSHIBA has the latest available firmware version
• Scarlett 2i2 USB has no available firmware updates
• System Firmware has the latest available firmware version
• UEFI Device Firmware has the latest available firmware version
• VMM3320 has no available firmware updates
20KHCTO1WW
│
└─UEFI Device Firmware:
  │   Device ID:           e563ad307df81c99f0de8c26292afd71cf409673
  │   Current version:     184.70.3626
  │   Minimum Version:     184.70.3626
  │   Vendor:              DMI:LENOVO
  │   GUID:                42a0a96e-c9f3-438f-9687-7826be33e4ce
  │   Device Flags:        • Internal device
  │                        • Updatable
  │                        • Requires AC power
  │                        • Supported on remote server
  │                        • Needs a reboot after installation
  │                        • Device is usable for the duration of the update
  │ 
  └─ThinkPad X1 Carbon 6th Corporate ME Update:
        New version:       184.77.3664
        Remote ID:         lvfs
        Summary:           Lenovo ThinkPad X1 Carbon 6th Corporate ME Firmware
        License:           Proprietary
        Size:              7,5 MB
        Vendor:            Lenovo Ltd.
        Flags:             is-upgrade
        Description:       Version 11.8.77.3664 (LVFS: 184.77.3664)
        
        2020.01 Q1 Intel Platform Update (IPU), formerly known as Quarterly security release (QSR)
        
        Several security fixes and enhancements are on this release.
        
        Security issues fixed:
        
         • CVE-2020-0531
         • CVE-2020-0532
         • CVE-2020-0533
         • CVE-2020-0535
         • CVE-2020-0536
         • CVE-2020-0537
         • CVE-2020-0538
         • CVE-2020-0539
         • CVE-2020-0540
         • CVE-2020-0545
         • CVE-2020-0594
         • CVE-2020-0595
         • CVE-2020-0596
         • CVE-2020-8674
`

nonsite, 2020-06-03:
Same here T480 Ubuntu 18.04

ryangwsimmons, 2020-06-07:
I'm having this problem as well on my T480 running Ubuntu 20.04.

omersi, 2020-06-08:
Same for Ubuntu 20.04 on Lenovo T490s.

pschichtel, 2020-06-11:
Probably the same issue on my thinkpad T480. I'm using secureboot. The updater to boot and find the capsules, but after the reboot the actual updater only flashes quickly and the boots the OS without actually installing the update.

OS: Archlinux
fwupd: 1.4.2

xdavidliu, 2020-06-12:
same exact issue for me; happening on thinkpad x1 carbon 7th gen, same issue both in Debian 10 buster stable and Ubuntu 20.04 LTS.

bullfroggie, 2020-06-15:
Same issue 

**Laptop:** ThinkPad T490s
**OS:**  Pop!_OS (20.04 LTS)

kmauleon, 2020-06-16:
> same exact issue for me; happening on thinkpad x1 carbon 7th gen, same issue both in Debian 10 buster stable and Ubuntu 20.04 LTS.

Hi...
May I know the result of this command.... thank you
fwupdmgr get-devices

Chuckaluphagus, 2020-06-16:
Hello kmauleon.  Here's the output from my laptop:

X1 Carbon Thunderbolt Controller
  DeviceId:             2b1d7ceac45e2eb07f5653bcef80c58c7657b14d
  Guid:                 2da42a33-cd30-5ef5-a8fb-2c800a4b760f
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|require-ac|supported|registered
  Vendor:               Lenovo
  VendorId:             TBT:0x0109
  Version:              43.00
  VersionFormat:        pair
  Icon:                 computer
  Created:              2020-06-12

20KG0022US System Firmware
  DeviceId:             1c53551e7da69d896138fac1ae131c83ad46d923
  Guid:                 a4b51dca-8f97-4310-8821-3330f83c9135
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.48
  VersionLowest:        0.0.1
  VersionFormat:        triplet
  Icon:                 computer
  Created:              2020-06-12

UEFI Device Firmware
  DeviceId:             29749712e93b58730ecd1b8537a5503ff5b7fb27
  Guid:                 9c9d9769-32fa-4841-b550-ea998e754e99
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              184.70.3626
  VersionLowest:        0.0.1
  VersionFormat:        triplet
  Icon:                 audio-card
  Created:              2020-06-12
  UpdateState:          success

UEFI Device Firmware
  DeviceId:             9698faabddf0d7b18925cfbbda95f8b0d0dacc53
  Guid:                 3babca5f-b2bf-4f4b-a72e-2bdc84eb4019
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.20
  VersionLowest:        0.1.20
  VersionFormat:        triplet
  Icon:                 audio-card
  Created:              2020-06-12

UEFI Device Firmware
  DeviceId:             9e329270a7a68d289c82fe77d32d02208ddf0890
  Guid:                 cea87551-1701-43fb-afbc-6e8ce9728345
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              0.73.4
  VersionLowest:        0.73.4
  VersionFormat:        triplet
  Icon:                 audio-card
  Created:              2020-06-12

INTEL SSDPEKKF512G8L
  DeviceId:             04e17fcf7d3de91da49a163ffe4907855c3648be
  Guid:                 79517f86-8df8-5d6e-a18b-33f0b36a78e9
  Guid:                 68db11e5-b0cf-5bc9-a94e-17e28496e505
  Guid:                 87fd9fec-f447-5ec6-a1cd-d55fac0d3d30
  Summary:              NVM Express Solid State Drive
  Plugin:               nvme
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Vendor:               Intel Corporation
  VendorId:             NVME:0x8086
  Version:              L11P
  VersionFormat:        plain
  Icon:                 drive-harddisk
  Created:              2020-06-12

kmauleon, 2020-06-16:
> Guid: 9c9d9769-32fa-4841-b550-ea998e754e99

Hi Chuckaluphagus
Thank you very much for your reply...
Based from Guid: 9c9d9769-32fa-4841-b550-ea998e754e99, this is X1 Carbon 6th not 7th... and issue is ongoing investigation by Lenovo... they already pulled out 184.77.3664 version in the meantime... thank you...

xdavidliu, 2020-06-16:
> > same exact issue for me; happening on thinkpad x1 carbon 7th gen, same issue both in Debian 10 buster stable and Ubuntu 20.04 LTS.
> 
> Hi...
> May I know the result of this command.... thank you
> fwupdmgr get-devices

Sure, [here it is from Ubuntu 20.04](https://pastebin.com/NaSrsJJE). I uninstalled Debian so can't get that

kmauleon, 2020-06-16:
> Sure, [here it is from Ubuntu 20.04](https://pastebin.com/NaSrsJJE). I uninstalled Debian so can't get that

Hi xdavidliu...
Based from GUID 9c9d9769-32fa-4841-b550-ea998e754e99, your system is also X1 Carbon 6th Gen... Let's wait for Lenovo owner's investigation on this... btw, need not install Debian, same command can work in Debian as well...but thank you very much for getting the information....

YegorTrembovetsky, 2020-06-18:
Same issue on Fedora 31/Lenovo T480

`client version:	1.3.9
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.3.9`

Cryt1c, 2020-06-20:
I have the same issue on a T490s with Ubuntu 20.04. Tried to upgrade fwupd - no luck.
Reading through this thread I only saw issue reports. Is someone looking into this issue?

kmauleon, 2020-06-22:
Hi Cryt1c...
Lenovo is investigating these issues... thank you very much for the attention....

kmauleon, 2020-06-22:
> I have the same issue on a T490s with Ubuntu 20.04. Tried to upgrade fwupd - no luck.
> Reading through this thread I only saw issue reports. Is someone looking into this issue?

Hi Cryt1c... sorry may I know the results of this command...thank you very much...
fwupdmgr get-devices

Cryt1c, 2020-06-23:
Hi kmauleon,

thanks for looking into this! Here is the output of the command:

20NYS02A00
│
├─Thunderbolt Controller:
│     Device ID:           8079efd674c6e461704ec24480b0bcccf6d640fa
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     20.00
│     Vendor:              Lenovo (TBT:0x0109)
│     GUIDs:               e773c51e-a20c-5b29-9f09-6bb0e0ef7560
│                          f1793149-6c12-5a7f-95db-9fefd116748d
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Device stages updates
│   
├─Embedded Controller:
│     Device ID:           09ce7908b2d90a3b53473c3f328410590a527f29
│     Current version:     0.1.18
│     Minimum Version:     0.1.18
│     Vendor:              DMI:LENOVO
│     GUID:                be9a5f93-930b-40bf-8864-815268ea4ea3
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─Intel Management Engine:
│     Device ID:           07edea1cf12d20814225b44de22b33758db1a1c8
│     Current version:     192.40.1433
│     Minimum Version:     0.0.1
│     Vendor:              DMI:LENOVO
│     GUID:                ef5cdc85-9cf6-469d-9cb7-920b7dd6672b
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─Prometheus:
│ │   Device ID:           51a5f0521bddd3e9d93388f831b2b5ffdf225d9a
│ │   Summary:             Fingerprint reader
│ │   Current version:     10.01.3158804
│ │   Vendor:              Synaptics (USB:0x06CB)
│ │   Install Duration:    2 seconds
│ │   GUIDs:               09e8bf16-3e69-50f5-bb66-c7a040248352
│ │                        8088f861-6318-5b1e-9ce4-fbddbedb09ac
│ │   Device Flags:        • Updatable
│ │                        • Supported on remote server
│ │                        • Cryptographic hash verification is available
│ │ 
│ └─Prometheus IOTA Config:
│       Device ID:         a717531e28f0288a4e7af638b8e6afc0a1c8743e
│       Current version:   0021
│       Minimum Version:   0021
│       Vendor:            Synaptics (USB:0x06CB)
│       GUID:              7c5a1e62-38fa-5859-9337-09dbac6377e4
│       Device Flags:      • Updatable
│                          • Supported on remote server
│     
├─System Firmware:
│     Device ID:           ca8e899e46f1da1776d16410bc92f2cf0d963433
│     Current version:     0.1.66
│     Vendor:              LENOVO (DMI:LENOVO)
│     GUID:                6300ba52-d9ea-4ca0-88f9-f89ef73cc07b
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           0a6ca0e0099a5836e3dec72fd5c67594f3fd9e2b
│     Current version:     16851095
│     Minimum Version:     1
│     Vendor:              DMI:LENOVO
│     GUID:                89aab014-a71d-4cae-8824-174808266539
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           b028a468cd8bdc225e88bdb99988cd7646b2687f
│     Current version:     4784136
│     Minimum Version:     4784136
│     Vendor:              DMI:LENOVO
│     GUID:                ee246b7d-a882-4e0e-984f-23ac1b0270db
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           06e92059ac0db8bc24337ed23cbd1991c4ca2c2f
│     Current version:     26403
│     Minimum Version:     1
│     Vendor:              DMI:LENOVO
│     GUID:                80c2e1e4-0a6a-42d0-b188-5a218e57c8de
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
└─WDC PC SN730 SDBQNTY-512G-1001:
      Device ID:           f2759da7fe8e0388c5f3601cb072f837b1070b03
      Summary:             NVM Express Solid State Drive
      Current version:     11130101
      Vendor:              Sandisk Corp (NVME:0x15B7)
      GUIDs:               a39943dd-3afb-54f8-b110-c5a21f071200
                           fccbb6ea-e20e-58ad-bf8a-7fb7d43ff4c2
                           06b4e2aa-91af-508b-b06e-65e3b3189e97
      Device Flags:        • Internal device
                           • Updatable
                           • Requires AC power
                           • Supported on remote server
                           • Needs a reboot after installation
                           • Device is usable for the duration of the update

kmauleon, 2020-06-23:
> ef5cdc85-9cf6-469d-9cb7-920b7dd6672

Thank you very much for the information...
with Secure Boot disabled, and with AC power, please try to run the following commands...

fwupdmgr refresh
fwupdmgr update

omersi, 2020-06-23:
Hi @kmauleon 

See `fwupdmgr get-devices ` output

```

{

  "ReportVersion": 2,
  "MachineId": "1bcc2ccf5c10ce6e3e3ed4bfca639a4fe1e9676f97650457294ba957eb723567",
  "Metadata": {
    "DistroId": "ubuntu",
    "DistroVersion": "20.04"
  },
  "Reports": [
    {
      "Checksum": "d792bd37bf95ac386dd581bd58d9561db7a26c12",
      "UpdateState": 2,
      "Guid": [
        "7c5a1e62-38fa-5859-9337-09dbac6377e4"
      ],
      "Plugin": "synaptics_prometheus",
      "VersionOld": "0014",
      "VersionNew": "0021",
      "Flags": 4194338,
      "Created": 1592741362,
      "Modified": 1592741455,
      "Metadata": {
        "RuntimeVersion(com.redhat.fwupdate)": "12",
        "RuntimeVersion(org.freedesktop.fwupd)": "1.3.9",
        "CompileVersion(org.freedesktop.gusb)": "0.3.4",
        "CpuArchitecture": "x86_64",
        "RuntimeVersion(org.freedesktop.gusb)": "0.3.4",
        "RuntimeVersion(org.kernel)": "5.4.0-37-generic",
        "DistroId": "ubuntu",
        "DistroVersion": "20.04",
        "RuntimeVersion(org.freedesktop.appstream-glib)": "0.7.14",
        "CompileVersion(org.freedesktop.fwupd)": "1.3.9",
        "BootTime": "1592741272",
        "CompileVersion(com.redhat.efivar)": "37",
        "RuntimeVersion(com.dell.libsmbios)": "2.4",
        "CompileVersion(com.redhat.fwupdate)": "12"
      }
    },
    {
      "Checksum": "afd95fd625bd2165a2bca50f39dfd02917d6cf67",
      "UpdateState": 2,
      "Guid": [
        "09e8bf16-3e69-50f5-bb66-c7a040248352"
      ],
      "Plugin": "synaptics_prometheus",
      "VersionOld": "10.01.2923304",
      "VersionNew": "10.01.3158804",
      "Flags": 37748770,
      "Created": 1592741362,
      "Modified": 1592741466,
      "Metadata": {
        "RuntimeVersion(com.redhat.fwupdate)": "12",
        "RuntimeVersion(org.freedesktop.fwupd)": "1.3.9",
        "CompileVersion(org.freedesktop.gusb)": "0.3.4",
        "CpuArchitecture": "x86_64",
        "RuntimeVersion(org.freedesktop.gusb)": "0.3.4",
        "RuntimeVersion(org.kernel)": "5.4.0-37-generic",
        "DistroId": "ubuntu",
        "DistroVersion": "20.04",
        "RuntimeVersion(org.freedesktop.appstream-glib)": "0.7.14",
        "CompileVersion(org.freedesktop.fwupd)": "1.3.9",
        "BootTime": "1592741272",
        "CompileVersion(com.redhat.efivar)": "37",
        "RuntimeVersion(com.dell.libsmbios)": "2.4",
        "CompileVersion(com.redhat.fwupdate)": "12"
      }
    }
  ]
}
```

kmauleon, 2020-06-23:
> Hi @kmauleon
>
> See `fwupdmgr get-devices ` output
>
> ```
>
> {
>
>   "ReportVersion": 2,
>   "MachineId": "1bcc2ccf5c10ce6e3e3ed4bfca639a4fe1e9676f97650457294ba957eb723567",
>   "Metadata": {
>     "DistroId": [...]

Hi Omersi...
sorry but I was hoping to see something as below...
├─Intel Management Engine:
│ Device ID: 07edea1cf12d20814225b44de22b33758db1a1c8
│ Current version: 192.40.1433
│ Minimum Version: 0.0.1
│ Vendor: DMI:LENOVO
│ GUID: ef5cdc85-9cf6-469d-9cb7-920b7dd6672b
│ Device Flags: • Internal device
│ • Updatable
│ • Requires AC power
│ • Supported on remote server
│ • Needs a reboot after installation
│ • Device is usable for the duration of the update

omersi, 2020-06-23:
> Hi Omersi...
> sorry but I was hoping to see something as below...
> ├─Intel Management Engine:
> │ Device ID: 07edea1cf12d20814225b44de22b33758db1a1c8
> │ Current version: 192.40.1433
> │ Minimum Version: [...]

├─System Firmware:
│     Device ID:           ca8e899e46f1da1776d16410bc92f2cf0d963433
│     Current version:     0.1.53
│     Vendor:              LENOVO (DMI:LENOVO)
│     GUID:                6300ba52-d9ea-4ca0-88f9-f89ef73cc07b
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update

kmauleon, 2020-06-23:
> ├─System Firmware:
> │ Device ID: ca8e899e46f1da1776d16410bc92f2cf0d963433
> │ Current version: 0.1.53
> │ Vendor: LENOVO (DMI:LENOVO)
> │ GUID: 6300ba52-d9ea-4ca0-88f9-f89ef73cc07b
> │ Device Flags: • [...]

Hi Omersi...
This is for system firmware... do you see the issue with ME FW as well?
If yes, I may need the info for MEFW
If no, since this is system firmware related, please open a separate issue for proper tracking...
thank you very much...

mayk78, 2020-06-23:
With ubuntu 19.10 and just upgraded to 20.04 , still the same issue.

[image]

20NX0076MH
│
├─Thunderbolt Controller:
│     Device ID:           98f9669c0b22c37b854aad6800ba456c29379133
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     20.00
│     Vendor:              Lenovo (TBT:0x0109)
│     GUIDs:               e773c51e-a20c-5b29-9f09-6bb0e0ef7560 ← TBT-01091806-native
│                          f1793149-6c12-5a7f-95db-9fefd116748d ← TBT-01091806-native-controller0-0
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Device stages updates
│   
├─Prometheus:
│ │   Device ID:           d432baa2162a32c1554ef24bd8281953b9d07c11
│ │   Summary:             Fingerprint reader
│ │   Current version:     10.01.3158804
│ │   Vendor:              Synaptics (USB:0x06CB)
│ │   Install Duration:    2 seconds
│ │   Serial Number:       9150547460203
│ │   GUIDs:               09e8bf16-3e69-50f5-bb66-c7a040248352 ← USB\VID_06CB&PID_00BD&REV_0000
│ │                        8088f861-6318-5b1e-9ce4-fbddbedb09ac ← USB\VID_06CB&PID_00BD
│ │   Device Flags:        • Updatable
│ │                        • Supported on remote server
│ │                        • Cryptographic hash verification is available
│ │ 
│ └─Prometheus IOTA Config:
│       Device ID:         8dcffb5d059857368caa2b7b22371ebd831b0c0c
│       Current version:   0021
│       Minimum Version:   0021
│       Vendor:            Synaptics (USB:0x06CB)
│       GUID:              7c5a1e62-38fa-5859-9337-09dbac6377e4 ← USB\VID_06CB&PID_00BD-cfg
│       Device Flags:      • Updatable
│                          • Supported on remote server
│     
├─System Firmware:
│     Device ID:           ca8e899e46f1da1776d16410bc92f2cf0d963433
│     Current version:     0.1.62
│     Vendor:              LENOVO (DMI:LENOVO)
│     GUID:                6300ba52-d9ea-4ca0-88f9-f89ef73cc07b
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           07edea1cf12d20814225b44de22b33758db1a1c8
│     Current version:     192.40.1433
│     Minimum Version:     0.0.1
│     Vendor:              DMI:LENOVO
│     Update State:        success
│     Last modified:       2020-06-23 07:18
│     GUIDs:               ef5cdc85-9cf6-469d-9cb7-920b7dd6672b
│                          056f0b06-1562-50f4-aea6-0d52f6fd4be5 ← UEFI\RES_{EF5CDC85-9CF6-469D-9CB7-920B7DD6672B}
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
[... 113 lines cut ...]
                               "VersionNew" : "10.01.3158804",
                               "Flags" : 37748770,
                               "Created" : 1592896092,
                               "Modified" : 1592896589,
                               "Metadata" : {
                                 "RuntimeVersion(com.redhat.fwupdate)" : "12",
                                 "RuntimeVersion(org.freedesktop.fwupd)" : "1.3.9",
                                 "CompileVersion(org.freedesktop.gusb)" : "0.3.4",
                                 "CpuArchitecture" : "x86_64",
                                 "RuntimeVersion(org.freedesktop.gusb)" : "0.3.4",
                                 "RuntimeVersion(org.kernel)" : "5.4.0-37-generic",
                                 "DistroId" : "ubuntu",
                                 "DistroVersion" : "20.04",
                                 "RuntimeVersion(org.freedesktop.appstream-glib)" : "0.7.14",
                                 "CompileVersion(org.freedesktop.fwupd)" : "1.3.9",
                                 "BootTime" : "1592857892",
                                 "CompileVersion(com.redhat.efivar)" : "37",
                                 "RuntimeVersion(com.dell.libsmbios)" : "2.4",
                                 "CompileVersion(com.redhat.fwupdate)" : "12"
                               }
                             }
                           ]
                         }
Proceed with upload? [Y|n]: y
Successfully uploaded 1 report

kmauleon, 2020-06-23:
> With ubuntu 19.10 and just upgraded to 20.04 , still the same issue.
>
> [image]
>
> 20NX0076MH
> │
> ├─Thunderbolt Controller:
> │ Device ID: 98f9669c0b22c37b854aad6800ba456c29379133
> │ Summary: Unmatched [...]

Thank you very much for the information...
with Secure Boot disabled, and with AC power, please try to run the following commands...

fwupdmgr refresh
fwupdmgr update

mayk78, 2020-06-23:
>     Screenshot from 2020-06-23 09-22-51
> Bug from
> "name": "View Issue" }, "description": "View this Issue on GitHub",
> "publisher": { "@type": "Organization", "name": "GitHub", "url":

Hi,

yes thank you.. I tried these commands earlier on 19.10 , to no avail.  
Now on 20.04 , it seems to work. The firmware got applied (it mentioned 
it during a reboot).

thank you.

slavoroi, 2020-06-23:
Just tried:
```
fwupdmgr refresh
fwupdmgr update
```
ubuntu 20.04 and this update does not pop up any more.

neuged, 2020-06-23:
I tried the same ( also Ubuntu 20.04, also used `fwupdmgr refresh` and `fwupdmgr update`), but while the firmware update is mentioned in the reboot, it is apparently not applied.

The "Ubuntu Software" application still lists the firmware update in the Updates tab and executing the above command once more, also tries to install the update again, which should not happen, since it was just installed.

kmauleon, 2020-06-23:
> I tried the same ( also Ubuntu 20.04, also used `fwupdmgr refresh` and `fwupdmgr update`), but while the firmware update is mentioned in the reboot, it is apparently not applied.
> 
> The "Ubuntu Software" application still lists the firmware update in the Updates tab and executing the above commands once more, also tries to install the update again, which should not happen, since it was just installed.

Hi neuged...
May I know results of the following command... 

fwupdmgr get-devices

Cryt1c, 2020-06-24:
> > ef5cdc85-9cf6-469d-9cb7-920b7dd6672
> 
> Thank you very much for the information...
> with Secure Boot disabled, and with AC power, please try to run the following commands...
> 
> fwupdmgr refresh
> fwupdmgr update

I tried this but no avail. The system still suggests to install the update and I can run the commands again with the same result.

I also get the same splash screen on boot as the OP with the contradictory instructions "Press a key to continue" and "Please wait while we install a system update" on startup after running the commands. Usually, when an update works I get a slightly different splash screen without the "Press a key to continue" and a proper loading bar. The "Please wait while we install a system update" and the Lenovo logo would be the same though.

kmauleon, 2020-06-24:
> > > ef5cdc85-9cf6-469d-9cb7-920b7dd6672
> > 
> > 
> > Thank you very much for the information...
> > with Secure Boot disabled, and with AC power, please try to run the following commands...
> > fwupdmgr refresh
> > fwupdmgr update
> 
> I tried this but to no avail. The system still suggests to install the update and I can run the commands again with the same result.
> 
> I also get the same splash screen on boot as the OP with the contradictory instructions "Press a key to continue" and "Please wait while we install a system update" on startup after running the commands. Usually, when an update works I get a slightly different splash screen without the "Press a key to continue" and a proper loading bar. The "Please wait while we install a system update" and the Lenovo logo would be the same though.

Hi Cryt1c...
When "press a key to continue" appears it means that system update has finished so press any key.... after this, the system should boot... please go to bios setuo to confirm MEFW version ... if updates did not work... please try these commands first...

apt install shim-signed
apt install fwupd
apt install fwupdate

(note: information will be displayed accordingly if apt already installed in the system)

execute the following commands again

fwupdmgr refresh
fwupdmgr update

neuged, 2020-06-24:
> May I know results of the following command...
> 
> fwupdmgr get-devices

Yes! Thanks for looking into this, btw. I put the output of "get-devices" into this gist:

https://gist.github.com/neuged/fd9aef26de37975e74e7078b608e9fbd

YegorTrembovetsky, 2020-06-24:
Hey developers, thank you for fixing the issue!

kmauleon, 2020-06-24:
> > May I know results of the following command...
> > fwupdmgr get-devices
> 
> Yes! Thanks for looking into this, btw, @kmauleon ! I put the output of "get-devices" into this gist:
> 
> https://gist.github.com/neuged/fd9aef26de37975e74e7078b608e9fbd

Hi neuged...
particular update is intended for T490s... but since GUID is the same for MEFW, thus update shows... i am running out of options but may i know the processor of the system? 
and can you please try these commands first...

apt install shim-signed
apt install fwupd
apt install fwupdate

(note: information will be displayed accordingly if apt already installed in the system)

execute the following commands again

fwupdmgr refresh
fwupdmgr update

[56 more replies]

closed 2020-12-18 by hughsie