fwupd/firmware-lenovo #32: Tracking Issue: Critical Intel Thunderbolt Firmware Updates - ThinkPad

ohazi, 2020-01-24:
This issue from Lenovo looks pretty bad:
https://support.lenovo.com/ca/en/solutions/ht508988

It's not clear what the actual problem is, but some of the individual updates mentioned "ROM wear out," so it might be related to excessive writes to flash somewhere. Failure of this ROM can apparently cause error messages, Thunderbolt chipset failure, or even boot delay / failure. 

The table below is a summary of firmware availability for affected machines on LVFS, as of 2020-01-23. 

I have more experience with embedded device firmware than PC firmware, so apologies if this comes off as naive: This table seems... haphazardly populated for a bug that can allegedly brick a machine, especially since Lenovo already has Windows utilities to update the firmware on each of the affected machine types. Is the process of writing the firmware image to the correct location dramatically different on different machine types?

I noticed that for each of the affected machines, LVFS either has Thunderbolt firmware version 20 / 43 (the version that fixes the bug), or no Thunderbolt firmware at all. There were one or two cases where LVFS had a version released a month or two earlier, but none that were *really* old. Have we not been updating Thunderbolt firmware until now? Maybe it didn't seem like there was a compelling reason to do so...?

I also noticed this issue: https://github.com/fwupd/fwupd/issues/1676
and this comment: "DO NOT FORCE UPDATE Thunderbolt Controller. This may damage the firmware." (https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN1MTF.firmware)
as well as notes from Lenovo about needing to update the driver before updating the firmware (at least on Windows). 

All of these things lead me to believe that we might need more than just the raw firmware image, and that the update process may need additional work and/or testing. 

So I guess there's two things I'm interested in tracking here:
1. Can we get firmware images for all of the remaining devices onto LVFS
2. What, if anything, should be done to ensure that the firmware images can be properly flashed on each device. Do the fwupd developers have to deal with this, or is it something that Lenovo provides when they add firmware to LVFS?

| Name | Required Firmware | Latest Firmware on LVFS | LVFS Link |
|-|-|-|-|
| X1 Tablet 3rd | 43 | None? |
| X1 Yoga 2nd | 43 | None |
| X1 Yoga 3rd | 43 | 43 | https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN25TF.firmware, https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN25TF_AssistMode.firmware (testing) |
| X1 Yoga 4th | 43 | 43 | https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN2HTF.firmware |
| T590 | 20 | None |
| T580 | 20 | 20 | https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN27TF.firmware | 
| T570 | 20 | 20 | https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN1VTF.firmware |
| P43s | 20 | None |
| P51s | 20 | 20 | https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN1VTF.firmware |
| P52s | 20 | 20 | https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN27TG.firmware, https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN27TG_AssistMode.firmware |
| P53s | 20 | None |
| T490 | 20 | None |
| T490s | 20 | None |
| T480 | 20 | None |
| T480s | 20 | 20 | https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN22TF.firmware |
| X1 Carbon 7th | 43 | 43 | https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN2HTF.firmware | 
| X1 Carbon 6th | 43 | None |
| X1 Carbon 5th | 43 | 43 (testing) | https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN1MTF.firmware (testing since 2019-07-01) |
| T470 / ThinkPad 25 | 20 | None |
| T470s | 20 | None |
| X380 Yoga | 20 | None? (X380 has version 20, but nothing for X380 Yoga) |
| X390 Yoga | 20 | None? (X390 has version 20, but nothing for X390 Yoga) |
| Yoga 370 | 20 | None |
| P52 | 45 | None |
| P51 | 43 | None |
| P71 | 43 | None |
| X390 | 20 | 20 | https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN2LTF_AssistMode.firmware, https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN2LTF.firmware |
| X280 | 20 | 20 | https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN20TF.firmware |

*: Some devices appear to have two firmware entries, com.lenovo.ThinkPad???.firmware and com.lenovo.ThinkPad???_AssistMode.firmware. Does anyone know what _AssistMode is for?

AlexOwen, 2020-01-24:
_AssistMode is an option in the BIOS that changes where resources are handled, as confirmed by Lenovo here: https://forums.lenovo.com/t5/Other-Linux-Discussions/Thunderbolt-BIOS-Assist-Mode-clearification-needed/m-p/4586530/highlight/true#M14029

> In BIOS Assist Mode, memory resource arraignment is done by BIOS. 
> In Native mode, memory resource arrangement is done by OS.
> 
> There are two different firmwares - one for each mode - as they aren't compatible between the two modes. LVFS should take care of getting you the right one now as mentioned above.
> 
> Also it was noted that native mode (assist disabled) is recommended. It's the mainstream setting and is what Intel are focussing on for testing and support.

ohazi, 2020-01-24:
@AlexOwen That makes sense, thanks. 

I noticed assist mode mentioned again in the summary thread here: https://www.reddit.com/r/thinkpad/comments/eszg2o/thinkpad_tb3_failure_what_we_know_repair/
and then it clicked that I had seen it before in the bios menu but never knew what it was for.

acidtv, 2020-01-27:
Correct me if I'm wrong, but by the looks of it the X390 Yoga update is actually already on LVFS. 

The page you're linking to from the X390 non-yoga contains version N2LTF08W, which is the yoga update according to this lenovo page: https://pcsupport.lenovo.com/us/en/products/laptops-and-netbooks/thinkpad-x-series-laptops/thinkpad-x390-yoga/downloads/DS541974

The update for the X390 non-yoga (and T490s) is N2JTF10W which is not on LVFS: https://pcsupport.lenovo.com/us/en/products/laptops-and-netbooks/thinkpad-t-series-laptops/thinkpad-t490s-type-20nx-20ny/downloads/DS541365

ohazi, 2020-01-28:
@acidtv I think you're right -- I put together the table by string searching for the machine name on the LVFS webpage, but some of the names don't quite match. The correct approach is probably to match Lenovo's Windows release code with the equivalent code on LVFS. There's at least one (X1 Carbon 7th) where the filename doesn't appear to match the entry name. It doesn't help that a few of the released firmware images (either for Windows or on LVFS) are newer than the version in the "Use version of higher" column of Lenovo's primary Thunderbolt critical update page, so you'd have to click through to each Windows download page. 

| Name | Required Firmware | Firmware on LVFS | Firmware on LVFS (AssistMode) | Comments |
|-|-|-|-|-|
| X1 Tablet 3rd | 43 |
| X1 Yoga 2nd | 43 |
| X1 Yoga 3rd | 43 | 43 | 43 (testing) |
| X1 Yoga 4th | 43 | 43 |
| T590 | 20 |
| T580 | 20 | 20 | 20 | 
| T570 | 20 | 20 (testing) |
| P43s | 20 |
| P51s | 20 | 20 |
| P52s | 20 | 20 | 20 |
| P53s | 20 |
| T490 | 20 |
| T490s | 20 |
| T480 | 20 |
| T480s | 20 | 20 |
| X1 Carbon 7th | 43 | 43 | | LVFS page says it's for X1 Carbon 7th, but filename suggests it's for 5th? https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN2HTF.firmware | 
| X1 Carbon 6th | 43 |
| X1 Carbon 5th | 43 | 43 |
| T470 / ThinkPad 25 | 20 |
| T470s | 20 |
| X380 Yoga | 20 | 20 | | Listed under "Lenovo Ltd. ThinkPad X380" |
| X390 Yoga | 20 | 20 | | Listed under "Lenovo Ltd. ThinkPad X390" |
| Yoga 370 | 20 |
| P52 | 45 |
| P51 | 43 |
| P71 | 43 |
| X390 | 20 |
| X280 | 20 | 20 |

jankratochvil, 2020-02-15:
Not sure if interesting but on Fedora 31 x86_64 MATE non-UEFI boot (and I think BIOS Assist disabled, I would have to verify) it was successful:
```
fwupdmgr update
Upgrade available for Thunderbolt Controller from 33.00 to 43.00
Thunderbolt Controller and all connected devices may not be usable while updating. Continue with update? [Y|n]: y
Downloading 43.00 for Thunderbolt Controller...
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Updating Thunderbolt Controller…                                 ]
Restarting device…       [***************************************]
Successfully installed firmware
```

AlexOwen, 2020-02-15:
@jankratochvil could you confirm which laptop this was on please?

RaitoBezarius, 2020-02-24:
What to do on T480? I have a TB3 dock and want to upgrade ASAP

AlexOwen, 2020-02-24:
This is not a support forum.

However the easiest way is to install Windows and use the downloads on
Lenovo's support site.

RaitoBezarius, 2020-02-24:
@AlexOwen Sure, but how hard is it to extract from the Windows' version of the firmware to package it into something that fwupd would accept?

matiargs, 2020-04-07:
Is there a way to force a `fwupdmgr update`? I only have 18% of battery left, when I run the update, the system start to add new capsules but then shows a battery low message and restart. The battery is not charging.

AlexOwen, 2020-04-07:
Adding --force may work, or --no-safety-check

matiargs, 2020-04-07:
> Adding --force may work, or --no-safety-check
> […](#)

Thanks, but still viewing "check power failure" message, and firmware doesn't update.

AlexOwen, 2020-04-07:
If it has already failed then there is probably no point doing a firmware
update. AFAIK it's a preventative measure.

axelsimon, 2020-10-06:
I think i was hit by this issue.
I have this error in my fwupdmgr history
```
$ sudo fwupdmgr get-history 
[sudo] password for axsimon: 
20KGS23S0P
│
├─X1 Carbon Thunderbolt Controller:
│ │   Device ID:           56e8795e4a366aab311bdbd752a3ad6da113906e
│ │   Previous version:    33.00
│ │   Update State:        failed
│ │   Update Error:        could not validate firmware: FW image image not compatible to this controller (Native). See https://github.com/hughsie/fwupd/wiki/Thunderbolt:-Validation-failed-or-unknown-device for more information.
│ │   GUID:                5ff9544e-fce2-5fd9-9cd2-671c95ac91dc
│ │   Device Flags:        • Internal device
│ │                        • Updatable
│ │                        • Requires AC power
│ │ 
│ └─  New version:       43.0.0
│       Licence:           Unknown
│     
├─X1 Carbon Thunderbolt Controller:
│ │   Device ID:           56e8795e4a366aab311bdbd752a3ad6da113906e
│ │   Previous version:    33.00
│ │   Update State:        success
│ │   GUID:                5ff9544e-fce2-5fd9-9cd2-671c95ac91dc
│ │   Device Flags:        • Internal device
│ │                        • Updatable
│ │                        • Requires AC power
│ │ 
│ └─  New version:       43.00
│       Licence:           Unknown
```

I think i just applied updates as they were offered by my system (Fedora) and didn't do or change anything special. Not until a long time later, when i tried to use my dock again and discovered it didn't work.

My thunderbolt controller now doesn't appear, neither using `fwupdmgr get-devices show-all-devices` nor `lspci -vvvv`.

Looks like my TB controller is just done for and i need to get a motherboard replacement, right?
Or is there something i can try like booting into Windows 10 to try and apply fw updates from the Windows side of things?

Thanks!

mrhpearson, 2020-10-06:
Hi @axelsimon 

Yeah - that doesn't look good :( Especially as you were using the older firmware that had the issue in it).

Can you see if Thunderbolt is showing up in the BIOS configuration (F1 during early boot) - just as a sanity check. If the controller is dead then Windows won't be able to help and I think your best course of action is to call Lenovo support for HW support.

Mark

axelsimon, 2020-10-08:
Thanks a lot for your advice and help @mrhpearson.

Just the other day (when i wrote the above message), i had Thunderbolt-related options in the BIOS. That being said, do you mean something more specific?

mrhpearson, 2020-10-08:
I was guessing a bit - poking around in the BIOS on my machine I'm not convinced it was the most helpful suggestion. I was hoping it might help show that if the BIOS couldn't see/configure the thunderbolt we could conclude it was a HW problem as we suspect.

I've sent a note to the thunderbolt experts here for their opinion. Because of the Chinese golden week holiday I probably won't have an update until next week I'm afraid. I'll let you know when I hear back from them.

Mark

mrhpearson, 2020-10-14:
Hi @axelsimon,
Conclusion from the experts was that you have hit the TB failure condition so this is a hardware failure. You should contact Lenovo support and they should be able to help you out.
As a note - the firmware with the fix has been delivered for all platforms so I think we can close this ticket. Let me know if any objections.
Thank
Mark

axelsimon, 2020-10-14:
Thanks a lot for following up on this @mrhpearson, it is much appreciated.

I'll get in touch with Lenovo / IT support and see how that goes.

Thanks!

operator-name, 2020-10-17:
I see this has been closed. Does that mean they have released their firmware through fwupd and the issue has been resolved?

mrhpearson, 2020-10-17:
Yes - please let me know if you find any problems.
Mark

ohazi, 2020-10-18:
I still don't see T470 firmware: #30

ohazi, 2020-10-18:
One more time:

| Name | Required Firmware | Firmware on LVFS | Firmware on LVFS (AssistMode) | Comments |
|-|-|-|-|-|
| X1 Tablet 3rd | 43 | **MISSING?**
| X1 Yoga 2nd | 43 | 43 |
| X1 Yoga 3rd | 43 | 43 |
| X1 Yoga 4th | 43 | 43 |
| T590 | 20 | 20 |
| T580 | 20 | 20 | 20 | 
| T570 | 20 | 20 |
| P43s | 20 | 20 |
| P51s | 20 | 20 |
| P52s | 20 | 20 | 20 |
| P53s | 20 | 20 |
| T490 | 20 | 20 |
| T490s | 20 | 20 |
| T480 | 20 | 20 |
| T480s | 20 | 20 |
| X1 Carbon 7th | 43 | 44 |
| X1 Carbon 6th | 43 | 43 | 43 |
| X1 Carbon 5th | 43 | 43 |
| T470 / ThinkPad 25 | 20 | **MISSING**
| T470s | 20 | 20 |
| X380 Yoga | 20 | 20 | | Listed under "Lenovo Ltd. ThinkPad X380" |
| X390 Yoga | 20 | 20 | | Listed under "Lenovo Ltd. ThinkPad X390" |
| Yoga 370 | 20 | **MISSING**
| P52 | 45 | 45 |
| P51 | 43 | 43 |
| P71 | 43 | 43 |
| X390 | 20 | 20 |
| X280 | 20 | 20 |

@hughsie Could you re-open this until the last few missing ones are confirmed?
@mrhpearson Would you mind asking around to see what happened with the T470 update? This one seems to have been an outlier in taking an abnormally long time, and now the firmware release that was in the 'testing' branch for months seems to be gone entirely.

mrhpearson, 2020-10-19:
Thanks @ohazi 
- X1 Tablet 3rd - I don't believe this is a Linux supported system unforunately so I don't have many levers to pull to get it up on LVFS. I can ask, but as it's not supported I can't really claim it's missing.
- T470 - this one is weird, I have a fwupd link, but it's not accesible so it must still be marked as under testing. I have a sneaking suspicion they forgot to enable it and I missed that when checking them. I'll chase that one down and I've re-opened the bug until we have that resolved.
- Yoga 370: is this not it:https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadR0HTF19W.firmware ? 

Thanks for the details though - if I've missed anything else let me know.
Mark

mrhpearson, 2020-10-20:
Hi @ohazi 
The T470 is now available - https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadNNQTF.firmware
Turns out it hadn't been set to stable, apologies for that.

If you can confirm on the 370 I'd like to close this ticket :)

Tanks
Mark

ohazi, 2020-10-20:
Thanks Mark!

I see the T470 and Yoga 370 firmware. If The X1 Tablet 3rd gen isn't supported, then I agree that we're finally good to close this ticket. I really appreciate your help with all of this on the Lenovo side. 

I missed the 370 because I searched for "yoga 370" but the search term that would have found it was "yoga370" (no space). 

I think the actual fwupdmgr utility has a more reasonable way to find firmware images that match a given machine type -- it would be nice if there were a way to do this on the LVFS website instead of relying on string matching, since vendors can and do name their firmware uploads arbitrarily. @hughsie do you think something like this would be worthwhile?

closed 2020-10-20 by ohazi