fwupd/firmware-lenovo #34: Could not apply capsule update: Not Found - ThinkPad T480 Consumer ME Update 184.70.3626

matthkoch, 2020-01-30:
Hi fwupd Team,
i'm trying to update my firmware.

```shell
$ fwupdmgr get-history
UEFI Device Firmware
  DeviceId:             ea0fd60df22477668e549841f6a22a8c31cae587
  Guid:                 9e21f98b-fe98-455c-b388-da5450ab6979
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|needs-reboot
  Version:              184.55.3510
  Created:              2020-01-30
  Modified:             2020-01-30
  UpdateState:          needs-reboot

[Release]
  RemoteId:             lvfs
  Version:              184.70.3626
  Checksum:             SHA1(a58ca3aed38409e5047d26e76f8e16efc95396d4)
  TrustFlags:           none
  CpuArchitecture:      x86_64
  RuntimeVersion(org.freedesktop.fwupd): 1.0.9
  CompileVersion(org.freedesktop.fwupd): 1.0.9
  CompileVersion(com.redhat.fwupdate): 10
  SecureBoot:           Disabled
  CompileVersion(org.freedesktop.gusb): 0.2.11
  CompileVersion(org.freedesktop.appstream-glib): 0.7.7
  ESPMountPoint:        /boot/efi
  KernelVersion:        4.15.0-76-generic
  UEFIUXCapsule:        Enabled
  RuntimeVersion(com.dell.libsmbios): 2.4
  BootTime:             1580369350
  CompileVersion(com.redhat.efivar): 34
```

```shell
$ fwupdmgr --version
client version: 1.0.9
daemon version: 1.0.9
compile-time dependency versions
        appstream-glib: 0.7.7
        gusb:   0.2.11
        fwupdate:       10
        efivar: 34
```

```shell
$ fwupdmgr get-devices
T480 Thunderbolt Controller
  DeviceId:             6a91208949d8f32757d80d1ec9b64eb71198415e
  Guid:                 fe328c2d-5300-597b-a21d-991116db8ab6
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|registered
  Vendor:               Lenovo
  VendorId:             TBT:0x0109
  Version:              14.00
  Icon:                 computer
  Created:              2020-01-30

UEFI Device Firmware
  DeviceId:             8e6d0ecd7a9737ba818ed0a90fb863aaa9af5a3c
  Guid:                 e0049080-09bd-473a-a02c-19aa0c8dad78
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              0.73.4
  VersionLowest:        0.73.4
  Icon:                 audio-card
  Created:              2020-01-30

UEFI Device Firmware
  DeviceId:             0a6ca0e0099a5836e3dec72fd5c67594f3fd9e2b
  Guid:                 89aab014-a71d-4cae-8824-174808266539
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              2.0.33505
  VersionLowest:        0.0.1
  Icon:                 audio-card
  Created:              2020-01-30

UEFI Device Firmware
  DeviceId:             ea0fd60df22477668e549841f6a22a8c31cae587
  Guid:                 9e21f98b-fe98-455c-b388-da5450ab6979
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              184.55.3510
  VersionLowest:        0.0.1
  Icon:                 audio-card
  Created:              2020-01-30

UEFI Device Firmware
  DeviceId:             2ea1b362db45581c8be7dc1094a973e2a0b45fec
  Guid:                 815e2465-da45-4b49-93c3-1517e649e1e6
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              1.3.1081
  VersionLowest:        0.0.1
  Icon:                 audio-card
  Created:              2020-01-30

UEFI Device Firmware
  DeviceId:             c19580db344e418cf2b23297bcf46d75c06e432a
  Guid:                 b87a926d-189c-49a1-b1ff-921099b1de89
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.20
  VersionLowest:        0.1.20
  Icon:                 audio-card
  Created:              2020-01-30

20L5S1L500 System Firmware
  DeviceId:             48477ba14bd9bd0c97323a07caa54b7766f917c2
  Guid:                 dc7f0308-1ef8-4774-9ba5-89a58c4d731c
  Guid:                 230c8b18-8d9b-53ec-838b-6cfc0383493a
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.30
  VersionLowest:        0.0.1
  Icon:                 computer
  Created:              2020-01-30
```

```shell
$ efibootmgr -v
BootNext: 0001
BootCurrent: 0000
Timeout: 0 seconds
BootOrder: 0000,0017,0018,0019,001A,001B,001C,001D,001E,001F,0024,0001
Boot0000* ubuntu        HD(1,MBR,0xb95b906d,0x800,0xf3800)/File(\EFI\ubuntu\shimx64.efi)
Boot0001* Linux-Firmware-Updater \fwupx64.efi   HD(1,MBR,0xb95b906d,0x800,0xf3800)/File(\EFI\ubuntu\shimx64.efi)\.f.w.u.p.x.6.4...e.f.i...
Boot0010  Setup FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
Boot0011  Boot Menu     FvFile(126a762d-5758-4fca-8531-201a7f57f850)
Boot0012  Diagnostic Splash Screen      FvFile(a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380)
Boot0013  Lenovo Diagnostics    FvFile(3f7e615b-0d45-4f80-88dc-26b234958560)
Boot0014  Startup Interrupt Menu        FvFile(f46ee6f4-4785-43a3-923d-7f786c3c8479)
Boot0015  Rescue and Recovery   FvFile(665d3f60-ad3e-4cad-8e26-db46eee9f1b5)
Boot0016  MEBx Hot Key  FvFile(ac6fd56a-3d41-4efd-a1b9-870293811a28)
Boot0017* USB CD        VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,86701296aa5a7848b66cd49dd3ba6a55)
Boot0018* USB FDD       VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,6ff015a28830b543a8b8641009461e49)
Boot0019* NVMe1 VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a401)
Boot001A* NVMe0 VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a400)
Boot001B* ATA HDD0      VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f602)
Boot001C* USB HDD       VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,33e821aaaf33bc4789bd419f88c50803)
Boot001D* PCI LAN       VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
Boot001E  Other CD      VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a35406)
Boot001F  Other HDD     VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f606)
Boot0020* USBR BOOT CDROM       PciRoot(0x0)/Pci(0x14,0x0)/USB(11,1)
Boot0021* USBR BOOT Floppy      PciRoot(0x0)/Pci(0x14,0x0)/USB(11,0)
Boot0022* ATA HDD       VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f6)
Boot0023* ATAPI CD      VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a354)
Boot0024* PCI LAN       VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
```

```shell
$ efivar -l | grep fw
0abba7dc-e516-4167-bbf5-4d9d1c739416-fwupdate-9e21f98b-fe98-455c-b388-da5450ab6979-0
```

```shell
$ tree /boot
/boot
├── config-4.15.0-72-generic
├── config-4.15.0-74-generic
├── config-4.15.0-76-generic
├── efi
│   └── EFI
│       ├── BOOT
│       │   ├── BOOTX64.EFI
│       │   └── fbx64.efi
│       └── ubuntu
│           ├── BOOTX64.CSV
│           ├── fw
│           │   └── fwupdate-8dQKVO.cap
│           ├── fwupx64.efi
│           ├── grub.cfg
│           ├── grubx64.efi
│           ├── mmx64.efi
│           └── shimx64.efi
```
When i'm rebooting the startupscreen shows the following picture:
[image]

```shell
$ fwupdmgr get-updates
No upgrades for UEFI Device Firmware: A reboot is pending
No upgrades for UEFI Device Firmware, current is 0.1.20: 0.1.19=older, 0.1.18=older, 0.1.17=older, 0.1.16=older, 0.1.14=older
No upgrades for 20L5S1L500 System Firmware, current is 0.1.30: 0.1.28=older, 0.1.26=older, 0.1.25=older, 0.1.24=older, 0.1.23=older
```

BIOS already updated via usb stick.

sdeleeuw, 2020-02-18:
Same issue here,

I posted on the Lenovo Forums about it:
https://forums.lenovo.com/t5/Other-Linux-Discussions/Firmware-upgrade-not-applied/m-p/4650063/highlight/false#M14439

kmauleon, 2020-10-20:
hi... latest MEFW 184.79.3722 has been posted already... please check with this one.  Thank you very much... https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN24RN.firmware

kmauleon, 2020-11-16:
hi... as per owner's remarks... please try as below.  thank you very much...
**we can see error message in fail report such as: failed to write data to efivarfs: Error writing to file descriptor: No space left on device, I think user can check the environment  first when meet error. Please update fwupd tool and use the latest MEFW 184.79.3722 try again.**

hughsie, 2020-11-16:
> Error writing to file descriptor: No space left on device

This normally indicates that something was logging to an EFI variable and used up all the space. Perhaps try listing /sys/firmware/efi/efivars/ and sorting by size?