fwupd/firmware-lenovo #35: Missing firmware: T560 / P1

vrilly, 2020-01-31:
For missing firmware please provide the following items:

**Link to firmware update at OEM website**
https://pcsupport.lenovo.com/us/en/products/laptops-and-netbooks/thinkpad-t-series-laptops/thinkpad-t560/downloads/ds112310
**Firmware release date (if available on OEM website)**
13 Jan 2020
**Link to the device page on LVFS.**
https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN1KET.firmware

In the cabinet provided by lenovo there seems to be a mistake:
```
  <provides>
    <firmware type="flashed">42f29620-3d63-4f09-950b-9b7055570f28</firmware>
  </provides>
```
It refers to the GUID of the T460s, while the firmware.bin is actually the correct payload for the T560. I used the firmware.bin with the T560 GUID (0b145790-0d11-4dee-9c2a-19ddbcbb200c) using fwupdate which installed fine on my machine.

kmauleon, 2020-10-20:
hi @vrilly ...sorry for being so late... but firmware for T560/P50s is in this location >> https://fwupd.org/lvfs/search?value=T560
For clarity though, can you please show result of following command:
     fwupdmgr get-devices

vrilly, 2020-10-20:
See https://gist.github.com/vrilly/908a461e682cbbddff4f4b7f8ca09a83

The situation on the T560/P50s on LVFS still persists, currently there's still an outdated cabinet with the wrong metadata on the repos. Rendering fwupd useless for this device.

The interesting part is lenovo does offer an updated cabinet file on their [own website](https://pcsupport.lenovo.com/us/en/products/laptops-and-netbooks/thinkpad-t-series-laptops/thinkpad-t560/downloads/ds112310) but did not publish it on LVFS

kmauleon, 2020-10-20:
hi @vrilly, we are targeting to release MEFW 184.79.3722 by November beginning... can you please wait a little more and check again.... but it would be helpful if you can show result of
fwupdmgr get-devices
thank you very much....

vrilly, 2020-10-21:
```20FHS00601
│
├─Intel Management Engine:
│     Device ID:           570d24e21cec6aadc4c7019ea2d7cfc16c22191b
│     Current version:     184.79.3722
│     Minimum Version:     0.0.1
│     Vendor:              DMI:LENOVO
│     GUID:                a7c68c44-60f0-431c-9242-926912892f28
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─Samsung SSD 860 EVO 1TB:
│     Device ID:           e421b2fc248391f6fe3e55ddbb3c9043be068bd0
│     Summary:             ATA Drive
│     Current version:     RVT04B6Q
│     Vendor:              Samsung (ATA:0x144D)
│     Serial Number:       S3Z9NB0M710931R
│     GUIDs:               f97c3c3c-efc9-56aa-8eb8-cfaae08fd615 ← IDE\Samsung_SSD_860_EVO_1TB_________________RVT04B6Q
│                          8f72ef92-2740-56c2-8025-bac145539815 ← IDE\0Samsung_SSD_860_EVO_1TB_________________
│                          bcaa2f53-dc18-5430-afaa-561ca1cff361 ← Samsung SSD 860 EVO 1TB
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─System Firmware:
│     Device ID:           3762edb590127a8d7498cf374bd920ce65fc54c2
│     Current version:     0.1.36
│     Minimum Version:     0.1.22
│     Vendor:              LENOVO (DMI:LENOVO)
│     GUID:                0b145790-0d11-4dee-9c2a-19ddbcbb200c
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
│   
└─UEFI Device Firmware:
      Device ID:           3b6d9c846a832bbcf3af788610c2baaea129f9bb
      Current version:     65544
      Minimum Version:     1
      Vendor:              DMI:LENOVO
      GUID:                2433c86b-6d72-4330-84ef-1972b1fe6e27
      Device Flags:        • Internal device
                           • Updatable
                           • Requires AC power
                           • Needs a reboot after installation
                           • Device is usable for the duration of the update```

kmauleon, 2020-10-21:
hi @vrilly I see that you already have the latest firmware so I guess we can close this issue... fyi...windows version is released 1-2 months ahead of LVFS, thus, you don't see it yet in LVFS site... please re open if you have any concern... thank you very much

closed 2020-10-21 by kmauleon