fwupd/firmware-lenovo #38: Lenovo X1CG6 'TPM PCR0 differs from reconstruction'

LinuxOnTheDesktop, 2020-04-01:
```
~ $ fwupdmgr --version
client version:	1.3.9
compile-time dependency versions
	gusb:	0.2.11
	efivar:	34
daemon version:	1.3.9
# /snap/fwupd/1704/bin/fwupdtpmevlog # I have that same file also at: /snap/fwupd/1501/libexec/fwupd/fwupdtpmevlog
PCR:                     BIOS (0)
Type:                    0x8
Description:             EV_S_CRTM_VERSION
ChecksumSha1:            fd2750786a0206674ebf2c8431e3cde0cb87151f8b82266f8a52e02997a9a6c1
BlobStr:                 N.2.3.E.T.7.0.W. ...

PCR:                     BIOS (0)
Type:                    0x80000008
Description:             EV_EFI_PLATFORM_FIRMWARE_BLOB
ChecksumSha1:            dc53a387ba8c7f0fcf8199a326009fd89378504c4bc16d15293c8ad97a7b5dbb

PCR:                     BIOS (0)
Type:                    0x80000008
Description:             EV_EFI_PLATFORM_FIRMWARE_BLOB
ChecksumSha1:            51485e22d314fac0d88822950696b312b6537bfb9ca6abe38f91f81209ccd9c8

PCR:                     BIOS (0)
Type:                    0x80000008
Description:             EV_EFI_PLATFORM_FIRMWARE_BLOB
ChecksumSha1:            4ef25de0d166620a0efff1e46f221a5383c9a5ce4129e8bc17e1f31b6c92a49a

PCR:                     BIOS (0)
Type:                    0x80000008
Description:             EV_EFI_PLATFORM_FIRMWARE_BLOB
ChecksumSha1:            cbca1b8cd59a03abb2abd8bbb510e5761a57bd18134d86f31e8059d2ea013264
BlobStr:                 ..........C.....

PCR:                     BIOS (0)
Type:                    0x1
Description:             EV_POST_CODE
ChecksumSha1:            bfc46da9ef25182f848dd38b96728eaa41409bb3c7c8db4b6b2e4019dbd1a107
BlobStr:                 ACPI DATA

PCR:                     BIOS (0)
Type:                    0x1
Description:             EV_POST_CODE
ChecksumSha1:            1efad6adef14bd2607d26be9cdf8dbc7d26a6f2db9452d685ec64e0dc8c60d35
BlobStr:                 ACPI DATA

PCR:                     BIOS (0)
Type:                    0x4
Description:             EV_SEPARATOR
ChecksumSha1:            df3f619804a92fdb4057192dc43dd748ea778adc52bc498ce80524c014b81119

PCR:                     BIOS Configuration (1)
Type:                    0x80000001
Description:             EV_EFI_VARIABLE_DRIVER_CONFIG
ChecksumSha1:            df7d7c617a9095e09f8d1d42ce005503ce5adc89fc0423672bf00ebb0fd0699c
BlobStr:                 ..M*.A.E.o-.4..e................L.e.n.o.v.o.C.o.n.f.i.g......................................................................................................................................................................................................................................................

PCR:                     BIOS Configuration (1)
Type:                    0x80000001
Description:             EV_EFI_VARIABLE_DRIVER_CONFIG
ChecksumSha1:            26e0d9abda13584bd8010fb2b298030dd55a76994c4cfc423a6bf81c810538bd
BlobStr:                 ....O..L....A..I................L.e.n.o.v.o.S.e.c.u.r.i.t.y.C.o.n.f.i.g............................................................................................................................................

PCR:                     BIOS Configuration (1)
Type:                    0x4
Description:             EV_SEPARATOR
ChecksumSha1:            df3f619804a92fdb4057192dc43dd748ea778adc52bc498ce80524c014b81119

PCR:                     Option ROMs (2)
[... 138 lines cut ...]
Type:                    0x80000001
Description:             EV_EFI_VARIABLE_DRIVER_CONFIG
ChecksumSha1:            a044b4ce4a4dca9af312c897dc56ee1727c385eb88f7cfb9092b8265029d5b1e
BlobStr:                 ....:=.E.....geo................d.b.x.&...LP.@..A.6.C(........0......wY.2M.`(...xK...i1.........O..R.m.@..`MA..e.....wY.2M.`(...xK./........r($...E4.[..$k;.}.n..z...wY.2M.`(...xK.......-...*o..s..>d.,N..gyj.......wY.2M.`(...xK63..M...x.bd..Y.W...C.&`H.X....v...wY.2M.`(...xK....Kle.. ..q..R0!.b..<H..k)Z+.....wY.2M.`(...xK..h.AFb..?i..nk.b.0.|o..x.....4....wY.2M.`(...xK...F..d.W.Xm........9y..2t-..S.f...wY.2M.`(...xKX......YC.._%...?.LX.^....)uh..q...wY.2M.`(...xKS.....!.....%.w..]o.....P."..Y.....wY.2M.`(...xK.&.~.jq..$...|.e.,.:{k%}...`.^.....wY.2M.`(...xK.c.(.~.S.d-.}.3..*...`....,2...m...wY.2M.`(...xK)..R.<:..,..n.`|.<.....eu\....JD...wY.2M.`(...xK....i.3@.>..h2.....'%'..=I..r..L...wY.2M.`(...xK.^....T..`.....< ......k......R8...wY.2M.`(...xK....Xdo...y..(....#g....+6...9.....wY.2M.`(...xK.._NQ...x.m...%.......or.xRY.e.&...wY.2M.`(...xK..C...z..0..eu1.{.............ct...wY.2M.`(...xK...9v-.6.=...c.qZ9....F\`.lk.......wY.2M.`(...xK....o)..o3.}r..K....H.:*...?O......wY.2M.`(...xK......!H...62u.>.......[1.R.*.[....wY.2M.`(...xK.o.....N0;t........+.o..t.!...h....wY.2M.`(...xK.N:.[C.....@O.4.=.9bg......#.......wY.2M.`(...xK.34)..b....>.H...-..ImT.....d......wY.2M.`(...xK+..&B...6_.K..'.l..Kzo.D./k..i.9...wY.2M.`(...xK+.,.....'.R..*].I.Z+.R.]fb....U....wY.2M.`(...xK,s.3%.m......<[.UY....P.P...R..}...wY.2M.`(...xK.p.g...sQ.......p.W.2..#....+Q.}...wY.2M.`(...xK0f(.Tw0W(.JF}..8zT.i.v..^u.........wY.2M.`(...xK6......A...wz./.^g.4g^..^i5........wY.2M.`(...xK8A.!6....\...!`9MlN.g`.....b..[....wY.2M.`(...xK?....>..TR..^.....mt:syqU.p.j.>s...wY.2M.`(...xKC......c.|....C.-/....&.z.K..u.....wY.2M.`(...xKG..a'.....:k.,....Zmk.6!h..,.*Z....wY.2M.`(...xKQ.1.s....>..!"...Ty..........0a5...wY.2M.`(...xKZ.I..U...9..[..B.,/.g...g6..A.+\...wY.2M.`(...xKk...x.A....{.^.`..G........r../f...wY.2M.`(...xKl.TG..Y..Q.&.l...+..585.r..........wY.2M.`(...xKo.(.q.......{...d|.e......&..:x^...wY.2M.`(...xKq..o."I~T.Fb.$.... w...h......cu...wY.2M.`(...xKrk>.T.j0..=.....p....p..q.-..,#....wY.2M.`(...xKr...g.].V....;....2.....^/m..(.....wY.2M.`(...xKx'..6,..q}.....C..q.Z..H.[...K.....wY.2M.`(...xK...e....k.).T...S.........;.3......wY.2M.`(...xK..;....C........YA.=...Xo+.V7W_g...wY.2M.`(...xK.Z......~.O..G.q.."8b....:....=....wY.2M.`(...xK..HY.........jag..z.n.F.d.r!.YE....wY.2M.`(...xK..4...........e...;=.<5.P_.{.c.!...wY.2M.`(...xK..........se.(.Q..<.Pm........H....wY.2M.`(...xK...c.....t...M.....so..C.fd..1ZB...wY.2M.`(...xK.Ji.1ah.U....`..........f......4...wY.2M.`(...xK....6U....G'Yyk.. .T...iuLHH.t.....wY.2M.`(...xK../P.N.....~.N....]...o..+...]E....wY.2M.`(...xK.h&..m&...h\..}..;M.=......`.<W ...wY.2M.`(...xK...1Q'.s....g.9.1..g0:3"..7...Z....wY.2M.`(...xK.....},....3:..OgQ.......D..L@.....wY.2M.`(...xK.O..6c..h..;.7........*9..h....U...wY.2M.`(...xK..x...J.3!c..5...,3....p.L.5.'W6...wY.2M.`(...xK.z....._..Km.;..vfh..U$|..(7..L....wY.2M.`(...xK...h..fH.....Q...j.$..y..b.........wY.2M.`(...xK....Gu.....".......F,.....].3......wY.2M.`(...xK......*..(...L....[.'(.a...........wY.2M.`(...xK..3f......T.....s.&.........g......wY.2M.`(...xK.k..@...vX....QJI`O........n.x.....wY.2M.`(...xK.;..Y.|....J...>..$Q?.eYW.5.).@....wY.2M.`(...xK....5.g+6~O...Iia]..J.lrMB.........wY.2M.`(...xK.,".;VB.\....G.YG8......D.oY.......wY.2M.`(...xK..n=)...t=.J..........2@...........wY.2M.`(...xK.c.Ox,..........7`..X.b...f..nm....wY.2M.`(...xK...2...KmH],qgr..RY..\..u."....6...wY.2M.`(...xK...aJ.~.......U.......n.E.AR'..[...wY.2M.`(...xKU....=..HZ..7.?...=.....|....c.....wY.2M.`(...xKw.......^;.....b.x...S^.......k/...wY.2M.`(...xK.<.9"...`tFu.7....Z...G/.4.q.9.....wY.2M.`(...xK;..S>......#...A..r.y....-...6.....wY.2M.`(...xK......Q.3@....H..rRj..R.......`I...wY.2M.`(...xKdW[..x....V.4.R.k...D.xYu..N-d.E...wY.2M.`(...xKE...u...H.7R}d..dM...<..$.M.ig..

PCR:                     Platform manufacturer specific measurements (7)
Type:                    0x4
Description:             EV_SEPARATOR
ChecksumSha1:            df3f619804a92fdb4057192dc43dd748ea778adc52bc498ce80524c014b81119

PCR:                     Static operating system (14)
Type:                    0xd
ChecksumSha1:            a7eea77f52839e64cc3426435290ed8b6405a036bf6ca11a16255a374794c012
BlobStr:                 MokList..

PCRs:
  0:                     d83f0bc616d5a3bf6c6d2d32139a5a1bf12e285b
  1:                     c5a4c60e7fe971b8894c1d741a412e1970a0f288
  2:                     3a4285c323c5af121f270900e88ffeefb2a34233
  3:                     3a4285c323c5af121f270900e88ffeefb2a34233
  4:                     a4da6aab4881a3d5dbc04d9d36d6bb6505a0257a
  5:                     83d8fff7891d6efcb52b3ea7f3960bb44d6f891d
  6:                     3a4285c323c5af121f270900e88ffeefb2a34233
  7:                     df9e9ac086fa686e4f9a3fb2c964a3276dacb090
```

hughsie, 2020-04-28:
Could you please attach the `fwupdmgr get-devices --show-all-devices` output to this bug so we can further analyse the problem. Thanks!

LinuxOnTheDesktop, 2020-04-28:
[output.txt](https://github.com/fwupd/missing-firmware-lenovo-thinkpad/files/4546229/output.txt)