fwupd/firmware-lenovo #40: Lenovo T480s PCR0 Difffers from reconstruction

s950449, 2020-04-01:
To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version:	1.3.8
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.3.8
```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

```shell
$ fwupdmgr get-devices
20L7CTO1WW
│
├─INTEL SSDPEKKW512G8:
│     Device ID:           c82980cd3fe66f7d0a8bb83b353cbc271177efd4
│     Summary:                NVM Express Solid State Drive
│     Current version:     004C
│     Vendor:              Intel Corporation (NVME:0x8086)
│     GUIDs:               79517f86-8df8-5d6e-a18b-33f0b36a78e9
│                          68db11e5-b0cf-5bc9-a94e-17e28496e505
│                          4d10db12-e333-5e97-9ef3-de112dd88175
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─System Firmware:
│     Device ID:           e7cc234e82d08b6a14e246beb6b5f8142091e570
│     Current version:     0.1.38
│     Minimum Version:     0.0.1
│     Vendor:              LENOVO (DMI:LENOVO)
│     Update Error:        TPM PCR0 differs from reconstruction, please report!
│     GUID:                ebfe8df8-dee7-4692-a721-cbcf5095c5cf
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           cf3ad1e5e6838b09cba64bb5ccb1142dd44995d2
│     Current version:     184.70.3626
│     Minimum Version:     184.70.3626
│     Vendor:              DMI:LENOVO
│     GUID:                676af093-2a5c-4238-9c29-db8063a33532
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
└─UEFI Device Firmware:
      Device ID:           323bda90831520867f8c02544efb9711dfca9fb6
      Current version:     0.1.20
      Minimum Version:     0.1.20
      Vendor:              DMI:LENOVO
      GUID:                4d254d6e-cd67-477b-97d5-bc3048af45c4
      Device Flags:        • Internal device
                           • Updatable
[... 615 lines cut ...]
│   │   ├── arch-git.conf
│   │   └── arch-mainline.conf
│   ├── loader.conf
│   └── random-seed
├── System Volume Information
│   ├── IndexerVolumeGuid
│   └── WPSettings.dat
├── vmlinuz-linux
├── vmlinuz-linux-git
└── vmlinuz-linux-mainline

```

Please answer the following questions:

- Operating system and version:
Arch Linux (Linux 5.6.0-rc3-1-mainline)
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc):
pacman
- Have you tried rebooting?
Yes
- Are you using an NVMe disk?
Yes
- Is secure boot enabled (only for the UEFI plugin)?
No

hughsie, 2020-04-28:
Could you please attach the `fwupdmgr get-devices --show-all-devices` output to this bug so we can further analyse the problem. Thanks!

s950449, 2020-04-30:
https://github.com/fwupd/missing-firmware-lenovo-thinkpad/files/4557357/output.txt
Note: I added a WD SN520 ssd last weekend.