fwupd/firmware-lenovo #41: TPM PCR0 differs from reconstruction

sgraf-pub, 2020-04-01:
In the middle of get-devices output (see below), there is sentence:
```shell
Update Error:        TPM PCR0 differs from reconstruction, please report!
```

I was able to update firmware without any issue (with disabled secure boot, I enabled it only recently).

To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version:	1.3.8
compile-time dependency versions
	gusb:	0.3.3
	efivar:	37
daemon version:	1.3.8
```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

```shell
$ fwupdmgr get-devices
20HMS1FC02
│
├─Lenovo ThinkPad Dock:
│     Device ID:           73ef80b60058b4f18549921520bfd94eaf18710a
│     Summary:             USB 3.x Hub
│     Current version:     50.41
│     Vendor:              LENOVO (USB:0x17EF)
│     Install Duration:    10 seconds
│     GUIDs:               3a5bdad1-2726-5194-aa7d-04e1daf9946b
│                          8d708000-7d3b-5e85-ac9d-8ef32fb2d445
│                          1df23e1b-0cb9-52e0-9b9d-f8ed96aac18e
│                          6201fecc-1641-51f6-a6d2-38a06d5476bf
│                          b435be36-7b59-5463-ab34-abfe22cf7a53
│                          06cdabc6-2059-5daa-87ab-2f6fc31fd74b
│                          8da160e7-8f9a-5eba-82c0-5f18ecaac31e
│     Device Flags:        • Updatable
│                          • Cryptographic hash verification is available
│   
├─SAMSUNG MZVLW256HEHP-000L7:
│     Device ID:           04e17fcf7d3de91da49a163ffe4907855c3648be
│     Summary:             NVM Express Solid State Drive
│     Current version:     4L7QCXB7
│     Vendor:              Samsung Electronics Co Ltd (NVME:0x144D)
│     GUIDs:               f87b9ac8-1cb3-5c0a-ae57-7144f211fe5e
│                          5b3df2da-f745-5fd0-81de-5dafd7f0bf8c
│                          257d6faa-82f3-53e2-afa4-f6b9adc9595d
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─System Firmware:
│     Device ID:           8c997bbd20dfdf9c471aaae577d182d9ae77bca7
│     Current version:     0.1.37
│     Minimum Version:     0.1.10
│     Vendor:              LENOVO (DMI:LENOVO)
│     Update Error:        TPM PCR0 differs from reconstruction, please report!
│     GUID:                14cc970e-c105-4eba-a704-448dde9de64d
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           caf28180ce03eb0bfaac417775de3fe0f6bd61bc
│     Current version:     182.29.3287
│     Minimum Version:     0.0.1
│     Vendor:              DMI:LENOVO
│     GUID:                9a7f2771-7ce4-4ff1-892b-c2ed700b77de
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           4144f261efb14c45aa79a27b4c4d3bf228f6cdb9
│     Current version:     0.1.17
│     Minimum Version:     0.0.1
│     Vendor:              DMI:LENOVO
│     GUID:                18dfeb28-b8a4-4cec-97ce-b0599416a13e
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
└─VMM2322:
      Device ID:           585c346c009da3cc19a400fb59a4ed8d0cbd0151
      Summary:             Multi-Stream Transport Device
      Current version:     2.33.00
      Vendor:              Synaptics (DRM_DP_AUX_DEV:0x06CB)
      GUIDs:               49ec4eb4-c02b-58fc-8935-b1ee182405c7
                           8eba3da0-2998-58bf-a7d7-210c15b3a3a8
                           75f1e01e-356f-5534-a8be-211d879838f1
      Device Flags:        • Updatable
```

```shell
$ efibootmgr -v
BootCurrent: 0000
Timeout: 0 seconds
BootOrder: 0000,0017,0018,0019,001A,001B,001C,001D
Boot0000* FedoraCould not parse device path: Invalid argument
```

```shell
$ efivar -l | grep fw
bash: efivar: command not found
```
I'm on Fedora Silverblue and it looks like efivar is not part of default ostree.

```shell
$ tree /boot
/boot
├── efi
│   └── EFI
│       ├── BOOT
│       │   ├── BOOTX64.EFI
│       │   └── fbx64.efi
│       └── fedora
│           ├── BOOTX64.CSV
│           ├── fonts
│           ├── fw
│           │   └── fwupd-14cc970e-c105-4eba-a704-448dde9de64d.cap
│           ├── fwupdx64.efi
│           ├── grub.cfg
│           ├── grub.cfg.old
│           ├── grubenv
│           ├── grubenv9e7a53
│           ├── grubx64.efi
│           ├── mmx64.efi
│           ├── shim.efi
│           ├── shimx64.efi
│           └── shimx64-fedora.efi
├── grub2
│   ├── grubenv -> ../efi/EFI/fedora/grubenv
│   └── themes
│       └── system
│           ├── background.png
│           └── fireworks.png
├── loader -> loader.1
├── loader.1
│   └── entries
│       ├── ostree-1-fedora.conf
│       └── ostree-2-fedora.conf
├── lost+found
└── ostree
    ├── fedora-2352c568b9ef3f276e9856b04fbf9cee72f409cc949b59144ffa410bd8f04691
    │   ├── initramfs-5.5.5-200.fc31.x86_64.img
    │   └── vmlinuz-5.5.5-200.fc31.x86_64
    └── fedora-aeadc68d71e6ffe40a42cde3022edeaae7ad06af186633d77042a436dff0491e
        ├── initramfs-5.6.0-0.rc2.git0.1.fc32.x86_64.img
        └── vmlinuz-5.6.0-0.rc2.git0.1.fc32.x86_64

16 directories, 23 files
```

Please answer the following questions:

- Operating system and version:
Fedora Silverblue 32 (reproducible also on 31)
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc):
It was already part of distribution.
- Have you tried rebooting?
YES
- Are you using an NVMe disk?
YES
- Is secure boot enabled (only for the UEFI plugin)?
YES (enabled only after latest firmware update)

hughsie, 2020-04-28:
Could you please attach the `fwupdmgr get-devices --show-all-devices` output to this bug so we can further analyse the problem. Thanks!

sgraf-pub, 2020-04-29:
`
$ fwupdmgr get-devices --show-all-devices
20HMS1FC02
│
├─Lenovo ThinkPad Dock:
│     Device ID:           23d940eac81a886d3523a7d731f2b68a2bedb0f7
│     Summary:             USB 3.x Hub
│     Current version:     50.41
│     Vendor:              LENOVO (USB:0x17EF)
│     Install Duration:    10 seconds
│     GUIDs:               3a5bdad1-2726-5194-aa7d-04e1daf9946b
│                          8d708000-7d3b-5e85-ac9d-8ef32fb2d445
│                          1df23e1b-0cb9-52e0-9b9d-f8ed96aac18e
│                          b435be36-7b59-5463-ab34-abfe22cf7a53
│                          06cdabc6-2059-5daa-87ab-2f6fc31fd74b
│                          8da160e7-8f9a-5eba-82c0-5f18ecaac31e
│     Device Flags:        • Updatable
│                          • Cryptographic hash verification is available
│   
├─Embedded Controller:
│     Device ID:           4144f261efb14c45aa79a27b4c4d3bf228f6cdb9
│     Current version:     0.1.17
│     Minimum Version:     0.0.1
│     Vendor:              DMI:LENOVO
│     GUID:                18dfeb28-b8a4-4cec-97ce-b0599416a13e
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─HD Graphics 620:
│     Device ID:           bbbf1ce3d1cf15550c3760b354592040292415bb
│     Current version:     02
│     Vendor:              Intel Corporation (PCI:0x8086)
│     GUIDs:               2886e312-afd8-5510-8993-12d568a85e00
│                          38f6c009-a25c-5b50-b3b5-fef4f9c6b846
│     Device Flags:        • Internal device
│                          • Cryptographic hash verification is available
│   
├─Intel AMT [unprovisioned]:
│     Device ID:           e2623122c99d58220498aacbfcfdb1baebbae3c5
│     Summary:             Hardware and firmware technology for remote out-of-band management
│     Current version:     11.6.29.3287
│     Bootloader Version:  11.6.29.3287
│     Vendor:              Intel Corporation
│     GUID:                2800f812-b7b4-2d4b-aca8-46e0ff65814c
│     Device Flags:        • Internal device
│   
├─Intel(R) Core™ i5-7300U CPU @ 2.60GHz:
│     Device ID:           4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:     0xca
│     Vendor:              GenuineIntel
│     GUID:                b9a2dd81-159e-5537-a7db-e7101d164d3f
│     Device Flags:        • Internal device
│   
├─SAMSUNG MZVLW256HEHP-000L7:
│     Device ID:           04e17fcf7d3de91da49a163ffe4907855c3648be
│     Summary:             NVM Express Solid State Drive
│     Current version:     4L7QCXB7
│     Vendor:              Samsung Electronics Co Ltd (NVME:0x144D)
│     GUIDs:               f87b9ac8-1cb3-5c0a-ae57-7144f211fe5e
│                          5b3df2da-f745-5fd0-81de-5dafd7f0bf8c
│                          257d6faa-82f3-53e2-afa4-f6b9adc9595d
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─System Firmware:
│     Device ID:           8c997bbd20dfdf9c471aaae577d182d9ae77bca7
│     Current version:     0.1.38
│     Minimum Version:     0.1.10
│     Vendor:              LENOVO (DMI:LENOVO)
│     GUID:                14cc970e-c105-4eba-a704-448dde9de64d
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
│   
├─TPM:
│ │   Device ID:           c6a80ac3a22083423992a3cb15018989f37834d6
│ │   Current version:     7.61.10.57600
│ │   Vendor:              Infineon (TPM:IFX)
│ │   GUIDs:               ff71992e-52f7-5eea-94ef-883e56e034c6
│ │                        5eebb112-75ad-5536-b173-a11eb3399402
│ │                        ddf995da-1b32-5a8a-bc1b-8d5af4b38b51
│ │                        6d81ab63-db2e-50ac-934f-6be9accf5e02
│ │                        301555de-680d-5ddc-b995-7553fc9138f1
│ │   Device Flags:        • Internal device
│ │ 
│ └─Event Log:
│       Device ID:         58bd405f31c48e6eca290b425f530a94c91e955c
│       GUID:              a25657fe-b5dc-5be0-8b78-8b9dfec678ff
│       Device Flags:      • Internal device
│     
├─UEFI Device Firmware:
│     Device ID:           caf28180ce03eb0bfaac417775de3fe0f6bd61bc
│     Current version:     3055357143
│     Minimum Version:     1
│     Vendor:              DMI:LENOVO
│     GUID:                9a7f2771-7ce4-4ff1-892b-c2ed700b77de
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
└─VMM2322:
      Device ID:           585c346c009da3cc19a400fb59a4ed8d0cbd0151
      Summary:             Multi-Stream Transport Device
      Current version:     2.33.00
      Vendor:              Synaptics (DRM_DP_AUX_DEV:0x06CB)
      GUIDs:               8eba3da0-2998-58bf-a7d7-210c15b3a3a8
                           75f1e01e-356f-5534-a8be-211d879838f1
                           49ec4eb4-c02b-58fc-8935-b1ee182405c7
      Device Flags:        • Updatable
`