fwupd/firmware-lenovo #42: Thinkpad T480S — TPM PCR0 differs from reconstruction

a-vrma, 2020-04-13:
**Describe the bug**
fwupdmgr get-devices shows

> TPM PCR0 differs from reconstruction, please see https://github.com/fwupd/fwupd/wiki/TPM-PCR0-differs-from-reconstruction

**Steps to Reproduce**
Arch Linux on T480s with Intel i5 on the `linux-lts` kernel. Have run fwupd
multiple times. Every time after the update reboot I would get a black screen
so I just held down power button to shut down and then pressed it again and
everything worked fine.

**Expected behavior**
No errors.

**fwupd version information**
```
client version: 1.3.9
compile-time dependency versions
        gusb:   0.3.4
        efivar: 37
daemon version: 1.3.9
```

Installed with `pacman`.

**fwupd device information**
<https://0x0.st/iQ-s.txt>

**System UEFI configuration**
efibootmgr stuff:
<https://0x0.st/iQ-o.txt>

efivar stuff
<https://0x0.st/iQ-X.txt>

`/boot` directory:
<https://0x0.st/iQ-8.txt>

**Additional questions**
- Operating system and version: Arch Linux with Linux LTS kernel
- Have you tried rebooting? Yes
- Is this a regression? Not sure, I just noticed it.
- Are you using an NVMe disk? Yes.
- Is secure boot enabled? No.

The output of fwupdtpmevlog: <https://0x0.st/iQ-P.log>.

hughsie, 2020-04-28:
Could you please attach the `fwupdmgr get-devices --show-all-devices` output to this bug so we can further analyse the problem. Thanks!

a-vrma, 2020-04-28:
>Could you please attach the fwupdmgr get-devices --show-all-devices output to this bug so we can further analyse the problem. Thanks!

```
20L7CTO1WW
│
├─Thunderbolt Controller:
│     Device ID:           a71027b1bf667cf254144b2a577cbe56c9380fc5
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     17.00
│     Vendor:              Lenovo (TBT:0x0109)
│     GUIDs:               3b125a20-5ed2-5c0f-b22d-ab0e0b687bbc
│                          6783c3fd-8932-5306-bb7f-4dfec968e44b
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Device stages updates
│   
├─Embedded Controller:
│     Device ID:           323bda90831520867f8c02544efb9711dfca9fb6
│     Current version:     0.1.20
│     Minimum Version:     0.1.20
│     Vendor:              DMI:LENOVO
│     GUID:                4d254d6e-cd67-477b-97d5-bc3048af45c4
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─Intel Management Engine:
│     Device ID:           cf3ad1e5e6838b09cba64bb5ccb1142dd44995d2
│     Current version:     184.70.3626
│     Minimum Version:     184.70.3626
│     Vendor:              DMI:LENOVO
│     GUID:                676af093-2a5c-4238-9c29-db8063a33532
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─Intel(R) Core™ i5-8250U CPU @ 1.60GHz:
│     Device ID:           4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:     0xca
│     Vendor:              GenuineIntel
│     GUID:                b9a2dd81-159e-5537-a7db-e7101d164d3f
│     Device Flags:        • Internal device
│   
├─PCIe SSD:
│     Device ID:           c82980cd3fe66f7d0a8bb83b353cbc271177efd4
│     Summary:             NVM Express Solid State Drive
│     Current version:     ECFM12.2
│     Vendor:              Phison Electronics Corporation (NVME:0x1987)
│     GUIDs:               8cb1e2fe-eb01-5508-9fb3-98add4bb7c34
│                          a44eb54c-5441-56f2-8cc0-5e48964c6457
│                          8d128eab-f266-513f-81e7-910de65fd73a
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs shutdown after installation
│                          • Device is usable for the duration of the update
│   
├─System Firmware:
│     Device ID:           e7cc234e82d08b6a14e246beb6b5f8142091e570
│     Current version:     0.1.38
│     Minimum Version:     0.0.1
│     Vendor:              LENOVO (DMI:LENOVO)
│     Update Message:      TPM PCR0 differs from reconstruction, please see https://github.com/fwupd/fwupd/wiki/TPM-PCR0-differs-from-reconstruction
│     GUID:                ebfe8df8-dee7-4692-a721-cbcf5095c5cf
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
│   
├─TPM:
│ │   Device ID:           c6a80ac3a22083423992a3cb15018989f37834d6
│ │   Current version:     7.63.13.6400
│ │   Vendor:              Infineon (TPM:IFX)
│ │   GUIDs:               ff71992e-52f7-5eea-94ef-883e56e034c6
│ │                        5eebb112-75ad-5536-b173-a11eb3399402
│ │                        ddf995da-1b32-5a8a-bc1b-8d5af4b38b51
│ │                        6d81ab63-db2e-50ac-934f-6be9accf5e02
│ │                        301555de-680d-5ddc-b995-7553fc9138f1
│ │   Device Flags:        • Internal device
│ │ 
│ └─Event Log:
│       Device ID:         58bd405f31c48e6eca290b425f530a94c91e955c
│       GUID:              a25657fe-b5dc-5be0-8b78-8b9dfec678ff
│       Device Flags:      • Internal device
│     
└─UHD Graphics 620:
      Device ID:           bbbf1ce3d1cf15550c3760b354592040292415bb
      Current version:     07
      Vendor:              Intel Corporation (PCI:0x8086)
      GUIDs:               fed2efa4-7045-55a1-b4fc-b29283d59fe5
                           8b72a10c-1279-5f8e-a28a-34fb11a58240
      Device Flags:        • Internal device
                           • Cryptographic hash verification is available
    
```

a-vrma, 2020-04-30:
I'm switching OS soon so I won't be able to give more logs after about a week

a-vrma, 2021-01-11:
Might have been a hardware issue since it went away after RMA.

closed 2021-01-11 by a-vrma