fwupd/firmware-lenovo #43: Hard freeze when upgrading firmware on T470s

jadahl, 2020-04-28:
After having upgraded firmware for the T470s in GNOME Software I rebooted to make the update take effect.

The update was initiated, and I saw the Lenovo logo as well as a message about firmware being updated.

After a short amount of time, the screen turned black. Thinking it was working and just failing to provide feedback I let it be for a couple of hours or three. Coming back it was still showing a black screen but otherwise being turned on (USB devices was lit up, etc), so I assumed it got stuck.

I held the power button for a while, but it didn't reboot or anything, just continued to be stuck.

Eventually I had to unscrew and remove both batteries inside the laptop to make it unstuck. After reattaching the batteries, the laptop booted up without issues.

`fwupdtool get-history` gives me this:

```
Loading…                 [***************************************]
20HGS22D1B
│
├─Thunderbolt Controller:
│ │   Device ID:           28c400a73ef3b76247eb086461a6dbf24cbbd60c
│ │   Previous version:    04.00
│ │   Update State:        failed
│ │   Update Error:        failed to get device after update: failed to wait for detach replug: device 28c400a73ef3b76247eb086461a6dbf24cbbd60c did not come back
│ │   Last modified:       2020-04-24 06:58
│ │   GUID:                96c3f95f-611a-56f8-be95-0f912cb84657
│ │   Device Flags:        • Internal device
│ │                        • Requires AC power
│ │                        • Hardware is waiting to be replugged
│ │                        • Device stages updates
│ │ 
│ └─  New version:       20.00
│       License:           Unknown
│     
├─System Firmware:
│ │   Device ID:           84149fb78009cf27cc1dd5520911f46f8792dbfe
│ │   Previous version:    0.1.34
│ │   Update State:        needs-reboot
│ │   Last modified:       2020-04-24 07:43
│ │   GUID:                7a176688-0960-47ba-931b-7829849e8347
│ │   Device Flags:        • Internal device
│ │                        • Updatable
│ │                        • Requires AC power
│ │                        • Supported on remote server
│ │                        • Needs a reboot after installation
│ │                        • Cryptographic hash verification is available
│ │                        • Device is usable for the duration of the update
│ │ 
│ └─ThinkPad T470s System Update:
│       New version:       0.1.37
│       Remote ID:         lvfs
│       Summary:           Lenovo ThinkPad T470s System Firmware
│       License:           Proprietary
│       Size:              9,4 MB
│       Vendor:            Lenovo Ltd.
│       Flags:             is-upgrade
│       Description:       Lenovo ThinkPad T470s System Firmware Version 1.37. Update includes security fixes.
│     
├─UEFI Device Firmware:
│ │   Device ID:           cbede50560aaa37366326cd54923421517427f39
│ │   Previous version:    184.65.3590
│ │   Update State:        success
│ │   Last modified:       2020-04-24 07:44
│ │   GUID:                e9124c4a-fdff-42e5-b2ad-f745db345953
│ │   Device Flags:        • Internal device
│ │                        • Updatable
│ │                        • Requires AC power
│ │                        • Supported on remote server
│ │                        • Needs a reboot after installation
│ │                        • Device is usable for the duration of the update
│ │ 
│ └─ThinkPad T470s Corporate ME Update:
│       New version:       184.70.3626
│       Remote ID:         lvfs
│       Summary:           Lenovo ThinkPad T470s Corporate ME Firmware
│       License:           Proprietary
│       Size:              7,5 MB
│       Vendor:            Lenovo Ltd.
│       Description:       • 8 Q2'19 Intel Platform Update
│       
│       Version 11.8.70.3626 (LVFS: 184.70.3626)
│       
│       Problem Fixes
│       
│        • Intel CSME PSIRT-TA-201905-011 Q2 2019 IPU:
│       
│       Security issues fixed:
│       
│        • CVE-2019-0168
│        • CVE-2019-0169
│        • CVE-2019-11087
│        • CVE-2019-11090
│        • CVE-2019-11101
│        • CVE-2019-11104
│        • CVE-2019-11106
│        • CVE-2019-11110
│        • CVE-2019-11131
│        • CVE-2019-11132
│     
└─UEFI Device Firmware:
  │   Device ID:           03a6930fc4c833443a7502dde379b98b961c8c57
  │   Previous version:    0.1.20
  │   Update State:        needs-reboot
  │   Last modified:       2020-04-24 07:44
  │   GUID:                e2acd8e4-a376-47bb-a316-4f1e62a8ca1f
  │   Device Flags:        • Internal device
  │                        • Updatable
  │                        • Requires AC power
  │                        • Supported on remote server
  │                        • Needs a reboot after installation
  │                        • Device is usable for the duration of the update
  │ 
  └─ThinkPad T470s Embedded Controller Update:
        New version:       0.1.21
        Remote ID:         lvfs
        Summary:           Lenovo ThinkPad T470s Embedded Controller Firmware
        License:           Proprietary
        Size:              791,8 kB
        Vendor:            Lenovo Ltd.
        Flags:             is-upgrade
        Description:       Lenovo ThinkPad T470s Embedded Controller Firmware version 1.21.Fixed an issue where monitors attached to ThinkPad Docking Stations are
        
        disconnected when battery charging is reached to 100%.
      
```

lijunwang1990, 2020-04-29:
Hi, can you provide your test steps? and which firmware do you update? BIOS/ME/TPM?
we will try to duplicate the issue on our side.Thank you.

jadahl, 2020-04-29:
I think the above "history" log entry should describe what firmware update, shouldn't it?

The steps are just 1) click upgrade in GNOME Software, 2) reboot, the rest happened by itself.

lijunwang1990, 2020-04-30:
Did you try upgrading with the command: fwupdmgr refresh, fwupdmgr update?
Please try it if you haven't

jadahl, 2020-04-30:
I'm not sure what you are referring to. This is a laptop and it was connected to the power cable the whole time. It didn't have any other utilities connected from start to end.

Elpe-75, 2021-05-02:
Hi, I had the same problem today with my TP T470s. After a few seconds of firmware-update screen the screen suddenly turned black and nothing else happend. Keeping the power-button pressed had no effect. The computer did not shut down.

So I opend the TP and took out the smaller battery. After that the computer was off. I then rebooted and just after the start pressed enter to get into the bios. In the bios "Linux firmware updater" was numer one in the order of boot devices. I moved that from pos.1 to the last position (using the F6-key). I left the bios while saving the changes an the T470s booted into Linux again.

Done. 

Hope this helps. Wish everybody having the same issue goog luck!

I will call Lenovo tomorrow cause this can't stay as it is.