fwupd/firmware-lenovo #47: Thinkpad X1 Gen 6 Firmware Update Fails

greyspectrum, 2020-05-22:
The error is:

```
Error calling StartServiceByName for org.freedesktop.fwupd: Timeout was reached
```
I suspect this is getting a 404 on https://fwupd.org, when fwupd attempts to download the firmware.

The following updates fail:

Version 0.1.20:
Lenovo ThinkPad X1 Carbon 6th Embedded Controller Firmware 1.20

Version 184.77.3664:
Version 11.8.77.3664 (LVFS: 184.77.3664)

2020.01 Q1 Intel Platform Update (IPU), formerly known as Quarterly security release (QSR)

Version 0.1.48:
Lenovo ThinkPad X1 Carbon 6th System Firmware

```
20KH002HUS
│
├─Thunderbolt Controller:
│     Device ID:           5aedc5181884a45b9651f51e1995f95e2494ebe7
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     43.00
│     Vendor:              Lenovo (TBT:0x0109)
│     GUIDs:               2da42a33-cd30-5ef5-a8fb-2c800a4b760f
│                          96411853-9f56-53d2-b05c-a54fcad5ee62
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Device stages updates
│   
├─Embedded Controller:
│     Device ID:           9698faabddf0d7b18925cfbbda95f8b0d0dacc53
│     Current version:     0.1.8
│     Minimum Version:     0.1.8
│     Vendor:              DMI:LENOVO
│     Update State:        success
│     GUID:                3babca5f-b2bf-4f4b-a72e-2bdc84eb4019
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─Intel Management Engine:
│     Device ID:           e563ad307df81c99f0de8c26292afd71cf409673
│     Current version:     184.50.3460
│     Minimum Version:     0.0.1
│     Vendor:              DMI:LENOVO
│     Update State:        success
│     GUID:                42a0a96e-c9f3-438f-9687-7826be33e4ce
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─SAMSUNG MZVLB1T0HALR-000L7:
│     Device ID:           04e17fcf7d3de91da49a163ffe4907855c3648be
│     Summary:             NVM Express Solid State Drive
│     Current version:     4L2QEXA7
│     Vendor:              Samsung Electronics Co Ltd (NVME:0x144D)
│     GUIDs:               6e54c992-d302-59ab-b454-2d26ddd63e6d
│                          47335265-a509-51f7-841e-1c94911af66b
│                          08575ca8-c22c-540a-8ac9-e784c04f6e5a
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
└─System Firmware:
      Device ID:           1c53551e7da69d896138fac1ae131c83ad46d923
      Current version:     0.1.30
      Minimum Version:     0.0.1
      Vendor:              LENOVO (DMI:LENOVO)
      GUID:                a4b51dca-8f97-4310-8821-3330f83c9135
      Device Flags:        • Internal device
                           • Updatable
                           • Requires AC power
                           • Supported on remote server
                           • Needs a reboot after installation
                           • Cryptographic hash verification is available
                           • Device is usable for the duration of the update
    ```

hughsie, 2020-05-22:
Can you attach the log of `sudo /usr/libexec/fwupd/fwupd --verbose` please

greyspectrum, 2020-05-22:
Absolutely. Thank you.

[fwupd-log.txt](https://github.com/fwupd/firmware-lenovo-thinkpad/files/4669471/fwupd-log.txt)

LordChaos73, 2020-06-23:
Looks like the update that was pulled, is available again:

ThinkPad X1 Carbon 6th Consumer ME Update:
        New version:       184.77.3664
        Remote ID:         lvfs
        Summary:           Lenovo ThinkPad X1 Carbon 6th Consumer ME Firmware
        License:           Proprietary
        Size:              2,2 MB
        Created:           2016-07-08
        Urgency:           High
        Vendor:            Lenovo Ltd.
        Flags:             is-upgrade
        Description:       • 8 Intel Platform Update 2020.1
        
        Version 11.8.77.3664 (LVFS: 184.77.3664)

Let's see if this one works properly.

LordChaos73, 2020-06-23:
I can confirm the latest firmware update for the Intel ME works fine now.

closed 2020-06-23 by hughsie