fwupd/firmware-lenovo #49: Update fails on ubuntu 20 T460s

NadirDerradj, 2020-06-06:
After updating the firmware from software center and reboot it was blocked on this screen any help

kmauleon, 2020-06-23:
Hi NadirDerradj...
Thank you very much for your attention.  Can you please show results for the below command...
fwupdmgr get-devices

kmauleon, 2020-10-22:
target to release new versions by november... please wait...thank you very much

kmauleon, 2020-11-16:
hi @NadirDerradj please try with latest MEFW... thank you very much
    Corporate (VPro) >> https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN1CRM.firmware
    Consumer (NonVPro) >> https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN1CRN.firmware

kmauleon, 2020-12-23:
hi @NadirDerradj latest 11.8.80.3746 is available already.  please check with this.  thank you very much.
Corporate >> https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN1CRM.firmware
Consumer >> https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN1CRN.firmware

closed 2022-11-16 by kmauleon