fwupd/firmware-lenovo #51: ThinkPad X1 Carbon 6th Gen, System Firmware "TPM PCR0 differs from reconstruction"

renffah, 2020-06-12:
**Describe the bug**
Using a ThinkPad X1 Carbon 6th Gen running Ubuntu 20.04.

After replacing the _Absolute Persistence Module version_ (UEFI BIOS ID `N23ET73W`) with the _Non-Absolute Persistence Module version_ (UEFI BIOS ID `N23ET73P`) — remove Anti-Theft > Computrace — `fwupdmgr get-devices` shows "TPM PCR0 differs from reconstruction" error for the System Firmware.

Due to this update, the System firmware Device ID had changed from `1c53551e7da69d896138fac1ae131c83ad46d923` to `2c7a30f46dc2054be78c13cfaba2dc3ba2fee2b0` . So actually the error message is correct, but nothing else is reporting the TPM PCR0 differencies.

**fwupd version information**
```shell
$ fwupdmgr --version
client version:	1.3.9
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.3.9
```

**fwupd device information**
```shell
$ fwupdmgr get-devices --show-all-devices
20KGS2A900
│
├─Thunderbolt Controller:
│     Device ID:           21e826f9c336f5fc0edd43f53873d03fe9f47016
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     43.00
│     Vendor:              Lenovo (TBT:0x0109)
│     GUIDs:               2da42a33-cd30-5ef5-a8fb-2c800a4b760f
│                          96411853-9f56-53d2-b05c-a54fcad5ee62
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Device stages updates
│   
├─Intel AMT [unprovisioned]:
│     Device ID:           e2623122c99d58220498aacbfcfdb1baebbae3c5
│     Summary:             Hardware and firmware technology for remote out-of-band management
│     Current version:     11.8.70.3626
│     Bootloader Version:  11.8.70.3626
│     Vendor:              Intel Corporation
│     GUID:                2800f812-b7b4-2d4b-aca8-46e0ff65814c
│     Device Flags:        • Internal device
│   
├─Samsung SSD 970 EVO Plus 1TB:
│     Device ID:           04e17fcf7d3de91da49a163ffe4907855c3648be
│     Summary:             NVM Express Solid State Drive
│     Current version:     2B2QEXM7
│     Vendor:              Samsung Electronics Co Ltd (NVME:0x144D)
│     GUIDs:               6e54c992-d302-59ab-b454-2d26ddd63e6d
│                          47335265-a509-51f7-841e-1c94911af66b
│                          f05ecac9-5cc2-5f77-b23b-a08ad8879ada
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─System Firmware:
│     Device ID:           2c7a30f46dc2054be78c13cfaba2dc3ba2fee2b0
│     Current version:     0.1.48
│     Minimum Version:     0.0.1
│     Vendor:              LENOVO (DMI:LENOVO)
│     Update Error:        TPM PCR0 differs from reconstruction, please see https://github.com/fwupd/fwupd/wiki/TPM-PCR0-differs-from-reconstruction
│     GUID:                99d47dfe-68fc-4a6f-b541-32e003c886fb
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
[... 715 lines cut ...]
ChecksumSha1:            668b315729c39ad5f0a0b59f870b607776edeb425481b70a1de01b0a19c12e2d
BlobStr:                 (hd0,gpt2)/boot/grub/fonts/unicode.pf2.

PCR:                     Static operating system (9)
Type:                    0xd
ChecksumSha1:            d0930a7bb899bfdc208ada47413399f3ff14ab3af4e35984977922dc6fd676f4
BlobStr:                 /boot/vmlinuz-5.4.0-31-generic.

PCR:                     Static operating system (9)
Type:                    0xd
ChecksumSha1:            79a2b4a04741b0d4778e37d38fc00723a41be0bb7a05dc1ad8a911604bed3100
BlobStr:                 /boot/initrd.img-5.4.0-31-generic.

PCRs:
  0:                     983dc74ae97def517e644a0880d3c282139a3bf3
  1:                     efd8cf14d6eff89d8fb57b6d4b19be0e15c6509a
  2:                     3a4285c323c5af121f270900e88ffeefb2a34233
  3:                     3a4285c323c5af121f270900e88ffeefb2a34233
  4:                     8b3422f26c1b96b0d9ba01807f434ea69525b255
  5:                     e4d734cac5c78179d2d6e6ea4a6e7a8156e65488
  6:                     3a4285c323c5af121f270900e88ffeefb2a34233
  7:                     42c2439d553ea14f74bff3f1a776361daef85d0e
  8:                     b8b030911a65f5b80bf2af79168c46b88124b00d
  9:                     ce21ca5a122d7b38d5cfcf10147c0ad001bb83da
```

renffah, 2020-06-28:
After installing System Firmware 1.49 the error disappeared.

closed 2020-06-28 by renffah

intelfx, 2023-12-25:
Apologies for the off-topic.

@renffah Could you please clarify how did you manage to install the P-firmware in the first place? Is it possible to do via fwupd?

renffah, 2023-12-25:
It's a long time ago.
Downloaded the P firmware (afair a cab archive or extracted from the iso) and installed it, (free)dos or fwupdmgr.
Maybe I have it documented somewhere, but the X1 and me are in different places right now.