fwupd/firmware-lenovo #53: UEFI Device Firmware update fails on Lenovo X1 Yoga

caugner, 2020-06-12:
**Describe the bug**
The *UEFI Device Firmware from 184.65.3590 to 184.70.3626* fails with the following error message:

> `failed to write data to efivarfs: Error writing to file descriptor: No space left on device`

**Steps to Reproduce**

```terminal
# fwupdmgr update
No upgrades for Thunderbolt Controller: current version is 43.00: 43.00=same
No upgrades for SAMSUNG MZVLW1T0HMLH-000L7: current version is 7L7QCXY7: 7L7QCXY7=same
No upgrades for System Firmware: current version is 0.1.36: 0.1.36=same, 0.1.33=older, 0.1.32=older, 0.1.31=older, 0.1.30=older
Upgrade available for UEFI Device Firmware from 184.65.3590 to 184.70.3626
20JES03T00 must remain plugged into a power source for the duration of the update to avoid damage. Continue with update? [Y|n]: 
Downloading 184.70.3626 for UEFI Device Firmware...                                                                                                                                                                                                                           Decompressing…           [***************************************] ]                                                                                                                                                                                                          Authenticating…          [***************************************] ] ]                                                                                                                                                                                                            Updating UEFI Device Firmware…                                     ]                                                                                                                                                                                                            Scheduling…              [***************************************] ]
failed to write data to efivarfs: Error writing to file descriptor: No space left on device
```

**Expected behavior**
The update should execute without errors.

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version
client version: 1.3.7
compile-time dependency versions
        gusb:   0.3.3
        efivar: 37
daemon version: 1.3.7
```

*Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):*

Installed on NixOS using the ([fwupd](https://nixos.org/nixos/packages.html?attr=fwupd&channel=nixpkgs-unstable&query=fwupd)) package.

```shell
# which fwupdmgr
/run/current-system/sw/bin/fwupdmgr
nix# ls -l $(which fwupdmgr)
lrwxrwxrwx 1 root root 68 Jan  1  1970 /run/current-system/sw/bin/fwupdmgr -> /nix/store/yb5rcmvj4jf8knd6vqwx0maxkd40wsvn-fwupd-1.3.7/bin/fwupdmgr
```

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
# fwupdmgr get-devices --show-all-devices
20JES03T00
│
├─ThinkPad Thunderbolt 3 Dock:
│     Device ID:           9777fc479d2648dbd23b0eeb58716165dddb3911
│     Current version:     15.00
│     Vendor:              Lenovo (TBT:0x0108)
│     GUID:                df675f10-53ac-59ca-bddd-0a86ee492920 ← TBT-01081630
│     Device Flags:        • Updatable
│                          • Requires AC power
│                          • Device stages updates
│
├─Thunderbolt Controller:
│     Device ID:           1999ff120e8a1810e06390ede52770e6325b15ba
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     43.00
│     Vendor:              Lenovo (TBT:0x0109)
│     GUIDs:               eb0d102f-761b-5c79-8ae5-177beb5a01d4 ← TBT-01091603
│                          d714ec8b-471a-5497-9795-c4d728092bf0 ← TBT-01091603-0000:08:00.0
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
[... 109 lines cut ...]
│   │   ├── naclhjim9wgg575d70hlyvqkhif2026v-initrd-linux-5.6.11-initrd.efi
│   │   ├── vn3na8vbccyqgg2xgmmn7h1yzqyxgyim-linux-5.6.11-bzImage.efi
│   │   └── y01nqlpcrivay9q3immqqr40gqjjz6qi-initrd-linux-5.6.12-initrd.efi
│   └── systemd
│       └── systemd-bootx64.efi
└── loader
    ├── entries
    │   ├── nixos-generation-410.conf
    │   ├── nixos-generation-411.conf
    │   ├── nixos-generation-412.conf
    │   ├── nixos-generation-413.conf
    │   ├── nixos-generation-414.conf
    │   ├── nixos-generation-415.conf
    │   ├── nixos-generation-416.conf
    │   ├── nixos-generation-417.conf
    │   └── nixos-generation-418.conf
    └── loader.conf
```

**Additional questions**
- Operating system and version: **NixOS 20.03**
- Have you tried rebooting? **Yes**
- Is this a regression? **No**
- Are you using an NVMe disk? **Yes**
- Is secure boot enabled? **No**

hughsie, 2020-06-13:
> Error writing to file descriptor: No space left on device

I suspect your EFI partition is full. I'm guessing this works if you increase the amount of space there?

caugner, 2020-06-18:
@hughsie My boot partition seems to have almost 200M free space:

```terminal
$ df -h
...
/dev/nvme0n1p2           499M  151M  348M 30.3 [#########......................] /boot 
...
```

hughsie, 2020-06-18:
> failed to write data to efivarfs: Error writing to file descriptor

So, I misread this. It's not telling us that /boot or /boot/EFI is full, it's telling us that we can't write to the efivarfs. I don't think that's expected at all. @vathpela -- any ideas?

vathpela, 2020-06-18:
Usually either a) there are a pile of big variables like pstore dumps and such that haven't been cleaned up or b) the firmware garbage collection cycle will trigger next time you reboot after seeing that, and it'll work after that.

vathpela, 2020-06-18:
Can you show us what "ls -l /sys/firmware/efi/efivars/" says?

caugner, 2020-06-18:
> Can you show us what "ls -l /sys/firmware/efi/efivars/" says?

```terminal
$ ls -l /sys/firmware/efi/efivars/
total 0
-rw-r--r-- 1 root root   26 Jun 18 23:12 AppName-1fd8b79f-0be2-4d57-b241-81c5e24e01a1
-rw-r--r-- 1 root root   33 Jun 18 23:12 AppPlatform-1fd8b79f-0be2-4d57-b241-81c5e24e01a1
-rw-r--r-- 1 root root    5 Jun 18 23:12 AuthVarKeyDatabase-aaf32c78-947b-439a-a180-2e144ec37792
-rw-r--r-- 1 root root   46 Jun 18 23:12 Boot0010-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root   54 Jun 18 23:12 Boot0011-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root   84 Jun 18 23:12 Boot0012-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root   72 Jun 18 23:12 Boot0013-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root   80 Jun 18 23:12 Boot0014-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root   74 Jun 18 23:12 Boot0015-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root   60 Jun 18 23:12 Boot0016-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root   64 Jun 18 23:12 Boot0017-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root   66 Jun 18 23:12 Boot0018-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root   63 Jun 18 23:12 Boot0019-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root   69 Jun 18 23:12 Boot001A-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root   66 Jun 18 23:12 Boot001B-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root   66 Jun 18 23:12 Boot001C-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root   69 Jun 18 23:12 Boot001D-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root   71 Jun 18 23:12 Boot001E-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root   72 Jun 18 23:12 Boot001F-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root   74 Jun 18 23:12 Boot0020-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root   66 Jun 18 23:12 Boot0021-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root   68 Jun 18 23:12 Boot0022-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root    6 Jun 18 23:12 BootCurrent-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root    8 Jun 18 23:12 BootOptionSupport-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root   20 Jun 18 23:12 BootOrder-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root   28 Jun 18 23:12 BootOrderDefault-0b7646a4-6b44-4332-8588-c8998117f2ef
-rw-r--r-- 1 root root    5 Jun 18 23:12 BootState-60b5e939-0fcf-4227-ba83-6bbed45bc0e3
-rw-r--r-- 1 root root   28 Jun 18 23:12 CapsuleLongModeBuffer-711c703f-c285-4b10-a3b0-36ecbd3c8be2
-rw-r--r-- 1 root root    8 Jun 18 23:12 certdb-d9bee56e-75dc-49d9-b4d7-b534210f637a
-rw-r--r-- 1 root root  184 Jun 18 23:12 ConIn-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root   38 Jun 18 23:12 ConInDev-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root   56 Jun 18 23:12 ConOut-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root   34 Jun 18 23:12 ConOutDev-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root  522 Jun 18 23:12 CpuSetup-b08f97ff-e6e8-4193-a997-5e9e9b0adb32
-rw-r--r-- 1 root root   14 Jun 18 23:12 CpuSetupVolatileData-b08f97ff-e6e8-4193-a997-5e9e9b0adb32
-rw-r--r-- 1 root root    6 Jun 18 23:12 DIAGSPLSHSCRN-a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380
-rw-r--r-- 1 root root  325 Jun 18 23:12 dump-type0-10-1-1572284543-C-cfc8fc79-be2e-4ddc-97f0-9f98bfe298a0
-rw-r--r-- 1 root root  410 Jun 18 23:12 dump-type0-10-1-1574121354-C-cfc8fc79-be2e-4ddc-97f0-9f98bfe298a0
-rw-r--r-- 1 root root  331 Jun 18 23:12 dump-type0-10-1-1574153345-C-cfc8fc79-be2e-4ddc-97f0-9f98bfe298a0
-rw-r--r-- 1 root root  333 Jun 18 23:12 dump-type0-10-1-1574290541-C-cfc8fc79-be2e-4ddc-97f0-9f98bfe298a0
-rw-r--r-- 1 root root  567 Jun 18 23:12 dump-type0-10-1-1574438134-C-cfc8fc79-be2e-4ddc-97f0-9f98bfe298a0
-rw-r--r-- 1 root root  596 Jun 18 23:12 dump-type0-10-1-1574592325-C-cfc8fc79-be2e-4ddc-97f0-9f98bfe298a0
-rw-r--r-- 1 root root  327 Jun 18 23:12 dump-type0-10-2-1572284543-C-cfc8fc79-be2e-4ddc-97f0-9f98bfe298a0
-rw-r--r-- 1 root root  406 Jun 18 23:12 dump-type0-10-2-1574121355-C-cfc8fc79-be2e-4ddc-97f0-9f98bfe298a0
-rw-r--r-- 1 root root  331 Jun 18 23:12 dump-type0-10-2-1574153345-C-cfc8fc79-be2e-4ddc-97f0-9f98bfe298a0
-rw-r--r-- 1 root root  334 Jun 18 23:12 dump-type0-10-2-1574290541-C-cfc8fc79-be2e-4ddc-97f0-9f98bfe298a0
-rw-r--r-- 1 root root  567 Jun 18 23:12 dump-type0-10-2-1574438135-C-cfc8fc79-be2e-4ddc-97f0-9f98bfe298a0
-rw-r--r-- 1 root root  329 Jun 18 23:12 dump-type0-11-1-1572284543-C-cfc8fc79-be2e-4ddc-97f0-9f98bfe298a0
-rw-r--r-- 1 root root  401 Jun 18 23:12 dump-type0-11-1-1574121354-C-cfc8fc79-be2e-4ddc-97f0-9f98bfe298a0
-rw-r--r-- 1 root root  657 Jun 18 23:12 dump-type0-11-1-1574153345-C-cfc8fc79-be2e-4ddc-97f0-9f98bfe298a0
-rw-r--r-- 1 root root  400 Jun 18 23:12 dump-type0-11-1-1574290541-C-cfc8fc79-be2e-4ddc-97f0-9f98bfe298a0
-rw-r--r-- 1 root root  571 Jun 18 23:12 dump-type0-11-1-1574438134-C-cfc8fc79-be2e-4ddc-97f0-9f98bfe298a0
-rw-r--r-- 1 root root  866 Jun 18 23:12 dump-type0-11-1-1574592325-C-cfc8fc79-be2e-4ddc-97f0-9f98bfe298a0
-rw-r--r-- 1 root root  498 Jun 18 23:12 dump-type0-1-1-1572284543-C-cfc8fc79-be2e-4ddc-97f0-9f98bfe298a0
-rw-r--r-- 1 root root  542 Jun 18 23:12 dump-type0-1-1-1574121354-C-cfc8fc79-be2e-4ddc-97f0-9f98bfe298a0
-rw-r--r-- 1 root root  494 Jun 18 23:12 dump-type0-1-1-1574153345-C-cfc8fc79-be2e-4ddc-97f0-9f98bfe298a0
[... cut ...]
-rw-r--r-- 1 root root   20 Jun 18 23:12 Tcg2PhysicalPresence-aeb9c5c1-94f1-4d02-bfd9-4602db2d3c54
-rw-r--r-- 1 root root    8 Jun 18 23:12 Tcg2PhysicalPresenceFlags-aeb9c5c1-94f1-4d02-bfd9-4602db2d3c54
-rw-r--r-- 1 root root    6 Jun 18 23:12 Timeout-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root   28 Jun 18 23:12 UCR-14a22a97-8424-489e-9ead-dc09255658b5
-rw-r--r-- 1 root root    5 Jun 18 23:12 VarErrorFlag-04b37fe8-f6ae-480b-bdd5-37d98c5e89aa
-rw-r--r-- 1 root root    5 Jun 18 23:12 VendorKeys-8be4df61-93ca-11d2-aa0d-00e098032b8c
```

caugner, 2020-06-18:
@vathpela Can I safely delete all those dumps?

vathpela, 2020-06-18:
I honestly don't know what they're from. I would *guess* yes... But I'm just guessing. Note you'll have to "chattr -i" them first.

vathpela, 2020-06-18:
Oh nevermind, those are how pstore dumps are named now. Yes, safe to delete those.

caugner, 2020-06-18:
Thanks, just deleted them (after making a backup of the efivars folder).

Unfortunately it didn't solve the issue. In the UI I shortly get the "No space left on device" error and then `GetUpdates not supported by backend`:

[image]

Same on the command line:

```
fwupdmgr update
• Thunderbolt Controller has the latest available firmware version
• SAMSUNG MZVLW1T0HMLH-000L7 has the latest available firmware version
• System Firmware has the latest available firmware version
Upgrade available for UEFI Device Firmware from 184.65.3590 to 184.70.3626
20JES03T00 must remain plugged into a power source for the duration of the update to avoid damage. Continue with update? [Y|n]: 
Downloading 184.70.3626 for UEFI Device Firmware...
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Updating UEFI Device Firmware…                                   ]
Scheduling…              [***************************************]
failed to write data to efivarfs: Error writing to file descriptor: No space left on device
```

Will try if the issue resolves after a reboot.

caugner, 2020-06-18:
**Update**: After a reboot, the UEFI Device Firmware installed properly! :tada: 

Thank you so much @hughsie and @vathpela! :1st_place_medal:

closed 2020-06-18 by caugner

hughsie, 2020-06-19:
@vathpela do you think fwupd should automatically delete the dumps if out of space? Or I guess we can give some sort of better error message back if out of space and a lot of different dumps exist.

vathpela, 2020-06-19:
Yes. Imnsho efivars pstore dumps are a "look what I can do" mistake, and they always fail this way. Distros that aren't LFS-like universally turn them off, and reports like this are a result of source distros being irresponsibly cavalier about their firmware interactions.

hughsie, 2020-06-19:
@vathpela so deleting anything `dump-type0*` unconditionally doesn't sound like it's a long term solution. Is there anything we can return to the user in the error message to say "XXX is enabled, see $THISURL for how to disable it"? Thanks.

caugner, 2020-06-19:
Can those dumps be disabled in `daemon.conf` or `uefi.conf`?

If so, I could add that option to the NixOS package I'm using: https://github.com/NixOS/nixpkgs/blob/release-20.03/nixos/modules/services/hardware/fwupd.nix#L100

ChauhanT, 2022-02-17:
Hi, faced this issue on Debian 11 and trashed the dump-* files on instinct (really, really bad practice, I know =.="). Worked after reboot. I don't know enough about bootup, but is this something that should be reported upstream to Debian ? If so, any tips/suggestions on where to report would be awesome. Thanks!