fwupd/firmware-lenovo #54: X1 Carbon Gen8 won't update Embedded Controller firmware

inffy, 2020-06-12:
Describe the bug
I have firmware updates offered when I run fwupmgr :

> fwupdmgr get-updates
> • Thunderbolt Controller has no available firmware updates
> • Prometheus has no available firmware updates
> • Prometheus IOTA Config has no available firmware updates
> • SAMSUNG MZVLB512HBJQ-000L7 has no available firmware updates
> • System Firmware has the latest available firmware version
> • Touchpad has the latest available firmware version
> • UEFI Device Firmware has no available firmware updates
> • UEFI Device Firmware has no available firmware updates
> 20U9CTO1WW
> │
> └─Embedded Controller:
>   │   Device ID:           0b3db288d2c013fcac57525c52aa6dc2c9934157
>   │   Current version:     0.1.2
>   │   Minimum Version:     0.1.2
>   │   Vendor:              DMI:LENOVO
>   │   Update State:        success
>   │   GUID:                1ebe8ad3-3d40-4178-92ce-bbb3d0906332
>   │   Device Flags:        • Internal device
>   │                        • Updatable
>   │                        • Requires AC power
>   │                        • Supported on remote server
>   │                        • Needs a reboot after installation
>   │                        • Device is usable for the duration of the update
>   │ 
>   └─ThinkPad X1 Carbon Gen 8 /X1 Yoga Gen 5 Embedded Controller Update:
>         New version:       1.0.02
>         Remote ID:         lvfs
>         Summary:           Lenovo ThinkPad X1 Carbon Gen 8 /X1 Yoga Gen 5 Embedded Controller Firmware
>         License:           Proprietary
>         Size:              767,1 kB
>         Created:           2020-04-17
>         Urgency:           High
>         Vendor:            Lenovo Ltd.
>         Flags:             is-upgrade
>         Description:       Lenovo ThinkPad X1 Carbon Gen 8 /X1 Yoga Gen 5 Embedded Controller Firmware 1.02. The computer will be restarted automatically after updating EC completely. Do NOT turn off your computer or remove the AC adapter while the update is in progress.
> 

After fwupdmgr update laptop asks to reboot. I reboot and laptop first goes to the lenovo logo and in the up corner says has found new capsules. After that it seems to start flashing the new firmware (I can see a progress bar fill up) and after completion restart again and show "Flashing Embedded controller" screen and then after completing reboots normally to Arch.

I then checked fwupdmgr again and it again shows and wants to update the same firmware

**Steps to Reproduce**
sudo fwupmgr update
reboot

**Expected behavior**

After the reboot running the update command returns that I'm up to date.

**fwupd version information**
Please provide the version of the daemon and client.

> client version:	1.4.2
> compile-time dependency versions
> 	gusb:	0.3.4
> 	efivar:	37
> daemon version:	1.4.2
> 

Installed through pacman

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

> 20U9CTO1WW
> │
[... 611 lines cut ...]
> │       ├── zfsinfo.mod
> │       ├── zfs.mod
> │       └── zstd.mod
> ├── initramfs-linux-fallback.img
> ├── initramfs-linux.img
> ├── intel-ucode.img
> └── vmlinuz-linux

**Additional questions**

    Operating system and version:
    Arch Linux (EndeavourOS)
    Have you tried rebooting?

many times

    Are you using an NVMe disk?

Yes

    Is secure boot enabled?
    
No

The firmware in question: https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN2WET.firmware

ChiWei-Chen, 2020-06-19:
@inffy 
Fixed package already uploaded to LVFS.
https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN2WHT.firmware
The latest version is N2WEHT08W(1.02) which is also the production level version. 
If it was the one installed on your X1 Carbon 8th the system, there is no need to update for now. 
Could you check with it again? Thanks.

inffy, 2020-06-19:
This is fixed now

closed 2020-06-19 by inffy