fwupd/firmware-lenovo #55: fwupd does not install anything on Fedora 32

xvitaly, 2020-06-12:
**Describe the bug**
Fwupd does not install anything. It found updates for my ThinkPad T480 (Intel ME update and UEFI BIOS firmware update) and requested a reboot.

After rebooting, I see only the message "Adding update capsule", then system hangs for 2-3 minutes and reboots again. No updates have been installed.

After booting back to Fedora fwupd will find these updates again and again.

Log:

```text
$ fwupdmgr update 
• Thunderbolt Controller has the latest available firmware version
• Embedded Controller has the latest available firmware version
Upgrade available for Intel Management Engine from 184.70.3626 to 184.77.3664
20L50007RT must remain plugged into a power source for the duration of the update to avoid damage. Continue with update? [Y|n]: y
Downloading 184.77.3664 for Intel Management Engine...
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Updating Intel Management Engine…                                ]
Scheduling…              [***************************************]
Successfully installed firmware
• SAMSUNG MZVLW256HEHP-000L7 has no available firmware updates
Upgrade available for System Firmware from 0.1.31 to 0.1.33
20L50007RT must remain plugged into a power source for the duration of the update to avoid damage. Continue with update? [Y|n]: y
Downloading 0.1.33 for System Firmware...
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Updating System Firmware…[          |                            ]
Scheduling…              [***************************************]
Successfully installed firmware: TPM PCR0 differs from reconstruction, please see https://github.com/fwupd/fwupd/wiki/TPM-PCR0-differs-from-reconstruction
• UEFI Device Firmware has no available firmware updates
• UEFI Device Firmware has no available firmware updates
• UEFI Device Firmware has no available firmware updates

An update requires a reboot to complete. Restart now? [y|N]:
```

**Steps to Reproduce**
1. fwupdmgr refresh
2. fwupdmgr update

**Expected behavior**
Successful update.

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version
client version: 1.4.4
compile-time dependency versions
        gusb:   0.3.4
        efivar: 37
daemon version: 1.4.4
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): dnf.

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
20L50007RT
│
├─Thunderbolt Controller:
│     Device ID:           c994c96d4262316c36825e4480dc79e618baa1e7
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     20.00
│     Vendor:              Lenovo (TBT:0x0109)
│     GUIDs:               fe328c2d-5300-597b-a21d-991116db8ab6
[... 218 lines cut ...]
│           └── fireworks.png
├── initramfs-0-rescue-d81e28cb17d24c1e9176a83c32f34d23.img
├── initramfs-5.6.14-300.fc32.x86_64.img
├── initramfs-5.6.16-300.fc32.x86_64.img
├── loader
│   └── entries
│       ├── d81e28cb17d24c1e9176a83c32f34d23-0-rescue.conf
│       ├── d81e28cb17d24c1e9176a83c32f34d23-5.6.14-300.fc32.x86_64.conf
│       └── d81e28cb17d24c1e9176a83c32f34d23-5.6.16-300.fc32.x86_64.conf
├── lost+found
├── System.map-5.6.14-300.fc32.x86_64
├── System.map-5.6.16-300.fc32.x86_64
├── vmlinuz-0-rescue-d81e28cb17d24c1e9176a83c32f34d23
├── vmlinuz-5.6.14-300.fc32.x86_64
└── vmlinuz-5.6.16-300.fc32.x86_64

18 directories, 41 files
```

**Additional questions**
- Operating system and version: Fedora 32.
- Have you tried rebooting? Yes.
- Is this a regression? Yes.
- Are you using an NVMe disk? Yes.
- Is secure boot enabled? Yes.

johannbg, 2020-06-12:
Same issue for Lenovo ThinkPad X1 Carbon 6th on F30 when trying to upgrade to 184.77.3664 ( seems to be the same update as well ) except I see the Lenovo Logo and installing firmware for a second before it drops out and reboots so presumably the firmware update itself is b0rked...
Fedora 30
FWUP version 1.2.12

johannbg, 2020-06-12:
@hughsie at quick glance issues #53 #54 #55 #56 #57 #58 and probably more are all duplicates are you sure that the update itself ( 184.77.3664 ) is not broken and dont you want to merge all these issues to one? ( from my pov this is already getting out of hand, the team will end up spending more time going through the backlog responding/closing duplicates than doing meaningful work + duplicates can overshadow other issues)

hughsie, 2020-06-13:
Deduping with https://github.com/fwupd/firmware-lenovo-thinkpad/issues/56 as it has more debugging there.

closed 2020-06-13 by hughsie