fwupd/firmware-lenovo #56: My lenovo X1 reboots and updates are not applied.

lhirlimann, 2020-06-12:
**Describe the bug**
I have firmware updates offered when I run fwupmgr :
```
[ludovic@saraan]~% sudo fwupdmgr update 
• Thunderbolt Controller has the latest available firmware version
• Embedded Controller has the latest available firmware version
Upgrade available for Intel Management Engine from 184.70.3626 to 184.77.3664
20KHCTO1WW must remain plugged into a power source for the duration of the update to avoid damage. Continue with update? [Y|n]: y
Downloading 184.77.3664 for Intel Management Engine...
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Updating Intel Management Engine…                                ]
Scheduling…              [***************************************]
Successfully installed firmware
• SAMSUNG MZVLB512HAJQ-000L7 has the latest available firmware version
Upgrade available for System Firmware from 0.1.46 to 0.1.48
20KHCTO1WW must remain plugged into a power source for the duration of the update to avoid damage. Continue with update? [Y|n]: y
Downloading 0.1.48 for System Firmware...
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Updating System Firmware…[            /                          ]
Scheduling…              [***************************************]
Successfully installed firmware

An update requires a reboot to complete. Restart now? [y|N]: n
```
When I reboot, I see messages about new capsule being found - it takes a while and then boot continues. firmware update don't seem to have been applied.

**Steps to Reproduce**
sudo fwupmgr update
reboot

**Expected behavior**

After the reboot running the update command returns that I'm up to date.

**fwupd version information**
Please provide the version of the daemon and client.
```shell
[ludovic@saraan]~% fwupdmgr --version
client version:	1.4.2
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.4.2
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

dnf with enable-testing

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
[ludovic@saraan]~% fwupdmgr get-devices --show-all-devices
20KHCTO1WW
│
├─Thunderbolt Controller:
│     Device ID:           199c280418de0aa1347fc161a9df3c68dd5b0848
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     43.00
│     Vendor:              Lenovo (TBT:0x0109)
│     GUIDs:               2da42a33-cd30-5ef5-a8fb-2c800a4b760f
│                          96411853-9f56-53d2-b05c-a54fcad5ee62
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Device stages updates
[... 232 lines cut ...]
├── vmlinuz-0-rescue-881a595ec75941a3b85457f68d10cd38
├── vmlinuz-5.6.11-300.fc32.x86_64
├── vmlinuz-5.6.12-300.fc32.x86_64
└── vmlinuz-5.6.13-300.fc32.x86_64

6 directories, 77 files
```

**Additional questions**
- Operating system and version:
Fedora 32 + some updates-testing
- Have you tried rebooting?

twice

- Is this a regression?

I think so.

- Are you using an NVMe disk?

I believe so.

- Is secure boot enabled?
I think so.

johannbg, 2020-06-12:
(X1CG6) Got a start invalid parameter error. Did a factory bios configuration reset and still got the error so that workaround does not work. The boot order lock was not enable as well.

xvitaly, 2020-06-13:
How can I remove these broken UEFI capsules and UEFI Boot entry?

kmauleon, 2020-06-15:
> Perhaps it will be useful to collate the following information from the material people have posted on this page.
> 
> **Machines affected**
> 
> * T480
> * X1CG6
> * X1CG7
> 
> **OSs affected**
> 
> * Arch
> * Fedora
> * Mint
> * NixOS
> * Pop
> * Ubuntu
> 
> **Has this problem occurred before, with Lenovo computers?** Yes
> 
> **Reports of whether the affected machine has 'boot order lock' set in the BIOS**
> _Set/enabled_: none
> _Not set/disabled_: five [but I am going to stop updating this total now]
> 
> EDITED for grammar and to add information about 'boot order lock'.

Hi,
Are you sure X1CG7 has this problem?  MEFW version of this system should be 192.xx.xxxx
And I have not set the latest version to stable yet... please confirm...Thank you very much.

daenney, 2020-06-15:
Looks like the update has been pulled? I'm no longer seeing it when running `fwupdmgr get-updates` for my X1C6.

LordChaos73, 2020-06-15:
> Looks like the update has been pulled? I'm no longer seeing it when running `fwupdmgr get-updates` for my X1C6.

Yes, this seems the cases indeed.

vancanhuit, 2020-06-15:
That's weird. The latest firmware version should be `184.77.3664`. After running `fwupdmgr refresh --force` and `fwupdmgr update`, I don't see any updates.

hughsie, 2020-06-15:
> Looks like the update has been pulled

Lenovo are investigating.

zalameedi, 2020-06-16:
Same problem.. T480. Thanks to those who made Lenovo aware!

keithy999, 2020-06-16:
FWIW, after running fwupdmgr refresh --force and fwupdmgr update I still see the update and it still goes through the motions but doesn't install.  Running arch on X1CG6

vancanhuit, 2020-06-22:
Great! I've successfully installed the latest update of the Intel Platform Update (version184.77.3664)  on my ThinkPad T480.

lhirlimann, 2020-06-22:
ditto. Report sent.

closed 2020-06-22 by lhirlimann

LordChaos73, 2020-06-22:
I just did a refresh, there seem to be be no update yet (X1CG6):

• Intel Management Engine has the latest available firmware version

Current version:     184.70.3626
Minimum Version:     184.70.3626

LordChaos73, 2020-06-23:
A new firmware version was uploaded, which seems to work fine now:

├─Intel Management Engine:
│     Device ID:           29749712e93b58730ecd1b8537a5503ff5b7fb27
│     Current version:     184.77.3664
│     Minimum Version:     184.77.3664
│     Vendor:              DMI:LENOVO
│     GUID:                9c9d9769-32fa-4841-b550-ea998e754e99
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update

shyukri, 2020-06-25:
I've manually downloaded the update for my t480 from https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN24RN.firmware and 
```
sudo fwupdmgr install *.cab
``` 
did the work

ackerleytng, 2020-06-28:
I did

```
sudo fwupdmgr refresh --force
sudo fwupdmgr update
```

and it still didn't update, but the following did result in a successful update

```
sudo fwupdmgr clear-history
sudo fwupdmgr clear-cache
sudo rm -rf /root/.cache/fwupd
sudo fwupdmgr refresh --force
sudo fwupdmgr update
```

HLeithner, 2020-06-30:
Selecting "Linux Firmware Updater" as Boot device options worked for me on a T470

fcelda, 2020-06-30:
This issue is resolved for me (X1 Carbon). It seems it really was a faulty update package.

```
Devices that have been updated successfully:

 • Intel Management Engine (184.70.3626 → 184.77.3664)
```

markuspeloquin, 2020-07-01:
**Lenovo ThinkPad T480s Corporate ME Firmware**, Fedora 31, fwupdmgr 1.3.10

I have never been able to update this particular firmware and nobody's solutions above have helped:

* Disable boot order lock (nope, already disabled)
* Tried downloading the cab and `fwupdmgr install *.cab`: https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN22RM.firmware
* Boot into windows 10 and install [Intel Management Engine 11.8 Firmware](https://pcsupport.lenovo.com/us/en/products/laptops-and-netbooks/thinkpad-t-series-laptops/thinkpad-t480s-type-20l7-20l8/downloads/driver-list/component?name=Chipset)
* Set BIOS settings to defaults
* Clearing /root/.cache/fwupd, ~/.cache/fwupd, which I didn't really think would fix anything anyway

I've tried tons of stuff in the past. Removing the UEFI fw update boot entries. Select the firmware update from the boot device menu. Downgrading (which didn't change anything at all). Reinstalling current version.

```
% fwupdmgr --version
client version:	1.3.10
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.3.10
```

kmauleon, 2020-07-01:
> I have the same issue on my T480s (it says _Adding new capsule_ but after reboot the update is shown again). [Someone on reddit said that it may have to do with deactivated IME](https://www.reddit.com/r/LinuxOnThinkpad/comments/gpbuiz/cannot_update_intel_management_engine_on_t480s/frndiav?utm_source=share&utm_medium=web2x).

Hi ...I referenced a new issue for this one for proper tracking... thank you very much....https://github.com/fwupd/firmware-lenovo-thinkpad/issues/61#issue-649397218

keithy999, 2020-07-02:
Hmm. tried all suggestions above on my thinkpad X1C6 to no avail

If I enable IME in BIOS, laptop won't boot :(

Running up to date arch linux

kmauleon, 2020-07-02:
> Hmm. tried all suggestions above on my thinkpad X1C6 to no avail
> 
> If I enable IME in BIOS, laptop won't boot :(
> 
> Running up to date arch linux

hi keithy,
sorry but please try the following again... this should work since new capsule was uploaded already...
please feedback the result of the following commands... thank you very much....

fwupdmgr refresh --force
fwupdmgr get-devices

belminf, 2020-07-05:
On my Thinkpad X1 C6, I am still experiencing the same issues as I reported #58. However, now it's attempting to upgrade from 184.55.3510 to 184.77.3664.

I tried @ackerleytng suggestion of:

```
sudo fwupdmgr clear-history
# The following did not work though, commented out
# sudo fwupdmgr clear-cache
sudo rm -rf /root/.cache/fwupd
sudo fwupdmgr refresh --force
sudo fwupdmgr update
```

After a reboot, I still did not have the update apply. Any hints or request for additional data points to help narrow this down would be appreciated.

kmauleon, 2020-07-06:
hi @belminf 
upon checking from your previous report #58 , I noticed that system is showing 2 GUIDs for MEFW... this is weird since it should only have 1 GUID... can you please confirm ...Thank you very much...

fwupdmgr get-devices

belminf, 2020-07-06:
Thanks for taking a look @kmauleon.

Seems like there's only 1 GUID now although I don't know what changed since I submitted that issue. Here's the output of `fwupdmgr get-devices --show-all-devices`. LMK if anything else would be helpful:

```
20KH002KUS
│
├─Thunderbolt Controller:
│     Device ID:           76755e859785d05252ff0ed3740d95e4dea9e1f7
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     43.00
│     Vendor:              Lenovo (TBT:0x0109)
│     GUIDs:               5ff9544e-fce2-5fd9-9cd2-671c95ac91dc
│                          b8360cb5-b152-506a-9990-49d7bbbc3b88
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Device stages updates
│   
├─USB3.0 Hub:
│     Device ID:           da77984c82b59c6fc69516431f467fd9a8d39a7f
│     Summary:             USB 3.x Hub
│     Current version:     3.83
│     Vendor:              VIA Labs, Inc. (USB:0x2109)
│     Install Duration:    15 seconds
│     GUIDs:               95d09348-73f7-5964-ae90-54c6b1ad5bbf
│                          26f33695-3a3e-5c08-badb-f6141390ebd9
│                          10eb3a15-c177-5810-af53-1963e9200e65
│                          9b27397e-3d9d-5466-bb1c-592066d7a037
│                          9e4db2b9-3a4f-5548-84e7-d8f71b68c1bb
│                          022d2f73-4826-546a-ba0f-62579ea848ea
│     Device Flags:        • Updatable
│                          • Cryptographic hash verification is available
│                          • Device stages updates
│                          • Device can recover flash failures
│   
├─Embedded Controller:
│     Device ID:           9698faabddf0d7b18925cfbbda95f8b0d0dacc53
│     Current version:     0.1.20
│     Minimum Version:     0.1.20
│     Vendor:              DMI:LENOVO
│     GUID:                3babca5f-b2bf-4f4b-a72e-2bdc84eb4019
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─Intel AMT [unprovisioned]:
│     Device ID:           e2623122c99d58220498aacbfcfdb1baebbae3c5
│     Summary:             Hardware and firmware technology for remote out-of-band management
│     Current version:     11.8.55.3510
│     Bootloader Version:  11.8.55.3510
│     Vendor:              Intel Corporation
│     GUID:                2800f812-b7b4-2d4b-aca8-46e0ff65814c
│     Device Flags:        • Internal device
│   
├─Intel Management Engine:
│     Device ID:           e563ad307df81c99f0de8c26292afd71cf409673
│     Current version:     184.55.3510
│     Minimum Version:     0.0.1
│     Vendor:              DMI:LENOVO
│     Update State:        success
│     GUID:                42a0a96e-c9f3-438f-9687-7826be33e4ce
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
[... 76 lines cut ...]
│     Vendor:              VIA Labs, Inc. (USB:0x2109)
│     Install Duration:    15 seconds
│     GUIDs:               db31bd61-b70f-553d-a94f-c579fa11bf40
│                          3f0581b6-15ed-5f87-91eb-296db71e0838
│                          4ae82b4d-0b4c-537a-b242-f689010b7571
│                          21f194f1-3152-5555-b816-a7596f49e897
│                          bd26f184-e083-5c0c-89c0-0b84d5092660
│                          8453111a-0e73-51f4-877f-fa340d8b2c03
│     Device Flags:        • Updatable
│                          • Cryptographic hash verification is available
│                          • Device stages updates
│                          • Device can recover flash failures
│   
└─Unifying Receiver:
      Device ID:           159fd929f7a8953aa5c72ab84ca4a0dfb570a304
      Summary:             A miniaturised USB wireless receiver
      Current version:     RQR12.09_B0030
      Bootloader Version:  BOT01.04_B0016
      Vendor:              USB:0x046D
      Install Duration:    30 seconds
      GUIDs:               9d131a0c-a606-580f-8eda-80587250b8d6
                           fcf55bf5-767b-51ce-9c17-f6f538c4ee9f
                           279ed287-3607-549e-bacc-f873bb9838c4
      Device Flags:        • Updatable
    ```

keithy999, 2020-07-06:
> hi keithy,
> sorry but please try the following again... this should work since new capsule was uploaded already...
> please feedback the result of the following commands... thank you very much.... [...]

here you go:

fwupdmgr refresh --force
Fetching signature https://cdn.fwupd.org/downloads/firmware.xml.gz.jcat
Downloading…             [***************************************]
Fetching metadata https://cdn.fwupd.org/downloads/firmware-00674-stable.xml.gz
Downloading…             [***************************************]
Successfully downloaded new metadata: 4 local devices supported
[keithy@keith-pc-endeavour ~]$ fwupdmgr get-devices
20KHCTO1WW
│
├─Thunderbolt Controller:
│     Device ID:           b023070274a3e356d02ca6222c1665c4eebbbe8c
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     43.00
│     Vendor:              Lenovo (TBT:0x0109)
│     GUIDs:               5ff9544e-fce2-5fd9-9cd2-671c95ac91dc
│                          b8360cb5-b152-506a-9990-49d7bbbc3b88
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Device stages updates
│   
├─Embedded Controller:
│     Device ID:           9698faabddf0d7b18925cfbbda95f8b0d0dacc53
│     Current version:     0.1.20
│     Minimum Version:     0.1.20
│     Vendor:              DMI:LENOVO
│     GUID:                3babca5f-b2bf-4f4b-a72e-2bdc84eb4019
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─INTEL SSDPEKKF256G8L:
│     Device ID:           04e17fcf7d3de91da49a163ffe4907855c3648be
│     Summary:             NVM Express Solid State Drive
│     Current version:     L08P
│     Vendor:              Intel Corporation (NVME:0x8086)
│     GUIDs:               79517f86-8df8-5d6e-a18b-33f0b36a78e9
│                          68db11e5-b0cf-5bc9-a94e-17e28496e505
│                          cdc79848-b022-5a05-a77a-862f21df0c29
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─Intel Management Engine:
│     Device ID:           e563ad307df81c99f0de8c26292afd71cf409673
│     Current version:     184.65.3590
│     Minimum Version:     184.65.3590
│     Vendor:              DMI:LENOVO
│     Update State:        success
│     GUID:                42a0a96e-c9f3-438f-9687-7826be33e4ce
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
└─System Firmware:
      Device ID:           1c53551e7da69d896138fac1ae131c83ad46d923
      Current version:     0.1.48
      Minimum Version:     0.0.1
      Vendor:              LENOVO (DMI:LENOVO)
      GUID:                a4b51dca-8f97-4310-8821-3330f83c9135
      Device Flags:        • Internal device
                           • Updatable
                           • Requires AC power
                           • Supported on remote server
                           • Needs a reboot after installation
                           • Cryptographic hash verification is available
                           • Device is usable for the duration of the update

kmauleon, 2020-07-06:
> 42a0a96e-c9f3-438f-9687

thank you very much @keithy999 ... please make sure the following are updated
     shim / shim-signed
     fwupd
     fwupdate

then please try to execute below command again
     fwupdmgr update

kmauleon, 2020-07-06:
> Thanks for taking a look @kmauleon.
>
> Seems like there's only 1 GUID now although I don't know what changed since I submitted that issue. Here's the output of `fwupdmgr get-devices [...]

hi @belminf ...thank you very much  ... please make sure the following are updated
     shim / shim-signed
     fwupd
     fwupdate

then please try to execute below commands again
    fwupdmgr refresh
    fwupdmgr update

keithy999, 2020-07-07:
> > 42a0a96e-c9f3-438f-9687
> 
> thank you very much @keithy999 ... please make sure the following are updated
> shim / shim-signed
> fwupd
> fwupdate
> 
> then please try to execute below command again
> fwupdmgr update

OK, can't find (Arch) package for shim

shim-signed wasn't installed - so I did!

fwupd was already up to date

can't find package for fwupdate

tried running the commands again - still goes through the motions but doesn't get installed :(

kmauleon, 2020-07-08:
> OK, can't find (Arch) package for shim
>
> shim-signed wasn't installed - so I did!
>
> fwupd was already up to date
>
> can't find package for fwupdate
>
> tried running the commands again - still goes through [...]

hi @keithy999 
fwupdate is required but i am not familiar with arch linux T_T

kmauleon, 2020-07-08:
hi @keithy999 
can you please check if this is helpful >> https://daenney.github.io/2018/04/07/arch-linux-firmware-updates
thanks!

belminf, 2020-07-08:
@kmauleon I had the same results as @keithy999--i.e., no difference was made when I ran through the steps you shared.

I looked at the [webpage you shared ](https://daenney.github.io/2018/04/07/arch-linux-firmware-updates). A few notes:

* My ESP is `/boot`.
* The webpage mentions installing `fwupdate` but I don't see that package. I do have `fwupd` install.
* Neither`/usr/lib/fwupdate/EFI` nor `/usr/share/doc/fwupdate/esp-as-boot.hook` exist on my system.

Can you advise? And thank you for helping us with this!

kmauleon, 2020-07-08:
hi @belminf sorry not really familiar with arch linux..... but let me pull in some help ^_^
hi @hughsie sorry but any advise with @belminf's case....how can he install fwupdate in arch linux?
thank you very much in advanced....

vladbejenaru, 2020-07-21:
I have similar behaviour on 2 machines: a T460s and a T480. Would be glad to provide more details if required:
```
client version:	1.3.11
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.3.11
```

kmauleon, 2020-07-22:
> I have similar behaviour on 2 machines: a T460s and a T480. Would be glad to provide more details if required:
> 
> ```
> client version:	1.3.11
> compile-time dependency versions
> 	gusb:	0.3.4
> 	efivar:	37
> daemon version:	1.3.11
> ```

hi @vladbejenaru sorry but can you please give more details for T460s.... thank you very much...
OS >>
result of this command >> fwupdmgr get-devices

vladbejenaru, 2020-07-22:
hi @kmauleon . Here you go:

```
20FAS3U001
│
├─System Firmware:
│     Device ID:           444a229da2b21bc0d24faf814d1c58c5bff3315f
│     Current version:     0.1.49
│     Minimum Version:     0.1.22
│     Vendor:              LENOVO (DMI:LENOVO)
│     GUID:                42f29620-3d63-4f09-950b-9b7055570f28
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
│   
├─TOSHIBA THNSFJ256GDNU A:
│     Device ID:           e421b2fc248391f6fe3e55ddbb3c9043be068bd0
│     Summary:             ATA Drive
│     Current version:     JYLA1102
│     GUIDs:               f260d559-ffd1-56ce-88a6-fc38c53d8e9d
│                          9bc9dfb3-2789-5a46-afff-739966fddfd9
│                          9d768a22-7625-501d-bfa7-7155287899e0
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           aff24b50c0a8c292c6c2a81d3827c36d7a86f62a
│     Current version:     176.16.1000
│     Minimum Version:     0.0.1
│     Vendor:              DMI:LENOVO
│     GUID:                03d3297b-3851-4379-95ad-12e21a96c80a
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           33bc73bf0f48eb6177a007e9351e22d5bce2c04d
│     Current version:     0.1.14
│     Minimum Version:     0.0.1
│     Vendor:              DMI:LENOVO
│     GUID:                ffaf3e12-a6ea-4352-851b-76c6455913c6
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
└─Unifying Receiver:
      Device ID:           77a8cb94164eb0e0db95b8da7db67527f4d238aa
      Summary:             A miniaturised USB wireless receiver
      Current version:     RQR12.10_B0032
      Bootloader Version:  BOT01.02_B0014
      Vendor:              USB:0x046D
      Install Duration:    30 seconds
      GUIDs:               9d131a0c-a606-580f-8eda-80587250b8d6
                           fcf55bf5-767b-51ce-9c17-f6f538c4ee9f
                           279ed287-3607-549e-bacc-f873bb9838c4
      Device Flags:        • Updatable
                           • Supported on remote server
```

kmauleon, 2020-07-23:
> hi @kmauleon . Here you go. OS is siduction, debian based on sid.
> 
> ```
> 20FAS3U001
> │
> ├─System Firmware:
> │     Device ID:           444a229da2b21bc0d24faf814d1c58c5bff3315f
> │     Current version:     0.1.49
> │     Minimum Version:     0.1.22
> │     Vendor:              LENOVO (DMI:LENOVO)
> │     GUID:                42f29620-3d63-4f09-950b-9b7055570f28
> │     Device Flags:        • Internal device
> │                          • Updatable
> │                          • Requires AC power
> │                          • Supported on remote server
> │                          • Needs a reboot after installation
> │                          • Cryptographic hash verification is available
> │                          • Device is usable for the duration of the update
> │   
> ├─TOSHIBA THNSFJ256GDNU A:
> │     Device ID:           e421b2fc248391f6fe3e55ddbb3c9043be068bd0
> │     Summary:             ATA Drive
> │     Current version:     JYLA1102
> │     GUIDs:               f260d559-ffd1-56ce-88a6-fc38c53d8e9d
> │                          9bc9dfb3-2789-5a46-afff-739966fddfd9
> │                          9d768a22-7625-501d-bfa7-7155287899e0
> │     Device Flags:        • Internal device
> │                          • Updatable
> │                          • Requires AC power
> │                          • Needs a reboot after installation
> │                          • Device is usable for the duration of the update
> │   
> ├─UEFI Device Firmware:
> │     Device ID:           aff24b50c0a8c292c6c2a81d3827c36d7a86f62a
> │     Current version:     176.16.1000
> │     Minimum Version:     0.0.1
> │     Vendor:              DMI:LENOVO
> │     GUID:                03d3297b-3851-4379-95ad-12e21a96c80a
> │     Device Flags:        • Internal device
> │                          • Updatable
> │                          • Requires AC power
> │                          • Supported on remote server
> │                          • Needs a reboot after installation
> │                          • Device is usable for the duration of the update
> │   
> ├─UEFI Device Firmware:
> │     Device ID:           33bc73bf0f48eb6177a007e9351e22d5bce2c04d
> │     Current version:     0.1.14
> │     Minimum Version:     0.0.1
> │     Vendor:              DMI:LENOVO
> │     GUID:                ffaf3e12-a6ea-4352-851b-76c6455913c6
> │     Device Flags:        • Internal device
> │                          • Updatable
> │                          • Requires AC power
> │                          • Supported on remote server
> │                          • Needs a reboot after installation
> │                          • Device is usable for the duration of the update
> │   
> └─Unifying Receiver:
>       Device ID:           77a8cb94164eb0e0db95b8da7db67527f4d238aa
>       Summary:             A miniaturised USB wireless receiver
>       Current version:     RQR12.10_B0032
>       Bootloader Version:  BOT01.02_B0014
>       Vendor:              USB:0x046D
>       Install Duration:    30 seconds
>       GUIDs:               9d131a0c-a606-580f-8eda-80587250b8d6
>                            fcf55bf5-767b-51ce-9c17-f6f538c4ee9f
>                            279ed287-3607-549e-bacc-f873bb9838c4
>       Device Flags:        • Updatable
>                            • Supported on remote server
> ```

hi @vladbejenaru 
T_T MEFW is very old.... initial release in LVFS was 184.55.3510 and those with version 184.50.3399 or higher can upgrade to this version.  Let me check with my colleagues on how we can resolve this.  Let me get back to you in a few days since they are on holidays currently.... please stay put... thank you very much....

kmauleon, 2020-07-28:
hi @vladbejenaru 
we are working on delivering you the older MEFW versions... this would take a while though since this isn't covered in our validation... will update you once we confirmed it's working on our side....Thank you very much.

pentatonicfunk, 2020-08-02:
Similar behavior on Thinkpad X1 Extreme 1st Gen. 
```
 hendrawan@fedora-thinkpad ❯ sudo fwupdmgr get-devices
20MF000BUS
│
├─Thunderbolt Controller:
│     Device ID:           33c9aaa39df63407868e077e39b6092e6e20f0e6
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     50.00
│     Vendor:              Lenovo (TBT:0x0109)
│     GUIDs:               b52afd68-e052-543c-92d5-de6c75349bb2 ← TBT-01091712-native
│                          f65fbe41-a3d5-530f-8636-55e4378fad4a ← TBT-01091712-native-controller0-0
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Device stages updates
│   
├─Embedded Controller:
│     Device ID:           6b3fcb33b92ea5cdd94954b276ad0b63ed14f9bb
│     Current version:     0.1.14
│     Minimum Version:     0.0.1
│     Vendor:              DMI:LENOVO
│     GUID:                74997a6b-1adf-4b12-b994-401f06ea8c72
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─Intel Management Engine:
│     Device ID:           bf8e97f32b43a34764c5200c7275e5254210924d
│     Current version:     192.35.1427
│     Minimum Version:     192.35.1427
│     Vendor:              DMI:LENOVO
│     Update State:        success
│     GUID:                b49ed72d-e186-4fa6-9761-3242da3c3be9
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─Samsung SSD 970 EVO Plus 500GB:
│     Device ID:           03281da317dccd2b18de2bd1cc70a782df40ed7e
│     Summary:             NVM Express Solid State Drive
│     Current version:     2B2QEXM7
│     Vendor:              Samsung Electronics Co Ltd (NVME:0x144D)
│     Serial Number:       S4EVNJ0N416584D
│     GUIDs:               6e54c992-d302-59ab-b454-2d26ddd63e6d ← NVME\VEN_144D&DEV_A808&REV_00
│                          47335265-a509-51f7-841e-1c94911af66b ← NVME\VEN_144D&DEV_A808
│                          173afe85-89dc-54b2-9a58-a5990e4e4b6a ← Samsung SSD 970 EVO Plus 500GB
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─System Firmware:
│     Device ID:           71519311a643244a2eaf392fbf17c968ec135dd3
│     Current version:     0.1.31
│     Minimum Version:     0.1.31
│     Vendor:              LENOVO (DMI:LENOVO)
│     GUID:                a95d69e7-2fa0-4849-9159-75645bc6e153
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           1a2df614d3488e4d0238fc21460adcbf6d256de2
│     Current version:     4784132
│     Minimum Version:     4784132
│     Vendor:              DMI:LENOVO
│     GUID:                cc426b6d-b6f9-496e-8676-82af7596d239
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
└─WDC PC SN720 SDAQNTW-512G-1001:
      Device ID:           c6a0cfba7c7d81e253fce571e1d1e9f6003ae1c7
      Summary:             NVM Express Solid State Drive
      Current version:     10190101
      Vendor:              Sandisk Corp (NVME:0x15B7)
      Serial Number:       184622423986
      GUIDs:               137520ce-3603-53e6-9165-56694ed744e7 ← NVME\VEN_15B7&DEV_5002&REV_00
                           ff2112dc-038c-596d-90ca-d43c5077c6ec ← NVME\VEN_15B7&DEV_5002
                           63af1441-fb24-5368-b450-886fbc98610b ← WDC PC SN720 SDAQNTW-512G-1001
      Device Flags:        • Internal device
                           • Updatable
                           • Requires AC power
                           • Supported on remote server
                           • Needs a reboot after installation
                           • Device is usable for the duration of the update
    
________________________________________________
```
Updates :
```
 hendrawan@fedora-thinkpad ❯ sudo fwupdmgr update         

• Thunderbolt Controller has the latest available firmware version
• Embedded Controller has the latest available firmware version
Upgrade available for Intel Management Engine from 192.35.1427 to 192.47.1524
20MF000BUS must remain plugged into a power source for the duration of the update to avoid damage. Continue with update? [Y|n]
```
Update and restart : 
- Boots show `Adding new capsule` `Reset system`
- After sometime it shows `Loading Txe image from capsule` `Error 0707 : Firmware update failed due to an internal error`

kmauleon, 2020-08-05:
hi @vladbejenaru ...old mefw versions have been uploaded as below so you could update to the latest... 

[image]

please execute the following... you may need to do this a few times since your current mefw version is very very old...thank you very much

fwupdmgr refresh
fwupdmgr update

vladbejenaru, 2020-08-12:
unfortunately I had to reinstall the OS and to move to Windows as the computer will change owners. I did not manage to test it... :-(
for what matters my new T480 that I will use makes this updates flawless from PopOS! Thanks for the good work and keep it up!

kmauleon, 2020-08-13:
@vladbejenaru haha ok no worries this will be helpful for those encountering the same concern with T460s ^_^

@lijunwang2020 can you please check issue with T480... thank you very much...

jorgelizandro, 2020-11-14:
I just got this issue with my Thinkpad X1 Carbon 5th Gen, the boot device is not locked down:
`fwupdmgr --version  
client version: 1.2.11
compile-time dependency versions
        gusb:   0.3.0
        efivar: 37
daemon version: 1.2.11

`

dkwo, 2020-11-21:
On my T460s with ME permanently disabled in BIOS, fwupd still suggests the update, and if I accept it and reboot, the system BIOS sees it and applies it, but when I run fwupd again after reboot, it says it was unable to apply ME update.
Shouldn't fwupd detect that the ME is disabled and just skip the update? or am I supposed to apply it anyway? in this case, why is it not working?

kmauleon, 2020-11-23:
hi @dkwo @jorgelizandro ... can you please open a new issue for this for proper tracking since this issue has been closed already. thank you very much.

federico-cuello-solarisbank, 2021-05-11:
This is still happening with `ThinkPad X1 Carbon 6th`,

```
$ fwupdmgr --version
client version: 1.5.8
compile-time dependency versions
        gusb:   0.3.5

daemon version: 1.5.8
```

`Upgrade available for Intel Management Engine from 184.79.3722 to 184.83.3874`

LinuxOnTheDesktop, 2021-05-13:
Cf. #3231.

Cf. also perhaps #3219 and possibly #683.

LinuxOnTheDesktop, 2021-08-01:
It is perhaps worth mentioning the following. _The relevant Lenovo 'pcsupport' webpages do not seem to acknowledge the existence_ of the following TPM update that is offered (as follows) by fwupd:

```
ThinkPad X1 Carbon 6th System Update:
        New version:       0.73.20
        Remote ID:         lvfs
        Summary:           Lenovo ThinkPad X1 Carbon 6th STM TPM Firmware
        License:           Proprietary
        Size:              439.6 kB
        Vendor:            Lenovo Ltd.
        Flags:             is-upgrade
        Description:       Lenovo ThinkPad X1 Carbon 6th STM TPM Firmware Version 73.20
```
It seems indeed that the Lenovo website offers that update _only to 4th and 5th generation X1s_.

zomp, 2021-08-04:
Today, Ubuntu 21.04 gave me `fwupd` and `fwupd-signed` updates and then the firmware was finally updated successfully. (ThinkPad X1 Carbon Gen 8)

niedzielski, 2022-01-09:
Thank you so much for fwupd!! I use it for everything.

I think I had this same issue on my ThinkPad X1 Yoga Gen 6 but for some reason the outdated firmware wasn't always reported so I mistakenly assumed it had upgraded a long time ago. By chance, I saw the outdated firmware in gnome-software's update tab.

What ultimately worked for me was disabling Secure Boot temporarily:
1. Disabled Secure Boot in BIOS settings (press enter at boot).
2. Boot.
3. Execute `sudo fwupdmgr update`.
4. Reboot and allow the firmware upgrade to be detected and automatically install. (This wasn't working when Secure Boot was enabled.)
5. Repeat steps 2-4 until no updates remain.
6. Re-enable Secure Boot.

peteryates, 2022-03-05:
@niedzielski - thanks, this worked for me too (X1 Carbon Gen 7)

LinuxOnTheDesktop, 2022-03-06:
Unfortunately, [niedzielski](https://github.com/niedzielski)'s putative fix does not work for me (X1 Carbon Gen 6).

ellisonleao, 2022-03-21:
I am getting this error on rebooting after the installation runs. I have also a X1 Carbon Gen 6. Any thoughts?

[image]

mrhpearson, 2022-03-22:
ShellFlash is what we use internally for updating the BIOS during the early stages. You shouldn't see if if you're installing a capsule update using LVFS as it's not packaged with the capsule etc.
Are you running from a USB stick or something like that?

ellisonleao, 2022-03-22:
@mrhpearson no, just tried to update with 

```
sudo fwupdmgr update
```

After the reboot, i am always getting this msg. I also have the secure boot disabled

shawnyeager, 2025-03-18:
I see this issue is closed, but I'm still encountering this problem as recently as today when attempting to install the [latest Intel ME update](https://pcsupport.lenovo.com/us/en/products/laptops-and-netbooks/thinkpad-x-series-laptops/thinkpad-x1-carbon-8th-gen-type-20u9-20ua/downloads/driver-list/component) on my X1 Carbon Gen 8.

Did someone find a fix, or did it age out, please?

LinuxOnTheDesktop, 2025-03-18:
&nbsp;&nbsp;&nbsp;&nbsp;I never found a fix. (Indeed: after multiple motherboard failures, farcically bad communication about those failures with Lenovo, and an offer of compensation upon which Lenovo partially reneged, I put aside my brick - as it had become - and bought a laptop from the company Framework. And, guess what, Framework is dire at firmware updates too, as is shown by a recent chorus of discontent on the Framework forum.)

kmauleon, 2025-03-18:
> I see this issue is closed, but I'm still encountering this problem as recently as today when attempting to install the [latest Intel ME [...]

@shawnyeager can you please open a separate ticket for proper tracking... thank you very much

shawnyeager, 2025-03-19:
> I never found a fix. (Indeed: after multiple motherboard failures, farcically bad communication about those failures with Lenovo, and an offer of compensation upon which Lenovo partially reneged, I [...]

Funny you should say. I bought a Framework 13" AMD last fall. Overall, I'm happy with it, but the way they handle firmware boggles.

shawnyeager, 2025-05-13:
@hughsie I've tried numerous times on the same machine with 2.0.9, and I'm still experiencing the same behavior as described above. I'll open a separate ticket.