fwupd/firmware-lenovo #58: Thinkpad X1 Carbon Gen 6 - Unable to apply Intel Management Engine updates

belminf, 2020-06-12:
**Describe the bug**

I perform an update and reboot but the update is still seen as available despite `Update State` showing `success` in `fwupdmgr get-updates` output.

This is to upgrade Intel Management Engine from 184.55.3510 to 184.70.3626.

**Steps to Reproduce**

1. Run `fwupdmgr refresh --force && fwupdmgr update`
2. Reboot
3. Check `fwupdmgr get-updates`

**Expected behavior**

I expect the update to apply or see something helpful in the debug output.

**fwupd version information**

```shell
$  fwupdmgr --version
client version: 1.4.0
compile-time dependency versions
        gusb:   0.3.4
        efivar: 37
daemon version: 1.4.0
```

Installed via `pacman`: https://www.archlinux.org/packages/?name=fwupd 

**fwupd device information**

```shell
$ fwupdmgr get-devices --show-all-devices

20KH002KUS
│
├─Thunderbolt Controller:
│     Device ID:           76755e859785d05252ff0ed3740d95e4dea9e1f7
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     43.00
│     Vendor:              Lenovo (TBT:0x0109)
│     GUIDs:               5ff9544e-fce2-5fd9-9cd2-671c95ac91dc
│                          b8360cb5-b152-506a-9990-49d7bbbc3b88
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Device stages updates
│   
├─Embedded Controller:
│     Device ID:           9698faabddf0d7b18925cfbbda95f8b0d0dacc53
│     Current version:     0.1.20
│     Minimum Version:     0.1.20
│     Vendor:              DMI:LENOVO
│     GUID:                3babca5f-b2bf-4f4b-a72e-2bdc84eb4019
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─Intel AMT [unprovisioned]:
│     Device ID:           e2623122c99d58220498aacbfcfdb1baebbae3c5
│     Summary:             Hardware and firmware technology for remote out-of-band management
│     Current version:     11.8.55.3510
│     Bootloader Version:  11.8.55.3510
│     Vendor:              Intel Corporation
│     GUID:                2800f812-b7b4-2d4b-aca8-46e0ff65814c
│     Device Flags:        • Internal device
[... 185 lines cut ...]

/boot/EFI
└── Arch
    ├── fw
    │   ├── fwupd-3b8c8162-188c-46a4-aec9-be43f1d65697.cap
    │   └── fwupd-42a0a96e-c9f3-438f-9687-7826be33e4ce.cap
    ├── fwupdx64.efi
    └── grubx64.efi
```

**Additional questions**
- Operating system and version: Archlinux
- Have you tried rebooting? Yes
- Is this a regression? No
- Are you using an NVMe disk? Yes
- Is secure boot enabled? No

**Additional info**

```
$ sudo fwupdate -L
../fwupd-1.4.0/plugins/uefi/efi/fwupdate.c:158:fwup_populate_update_table(): Found update fwupd-ux-capsule
../fwupd-1.4.0/plugins/uefi/efi/fwupdate.c:158:fwup_populate_update_table(): Found update fwupd-42a0a96e-c9f3-438f-9687-7826be33e4ce-0
../fwupd-1.4.0/plugins/uefi/efi/fwupdate.c:564:efi_main(): No updates to process.  Called in error
```

ackerleytng, 2020-06-13:
Same issue here. I was able to update all the other packages by installing them individually using `fwupdmgr install [FILE]`.

```
$ fwupdmgr get-updates
• Thunderbolt Controller has the latest available firmware version
• Embedded Controller has the latest available firmware version
• System Firmware has the latest available firmware version
• UEFI Device Firmware has no available firmware updates
• WDC PC SN720 SDAQNTW-512G-1001 has the latest available firmware version
20KHCTO1WW
│
└─Intel Management Engine:
  │   Device ID:           29749712e93b58730ecd1b8537a5503ff5b7fb27
  │   Current version:     184.70.3626
  │   Minimum Version:     184.70.3626
  │   Vendor:              DMI:LENOVO
  │   Update State:        success
  │   GUID:                9c9d9769-32fa-4841-b550-ea998e754e99
  │   Device Flags:        • Internal device
  │                        • Updatable
  │                        • Requires AC power
  │                        • Supported on remote server
  │                        • Needs a reboot after installation
  │                        • Device is usable for the duration of the update
  │ 
  └─ThinkPad X1 Carbon 6th Consumer ME Update:
        New version:       184.77.3664
        Remote ID:         lvfs
        Summary:           Lenovo ThinkPad X1 Carbon 6th Consumer ME Firmware
        License:           Proprietary
        Size:              2.2 MB
        Created:           2016-07-08
        Urgency:           High
        Vendor:            Lenovo Ltd.
        Flags:             is-upgrade
        Description:       Version 11.8.77.3664 (LVFS: 184.77.3664)
        
        2020.01 Q1 Intel Platform Update (IPU), formerly known as Quarterly security release (QSR)
        
        Several security fixes and enhancements are on this release.
        
        Security issues fixed:
        
         • CVE-2020-0531
         • CVE-2020-0532
         • CVE-2020-0533
         • CVE-2020-0535
         • CVE-2020-0536
         • CVE-2020-0537
         • CVE-2020-0538
         • CVE-2020-0539
         • CVE-2020-0540
         • CVE-2020-0545
         • CVE-2020-0594
         • CVE-2020-0595
         • CVE-2020-0596
         • CVE-2020-8674
```

My boot order lock is disabled, and I'm using an NVMe disk.

hughsie, 2020-06-13:
Deduping with https://github.com/fwupd/firmware-lenovo-thinkpad/issues/56 as it has more debugging there.

closed 2020-06-13 by hughsie