fwupd/firmware-lenovo #59: Firmware update fails with "start_image() returns unsupported" on Lenovo T470s

aravindhp, 2020-06-12:
```shell
$ fwupdmgr --version
client version:	1.3.3
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.3.3
```

```shell
$ fwupdmgr get-devices
20HGS22D00
│
├─Thunderbolt Controller:
│     Device ID:           9829435011463170d5c4af2a48d441d23f4f7a2e
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     20.00
│     Vendor:              Lenovo (TBT:0x0109)
│     GUIDs:               96c3f95f-611a-56f8-be95-0f912cb84657 ← TBT-01091609
│                          e2b1fec6-8e86-51f6-89a6-afd49e79c64b ← TBT-01091609-0000:03:00.0
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Device stages updates
│   
├─Samsung SSD 960 EVO 500GB:
│     Device ID:           e11623b2caa18fee292058a5c09ca4e6152f7ecf
│     Summary:             NVM Express Solid State Drive
│     Current version:     3B7QCXE7
│     Vendor:              Samsung Electronics Co Ltd (NVME:0x144D)
│     Serial Number:       S3X4NB0K232179V
│     GUIDs:               5b3df2da-f745-5fd0-81de-5dafd7f0bf8c ← NVME\VEN_144D&DEV_A804
│                          841f7890-a450-5aa7-8e16-9a4ad2bffedb ← Samsung SSD 960 EVO 500GB
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─System Firmware:
│     Device ID:           84149fb78009cf27cc1dd5520911f46f8792dbfe
│     Current version:     0.1.35
│     Minimum Version:     0.1.7
│     Vendor:              LENOVO
│     GUID:                7a176688-0960-47ba-931b-7829849e8347
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           cbede50560aaa37366326cd54923421517427f39
│     Current version:     184.50.3425
│     Minimum Version:     0.0.1
│     Update State:        success
│     GUID:                e9124c4a-fdff-42e5-b2ad-f745db345953
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
└─UEFI Device Firmware:
      Device ID:           03a6930fc4c833443a7502dde379b98b961c8c57
      Current version:     0.1.20
[... 120 lines cut ...]
20HGS22D00 must remain plugged into a power source for the duration of the update to avoid damage. Continue with update? [Y|n]: Y
Downloading 184.70.3626 for UEFI Device Firmware...
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Updating UEFI Device Firmware…                                   ]
Scheduling…              [***************************************]
Successfully installed firmware
No upgrades for UEFI Device Firmware, current is 0.1.20: 0.1.20=same

An update requires a reboot to complete. Restart now? [Y|n]: Y
```

On the reboot, I see the following which I have transcribed below:
```
dpath: \EFI\fedora\shimx64.efi
path: \fwupdx64.efi
Found update fwupd-e9124c4a-fdff-42e5-b2ad-f745db345953-0
Adding new capsule
Adding new capsule
Adding new capsule
WARNING: Could not query capsule capabilities: Unsupported
WARNING: Could not apply capsules: Unsupported
start_image() returned Unsupported
```
On rebooting I see the same available firmware update. Is there a workaround or fix for this?

kmauleon, 2020-07-01:
Hi....
ThinkPad T480s Latest Version: 184.77.3664 has been reuploaded... please check... thank you very much.

aravindhp, 2020-07-06:
I am still seeing the same issue on a Lenovo T470s

kmauleon, 2020-07-06:
> I am still seeing the same issue on a Lenovo T470s

hi @aravindhp ...called the attention of the owner... please wait for update...thank you very much

Crapshit, 2020-08-25:
Having the exact same issue on my ThinkPad X1 Carbon G7 running on Manjaro with latest fwupdmgr:

```
WARNING: Could not query capsule capabilities: Unsupported
WARNING: Could not apply capsules: Unsupported
```

```
[root@myhost ~]# fwupdmgr --version
client version:	1.4.5
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.4.5
```

kmauleon, 2020-08-26:
hi @Crapshit .. i am not familiar with Manjaro but can you please submit results of below command? thank you very much...

fwupdmgr get-devices

Crapshit, 2020-09-25:
@kmauleon Sorry for replying so late.

Here are my results:
`20QD00L7GE
│
├─Thunderbolt Controller:
│     Device ID:          6a48cb173143bfe843d2bd9b94eec6cd18663676
│     Zusammenfassung:    Unmatched performance for high-speed I/O
│     Current version:    44.00
│     Vendor:             Lenovo (TBT:0x0109)
│     GUIDs:              74cdddba-f7b6-5a31-a3dc-64186b5a5491 ← THUNDERBOLT\VEN_0109&DEV_1801&REV_00
│                         18bb00e3-5d7a-55d3-831e-0e000c184b0e ← THUNDERBOLT\VEN_0109&DEV_1801
│                         c7920601-0cda-507f-851e-92229eb1d370 ← TBT-01091801-native
│                         96c6512b-0138-51da-b02f-f44f2c4c7744 ← TBT-01091801-native-controller0-0
│     Device Flags:       • Internal device
│                         • Updatable
│                         • Requires AC power
│                         • Supported on remote server
│                         • Device stages updates
│   
├─Touchpad:
│     Device ID:          91961e6ddc950eb21e71db6f33ec09a133933a95
│     Current version:    1.3.3013412
│     Bootloader Version: 8.0.0
│     Vendor:             Synaptics (HIDRAW:0x06CB)
│     GUIDs:              10010313-27c0-5c7f-bd99-955028fd3a01 ← HIDRAW\VEN_06CB&DEV_CD8C&REV_00
│                         71463ccd-7553-5d7d-9fac-7ed8073a7853 ← HIDRAW\VEN_06CB&DEV_CD8C
│                         981824ee-4283-595b-a2e4-46eae92dba40 ← SYNAPTICS_RMI\TM3468-010
│                         0957b08b-b580-5c3a-8073-98b9bb15bc7f ← SYNAPTICS_RMI\TM3468
│     Device Flags:       • Internal device
│                         • Updatable
│                         • Supported on remote server
│   
├─UEFI dbx:
│     Device ID:          362301da643102b9f38477387e2193e57abaa590
│     Zusammenfassung:    UEFI Revocation Database
│     Current version:    77
│     Minimum Version:    77
│     Vendor:             UEFI:Linux Foundation
│     Install Duration:   1 second
│     GUIDs:              14503b3d-73ce-5d06-8137-77c68972a341 ← UEFI\CRT_A9087D1044AD18F7A94916D284CBC01827CF23CD8F60B79072C9CAA1FEF4D649
│                         5971a208-da00-5fce-b5f5-1234342f9cf7 ← UEFI\CRT_A9087D1044AD18F7A94916D284CBC01827CF23CD8F60B79072C9CAA1FEF4D649&ARCH_X64
│                         c6682ade-b5ec-57c4-b687-676351208742 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503
│                         f8ba2887-9411-5c36-9cee-88995bb39731 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503&ARCH_X64
│     Device Flags:       • Internal device
│                         • Updatable
│                         • Needs a reboot after installation
│   
└─WDC PC SN720 SDAQNTW-512G-1001:
      Device ID:          3743975ad7f64f8d6575a9ae49fb3a8856fe186f
      Zusammenfassung:    NVM Express Solid State Drive
      Current version:    10190101
      Vendor:             Sandisk Corp (NVME:0x15B7)
      Serial Number:      192649806942
      GUIDs:              137520ce-3603-53e6-9165-56694ed744e7 ← NVME\VEN_15B7&DEV_5002&REV_00
                          ff2112dc-038c-596d-90ca-d43c5077c6ec ← NVME\VEN_15B7&DEV_5002
                          63af1441-fb24-5368-b450-886fbc98610b ← WDC PC SN720 SDAQNTW-512G-1001
      Device Flags:       • Internal device
                          • Updatable
                          • Requires AC power
                          • Supported on remote server
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
    `

kmauleon, 2020-09-28:
hi @Crapshit ... it seems you already have the latest system firmware.  you may want to run the following commands as well to check for additional firmware updates if any... thank you very much

fwupdmgr refresh
fwupdmgr update

Crapshit, 2020-10-20:
@kmauleon: Right, I was on the latest and greates bios 1.38 for my Thinkpad X1 7th Gen.
Today I saw an update for my Thinkpad and today I tried updating it with Ubuntu 20.04 instead of Manjaro.
Ubuntu is using an older version of fwupdmgr, because it is not a rolling release and
this was the first time I did an update via fwupdmgr for my bios.

So right now I don't have any version numbers for you for fwupdmgr and other stuff, but I could do a lookup if you can say me what I should share for you.
The last bios from September was not possible to install on distribution Manjaro with latest and greatest patch status:

  Package      (ID)     UEFI BIOS (BIOS ID)  ECP (ECP ID)      Rev.  Issue Date
  --------------------  -------------------  ---------------   ----  ----------
  1.40      (N2HUR32W)  1.40 (N2HET57W)      1.21 (N2HHT38W)   01    2020/10/16
  1.38-1.21 (N2HUR30W)  1.38 (N2HET55W)      1.21 (N2HHT38W)   01    2020/09/24

And I also reset my bios to defaults before I tried to update it to 1.38 with Manjaro as read in other posts.

kmauleon, 2020-10-20:
hi @Crapshit ... i will forward your concern to owner but can you please show results of below command in Ubuntu... thank you very much...

fwupdmgr get-devices

kmauleon, 2020-10-22:
new versions target to be released within november... please wait.. thank you very much

kmauleon, 2021-02-04:
hi @Crapshit **X1 Carbon 7th** latest firmware available here >> https://fwupd.org/lvfs/search?value=X1+carbon+7th
thank you very much