fwupd/firmware-lenovo #60: Lenovo ThinkPad P52/P72 Management Engine Firmware 12.0.64.1551

nvaert1986, 2020-06-16:
The firmware in the subject is missing, which has already been available for Microsoft Windows since early May 2020 at: https://pcsupport.lenovo.com/us/en/products/laptops-and-netbooks/thinkpad-p-series-laptops/thinkpad-p52-type-20m9-20ma/downloads/driver-list/component?name=Chipset

Since the firmware contains quite a few CVE's that have been fixed, I'd consider it an important update.

kmauleon, 2020-06-16:
Hi nvaert1986

Thank you for your attention... it's currently on validation process... owner needs to make sure this will be installed successfully in the system...but they are working on it and this will be released soon.  Technically, windows version is released a few months earlier prior to Linux'....thank you very much for your patience...

nvaert1986, 2020-06-17:
> Hi nvaert1986
>
> Thank you for your attention... it's currently on validation process... owner needs to make sure this will be installed successfully in the system...but they are working on it and [...]

Thank you for providing me with an update regarding the information.

BTW: I noticed that the ALPS TrackPoint and ALPS TouchPad firmware and the Lenovo Thunderbolt 3 Docking Station were missing too. Are these firmware updates provided by ALPS / Your Thunderbolt Docking Station provider not supported?

kmauleon, 2020-06-17:
Hi nvaert1986

I am not sure with these firmware but I will try to check with our team and get back to you once I have some information....thank you very much

hughsie, 2020-06-17:
> Lenovo Thunderbolt 3 Docking Station

I've been working on this for many months now. https://github.com/fwupd/fwupd/pull/2187 is the last fix required, then it's over to Lenovo for testing and release.

kmauleon, 2020-06-18:
> Thank you for providing me with an update regarding the information.
>
> BTW: I noticed that the ALPS TrackPoint and ALPS TouchPad firmware and the Lenovo Thunderbolt 3 Docking Station were missing [...]

Hi nvaert1986,
Sorry but I don't have any idea about the other firmware...and was not able to find the appropriate person to ask... sorry was not of help for this one....

ChiWei-Chen, 2020-06-18:
Hi @nvaert1986,
We got some positive feedback regarding Thunderbolt 3 Docking Station firmware LVFS support, it is under validation. I'll keep you posted once it became available.

kmauleon, 2020-08-04:
hi @nvaert1986 
The new MEFW has been uploaded already...can you please check?  thank you very much...

nvaert1986, 2020-08-06:
> hi @nvaert1986
> The new MEFW has been uploaded already...can you please check? thank you very much...

@kmauleon Yeah I noticed it was uploaded a few weeks ago. I already updated back then and the update was successful.

kmauleon, 2020-08-07:
hi @nvaert1986 thank you very much for the confirmation...will now close this issue ^_^

closed 2020-08-07 by kmauleon