fwupd/firmware-lenovo #61: Lenovo ThinkPad T480s Corporate ME Firmware, Fedora 31, fwupdmgr 1.3.10

kmauleon, 2020-07-01:
**Lenovo ThinkPad T480s Corporate ME Firmware**, Fedora 31, fwupdmgr 1.3.10

I have never been able to update this particular firmware and nobody's solutions above have helped:

* Disable boot order lock (nope, already disabled)
* Tried downloading the cab and `fwupdmgr install *.cab`: https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN22RM.firmware
* Boot into windows 10 and install [Intel Management Engine 11.8 Firmware](https://pcsupport.lenovo.com/us/en/products/laptops-and-netbooks/thinkpad-t-series-laptops/thinkpad-t480s-type-20l7-20l8/downloads/driver-list/component?name=Chipset)
* Set BIOS settings to defaults
* Clearing /root/.cache/fwupd, ~/.cache/fwupd, which I didn't really think would fix anything anyway

I've tried tons of stuff in the past. Removing the UEFI fw update boot entries. Select the firmware update from the boot device menu. Downgrading (which didn't change anything at all). Reinstalling current version.

```
% fwupdmgr --version
client version:	1.3.10
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.3.10
```

**Update** I've also noticed that the *.cap file remains in /boot/efi/EFI/fedora/fw after the update runs. I removed it and re-ran `fwupdmgr update` and the md5sum didn't change. Also, the last few lines of the update (from a picture I took because I don't know how to get the text otherwise):

```
[...]
fwup_open_file(): Found device
fwup_open_file(): Found volume
fwup_open_file(): Found file
fwup_add_update_capsule(): Read file; 7526944 bytes
fwup_add_update_capsule(): updates guid: 39A39E45-38CE-4B05-809B-60351BF88EC4
fwup_add_update_capsule(): File guid: 39A39E45-38CE-4B05-809B-60351BF88EC4
efi_main(): n_updates: 1
fwup_apply_capsules(): QueryCapsuleCapabilities: Success max: 10485760 reset:0
fwup_apply_capsules(): Capsules: 1
efi_main(): Reset System
```

It paused before 'Reset System' for about a minute.

**Update 2** I noticed some use of `fwupdtool` above, which I hadn't tried. Running it against faeb51f0b8ed4f5486facc4cfb15980ee4485230c40066b0070f2ef7bcb4b34d-Lenovo-ThinkPad-T480s-CorporateMEFirmware-11.8.77.3664.cab gave `FuPluginUefi         missing or invalid embedded capsule header`. And installing with fwupdtool ended up failing as usual.

But then I tried running against the older firmware, which also previously failed. This no longer had the `missing or invalid embedded capsule header`. Upon rebooting, I got the same wall of text, but then it rebooted and **actually attempted** to install the firmware (spinner and all). It failed as I expected, but I hadn't noticed the error before, which flashed for about 1 second:

```
Error 8758: The image provided is not supported by the platform.
```

So basically it looks like the last two updates fail for different reasons.

_Originally posted by @markuspeloquin in https://github.com/fwupd/firmware-lenovo-thinkpad/issues/56#issuecomment-652618221_

kmauleon, 2020-07-01:
hi @markuspeloquin
I opened a new issue for this one for proper tracking.... thank you very much...

kmauleon, 2020-07-02:
hi @markuspeloquin...
can you please try again? owner just uploaded a new cab file just a while ago... thank you very much...

markuspeloquin, 2020-07-02:
Thanks, @kmauleon, I'll try it now. Sometimes I just think the issue that I was hoping would be the solution to my problem gets closed too early...

kmauleon, 2020-07-02:
> Thanks, @kmauleon, I'll try it now. Sometimes I just think the issue that I was hoping would be the solution to my problem gets closed too early...
> 
> **Update** So I ran `fwupdmgr update`, and it did download a new file (83666d6d33094c77a4494b615a00cda050890a4613dfc064ee507d83da5cf5fc-Lenovo-ThinkPad-T480s-CorporateMEFirmware-11.8.77.3664.cab). It rebooted and appeared to update the firmware, and it appeared (at the time), that nothing was wrong (except after rebooting, it at first attempted a PXE boot). I didn't get that `Error 8758` mentioned above.
> 
> But after a `fwupdmgr refresh --force`, I get:
> 
> ```
> % fwupdmgr get-updates
> • ThinkPad Thunderbolt 3 Dock has no available firmware updates
> • Thunderbolt Controller has no available firmware updates
> • INTEL SSDPEKKF512G8L has no available firmware updates
> • System Firmware has the latest available firmware version
> • UEFI Device Firmware has the latest available firmware version
> • VMM3320 has no available firmware updates
> 20L70026US
> │
> └─UEFI Device Firmware:
>   │   Device ID:           118b3d8cfafa8781890c4d3e23a4deb9fcb18e5a
>   │   Current version:     184.65.3590
>   │   Minimum Version:     184.65.3590
>   │   Vendor:              DMI:LENOVO
>   │   Update State:        success
>   │   GUID:                39a39e45-38ce-4b05-809b-60351bf88ec4
>   │   Device Flags:        • Internal device
>   │                        • Updatable
>   │                        • Requires AC power
>   │                        • Supported on remote server
>   │                        • Needs a reboot after installation
>   │                        • Device is usable for the duration of the update
>   │
>   ├─ThinkPad T480s Corporate ME Update:
>   │     New version:       184.77.3664
>   │     Remote ID:         lvfs
>   │     Summary:           Lenovo ThinkPad T480s Corporate ME Firmware
>   │     License:           Proprietary
>   │     Size:              7.5 MB
>   │     Vendor:            Lenovo Ltd.
>   │     Flags:             is-upgrade
>   │     Description:       • 8 Intel Platform Update 2020.1
>   │
>   │     Version 11.8.77.3664 (LVFS: 184.77.3664)
>   │
>   │     Problem Fixes
>   │
>   │      • Intel CSME PSIRT-TA-2019-10-001 (IPU 2020.1):
>   │      • Mitigated the following security vulnerabilities:
>   │
>   │     Security issues fixed:
>   │
>   │      • CVE-2020-0545
>   │      • CVE-2020-0542
>   │      • CVE-2020-0541
>   │      • CVE-2020-0540
>   │      • CVE-2020-0539
>   │      • CVE-2020-0538
>   │      • CVE-2020-0537
>   │      • CVE-2020-0536
>   │      • CVE-2020-0535
>   │      • CVE-2020-0534
>   │      • CVE-2020-0533
>   │      • CVE-2020-0532
>   │      • CVE-2020-0531
>   │
>   └─ThinkPad T480s Corporate ME Update:
>         New version:       184.70.3626
>         Remote ID:         lvfs
>         Summary:           Lenovo ThinkPad T480s Corporate ME Firmware
>         License:           Proprietary
>         Size:              7.5 MB
>         Vendor:            Lenovo Ltd.
>         Flags:             is-upgrade
>         Description:       • 8 Q2'19 Intel Platform Update
> 
>         Version 11.8.70.3626 (LVFS: 184.70.3626)
> 
>         Problem Fixes
> 
>          • Intel CSME PSIRT-TA-201905-011 Q2 2019 IPU:
> 
>         Security issues fixed:
> 
>          • CVE-2019-0168
>          • CVE-2019-0169
>          • CVE-2019-11087
>          • CVE-2019-11090
>          • CVE-2019-11101
>          • CVE-2019-11104
>          • CVE-2019-11106
>          • CVE-2019-11110
>          • CVE-2019-11131
>          • CVE-2019-11132
> ```

Hi markus...
after forcing "fwupdmgr refresh".... did you run "fwupdmgr update" or try updating via gnome software.... thank you very much....

markuspeloquin, 2020-07-06:
`fwupdmgr update` is all I ever use.

kmauleon, 2020-07-06:
calling the attention of the owner... please wait for update... thank you very much...

markuspeloquin, 2020-07-07:
Nothing really new to report. I tried updating from Windows again. I Also figured out how to get the logs, so it isn't transcribed from a picture I made. Pretty dull:

```
% sudo fwupdate -L
../plugins/uefi/efi/fwupdate.c:158:fwup_populate_update_table(): Found update fwupd-39a39e45-38ce-4b05-809b-60351bf88ec4-0
../plugins/uefi/efi/fwupdate.c:577:efi_main(): Adding new capsule
../plugins/uefi/efi/fwupdate.c:198:fwup_search_file(): Searching Device Path: HD(1,GPT,8AB270A4-00B2-4F10-91B6-8398F73354CB)/\EFI\fedora\fw\fwupd-39a39e45-38ce-4b05-809b-60351bf88ec4.cap...
../plugins/uefi/efi/fwupdate.c:218:fwup_search_file(): Device Path prepared: HD(1,GPT,8AB270A4-00B2-4F10-91B6-8398F73354CB)
../plugins/uefi/efi/fwupdate.c:228:fwup_search_file(): Device supporting SFSP: PciRoot(0)/Pci(0x1D,0x0)/Pci(0x0,0x0)/Path(131,23,0x010000005CD2E42A814071AF84012C00)/HD(1,GPT,8AB270A4-00B2-4F10-91B6-8398F73354CB)
../plugins/uefi/efi/fwupdate.c:231:fwup_search_file(): Comparing: HD(1,GPT,8AB270A4-00B2-4F10-91B6-8398F73354CB) and PciRoot(0)/Pci(0x1D,0x0)/Pci(0x0,0x0)/Path(131,23,0x010000005CD2E42A814071AF84012C00)/HD(1,GPT,8AB270A4-00B2-4F10-91B6-8398F73354CB)
../plugins/uefi/efi/fwupdate.c:231:fwup_search_file(): Comparing: HD(1,GPT,8AB270A4-00B2-4F10-91B6-8398F73354CB) and Pci(0x1D,0x0)/Pci(0x0,0x0)/Path(131,23,0x010000005CD2E42A814071AF84012C00)/HD(1,GPT,8AB270A4-00B2-4F10-91B6-8398F73354CB)
../plugins/uefi/efi/fwupdate.c:231:fwup_search_file(): Comparing: HD(1,GPT,8AB270A4-00B2-4F10-91B6-8398F73354CB) and Pci(0x0,0x0)/Path(131,23,0x010000005CD2E42A814071AF84012C00)/HD(1,GPT,8AB270A4-00B2-4F10-91B6-8398F73354CB)
../plugins/uefi/efi/fwupdate.c:231:fwup_search_file(): Comparing: HD(1,GPT,8AB270A4-00B2-4F10-91B6-8398F73354CB) and Path(131,23,0x010000005CD2E42A814071AF84012C00)/HD(1,GPT,8AB270A4-00B2-4F10-91B6-8398F73354CB)
../plugins/uefi/efi/fwupdate.c:231:fwup_search_file(): Comparing: HD(1,GPT,8AB270A4-00B2-4F10-91B6-8398F73354CB) and HD(1,GPT,8AB270A4-00B2-4F10-91B6-8398F73354CB)
../plugins/uefi/efi/fwupdate.c:240:fwup_search_file(): Match up! Returning \EFI\fedora\fw\fwupd-39a39e45-38ce-4b05-809b-60351bf88ec4.cap
../plugins/uefi/efi/fwupdate.c:300:fwup_open_file(): Found device
../plugins/uefi/efi/fwupdate.c:307:fwup_open_file(): Found volume
../plugins/uefi/efi/fwupdate.c:315:fwup_open_file(): Found file
../plugins/uefi/efi/fwupdate.c:415:fwup_add_update_capsule(): Read file; 7515600 bytes
../plugins/uefi/efi/fwupdate.c:416:fwup_add_update_capsule(): updates guid: 39A39E45-38CE-4B05-809B-60351BF88EC4
../plugins/uefi/efi/fwupdate.c:417:fwup_add_update_capsule(): File guid: 39A39E45-38CE-4B05-809B-60351BF88EC4
../plugins/uefi/efi/fwupdate.c:593:efi_main(): n_updates: 1
../plugins/uefi/efi/fwupdate.c:461:fwup_apply_capsules(): QueryCapsuleCapabilities: Success max: 1055363072 reset:1
../plugins/uefi/efi/fwupdate.c:463:fwup_apply_capsules(): Capsules: 1
../plugins/uefi/efi/fwupdate.c:613:efi_main(): Reset System
```

*Update Status: attempted* below is definitely not great:

```
% sudo fwupdate -i
failed: Error opening file /sys/firmware/efi/efivars/fwupd-ebfe8df8-dee7-4692-a721-cbcf5095c5cf-0-0abba7dc-e516-4167-bbf5-4d9d1c739416: No such file or directory
Information for the update status entry 1:
  Information Version: 7
  Firmware GUID: {39a39e45-38ce-4b05-809b-60351bf88ec4}
  Capsule Flags: 0x00032784x
  Hardware Instance: 0
  Update Status: attempted
  Capsule File Path: /EFI/fedora/fw/fwupd-39a39e45-38ce-4b05-809b-60351bf88ec4.cap

failed: Error opening file /sys/firmware/efi/efivars/fwupd-4d254d6e-cd67-477b-97d5-bc3048af45c4-0-0abba7dc-e516-4167-bbf5-4d9d1c739416: No such file or directory
```

kmauleon, 2020-07-07:
hi @markuspeloquin 
can you please try download the cab file itself from https://fwupd.org/lvfs/ and manually install them...
navigate to the location of the cab file and execute from the terminal..

fwupdmgr --allow-older --allow-reinstall install <cab file>

markuspeloquin, 2020-07-13:
Downloaded the cab from https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN22RM.firmware, same md5sum as the file already in ~/.cache/fwupd, same failure.

kmauleon, 2020-07-14:
hi @markuspeloquin checking with owner again.....

lijunwang1990, 2020-07-14:
HI @markuspeloquin ,
did you try to use manual update? download the cab file from LVFS and put it in your system project. such as "home"
use command:
fwupdmgr --allow-older --allow-reinstall install '/home/username/[Name].cab'
and the system will display:"An update requires a reboot to complete.Restart now? [Y|n]:"
press Y.

kmauleon, 2020-10-22:
hi @markuspeloquin ... new firmware versions available. please try the following again... thank you very much.
    fwupdmgr refresh
    fwupdmgr update

kmauleon, 2020-11-16:
hi @markuspeloquin ... please try with latest MEFW >> https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN22RM.firmware
thank you very much

kleinph, 2020-11-16:
@kmauleon unfortunately this didn't work for me. I see the firmware update screen when rebooting, but the update is still there:

```
$ fwupdmgr get-updates 
Devices with no available firmware updates: 
 • INTEL SSDPEKKF512G8L
Devices with the latest available firmware version:
 • Thunderbolt host controller
 • Embedded Controller
 • System Firmware
Unbekannt                [***************************************]
Successfully uploaded 1 report
20L7001LGE
│
└─Intel Management Engine:
  │   Device ID:          cf3ad1e5e6838b09cba64bb5ccb1142dd44995d2
  │   Current version:    184.70.3626
  │   Minimum Version:    184.70.3626
  │   Vendor:             DMI:LENOVO
  │   Update State:       success
  │   GUIDs:              676af093-2a5c-4238-9c29-db8063a33532
  │                       039bd398-f61a-5c46-b02c-b660d1cf4926
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Device is usable for the duration of the update
  │ 
  ├─ThinkPad T480s Consumer ME Update:
  │     New version:      184.79.3722
  │     Remote ID:        lvfs
  │     Zusammenfassung:  Lenovo ThinkPad T480s Consumer ME Firmware
  │     License:          Proprietary
  │     Size:             2,2 MB
  │     Created:          2016-07-08
  │     Urgency:          High
  │     Details:          https://pcsupport.lenovo.com/de/en/search?query=N22RN13W
  │     Vendor:           Lenovo Ltd.
  │     Flags:            is-upgrade
  │     Beschreibung:     
  │     • 8 Intel Platform Update 2020.1
  │     
  │     Version 11.8.79.3722 (LVFS: 184.79.3722)
  │     
  │     Problem Fixes
  │     
  │     • Mitigated the following security vulnerabilities:
  │     
  │     Security issues fixed:
  │     
  │     • CVE-2020-8758
  │   
  └─ThinkPad T480s Consumer ME Update:
        New version:      184.77.3664
        Remote ID:        lvfs
        Zusammenfassung:  Lenovo ThinkPad T480s Consumer ME Firmware
        License:          Proprietary
        Size:             2,2 MB
        Created:          2016-07-08
        Urgency:          High
        Details:          https://pcsupport.lenovo.com/de/en/search?query=N22RN13W
        Vendor:           Lenovo Ltd.
        Flags:            is-upgrade
        Beschreibung:     
        • 8 Intel Platform Update 2020.1
        
        Version 11.8.77.3664 (LVFS: 184.77.3664)
        
        Problem Fixes
        
        • Intel CSME PSIRT-TA-2019-10-001 (IPU 2020.1):
        • Mitigated the following security vulnerabilities:
        
        Security issues fixed:
        
        • CVE-2020-0531
        • CVE-2020-0532
        • CVE-2020-0533
        • CVE-2020-0534
        • CVE-2020-0535
        • CVE-2020-0536
        • CVE-2020-0537
        • CVE-2020-0538
        • CVE-2020-0539
        • CVE-2020-0540
        • CVE-2020-0541
        • CVE-2020-0542
        • CVE-2020-0545
```

kmauleon, 2020-11-16:
hi @kleinph can you please show result of below command... thank you very much
fwupdmgr get-devices

kleinph, 2020-11-16:
```
$ fwupdmgr get-devices 
20L7001LGE
│
├─Thunderbolt host controller:
│     Device ID:          f5401d94c86c965c7250fdd83decdbb60d2adaa9
│     Zusammenfassung:    Unmatched performance for high-speed I/O
│     Current version:    20.00
│     Vendor:             Lenovo (TBT:0x0109)
│     GUIDs:              3849839e-d8b2-5941-9e69-5acc397d3ed3
│                         7e3190a2-d5a5-5a9e-b582-494b47f7288c
│                         061f2847-061b-5cfb-9c39-3dbfe2b49d69
│                         65a3d897-4a18-5d27-83d8-eb8a231b7aa4
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Device stages updates
│   
├─Embedded Controller:
│     Device ID:          323bda90831520867f8c02544efb9711dfca9fb6
│     Current version:    0.1.20
│     Minimum Version:    0.1.20
│     Vendor:             DMI:LENOVO
│     GUIDs:              4d254d6e-cd67-477b-97d5-bc3048af45c4
│                         cc9cf2fe-ba41-532b-bc58-17ca9cc3c63d
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─INTEL SSDPEKKF512G8L:
│     Device ID:          c82980cd3fe66f7d0a8bb83b353cbc271177efd4
│     Zusammenfassung:    NVM Express Solid State Drive
│     Current version:    L08P
│     Vendor:             Intel Corporation (NVME:0x8086)
│     GUIDs:              f91a5c60-8696-539e-9d0a-57d194f74ac4
│                         396fb4fd-7656-564f-88b5-381e83ca30c0
│                         79517f86-8df8-5d6e-a18b-33f0b36a78e9
│                         68db11e5-b0cf-5bc9-a94e-17e28496e505
│                         87fd9fec-f447-5ec6-a1cd-d55fac0d3d30
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─Intel Management Engine:
│     Device ID:          cf3ad1e5e6838b09cba64bb5ccb1142dd44995d2
│     Current version:    184.70.3626
│     Minimum Version:    184.70.3626
│     Vendor:             DMI:LENOVO
│     Update State:       success
│     GUIDs:              676af093-2a5c-4238-9c29-db8063a33532
│                         039bd398-f61a-5c46-b02c-b660d1cf4926
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
└─System Firmware:
      Device ID:          e7cc234e82d08b6a14e246beb6b5f8142091e570
      Current version:    0.1.42
      Minimum Version:    0.0.1
      Vendor:             LENOVO (DMI:LENOVO)
      GUIDs:              ebfe8df8-dee7-4692-a721-cbcf5095c5cf
                          230c8b18-8d9b-53ec-838b-6cfc0383493a
                          2b79e0a0-0801-5fc9-a7a5-90374e924a9b
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Supported on remote server
                          • Needs a reboot after installation
                          • Cryptographic hash verification is available
                          • Device is usable for the duration of the update
```

kmauleon, 2020-11-16:
it's weird that you have 2 GUIDs for MEFW as below
│     GUIDs:              **676af093-2a5c-4238-9c29-db8063a33532**
│                         039bd398-f61a-5c46-b02c-b660d1cf4926
the first one should be the correct one... 
sorry but can you please try this again...
     **fwupdmgr refresh
     fwupdmgr update**
after update and before system reboots please check MEFW version in BIOS Setup
thank you very much

kleinph, 2020-11-16:
I am not sure, but I think MEFW version in BIOS/UEFI is 11.8.79.3722.

[image]

kmauleon, 2020-11-16:
hi @kleinph ... this means MEFW is already the latest one... it seems ESRT table was not updated correctly... currently this is a known issue and we are still checking on this... can we just wait for the next MEFW version targeted to be released by December end and see if behavior will change...  Thank you very much

kmauleon, 2021-01-04:
hi @kleinph please try latest mefw with following commands... thank you very much
**fwupdmgr --force refresh
fwupdmgr update**

kleinph, 2021-01-04:
Unfortunately this didn't work.

```
$ fwupdmgr get-updates 
Devices with no available firmware updates: 
 • USB3.1 Hub
 • INTEL SSDPEKKF512G8L
 • USB2.0 Hub
 • VMM5333
Devices with the latest available firmware version:
 • Thunderbolt host controller
 • Embedded Controller
 • System Firmware
20L7001LGE
│
└─Intel Management Engine:
  │   Device ID:          cf3ad1e5e6838b09cba64bb5ccb1142dd44995d2
  │   Current version:    184.70.3626
  │   Minimum Version:    184.70.3626
  │   Vendor:             DMI:LENOVO
  │   Update State:       success
  │   GUIDs:              676af093-2a5c-4238-9c29-db8063a33532
  │                       039bd398-f61a-5c46-b02c-b660d1cf4926
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Device is usable for the duration of the update
  │ 
  ├─ThinkPad T480s Consumer ME Update:
  │     New version:      184.80.3746
  │     Remote ID:        lvfs
  │     Zusammenfassung:  Lenovo ThinkPad T480s Consumer ME Firmware
  │     License:          Proprietary
  │     Size:             2,2 MB
  │     Created:          2016-07-08
  │     Urgency:          High
  │     Details:          https://pcsupport.lenovo.com/de/en/search?query=N22RN15W
  │     Vendor:           Lenovo Ltd.
  │     Flags:            is-upgrade
  │     Beschreibung:     
  │     • 8 Intel Platform Update 2020.1
  │     
  │     Version 11.8.80.3746 (LVFS: 184.80.3746)
  │     
  │     Problem Fixes
  │     
  │     • Intel CSME PSIRT-TA-2019-10-001 (IPU 2020.1):
  │     • Mitigated the following security vulnerabilities:
  │     
  │     Security issues fixed:
  │     
  │     • CVE-2020-12356
  │     • CVE-2020-12355
  │     • CVE-2020-12303
  │     • CVE-2020-12297
  │     • CVE-2020-8752
  │     • CVE-2020-8749
  │     • CVE-2020-8746
  │     • CVE-2020-8747
  │     • CVE-2020-8754
  │     • CVE-2020-8751
  │     • CVE-2020-8760
  │     • CVE-2020-8756
  │     • CVE-2020-8757
  │     • CVE-2020-8705
  │     • CVE-2020-8745
  │     • CVE-2020-8753
  │   
  ├─ThinkPad T480s Consumer ME Update:
  │     New version:      184.79.3722
  │     Remote ID:        lvfs
  │     Zusammenfassung:  Lenovo ThinkPad T480s Consumer ME Firmware
  │     License:          Proprietary
  │     Size:             2,2 MB
  │     Created:          2016-07-08
  │     Urgency:          High
  │     Details:          https://pcsupport.lenovo.com/de/en/search?query=N22RN13W
  │     Vendor:           Lenovo Ltd.
  │     Flags:            is-upgrade
  │     Beschreibung:     
  │     • 8 Intel Platform Update 2020.1
  │     
  │     Version 11.8.79.3722 (LVFS: 184.79.3722)
  │     
  │     Problem Fixes
  │     
  │     • Mitigated the following security vulnerabilities:
  │     
  │     Security issues fixed:
  │     
  │     • CVE-2020-8758
  │   
  └─ThinkPad T480s Consumer ME Update:
        New version:      184.77.3664
        Remote ID:        lvfs
        Zusammenfassung:  Lenovo ThinkPad T480s Consumer ME Firmware
        License:          Proprietary
        Size:             2,2 MB
        Created:          2016-07-08
        Urgency:          High
        Details:          https://pcsupport.lenovo.com/de/en/search?query=N22RN13W
        Vendor:           Lenovo Ltd.
        Flags:            is-upgrade
        Beschreibung:     
        • 8 Intel Platform Update 2020.1
        
        Version 11.8.77.3664 (LVFS: 184.77.3664)
        
        Problem Fixes
        
        • Intel CSME PSIRT-TA-2019-10-001 (IPU 2020.1):
        • Mitigated the following security vulnerabilities:
        
        Security issues fixed:
        
        • CVE-2020-0531
        • CVE-2020-0532
        • CVE-2020-0533
        • CVE-2020-0534
        • CVE-2020-0535
        • CVE-2020-0536
        • CVE-2020-0537
        • CVE-2020-0538
        • CVE-2020-0539
        • CVE-2020-0540
        • CVE-2020-0541
        • CVE-2020-0542
        • CVE-2020-0545
```

kmauleon, 2021-01-04:
Hi @kleinph thank you very much for the information... will forward this result to owner.  thank you very much...

lijunwang1990, 2021-01-05:
> Unfortunately this didn't work.
>
> ```
> $ fwupdmgr get-updates
> Devices with no available firmware updates:
>  • USB3.1 Hub
>  • INTEL SSDPEKKF512G8L
>  • USB2.0 Hub
>  • VMM5333
> Devices with the latest [...]

Hi @kleinph I'm the BIOS owner of T480s , I will try to build a trail bios to fix the ESRT issue. could you please help to verify it? as we can't duplicate this issue on our side, please give me your connect email if you can, I will send the bios to you. thank you.

kleinph, 2021-01-08:
Hi @kmauleon sure. What do you mean with "_connect_ email"?

kmauleon, 2021-01-11:
hi @kleinph I think owner @lijunwang1990 is asking for your contact email so he could directly send you the trial bios.... thank you very much...

lijunwang1990, 2021-01-21:
Hi @kleinph , how can i send trail BIOS to you?, could you please help to check this issue?

kleinph, 2021-01-21:
Hi @lijunwang1990 , I am sorry, I forgot about this. Can you send it to philipptheklein@gmail.com. Let me know when you got the e-mail address.

lijunwang1990, 2021-02-03:
Hi @kleinph I send the trail BIOS to your e-mail  philipptheklein@gmail.com, please help to check. thank you.

kmauleon, 2021-02-04:
update from BIOS Dev... thank you very much....
**no clear progress yet regarding BIOS ESRT problem investigation**

kleinph, 2021-02-15:
Is there some progress on this? Should I have received the BIOS?

 Also @lijunwang1990 can you please remove my e-mail from your comment?

kmauleon, 2021-02-17:
hi @kleinph have u received the trial BIOS sent to your personal email as mentioned by @lijunwang1990 ?
please advise... thank you very much.

kleinph, 2021-02-17:
No, I didn't received it. But maybe it was marked as spam and I missed it. Couldn't this be provided as a download on some official infrastructure?

kmauleon, 2021-02-17:
asking lijun if can upload here... please wait... thank you very much

kmauleon, 2021-02-23:
hi @kleinph please use the attached file as mentioned by @lijunwang1990 thank you very much
[Lenovo-ThinkPad-T480s-SystemFirmware-1.43T.cab.zip](https://github.com/fwupd/firmware-lenovo-thinkpad/files/6026330/Lenovo-ThinkPad-T480s-SystemFirmware-1.43T.cab.zip)

kleinph, 2021-02-24:
Unfortunately this didn't worked. But it seems as on reboot the firmware updater never runs. I also experienced this a few days ago with the normal update from fwupd.

I tried booting manually the Linux Firmware Updater and also deleting the files from ESP and removing the EFI var for he firmware updater and then rerun `fwupdmgr update` with no success (the files and EFI var are restored, but it does not boot into the updater).

kleinph, 2021-03-01:
With the newest update of fwupd (1.5.7), FW updates worked again. So I installed the provided system firmware, but the updates still appear. I also tried to install the updates from fwupd itself, but this didn't change anything.

lijunwang1990, 2021-03-02:
@kleinph thank you very much for your update,  the problem is two GUID displayed and still show older ME version even you update to newer version? Will the system constantly remind you to upgrade？

padok, 2021-03-02:
Updating to fwupd (1.5.7) fixed the issue for me.

`"HostSku" : "LENOVO_MT_20HE_BU_Think_FM_ThinkPad T470"`

kleinph, 2021-03-10:
Yes the update is always present if I run `fwupdmgr get-updates`. fwupd update didn't fixed this problem.

Here is the output:
```
20L7001LGE
│
└─Intel Management Engine:
  │   Device ID:          cf3ad1e5e6838b09cba64bb5ccb1142dd44995d2
  │   Current version:    184.70.3626
  │   Minimum Version:    184.70.3626
  │   Vendor:             DMI:LENOVO
  │   GUIDs:              676af093-2a5c-4238-9c29-db8063a33532
  │                       039bd398-f61a-5c46-b02c-b660d1cf4926
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Device is usable for the duration of the update
  │ 
  ├─ThinkPad T480s Consumer ME Update:
  │     New version:      184.82.3838
  │     Remote ID:        lvfs
  │     Zusammenfassung:  Lenovo ThinkPad T480s Consumer ME Firmware
  │     License:          Proprietary
  │     Size:             2,2 MB
  │     Created:          2016-07-08
  │     Urgency:          High
  │     Details:          https://pcsupport.lenovo.com/de/en/search?query=N22RN16W
  │     Vendor:           Lenovo Ltd.
  │     Flags:            is-upgrade
  │     Beschreibung:     
  │     • 8 Intel Platform Update 2020.1
  │     
  │     Version 11.8.82.3838 (LVFS: 184.82.3838)
  │     
  │     Problem Fixes
  │     
  │     • Intel CSME PSIRT-TA-00391 IPU 2020.2:
  │     • Mitigated the following security vulnerabilities:
  │   
  ├─ThinkPad T480s Consumer ME Update:
  │     New version:      184.80.3746
  │     Remote ID:        lvfs
  │     Zusammenfassung:  Lenovo ThinkPad T480s Consumer ME Firmware
  │     License:          Proprietary
  │     Size:             2,2 MB
  │     Created:          2016-07-08
  │     Urgency:          High
  │     Details:          https://pcsupport.lenovo.com/de/en/search?query=N22RN15W
  │     Vendor:           Lenovo Ltd.
  │     Flags:            is-upgrade
  │     Beschreibung:     
  │     • 8 Intel Platform Update 2020.1
  │     
  │     Version 11.8.80.3746 (LVFS: 184.80.3746)
  │     
  │     Problem Fixes
  │     
  │     • Intel CSME PSIRT-TA-2019-10-001 (IPU 2020.1):
  │     • Mitigated the following security vulnerabilities:
  │   
  ├─ThinkPad T480s Consumer ME Update:
  │     New version:      184.79.3722
  │     Remote ID:        lvfs
  │     Zusammenfassung:  Lenovo ThinkPad T480s Consumer ME Firmware
  │     License:          Proprietary
  │     Size:             2,2 MB
  │     Created:          2016-07-08
  │     Urgency:          High
  │     Details:          https://pcsupport.lenovo.com/de/en/search?query=N22RN13W
  │     Vendor:           Lenovo Ltd.
  │     Flags:            is-upgrade
  │     Beschreibung:     
  │     • 8 Intel Platform Update 2020.1
  │     
  │     Version 11.8.79.3722 (LVFS: 184.79.3722)
  │     
  │     Problem Fixes
  │     
  │     • Mitigated the following security vulnerabilities:
  │   
  └─ThinkPad T480s Consumer ME Update:
        New version:      184.77.3664
        Remote ID:        lvfs
        Zusammenfassung:  Lenovo ThinkPad T480s Consumer ME Firmware
        License:          Proprietary
        Size:             2,2 MB
        Created:          2016-07-08
        Urgency:          High
        Details:          https://pcsupport.lenovo.com/de/en/search?query=N22RN13W
        Vendor:           Lenovo Ltd.
        Flags:            is-upgrade
        Beschreibung:     
        • 8 Intel Platform Update 2020.1
        
        Version 11.8.77.3664 (LVFS: 184.77.3664)
        
        Problem Fixes
        
        • Intel CSME PSIRT-TA-2019-10-001 (IPU 2020.1):
        • Mitigated the following security vulnerabilities:
```

kmauleon, 2022-11-16:
Closing this as BIOS ESRT fix has been rolled out already.  Please reopen if you still see the problem.  Thank you very much.

closed 2022-11-16 by kmauleon

lijunwang1990, 2022-11-16:
您好，邮件已收到