fwupd/firmware-lenovo #63: ThinkPad T14s Embedded Controller update offered repeatedly

Yawning, 2020-07-14:
**Describe the bug**

The `ThinkPad T14s Gen 1 / ThinkPad X13 Gen 1 Embedded Controller Update` (1.0.04) either does not get installed correctly, or more likely has a malformed version preventing fwupd from registering the update as installed.

```
20T00023US
│
└─Embedded Controller:
  │   Device ID:           985c195dc0b7963d5d69f0983522e52ac37835b7
  │   Current version:     0.1.4
  │   Minimum Version:     0.1.4
  │   Vendor:              DMI:LENOVO
  │   Update State:        success
  │   GUID:                f79e7515-dc49-4426-a7a0-26dd66016ebc
  │   Device Flags:        • Internal device
  │                        • Updatable
  │                        • Requires AC power
  │                        • Supported on remote server
  │                        • Needs a reboot after installation
  │                        • Device is usable for the duration of the update
  │ 
  └─ThinkPad T14s Gen 1 / ThinkPad X13 Gen 1 Embedded Controller Update:
        New version:       1.0.04
        Remote ID:         lvfs
        Summary:           Lenovo ThinkPad T14s Gen 1 / ThinkPad X13 Gen 1 Embedded Controller Firmware
        License:           Proprietary
        Size:              767.1 kB
        Created:           2020-05-07
        Urgency:           High
        Vendor:            Lenovo Ltd.
        Flags:             is-upgrade
        Description:       Lenovo ThinkPad T14s Gen 1 / ThinkPad X13 Gen 1 Embedded Controller Firmware 1.04. The computer will be restarted automatically after updating EC completely. Do NOT turn off your computer or remove the AC adapter while the update is in progress.
```

My working hypothesis is that `0.1.4` (Current Version) is in fact `1.0.04` (New version).

**Steps to Reproduce**

1. Update a ThinkPad T14s.
2. Re-run `fwupdmgr get-updates`.
3. See the same embedded controller update still being offered.

Note: Repeating the update process appears to succeed, though the vendor's flash process likely is doing nothing since it's completes nearly instantly (though it does display the splash screen).

**Expected behavior**

The update to install, and not be offered repeatedly.

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version
client version: 1.4.4
compile-time dependency versions
        gusb:   0.3.4
        efivar: 37
daemon version: 1.4.4
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

Fedora 32 installation (fwupd-1.4.4-1.fc32.x86_64)

(Other answers omitted for brevity, the actual update process works.)

Yawning, 2020-07-15:
This looks like the same issue as #54, just with a different package.

Yawning, 2020-07-22:
While verifying that the system firmware was updated correctly, per the System Setup (BIOS) menu, the system indeed already is running the most up to date EC firmware.

Since I am resigned to having to deal with this issue for the foreseeable future, is there a way to suppress updates on a per-update-package basis, so fwupd handles ignoring the broken update package for me?

hughsie, 2020-07-27:
> is there a way to suppress updates on a per-update-package basis

See /etc/fwupd/daemon.conf you can add something to the blocklist.

Yawning, 2020-07-27:
> See /etc/fwupd/daemon.conf you can add something to the blocklist.

Unless the config shipped by Fedora is incomplete, there appears to be no way to reject specific updates, only all updates for specific devices by GUID.

Going from "having to explicitly not to update the EC firmware" to "having to check Lenovo's website and or github to see if there's a fixed package and or newer version" doesn't feel like much of an improvement to me.

hughsie, 2020-07-27:
> there appears to be no way to reject specific updates

We could certainly add such a feature in newer versions.

> doesn't feel like much of an improvement to me

Well, we're trying our best with limited resources. ¯\_(ツ)_/¯

Yawning, 2020-07-27:
> > there appears to be no way to reject specific updates
> 
> We could certainly add such a feature in newer versions.

https://github.com/fwupd/fwupd/issues/2280

Yawning, 2020-07-27:
I was hoping I could say "the newly released 0.1.06 package renders this moot", but apparently having a new version of the firmware available doesn't actually resolve this issue because fwupd insists on trying to install `0.1.04` before installing `0.1.06`.

```
20T00023US
│
└─Embedded Controller:
  │   Device ID:           985c195dc0b7963d5d69f0983522e52ac37835b7
  │   Current version:     0.1.4
  │   Minimum Version:     0.1.4
  │   Vendor:              DMI:LENOVO
  │   Update State:        success
  │   GUID:                f79e7515-dc49-4426-a7a0-26dd66016ebc
  │   Device Flags:        • Internal device
  │                        • Updatable
  │                        • Requires AC power
  │                        • Supported on remote server
  │                        • Needs a reboot after installation
  │                        • Device is usable for the duration of the update
  │ 
  ├─ThinkPad T14s Gen 1 / ThinkPad X13 Gen 1 Embedded Controller Update:
  │     New version:       1.0.04
  │     Remote ID:         lvfs
  │     Summary:           Lenovo ThinkPad T14s Gen 1 / ThinkPad X13 Gen 1 Embedded Controller Firmware
  │     License:           Proprietary
  │     Size:              767.1 kB
  │     Created:           2020-05-07
  │     Urgency:           High
  │     Vendor:            Lenovo Ltd.
  │     Flags:             is-upgrade
  │     Description:       Lenovo ThinkPad T14s Gen 1 / ThinkPad X13 Gen 1 Embedded Controller Firmware 1.04. The computer will be restarted automatically after updating EC completely. Do NOT turn off your computer or remove the AC adapter while the update is in progress.
  │   
  └─ThinkPad T14s Gen 1 / ThinkPad X13 Gen 1 Embedded Controller Update:
        New version:       0.1.06
        Remote ID:         lvfs
        Summary:           Lenovo ThinkPad T14s Gen 1 / ThinkPad X13 Gen 1 Embedded Controller Firmware
        License:           Proprietary
        Size:              767.1 kB
        Created:           2020-06-30
        Urgency:           High
        Vendor:            Lenovo Ltd.
        Flags:             is-upgrade
        Description:       Lenovo ThinkPad T14s Gen 1 / ThinkPad X13 Gen 1 Embedded Controller Firmware 1.06. The computer will be restarted automatically after updating EC completely. Do NOT turn off your computer or remove the AC adapter while the update is in progress. Fix the Battery FW update process.
```

Yawning, 2020-07-27:
Manually installing the 1.0.06 cab file downloaded from Lenovo's website (`N2YHT15W.cab`) results in:

```
20T00023US
│
└─Embedded Controller:
  │   Device ID:           985c195dc0b7963d5d69f0983522e52ac37835b7
  │   Current version:     0.1.6
  │   Minimum Version:     0.1.6
  │   Vendor:              DMI:LENOVO
  │   GUID:                f79e7515-dc49-4426-a7a0-26dd66016ebc
  │   Device Flags:        • Internal device
  │                        • Updatable
  │                        • Requires AC power
  │                        • Supported on remote server
  │                        • Needs a reboot after installation
  │                        • Device is usable for the duration of the update
  │ 
  └─ThinkPad T14s Gen 1 / ThinkPad X13 Gen 1 Embedded Controller Update:
        New version:       1.0.04
        Remote ID:         lvfs
        Summary:           Lenovo ThinkPad T14s Gen 1 / ThinkPad X13 Gen 1 Embedded Controller Firmware
        License:           Proprietary
        Size:              767.1 kB
        Created:           2020-05-07
        Urgency:           High
        Vendor:            Lenovo Ltd.
        Flags:             is-upgrade
        Description:       Lenovo ThinkPad T14s Gen 1 / ThinkPad X13 Gen 1 Embedded Controller Firmware 1.04. The computer will be restarted automatically after updating EC completely. Do NOT turn off your computer or remove the AC adapter while the update is in progress.
```

I assume this issue will resolve itself once the erroneous `1.0.04` goes away, or gets fixed.

Yawning, 2020-09-09:
This appears to be fixed by Lenovo re-uploading the 0.1.04 firmware update on 2020-09-08 09:56:04.967290 (`f9b9afca6213e5d88a2a0ca-Lenovo-ThinkPad-GX3A2-EmbeddedControllerFirmware-1.04.cab`), and can more than likely be closed.

hughsie, 2020-09-09:
> This appears to be fixed by Lenovo re-uploading the 0.1.04 firmware update

Great, thanks. I guess the system is working like designed :)

closed 2020-09-09 by hughsie