fwupd/firmware-lenovo #64: Thinkpad X1 Carbon Gen7: Update fails when trying to install multiple updates at once

mendel5, 2020-07-27:
The problem described in this issue happened to me almost a month ago. I'm sorry if it is lacking some details but maybe it is still helpful.

I have not used fwupdmgr on my laptop (Thinkpad X1 Carbon Gen7) for about 3 months. When I did it again there were four updates available. I remember one update was for the Thunderbolt controller and another one for some type of UEFI device. I also remember that two of the four updates specifically used UEFI update capsules.

I entered the following commands:
```
fwupdmgr refresh
fwupdmgr get-devices
fwupdmgr get-updates
fwupdmgr update
```

When fwupdmgr asked me whether to install all updates I pressed "Y" for all of the four updates. Then I restarted the computer to have the updates installed. When I ran the commands listed above again, two of the four updates have been successfully installed. The successful updates were the ones that did not use the UEFI update capsules. But the two updates using the UEFI update capsule were still shown as available so I concluded that they hadn't been installed.

The solution was to install both updates capsules on their own. First one of the capsule updates, then reboot, then the other one, then reboot, then it was done. Unfortunately I don't have any kind of output from fwupdmgr anymore.

I was expecting all four updates, two using UEFI update capsules and two without, to be installed with one single reboot. That was not possible so I first installed the two updates without capsules, rebooted, installed the first capsule, rebooted and then the second capsule. That was successful. Is it possible to install four updates, two of which are UEFI capsules, with just a single reboot?

hsanjuan, 2020-08-18:
I have experienced and can confirm this. Need to install things one by one. Always thought multiple upgrades on a single reboot were not really supported.

mendel5, 2020-08-23:
@hsanjuan I'm glad to hear from you. So it seems the issue is not on my side.

dailytabs, 2020-09-11:
1. Gen. 8 just got delivered today. Downloaded Fedora 32 iso and installed alongside windows.
3. ran `dnf update` and gnome-shell crashed with SIGSEV.
4. Logged back in, decided to use "Software" application.
5. This indicated 3 updates (vs the 1500+ that dnf was working on). All were firmware: 2 mentioned X1 Gen8/<something else> and the 3rd was intel. All were firmware-ish related and at least 1 of the 3 mentioned EFI/UEFI in its name/description.
6. Clicked "Update" on the intel one. Got a notification that it requires AC power.
7. Plugged in power cable, "Update", notification said it was done, then another asking me to reboot, I said later.
8. The Updates page refreshed and showed the same 3 updates.
9. Clicked "Update" on intel one again. Same thing, including the refresh.
10. Clicked "Update" on the other 2, sequentially. Same thing, except on the last one I accepted the reboot.
11. Lenovo image came up with Fedora icon.
12. Text started writing in the top-left, something about /uefi/Fedora/...there were abour 3-5 lines, I don't recall the text, but then it had 4 lines saying something like adding module.
13. Screen flashed. Might have flashed the login screen, wasn't looking.
14. Appeared to reboot.
15. Stuck on Lenovo logo, no icon or "loading" spinner, as usually happens on boot.

Been there for over 30 minutes now.
The "power" button does nothing, even holding for 10+ seconds.
The battery is not removable.
It's just going to sit there until the battery dies, then who knows?

kmauleon, 2021-01-04:
hi all discussing with @hughsie on how to update BIOS/EC separately from MEFW.... thank you very much....