fwupd/firmware-lenovo #65: Thinkpad X1 Carbon Gen 7: Consumer ME update fails to install

Yawning, 2020-07-27:
ThinkPad X1 Yoga 4th/ThinkPad X1 Carbon 7th Consumer ME Update Version 12.0.64.1551 (LVFS: 192.64.1551) fails to actually install.

1) `fwupdmgr upgrade`
2) The system reboots into the updater (black screen with Lenovo logo).
3) "Press a key to continue"
4) System reboots
5) Update not installed.

```
20QD0000US
│
└─Intel Management Engine:
  │   Device ID:           0ec6a2d51c2c55a649247d119ffb8e229168d6c9
  │   Current version:     192.47.1524
  │   Minimum Version:     192.47.1524
  │   Vendor:              DMI:LENOVO
  │   Update State:        success
  │   GUID:                c3e4be53-e714-4ea1-bb9c-7fe13a98b556
  │   Device Flags:        • Internal device
  │                        • Updatable
  │                        • Requires AC power
  │                        • Supported on remote server
  │                        • Needs a reboot after installation
  │                        • Device is usable for the duration of the update
  │ 
  └─ThinkPad X1 Yoga 4th/ThinkPad X1 Carbon 7th Consumer ME Update:
        New version:       192.64.1551
        Remote ID:         lvfs
        Summary:           Lenovo ThinkPad X1 Yoga 4th/X1 Carbon 7th Consumer ME Firmware
        License:           Proprietary
        Size:              3.0 MB
        Created:           2016-07-08
        Urgency:           High
        Vendor:            Lenovo Ltd.
        Flags:             is-upgrade
        Description:       • 0 Intel Platform Update 2020.1 (Maintenance Release)
        
        Version 12.0.64.1551 (LVFS: 192.64.1551)
        
        Problem Fixes
        
         • BIOS post will be over 30 sec after G3 reset.After flashing image by fpt.efi -f BIOS.rom -me and fpt.efi -greset
         • Intel CSME PSIRT-TA-2019-10-001(IPU 2020.1)
         • Mitigated the following security vulnerabilities under issues:
        
        Security issues fixed:
        
         • CVE-2020-0539
         • CVE-2020-0537
         • CVE-2020-0536
         • CVE-2020-0540
         • CVE-2020-0535
         • CVE-2020-0531
         • CVE-2020-0541
         • CVE-2020-0532
         • CVE-2020-0534
         • CVE-2020-0538
         • CVE-2020-0542
```

This is probably #56 but for the X1 Carbon Gen 7.

kmauleon, 2020-07-28:
hi @Yawning 
thank you very much for the attention... but the new version requires latest fwupd.... can you please try to update the following first...

shim-signed
fwupd
fwupdate

then execute the following

fwupdmgr refresh
fwupdmgr update

please let me know the results of this commands....thank you very much

Yawning, 2020-07-28:
According to my package manager I am running the latest available for F32.

fwupd-1.4.4-1.fc32.x86_64
shim-x64-15-8.x86_64
shim-ia32-15-8.x86_64

Do I need to run something more bleeding edge?

kmauleon, 2020-07-29:
hi @Yawning 
I have tried with Ubuntu/Debian/RHEL/Open Suse and updated OK.... I will try to download Fedora 32 and update you....
BTW, can you please send results of this command... Thank you very much

fwupdmgr get-devices

kmauleon, 2020-07-29:
hi @Yawning 
i did encounter same issue a while ago with SLED 15 SP2... upon further checking, the update seems to offer system fw 1.36 and ecfw  1.19 as well.  but i was able to successfully update mefw separately via manual update & gnome software.  when i tried to update system & ecfw first, system/ecfw updates did not push through. so i guess there's something about the latest system and ecfw updates. so can you please try updating mefw only via gnome software....let me know your results.  thank you very much.
NOTE: please make sure you have at least BIOS system 1.31 and ecfw 1.16 prior to updating mefw.....

Yawning, 2020-07-29:
```
20QD0000US
│
├─Thunderbolt Controller:
│     Device ID:           326b7fa241d54ea3aa982825ac3c295b21b29de8
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     43.00
│     Vendor:              Lenovo (TBT:0x0109)
│     GUIDs:               0f11f23b-ccfc-5bac-9a41-405429734903
│                          c14d72b3-bc0c-5cfb-be3d-8758a48ac8b9
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Device stages updates
│   
├─Embedded Controller:
│     Device ID:           c085a3cbf4dc1d60d110b7616816e2ffd279b64e
│     Current version:     0.1.19
│     Minimum Version:     0.1.19
│     Vendor:              DMI:LENOVO
│     GUID:                f72e048b-65bd-4e71-9071-1ac7045223e5
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─Intel Management Engine:
│     Device ID:           0ec6a2d51c2c55a649247d119ffb8e229168d6c9
│     Current version:     192.47.1524
│     Minimum Version:     192.47.1524
│     Vendor:              DMI:LENOVO
│     Update State:        success
│     GUID:                c3e4be53-e714-4ea1-bb9c-7fe13a98b556
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─Prometheus:
│ │   Device ID:           d432baa2162a32c1554ef24bd8281953b9d07c11
│ │   Summary:             Fingerprint reader
│ │   Current version:     10.01.3158804
│ │   Vendor:              Synaptics (USB:0x06CB)
│ │   Install Duration:    2 seconds
│ │   GUIDs:               09e8bf16-3e69-50f5-bb66-c7a040248352
│ │                        8088f861-6318-5b1e-9ce4-fbddbedb09ac
│ │   Device Flags:        • Updatable
│ │                        • Supported on remote server
│ │                        • Cryptographic hash verification is available
│ │ 
│ └─Prometheus IOTA Config:
│       Device ID:         8dcffb5d059857368caa2b7b22371ebd831b0c0c
│       Current version:   0021
│       Minimum Version:   0021
│       Vendor:            Synaptics (USB:0x06CB)
│       GUID:              7c5a1e62-38fa-5859-9337-09dbac6377e4
│       Device Flags:      • Updatable
│                          • Supported on remote server
│     
├─Samsung SSD 970 PRO 512GB:
│     Device ID:           3743975ad7f64f8d6575a9ae49fb3a8856fe186f
│     Summary:             NVM Express Solid State Drive
│     Current version:     1B2QEXP7
│     Vendor:              Samsung Electronics Co Ltd (NVME:0x144D)
│     GUIDs:               6e54c992-d302-59ab-b454-2d26ddd63e6d
│                          47335265-a509-51f7-841e-1c94911af66b
[... 49 lines cut ...]
      Vendor:              DMI:LENOVO
      GUID:                68d3b08f-6ed2-43ca-86f0-f1ad6e41b390
      Device Flags:        • Internal device
                           • Updatable
                           • Requires AC power
                           • Needs a reboot after installation
                           • Device is usable for the duration of the update
```

> so can you please try updating mefw only via gnome software.

Same result.  Reboots into the updater (black screen with Lenovo logo), "Press any key to continue." prompt appears after a while, update does not apply.

> NOTE: please make sure you have at least BIOS system 1.31 and ecfw 1.16 prior to updating mefw

The system in question is running BIOS 1.36 and EC FW 1.19.

And for completeness sake, this is the output of `fwupdmgr --version`:
```
client version:	1.4.4
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.4.4
```

kmauleon, 2020-07-30:
hi @Yawning 
"Press any key to continue" is expected... press enter then wait for system to reboot....thank you very much...

Yawning, 2020-07-31:
> "Press any key to continue" is expected... press enter then wait for system to reboot....thank you very much...

Ok.  The system reboots immediately instead of applying the update.  If there is a way to get more diagnostic information, please let me know.

kmauleon, 2020-08-03:
Can you please show result of the following again after executing "fwupdmgr update"....thank you very much...

fwupdmgr get-devices

kmauleon, 2020-08-03:
hi @Yawning 
I have tried with Fedora 32 and did not encounter any issue.  Please see attached images for your reference.  Thank you very much.
[fedora 32 mefw consumer update.zip](https://github.com/fwupd/firmware-lenovo-thinkpad/files/5013748/fedora.32.mefw.consumer.update.zip)

Yawning, 2020-08-03:
> Can you please show result of the following again after executing "fwupdmgr 
> update"....thank you very much...
> 
> fwupdmgr get-devices

The output is unchanged from what I posted 5 days ago, but I have more information that leaves me even more confused.

To double check things, I attempted to determine what ME firmware is actually present on my system.

 * System BIOS setup menu: `ME Firmware Version 12.0.64.1551`
 * Intel's CSME detection tool: `Version: 12.0.64.1551`

```
$ cat /sys/class/mei/mei0/fw_ver                                   
0:12.0.64.1551
0:12.0.64.1551
0:12.0.35.1427
```

I'm puzzled why fwupd insists that I am running `192.47.1524`, when the system bios, Intel's tool (that just reads sysfs if possible from what I can tell), and the kernel all think I am running something newer.

So, the good news is that my laptop's ME firmware is presumably up to date, and it probably updated successfully the first time I ran the update, before I even filed this issue.  The bad news is, fwupd is convinced that I am running an older version somehow.

kmauleon, 2020-08-04:
hi @Yawning ...have you tried clearing the fwupd cache?   but it's good to know that MEFW version was indeed updated.  Thank you very much for the information.

hi @hughsie any idea why fwupd still identifies that system has olde version even if it was actually updated?  please advise.  Thank you very much...

hughsie, 2020-08-04:
> any idea why fwupd still identifies that system has olde version even if it was actually updated

We're just reading out the version from the ESRT table;  e.g. doing the slightly convoluted version of `sudo cat /sys/firmware/efi/esrt/entries/entry*/fw_version` -- @Yawning can you get me the output of that please?

Yawning, 2020-08-04:
>  @Yawning can you get me the output of that please?

I assume this is the ESRT entry of interest based on the GUID.

```
==> /sys/firmware/efi/esrt/entries/entry1/capsule_flags <==
0x8010

==> /sys/firmware/efi/esrt/entries/entry1/fw_class <==
c3e4be53-e714-4ea1-bb9c-7fe13a98b556

==> /sys/firmware/efi/esrt/entries/entry1/fw_type <==
2

==> /sys/firmware/efi/esrt/entries/entry1/fw_version <==
3224307188

==> /sys/firmware/efi/esrt/entries/entry1/last_attempt_status <==
0

==> /sys/firmware/efi/esrt/entries/entry1/last_attempt_version <==
3224307188

==> /sys/firmware/efi/esrt/entries/entry1/lowest_supported_fw_version <==
3224307188
```

Is it possible for an update to get installed but the corresponding ESRT entry to not get updated?

kmauleon, 2020-08-05:
hi @Yawning since your MEFW was updated successfully, can we close this issue and track your fwupd issue in here >>> #67 ?
thank you very much...

Yawning, 2020-08-05:
> track your fwupd issue in here

Sure that sounds reasonable.

hughsie, 2020-08-05:
> ==> /sys/firmware/efi/esrt/entries/entry1/fw_version <==
> 3224307188

This is `192.47.1524`. @kmauleon I don't think the ESRT version was updated to the intended `192.64.1551`

kmauleon, 2020-08-05:
hi @Yawning ...I will keep this open for a while until we know the root cause... thank you very much...
hi @hughsie... I will consult our team regarding this matter.  thank you very much...

vin-d, 2020-10-06:
> > ==> /sys/firmware/efi/esrt/entries/entry1/fw_version <==
> > 3224307188
> 
> This is `192.47.1524`. @kmauleon I don't think the ESRT version was updated to the intended `192.64.1551`

exactly the same problem here and i'm on a P52

kmauleon, 2020-10-23:
hi.. can you please try this >> https://github.com/fwupd/firmware-lenovo-thinkpad/issues/70#issuecomment-714491086
downgrade / update seem to fix the issue as stated... thank you very much

Yawning, 2020-10-23:
```
Choose a device:
0.	Cancel
1.	c085a3cbf4dc1d60d110b7616816e2ffd279b64e (Embedded Controller)
2.	0ec6a2d51c2c55a649247d119ffb8e229168d6c9 (Intel Management Engine)
3.	d432baa2162a32c1554ef24bd8281953b9d07c11 (Prometheus)
4.	8dcffb5d059857368caa2b7b22371ebd831b0c0c (Prometheus IOTA Config)
5.	f09b748a98b31bb5c7c64d83f24ddbc80dff7c5f (System Firmware)
2
No downgrades for Intel Management Engine: current version is 192.47.1524: 192.64.1551=newer, 192.47.1524=same
```

fwupd is convinced I am running the earliest available version so it is not possible to downgrade (https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN2HRN.firmware).

Following the same train of thought, attempting to re-install the version that is listed in my ESRT via `fwupdmgr reinstall c3e4be53-e714-4ea1-bb9c-7fe13a98b556` reboots into the flashing app (with the Lenovo logo), but fails with `Error 8707: Firmware update failed due to an internal error`.

kmauleon, 2020-10-25:
hi @Yawning , can you please check if there are updates for BIOS and EC firmware.  please update those two first then update MEFW if successful.  thank you very much....
     **fwupdmgr refresh**
     **fwupdmgr update** _(update for BIOS & EC only)_
_system will restart, update, and reboot again if successful_
     **fwupdmgr update** _(update MEFW)_
_system will restart, update, and reboot again if successful_

Yawning, 2020-10-26:
According to `fwupdmgr update` I am running the latest EC and System (BIOS) firmware.

kmauleon, 2020-10-26:
hi @Yawning can you please show results of below command... thanks
**fwupdmgr report-history**

Yawning, 2020-10-27:
I have history disabled for privacy reasons (`No reports require uploading`).

Here is the `fwupdmgr get-devices` output that lists the versions of the various things:
```
20QD0000US
│
├─Embedded Controller:
│     Device ID:           c085a3cbf4dc1d60d110b7616816e2ffd279b64e
│     Current version:     0.1.20
│     Minimum Version:     0.1.20
│     Vendor:              DMI:LENOVO
│     GUID:                f72e048b-65bd-4e71-9071-1ac7045223e5
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─Intel Management Engine:
│     Device ID:           0ec6a2d51c2c55a649247d119ffb8e229168d6c9
│     Current version:     192.47.1524
│     Minimum Version:     192.47.1524
│     Vendor:              DMI:LENOVO
│     Update State:        success
│     GUID:                c3e4be53-e714-4ea1-bb9c-7fe13a98b556
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─Prometheus:
│ │   Device ID:           d432baa2162a32c1554ef24bd8281953b9d07c11
│ │   Summary:             Fingerprint reader
│ │   Current version:     10.01.3158804
│ │   Vendor:              Synaptics (USB:0x06CB)
│ │   Install Duration:    2 seconds
│ │   GUIDs:               09e8bf16-3e69-50f5-bb66-c7a040248352
│ │                        8088f861-6318-5b1e-9ce4-fbddbedb09ac
│ │   Device Flags:        • Updatable
│ │                        • Supported on remote server
│ │                        • Cryptographic hash verification is available
│ │ 
│ └─Prometheus IOTA Config:
│       Device ID:         8dcffb5d059857368caa2b7b22371ebd831b0c0c
│       Current version:   0021
│       Minimum Version:   0021
│       Vendor:            Synaptics (USB:0x06CB)
│       GUID:              7c5a1e62-38fa-5859-9337-09dbac6377e4
│       Device Flags:      • Updatable
│                          • Supported on remote server
│     
├─Samsung SSD 970 PRO 512GB:
│     Device ID:           3743975ad7f64f8d6575a9ae49fb3a8856fe186f
│     Summary:             NVM Express Solid State Drive
│     Current version:     1B2QEXP7
│     Vendor:              Samsung Electronics Co Ltd (NVME:0x144D)
│     GUIDs:               6e54c992-d302-59ab-b454-2d26ddd63e6d
│                          47335265-a509-51f7-841e-1c94911af66b
│                          d356e2f6-25bb-5121-b968-a05719ccdbfa
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─System Firmware:
│     Device ID:           f09b748a98b31bb5c7c64d83f24ddbc80dff7c5f
│     Current version:     0.1.40
│     Minimum Version:     0.0.1
│     Vendor:              LENOVO (DMI:LENOVO)
│     GUID:                4bbc40fa-f81e-4206-bc70-a1f7b744d964
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           9561e1f11a616405008effa52d976f04de1e3c7e
│     Current version:     16777224
│     Minimum Version:     1
│     Vendor:              DMI:LENOVO
│     GUID:                057d7512-072c-4b15-82b3-c5156b0f0daf
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           f61a138e8240c473a88d703babf7be5bc0e098e6
│     Current version:     4784136
│     Minimum Version:     4784136
│     Vendor:              DMI:LENOVO
│     GUID:                db246b7d-a882-4e0e-984f-23ac1b0270db
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
└─UEFI Device Firmware:
      Device ID:           97d441c3228fe8615f883b166a8809389803f9e2
      Current version:     26403
      Minimum Version:     1
      Vendor:              DMI:LENOVO
      GUID:                68d3b08f-6ed2-43ca-86f0-f1ad6e41b390
      Device Flags:        • Internal device
                           • Updatable
                           • Requires AC power
                           • Needs a reboot after installation
                           • Device is usable for the duration of the update
```

kmauleon, 2020-10-28:
hi @Yawning .. how about installing it via GNOME Software? Please check if you can see any available updates for MEFW there... thank you very much (latest should be **192.64.1551**)

Yawning, 2020-10-28:
I'm confused.

My system has 192.64.1551 of the ME firmware installed, but the updater failed to correctly update the ESRT entry.  Regardless of how the update package is launched (via fwupd directly or by GNOME software), this should not (and does not) change the fact that the update package can not and will not write the correct ESRT entry.

Since I am unable to downgrade my system's ME firmware and then reapply the update (`Error 8707`), it seems that all I can do is either:

 - Wait till a version of the 192.64.1551 update that can/will write an ESRT entry is published (unlikely).

 - Wait till a version of the ME firmware > 192.64.1551 is published, and hope the update both applies and correctly sets the ESRT entry.

kmauleon, 2020-10-29:
hi @Yawning , ESRT table should have been updated by BIOS so our BIOS team is looking into this... will ping you again once i got any updates from them... thank you very much for bearing with us.

kmauleon, 2020-11-16:
hi all can you please try with latest MEFW >> https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN2HRN.firmware
thank you very much...

Yawning, 2020-11-16:
```
$ fwupdmgr install ef3b7c15f5cec388f9a51bc5aaa08a5c05a2e6f1e3aa114a82524085c0debfc9-Lenovo-ThinkPad-X1Yoga4thX1Carbon7th-ConsumerMEFirmware-12.0.68.1606.cab
Decompressing…           [***************************************]
Not compatible with firmware version 192.47.1524, requires >= 192.64.1551
```

The ESRT being incorrect leads to fwupdmgr refusing to attempt to install the update.  None of the "Do what I say" looking options (`--force`, `--no-safety-check`, `--no-metadata-check`) alter the behavior.

kmauleon, 2020-11-16:
hi @Yawning can you please try with the following ... thank you very much
     fwupdmgr --allow-older --allow-reinstall install <cab file>

Yawning, 2020-11-17:
That doesn't change the behavior.  As far as I can tell, there is no way to bypass the check at https://github.com/fwupd/fwupd/blob/d483fcedd5e54b5fc805a9826e8568982c28cc31/src/fu-engine.c#L1179

kmauleon, 2020-11-17:
hi @Yawning ... i temporarily removed the version requirement. can you please try again? thank you very much

Yawning, 2020-11-18:
Well, there's good news and bad news.  The good news is, according to the system setup screen, the update applied, because it lists the new ME firmware version.  The bad news is, the ESRT entry did not get updated again, so fwupd is still convinced I am running `192.47.1524` when I am not.

The relevant parts from `fwupdmgr history` are probably where it says `Update State: failed` and `Update Error: failed to run update on reboot`.  I have tried to manually downgrade to `192.47.1524` again, but unsurprisingly the capsule still fails with `Error 8707`.  Reinstalling the latest update appears to succeed, but the ESRT does not get updated.

Since the ME firmware on the affected laptop is again updated to the latest version (regardless of what the ESRT says), I assume the best course of action is for me to set fwupd to ignore ME firmware updates till either a new system UEFI image that addresses this issue is released, or a new ME firmware update is made available.

kmauleon, 2020-12-23:
hi @Yawning latest 12.0.70.1652 is available already... can you please try with this? thank you very much...
https://fwupd.org/downloads/e78d2c1d0372734379e135b5fe26bdf479642529ea5136036cffd1d5f355429e-Lenovo-ThinkPad-X1Yoga4thX1Carbon7th-ConsumerMEFirmware-12.0.70.1652.cab

Yawning, 2020-12-23:
The good news is that the update succeeded without issue (as verified from the system BIOS setup screen).  The bad news is that fwupdmgr is still convinced that I am running the old version.

```
20QD0000US
│
├─Embedded Controller:
│     Device ID:          c085a3cbf4dc1d60d110b7616816e2ffd279b64e
│     Current version:    0.1.20
│     Minimum Version:    0.1.20
│     Vendor:             DMI:LENOVO
│     GUIDs:              f72e048b-65bd-4e71-9071-1ac7045223e5
│                         9cd54062-d764-51a7-a780-51a3695beae2
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─Intel Management Engine:
│     Device ID:          0ec6a2d51c2c55a649247d119ffb8e229168d6c9
│     Current version:    192.47.1524
│     Minimum Version:    192.47.1524
│     Vendor:             DMI:LENOVO
│     GUIDs:              c3e4be53-e714-4ea1-bb9c-7fe13a98b556
│                         35cbf8ae-bee1-52fa-bf0d-11f1fc4c12f5
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
```

```
└─Intel Management Engine:
  │   Device ID:          0ec6a2d51c2c55a649247d119ffb8e229168d6c9
  │   Previous version:   192.47.1524
  │   Update State:       failed
  │   Update Error:       failed to run update on reboot
  │   Last modified:      2020-12-23 11:33
  │   GUID:               c3e4be53-e714-4ea1-bb9c-7fe13a98b556
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Needs a reboot after installation
  │                       • Device is usable for the duration of the update
  │ 
  └─  New version:      192.70.1652
        License:          Unknown
```

I have not checked, but I assume this is still ESDT related issues.  This isn't urgent since my laptop is up to date, despite what fwupdmgr thinks, I hope you and the team have a happy holidays, and maybe we will have better luck with this next year.

kmauleon, 2020-12-24:
hi @Yawning thank you very much for the information... great that MEFW was updated to latest... unfortunately still have not received any updates from BIOS dev regarding ESRT table issue... anyway, let's keep this open for tracking.  Happy Christmas!

kmauleon, 2021-02-04:
update from BIOS Dev... thank you very much....
**no clear progress yet regarding BIOS ESRT problem investigation**

federico-cuello-solarisbank, 2021-05-11:
Hi, same problem here with a `ThinkPad X1 Carbon 6th`.
```
$ fwupdmgr --version
client version: 1.5.8
compile-time dependency versions
        gusb:   0.3.5

daemon version: 1.5.8
```

`Upgrade available for Intel Management Engine from 184.79.3722 to 184.83.3874`

But the BIOS setup screen show the new version as installed.

kmauleon, 2021-05-26:
> Hi, same problem here with a `ThinkPad X1 Carbon 6th`.
>
> ```
> $ fwupdmgr --version
> client version: 1.5.8
> compile-time dependency versions
>         gusb:   0.3.5
>
> daemon version: 1.5.8
> ```
>
> `Upgrade [...]

hi @federico-cuello-solarisbank may I ask you help in debugging the BIOS issue.  Can you please contact me >> kmauleon@lenovo.com thank you very much in advanced....

Yawning, 2021-08-17:
This still is an issue for me, after over a year and countless updates.  It's still not a big deal because the firmware does update, but it isn't any less annoying.

```
Upgrade available for Intel Management Engine from 192.47.1524 to 192.72.1757
20QD0000US must remain plugged into a power source for the duration of the update to avoid damage. Continue with update? [Y|n]: 
```

kmauleon, 2021-08-17:
hi @Yawning thank you very much for the update... unfortunately, I can't get BIOS dev to further investigate the ESRT table problem without recreating the issue.  I already have done a lot of testings for this one but still unable to recreate the issue T_T

Yawning, 2021-08-17:
> hi @Yawning thank you very much for the update... unfortunately, I can't get BIOS dev to further investigate the ESRT table problem without recreating the issue. I already have done a lot of testings for this one but still unable to recreate the issue T_T

That's fine, is there anything I can do to help with this?

kmauleon, 2021-08-18:
hi @Yawning if you still remember the specific config or scenario the first time you see the issue... we need to have the recreational procedure for this in order for BIOS Dev to investigate... if you have any information please advise... thank you very much

deriamis, 2022-01-26:
@kmauleon I just want to point out that *only* the Intel ME capsule doesn't update the ESRT for me. All other updates work. I'm quite sure my system has been in essentially the same configuration since before the problem started, and I am very diligent about running updates (including firmware) once per week. The only thing I noticed was that one day I was suddenly getting the same Intel ME update over and over again after it working perfectly fine previously. I'm willing to give configuration details if needed.

kmauleon, 2022-01-26:
hi @deriamis yes this is an existing issue, unfortunately ESRT table is managed by BIOS, thus, can't do anything on ME side.
Thank you very much.
hi @mrhpearson any update from BIOS dev regarding the ESRT issue?  please advise.  Thank you very much.

deriamis, 2022-01-27:
@kmauleon I think I've unintentionally fixed my problem, but I don't fully understand how. 

Through a series of complicated (and embarrassing) events, I managed to delete my GPT last night and corrupted my EFI system partition in the process. Everything else was fine, and I rebuilt the EFI system partition from scratch.

After rebuilding my EFI system partition, the Intel ME update installed properly - and it even had a different update screen. I think, in my case, this didn't have anything to do with the ESRT. I think something was wrong with a previous update, and something got "stuck" in my EFI system partition.

kmauleon, 2022-01-27:
hi @deriamis thank you very much for the update... this is interesting.  honestly it's hard to pinpoint which caused the failure since it was not recreated here. but based on investigation, fwupd was not getting the correct version.  fwupd is using ESRT table as reference and upon checking, ESRT table does not contain the correct version.  Thus, we have concluded that somehow this was caused by BIOS update since ESRT table is managed by BIOS.  but I am glad you managed to fix the problem...

deriamis, 2022-01-27:
@kmauleon Could it be that the ESRT may not be getting updated because there's something wrong with the update process in relation to what's on the EFI system partition? I don't know enough about how `fwupd` and update capsules work, but something was definitely different with how the update looked and behaved while it was running when it worked.

kmauleon, 2022-01-27:
@deriamis could it be but neither do I... we can't exactly say what's the cause since we never replicated the issue.  but maybe we can have some insight from @hughsie ... what do you think @hughsie ? ^_^

MfCrizz, 2022-04-28:
Hi, i have the same issue with a ThinkPad X1 Carbon 7th.
Tried on different OSs Ubuntu 22.04, Pop_OS 22.04, Pop_OS 21.10.
The ME seems updated but fwupdmgr reports an old version number.

After some googling i found this issue and here are my findings:

    - System BIOS setup menu: ME Firmware Version 12.0.85.1869
    - Intel's CSME detection tool: Version: 12.0.85.1869
    - fwupdmgr: 192.81.1753

```
sudo cat /sys/class/mei/mei0/fw_ver
0:12.0.85.1869
0:12.0.85.1869
0:12.0.40.1434
```

```
sudo cat /sys/firmware/efi/esrt/entries/entry?/fw_version
65587
3226535641
65561
16785559
4784192
26918
```

```
fwupdmgr --version 
runtime   org.freedesktop.fwupd         1.7.5
runtime   com.dell.libsmbios                2.4
compile   org.freedesktop.gusb           0.3.10
runtime   org.kernel                             5.15.0-27-generic
compile   org.freedesktop.fwupd         1.7.5
runtime   org.freedesktop.gusb           0.3.10
```

fwupdmgr output:
```
20QES01L00
│
└─Intel Management Engine:
  │   Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
  │   Summary:            UEFI ESRT device
  │   Current version:    192.81.1753
  │   Minimum Version:    192.81.1753
  │   Vendor:             Lenovo (DMI:LENOVO)
  │   Update State:       Success
  │   GUIDs:              c3e4be53-e714-4ea1-bb9c-7fe13a98b556
  │                       35cbf8ae-bee1-52fa-bf0d-11f1fc4c12f5 ← UEFI\RES_{C3E4BE53-E714-4EA1-BB9C-7FE13A98B556}
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Device is usable for the duration of the update
  │ 
  └─ThinkPad X1 Yoga 4th/ThinkPad X1 Carbon 7th:
        New version:      192.85.1869
        Remote ID:        lvfs
        Release ID:       11763
        Summary:          Lenovo ThinkPad X1 Yoga 4th/X1 Carbon 7th Consumer ME Firmware
        License:          Proprietary
        Size:             3,0 MB
        Created:          2016-07-08
        Urgency:          High
        Details:          https://pcsupport.lenovo.com/de/en/search?query=N2HRG32W
        Vendor:           Lenovo
        Release Flags:    • Is upgrade
        Description:      
        • 0 Intel Platform Update 2021.2 Product Version Maintenance Release
        
        Version 12.0.85.1869 (LVFS: 192.85.1869)
        
        Problem Fixes
        
        • Mitigated the following security vulnerabilities under issues. Please see fixed issues for details.
        
        Note: Please suspend or disable any disk encryption tool prior to update.
```

I hope this information helps, and if i can do something to help with this issue please let me know.
Thanks!

kmauleon, 2022-04-29:
BIOS dev team is still investigating this... unfortunately, the issue is not replicated thus taking it some time. @ChiWei-Chen do we have trial bios for x1 carbon 7 as well?  please advise.  thank you very much

AdnanHodzic, 2022-05-07:
Hi,

I too have the same issue on Ubuntu 22.04 with X1 Carbon Gen 7, for full reference please refer to: #222 

I also updated BIOS to the latest version, `N2HET68W (1.51)` and I successfully ran:

```
fwupdmgr --allow-older --allow-reinstall install 8aeedced34d7f64af0302b88b224dd0b5eba1370274c9918972f7c756126671a-Lenovo-ThinkPad-X1Yoga4thX1Carbon7th-ConsumerMEFirmware-12.0.85.1869.cab
```

After [getting the .cab file.](after I got the .cab file). However, as before after I run `fwupdmgr update` I still get:

```
Devices with no available firmware updates: 
 • INTEL SSDPEKKF512G8L
 • UEFI Device Firmware
 • UEFI Device Firmware
Devices with the latest available firmware version:
 • Embedded Controller
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Intel Management Engine from 192.81.1753 to 192.85.1869?             ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ • 0 Intel Platform Update 2021.2 Product Version Maintenance Release         ║
║                                                                              ║
║ Version 12.0.85.1869 (LVFS: 192.85.1869)                                     ║
║                                                                              ║
║ Problem Fixes                                                                ║
║                                                                              ║
║ • Mitigated the following security vulnerabilities under issues. Please      ║
║ see fixed issues for details.                                                ║
║                                                                              ║
║ Note: Please suspend or disable any disk encryption tool prior to update.    ║
║                                                                              ║
║ 20QD00L1MH must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: 
```

Looking forward to hearing any updates on this issue.

ChiWei-Chen, 2022-05-11:
Hi 
We're working on the similar issue #140 which is on different platform, we don't have trial BIOS for X1 Carbon7 so far.
I'll keep you update once the FW team fix this issue.
Thank you.

AdnanHodzic, 2022-05-14:
@ChiWei-Chen thanks for the update, looking forward to any updates you may have.

pixelb, 2022-05-20:
Same issue here on X1 Carbon 7.
Note the root device is encrypted using luks.
Some output from `fwupdmgr get-devices`:

```
Devices that were not updated correctly:

 • Intel Management Engine (192.70.1652 → 192.85.1869)

Devices that have been updated successfully:

 • WDC PC SN730 SDBQNTY-512G-1001 (11130101 → 11170101)
 • Embedded Controller (0.1.22 → 0.1.25)
 • System Firmware (0.1.45 → 0.1.51)
 • Embedded Controller (0.1.20 → 0.1.22)
 • System Firmware (0.1.43 → 0.1.45)
 • Prometheus (10.01.3158804 → 10.01.3478575)
```

kmauleon, 2022-05-23:
hi @pixelb can you please check the MEFW version displayed in BIOS Setuo Menu just to confirm if this is ESRT related issue.... thank you very much..,

pixelb, 2022-05-23:
BIOS menu reports 12.0.85.1869.
This corresponds to the version trying to be updated to (i.e. 192.85.1869)
That suggests the BIOS is updated? Or partially updated?
fwupd thinks the update has failed at least

kmauleon, 2022-05-24:
hi @pixelb you indeed hit on BIOS ESRT issue wherein values in ESRT table were not updated correctly.  this is still under investigation by BIOS team.  no worries since you have updated your MEFW successfully although I am sure it's bothersome that you are prompted to update MEFW even if you already have the latest version.... for now please ignore the notification for MEFW update.  thank you very much...

lhedrich, 2022-06-21:
On my X1Y4 after not being able to update the managment engine for a long time: Today i updated first the BIOS to version 0.1.51. Aftwards i was able with the fwupdmgr to upgrade the Intel Management engine from 192.47.XXX to 192.90.2072. Thanks for the thread and the bios update team...

hein-kr, 2022-06-21:
On my X1Y4 after not being able to update the managment engine for a long time: Today i updated first the BIOS to version 0.1.51. Aftwards i was able with the fwupdmgr to upgrade the Intel Management engine from 192.47.XXX to 192.90.2072. Thanks for the thread and the bios update team...

[14 more replies]

closed 2022-08-30 by ChiWei-Chen