fwupd/firmware-lenovo #66: TPM PCR0 differs from reconstruction

smlx, 2020-07-29:
**Describe the bug**
I am currently unable to upgrade the system firmware on my Lenovo T590.

I've tried both via Gnome Software and via `fwupdmgr update`, and the upgrade appears to be staged correctly, but then on reboot the machine just boots straight into my regular OS instead of applying the firmware upgrade.

`fwupdmgr get-devices` shows this error:
[image]

The requested `fwupdtpmevlog` file is here: [fwupdtpmevlog.log](https://github.com/fwupd/fwupd/files/4995481/fwupdtpmevlog.log)

**Steps to Reproduce**
Run `fwupdmgr update`.

**Expected behavior**
I expected the firmware to successfully upgrade as it usually does.

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version
client version:	1.3.11
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.3.11
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): apt

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
20N4CTO1WW
│
├─Thunderbolt Controller:
│     Device ID:           5fa92a49258d586d046af060fa8dd04f6a109d50
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     20.00
│     Vendor:              Lenovo (TBT:0x0109)
│     GUIDs:               52265728-359a-574c-9a6a-a23d3b1f952d
│                          f117e89e-a75f-5f33-9e8e-c4aded9cbadf
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Device stages updates
│   
├─INTEL SSDPEKKF512G8L:
│     Device ID:           f2759da7fe8e0388c5f3601cb072f837b1070b03
│     Summary:             NVM Express Solid State Drive
│     Current version:     L12P
│     Vendor:              Intel Corporation (NVME:0x8086)
│     GUIDs:               79517f86-8df8-5d6e-a18b-33f0b36a78e9
│                          68db11e5-b0cf-5bc9-a94e-17e28496e505
│                          87fd9fec-f447-5ec6-a1cd-d55fac0d3d30
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─Intel AMT [unprovisioned]:
│     Device ID:           e2623122c99d58220498aacbfcfdb1baebbae3c5
│     Summary:             Hardware and firmware technology for remote out-of-band management
│     Current version:     12.0.47.1524
│     Bootloader Version:  12.0.47.1524
│     Vendor:              Intel Corporation
│     GUID:                2800f812-b7b4-2d4b-aca8-46e0ff65814c
│     Device Flags:        • Internal device
│   
├─System Firmware:
│     Device ID:           6150dd1f7291b0709289ab8a53cc85a17e117ef2
│     Current version:     0.1.66
│     Minimum Version:     0.0.1
│     Vendor:              LENOVO (DMI:LENOVO)
│     Update State:        success
│     Update Error:        TPM PCR0 differs from reconstruction, please see https://github.com/fwupd/fwupd/wiki/TPM-PCR0-differs-from-reconstruction
│     GUID:                603baf73-b997-45b5-86b4-2f981a008e18
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
│   
├─TPM:
│ │   Device ID:           c6a80ac3a22083423992a3cb15018989f37834d6
│ │   Current version:     7.2.1.0
│ │   Vendor:              Nuvoton Technology (TPM:NTC)
│ │   GUIDs:               ff71992e-52f7-5eea-94ef-883e56e034c6
│ │                        fac1c8f3-73c8-5cd6-8330-07a3690b5140
│ │                        8d94eb9a-78aa-513d-af50-319e3b9f5210
│ │                        e9ccc1dc-960a-5e09-afe9-e59a904b776d
│ │                        f1c4013d-3f4c-5623-a40b-81ad96e23860
│ │   Device Flags:        • Internal device
│ │ 
│ └─Event Log:
│       Device ID:         58bd405f31c48e6eca290b425f530a94c91e955c
│       Vendor:            Nuvoton Technology (TPM:NTC)
│       GUID:              a25657fe-b5dc-5be0-8b78-8b9dfec678ff
│       Device Flags:      • Internal device
│     
├─UEFI Device Firmware:
│     Device ID:           c0649684d1e6688e8147fac95eccb4fdd18d05aa
│     Current version:     192.47.1524
│     Minimum Version:     192.47.1524
│     Vendor:              DMI:LENOVO
│     GUID:                2c0665e2-fdbd-495e-b8e4-69d92b9c119a
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           489f23b2ba9c1adf3e9f9f10598c98ba5c6bba39
│     Current version:     0.1.19
│     Minimum Version:     0.1.19
│     Vendor:              DMI:LENOVO
│     GUID:                38ea6335-29ca-417b-8cd4-6b4e5e866f92
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           88cf266a57697921241cc5f4b412c3f1b5a07780
│     Current version:     1.1.7
│     Minimum Version:     0.0.1
│     Vendor:              DMI:LENOVO
│     GUID:                a6634b3a-f446-42f0-a0b2-62c7d4dcb694
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           0eefc4f2a337dac7bd1d9b7cff714112a95d6f8b
│     Current version:     0.0.26885
│     Minimum Version:     0.0.1
│     Vendor:              DMI:LENOVO
│     GUID:                13fc227f-caf5-4baf-b177-82f587ab3582
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
└─UHD Graphics 620 (Whiskey Lake):
      Device ID:           bbbf1ce3d1cf15550c3760b354592040292415bb
      Current version:     02
      Vendor:              Intel Corporation (PCI:0x8086)
      GUIDs:               6986c21a-0f3f-5538-bcef-fe4e8cda7ab7
                           52aa8ac7-c837-561f-af32-b7979cd2f6ed
      Device Flags:        • Internal device
                           • Cryptographic hash verification is available
```

**Additional questions**
- Operating system and version: Debian testing.
- Have you tried rebooting? Yes.
- Is this a regression? I don't know.

smlx, 2020-10-09:
For anyone on Linux looking for a workaround, I did this:

1. find the bootable ISO firmware update for your model on the lenovo website.
2. convert it to USB bootable image: `geteltorito -o n2iur24w.img n2iur24w.iso`
3. `dd` the image onto a USB stick
4. reboot
5. hit enter at the Lenovo logo
6. hit F12 and select the USB HDD
7. select option 1 to read the warnings and then 2 to upgrade the firmware
8. wait for the machine to reboot a few times. after the first reboot the screen goes black for ~5min so don't stress.
9. eventually it will boot all the way into your OS

dieunb, 2021-04-27:
I did the same way as @smlx but still got that error
[image]