fwupd/firmware-lenovo #67: fwupd not recognizing the current version of my UEFI on a ThinkPad T480

chaitunya, 2020-07-31:
**Describe the bug**
According to fwupdmgr, I need to upgrade my UEFI firmware from 184.70.3626 to 184.77.3664. However, my firmware is already at the latest version, according to my BIOS: 
[image]

When I try to upgrade, it runs the full upgrade, but fwupd doesn't recognize the changes.

**Steps to Reproduce**
1. Run `sudo fwupdmgr upgrade`
2. Reboot computer, go through upgrade process
3. Run `fwupdmgr get-updates`
4. The UEFI firmware appears to be at 184.70.3626, not 184.70.3664

**Expected behavior**
When I run fwupdmgr get-updates, it should show that there are no updates, since my UEFI firmware is at the latest version.

**fwupd version information**
```
client version:	1.3.11
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.3.11
```
Installed via apt.

**fwupd device information**

```
20L5S01J00
│
├─Thunderbolt Controller:
│     Device ID:           82cc5614e1c0580da72730c1a44c805e55b559be
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     20.00
│     Vendor:              Lenovo (TBT:0x0109)
│     GUIDs:               7d131734-64ed-5865-a46e-a67cfb9b5a46
│                          e45313fd-8d5c-54e9-b175-1f15ffe6b05c
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Device stages updates
│   
├─System Firmware:
│     Device ID:           48477ba14bd9bd0c97323a07caa54b7766f917c2
│     Current version:     0.1.34
│     Minimum Version:     0.0.1
│     Vendor:              LENOVO (DMI:LENOVO)
│     GUID:                dc7f0308-1ef8-4774-9ba5-89a58c4d731c
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
│   
├─TOSHIBA KSG60ZMV256G M.2 2280 256GB:
│     Device ID:           64ec4cd1e1c9565e79b00f4e7221b1c689b33e96
│     Summary:             ATA Drive
│     Current version:     ABDA4101
│     GUIDs:               b8f80bb3-a668-5f78-ab29-0036b3b97ccf
│                          d2751458-58a0-58cd-b6f6-ebe88d68df08
│                          20bb3a66-e85a-5fdd-862d-be4d2bd6f383
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
[... 389 lines cut ...]
│       └── zstd.mod
├── initrd.img -> initrd.img-5.4.0-42-generic
├── initrd.img-5.4.0-40-generic
├── initrd.img-5.4.0-42-generic
├── initrd.img.old -> initrd.img-5.4.0-40-generic
├── lost+found [error opening dir]
├── memtest86+.bin
├── memtest86+.elf
├── memtest86+_multiboot.bin
├── System.map-5.4.0-40-generic
├── System.map-5.4.0-42-generic
├── vmlinuz -> vmlinuz-5.4.0-42-generic
├── vmlinuz-5.4.0-40-generic
├── vmlinuz-5.4.0-42-generic
└── vmlinuz.old -> vmlinuz-5.4.0-40-generic

5 directories, 297 files
```

**Additional questions**
- Operating system and version: Ubuntu 20.04.1 LTS
- Have you tried rebooting? Yes
- Is this a regression? It has been like this for a while
- Are you using an NVMe disk? Yes
- Is secure boot enabled? No

kmauleon, 2020-10-22:
hi @chaitunya someone mentioned similar issue and resolved it #70 
-------------------------------------------------------------
Ok guys, I resolved it! Here's how:
$ sudo fwupdmgr downgrade 
This should give you a list of your recent updates. Just downgrade the looping updates. Don't worry though. Once you've downgraded, just go ahead and run
$ sudo fwupdmgr update 
And that should be the end of your troubles. Or at least it was the end of my troubles. No more updates which can't be installed. fwupdmgr --help for other options.
-------------------------------------------------------------
can you please try and let us know if it works.... thank you very much...

kmauleon, 2020-10-22:
oooopppps sorry for the big fonts... didn't know what actually happened there.... guess due to "separation dash lines" i used

kmauleon, 2020-10-22:
hi @chaitunya  ... thanks so much for your work around.... hope it works with others... btw, did your BIOS version changed from last time when you encountered the issue?

@hughsie if fwupdmgr downgrade / update fixed the issue... should this be a glitch in fwupd?

chaitunya, 2020-10-23:
@kmauleon I was able to fix the UEFI issues by booting into Windows and updating from there. However, I still encounter issues with updating the Intel Management Engine. When I tried doing the commands you mentioned in your other comment, I encountered this:

```
➜  ~ sudo fwupdmgr downgrade
Choose a device:
0.	Cancel
1.	c19580db344e418cf2b23297bcf46d75c06e432a (Embedded Controller)
2.	ea0fd60df22477668e549841f6a22a8c31cae587 (Intel Management Engine)
3.	48477ba14bd9bd0c97323a07caa54b7766f917c2 (System Firmware)
2
No downgrades for Intel Management Engine: current version is 184.70.3626: 184.79.3722=newer, 184.77.3664=newer, 184.70.3626=same, 184.65.3590=lowest, 184.60.3561=lowest
```

Do you know how to fix this issue?

kmauleon, 2020-10-23:
how about the following commands...

fwupdmgr refresh
fwupdmgr update

kmauleon, 2021-01-04:
hi @chaitunya how about the following commands...

fwupdmgr --force refresh
fwupdmgr update

kmauleon, 2021-02-04:
update from BIOS Dev... thank you very much....
**no clear progress yet regarding BIOS ESRT problem investigation**

mrhpearson, 2022-02-01:
I have a Trial BIOS for the T480 to potentially fix this problem if anybody is able to try it. Please ping me directly (markpearson at lenovo dot com) so I can share the image and details.

kmauleon, 2022-11-16:
Closing this as BIOS ESRT fix has been rolled out already.  Please reopen if you still see the problem.  Thank you very much.

closed 2022-11-16 by kmauleon