fwupd/firmware-lenovo #68: Firmware update for Lenovo T580

clydeh7, 2020-08-10:
For missing firmware please provide the following items:

* Link to firmware update at OEM website
https://pcsupport.lenovo.com/us/en/products/laptops-and-netbooks/thinkpad-t-series-laptops/thinkpad-t580-type-20l9-20la/downloads/driver-list/component?name=BIOS%2FUEFI
* Firmware release date (if available on OEM website)
01 Jul 2020
* Link to the device page on LVFS.
https://fwupd.org/lvfs/devices/

kmauleon, 2021-02-01:
Latest System firmware is already available [here](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN27ET.firmware).  please check.  thank you very much.

closed 2021-02-01 by kmauleon

clydeh7, 2021-02-01:
Unfortunately, my t580 died during the weekend...On Friday, we were in a
video chat with friends and were watching the battery go from 80% to 5% in
about an hour while the device was plugged in.  Previously, I received a
message that the Lenovo provided AC cord was not the correct one for the
computer, so I assume part of the problem could be that the inline
transformer had somehow been compromised.  I've decided to retire my t580
to the garage with other electronic garbage that I will dispose of after 2
1/2 years of use.  I purchased the t580 because I had previously had a
great deal of respect for the name, believing that spending a little more
money would provide me with a device that was durable for the long term.
I've since purchased an HP computer at a little more than half the cost of
the t580 and will not be purchasing another lenovo product.

clyde hladky

kmauleon, 2021-02-02:
hi @clydeh7 very sorry for this unfortunate incident. i'll do my best to relay this to proper owner. thank you very much for the feedback. hope that you find yourself using lenovo products again in the future. again, thank you very much.