fwupd/firmware-lenovo #69: Lenovo X1CG7 cant install any updates

emvoo, 2020-08-24:
Ok so the problem is it seems I cannot update any available updates for my machine either through 
`fwupdmgr update`
or 
`fwupdmgr install *.cab`
or
`fwupdate -a c3e4be53-e714-4ea1-bb9c-7fe13a98b556 firmware.bin`

No matter what method of updating I use, I either get 
`Embedded Controller installed 1.0.7 => required 1.0.09`
 but cannot update Embedded Controller
```
Found update ....
Found update ....
Found update ....
Adding new capsule
Adding new capsule
Adding new capsule
WARNING: Could not query capsule capabilities: Unsupported
WARNING: Could not apply capsules: Unsupported
```
Im currently running
```
UEFI BIOS Version: N2HET35W (1.18)
Embedded Controller Version: N2HHT24W (1.07)
ME Firmware Version: 12.0.40.1433
```
available versions:
```
UEFI BIOS Version: 1.37
Embedded Controller Version: 1.19
ME Firmware Version: 12.0.64.1551
```
```
fwupdmgr get-devices --show-all-devices
20QD00KWUK
│
├─Thunderbolt Controller:
│     Device ID:           e4ac08c0c0440156a166927b1e4c561d5df8f57a
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     44.00
│     Vendor:              Lenovo (TBT:0x0109)
│     GUIDs:               74cdddba-f7b6-5a31-a3dc-64186b5a5491
│                          18bb00e3-5d7a-55d3-831e-0e000c184b0e
│                          c7920601-0cda-507f-851e-92229eb1d370
│                          96c6512b-0138-51da-b02f-f44f2c4c7744
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Device stages updates
│   
├─Embedded Controller:
│     Device ID:           c085a3cbf4dc1d60d110b7616816e2ffd279b64e
│     Current version:     0.1.7
│     Minimum Version:     0.1.7
│     Vendor:              DMI:LENOVO
│     Update State:        success
│     GUID:                f72e048b-65bd-4e71-9071-1ac7045223e5
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─INTEL SSDPEKKF512G8L:
│     Device ID:           3743975ad7f64f8d6575a9ae49fb3a8856fe186f
│     Summary:             NVM Express Solid State Drive
│     Current version:     L12P
│     Vendor:              Intel Corporation (NVME:0x8086)
[... 610 lines cut ...]
├── initramfs-linux-zen.img
├── intel-ucode.img
├── vmlinuz-linux
└── vmlinuz-linux-zen

10 directories, 362 files
```

Additional questions

Operating system and version: Arch Linux
Have you tried rebooting? Ohh yes
Is this a regression? Not sure
Are you using an NVMe disk? Yes
Is secure boot enabled? No
Boot Device Lock: Disabled
Tried through Linux-Firmware-Updater: Both yes and no
```
fwupdmgr --version
client version:	1.4.5
compile-time dependency versions
	gusb:	0.3.5
	efivar:	37
daemon version:	1.4.5
```

kmauleon, 2020-10-20:
hi @emvoo thank you very much for the attention... please see advise from owner as below...

Looks like ECFW is not installed by LVFS before. Would you please advise as follows?

1.	Download the latest ECFW package from Lenovo we site. (https://pcsupport.lenovo.com/uu/en/products/laptops-and-netbooks/thinkpad-x-series-laptops/thinkpad-x1-carbon-7th-gen-type-20r1-20r2/downloads/ds540232-bios-update-utility-bootable-cd-for-linux-windows-10-64-bit-thinkpad-x1-carbon-7th-gen-x1-yoga-4th-gen)
2.	Extract it on the target system.
3.	Update ECFW by the following command:

fwupdate -a f72e048b-65bd-4e71-9071-1ac7045223e5 firmware.bin

kmauleon, 2020-10-20:
please let us know how it goes.... thank you very much

kmauleon, 2020-11-16:
hi @emvoo any update?  please advise... thank you very much

kmauleon, 2022-11-16:
closing as no update... thank you very much

closed 2022-11-16 by kmauleon