fwupd/firmware-lenovo #70: Lenovo T480s - ME Firmware upgrade loop

galibozek, 2020-09-04:
I have problem with upgrade this firmware - https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN22RM.firmware
after upgrade I still see old version, in bios see new version

```
20L8S3DK1A
│
└─Intel Management Engine:
  │   Device ID:           118b3d8cfafa8781890c4d3e23a4deb9fcb18e5a
  │   Current version:     184.77.3664
  │   Minimum Version:     184.77.3664
  │   Vendor:              DMI:LENOVO
  │   Update State:        success
  │   GUID:                39a39e45-38ce-4b05-809b-60351bf88ec4
  │   Device Flags:        • Internal device
  │                        • Updatable
  │                        • Requires AC power
  │                        • Supported on remote server
  │                        • Needs a reboot after installation
  │                        • Device is usable for the duration of the update
  │ 
  └─ThinkPad T480s Corporate ME Update:
        New version:       184.79.3722
        Remote ID:         lvfs
        Summary:           Lenovo ThinkPad T480s Corporate ME Firmware
        License:           Proprietary
        Size:              7.5 MB
        Created:           2016-07-08
        Urgency:           High
        Vendor:            Lenovo Ltd.
        Flags:             is-upgrade
        Description:       • 8 Intel Platform Update 2020.1
        
        Version 11.8.79.3722 (LVFS: 184.79.3722)
        
        Problem Fixes
        
         • Mitigated the following security vulnerabilities:
        
        Security issues fixed:
        
         • CVE-2020-8758
```

`Arch Linux` and 
```
fwupdmgr --version
client version: 1.4.5
compile-time dependency versions
        gusb:   0.3.5
        efivar: 37
daemon version: 1.4.5
```

jvanschie, 2020-09-26:
I have the same issue. Did you find a workaround to run this update?

ChauhanT, 2020-10-16:
Same here. There are a couple of updates which seem to never get resolved, not matter how many times I install them :(

smalltimer, 2020-10-16:
Same here. There are a couple of updates which seem to never get resolved, not matter how many times I install them :(

jvanschie, 2020-10-16:
In the end, I've managed to install the update by a manual update of fwupdmgr.

However, now I am having issues with at the boot loader.
I can only boot my machine by disconnecting my external monitor and interrupt normal boot by pressing enter, and then escape to proceed.

Thinkpad T480 with Nvidia MX150.

Did anyone solved this?

ChauhanT, 2020-10-16:
@jvanschie Sorry to hear that. That does not sound very nice. I wonder how this can be fixed without any impact on the bootloader. Absolutely ridiculous that they would push firmware updates which cannot be installed.

jvanschie, 2020-10-20:
@ChauhanT yes it is ridiculous, I had to do a clean install of ubuntu. Now everything is working fine.

ChauhanT, 2020-10-20:
@jvanschie Interesting. So you no longer have two updates which just refuse to install? Or problems with your external monitor while booting? You mean an OS re-install fixed all the problems?

lijunwang1990, 2020-10-22:
> @ChauhanT yes it is ridiculous, I had to do a clean install of ubuntu. Now everything is working fine.

Hi @jvanschie can you help to check your ME version with fwupdmgr get-devices? is it right now?

ChauhanT, 2020-10-22:
Ok guys, I resolved it! Here's how:

`$ sudo fwupdmgr downgrade `

This should give you a list of your recent updates. Just downgrade the looping updates. Don't worry though. Once you've downgraded, just go ahead and run

`$ sudo fwupdmgr update `

And that should be the end of your troubles. Or at least it was the end of my troubles. No more updates which can't be installed. `fwupdmgr --help` for other options.

jvanschie, 2020-10-22:
@lijunwang1990 does this help you?

~$ fwupdmgr get-devices
```
20L5CTO1WW
│
├─Thunderbolt Controller:
│     Device ID:           28e21d40c02314b11f05ef3fe7414b603b1426a6
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     17.01
│     Vendor:              Lenovo (TBT:0x0109)
│     GUIDs:               7d131734-64ed-5865-a46e-a67cfb9b5a46
│                          e45313fd-8d5c-54e9-b175-1f15ffe6b05c
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Device stages updates
│   
├─LITEON CA3-8D512:
│     Device ID:           f2759da7fe8e0388c5f3601cb072f837b1070b03
│     Summary:             NVM Express Solid State Drive
│     Current version:     14979
│     Vendor:              Lite-On Technology Corporation (NVME:0x14A4)
│     GUIDs:               c8066dea-4bd3-50ad-9e08-d10bab80d671
│                          2f3deb1f-5b77-555e-9940-2148efd76a0b
│                          81dcded2-1e83-5c8e-8af2-3a1e7b2be38a
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─System Firmware:
│     Device ID:           48477ba14bd9bd0c97323a07caa54b7766f917c2
│     Current version:     0.1.35
│     Minimum Version:     0.0.1
│     Vendor:              LENOVO (DMI:LENOVO)
│     GUID:                dc7f0308-1ef8-4774-9ba5-89a58c4d731c
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           5decfff65e3391259bca4003a090e59bf50e3bea
│     Current version:     184.79.3722
│     Minimum Version:     0.0.1
│     Vendor:              DMI:LENOVO
│     GUID:                dd31d983-cf3b-4c84-acc9-70caa94e827b
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           c19580db344e418cf2b23297bcf46d75c06e432a
│     Current version:     0.1.20
│     Minimum Version:     0.1.20
│     Vendor:              DMI:LENOVO
│     GUID:                b87a926d-189c-49a1-b1ff-921099b1de89
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           b1825e2b03a9ebdea247a74f1f3ddee2a21ac565
│     Current version:     1.2.1081
│     Minimum Version:     0.0.1
│     Vendor:              DMI:LENOVO
│     GUID:                24aac206-3921-45c0-a79e-0a607b11aef0
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           a885679bdb552409a5a7e6245e7d6911b500f5cc
│     Current version:     1.4.1084
│     Minimum Version:     0.0.1
│     Vendor:              DMI:LENOVO
│     GUID:                0eeda41e-9d6c-4ea0-8163-4aa126a4b851
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
└─UEFI Device Firmware:
      Device ID:           8e6d0ecd7a9737ba818ed0a90fb863aaa9af5a3c
      Current version:     0.73.4
      Minimum Version:     0.73.4
      Vendor:              DMI:LENOVO
      GUID:                e0049080-09bd-473a-a02c-19aa0c8dad78
      Device Flags:        • Internal device
                           • Updatable
                           • Requires AC power
                           • Needs a reboot after installation
                           • Device is usable for the duration of the update
```

kmauleon, 2020-10-22:
> Ok guys, I resolved it! Here's how:
>
> `$ sudo fwupdmgr downgrade `
>
> This should give you a list of your recent updates. Just downgrade the looping updates. Don't worry though. Once you've downgraded, [...]

thank you very much for the work around... did BIOS version changed from last time?

ChauhanT, 2020-10-23:
Hi, I'm afraid I didn't check this. If you think the information is available in a log file somewhere, I can have a look..

kmauleon, 2020-10-25:
hi @ChauhanT , can you please show result of below command... thank you very much
     **fwupdmgr get-devices**

ChauhanT, 2020-10-26:
@kmauleon There you go. Hope this helps! I'm curious now. Would you know why I have 3 audio cards showing up here?

```
$ fwupdmgr get-devices
T480 Thunderbolt Controller
  DeviceId:             3f15b...
  Guid:                 7d131...
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|require-ac|registered
  Vendor:               Lenovo
  VendorId:             TBT:0x0109
  Version:              20.00
  VersionFormat:        pair
  Icon:                 thunderbolt
  Created:              2020-10-24

20L6S0KE00 System Firmware
  DeviceId:             48477...
  Guid:                 dc7f0...
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Checksum:             SHA1(f0f80c2a0b04c1f04f7b4a88dcd856f9195d2f2d)
  Checksum:             SHA256(e3ea843f5e416d9a49d00302530c43f4d64c953e17166aeaa3a63fc444c2c638)
  Vendor:               LENOVO
  Version:              0.1.35
  VersionLowest:        0.0.1
  VersionFormat:        triplet
  Icon:                 computer
  Created:              2020-10-24

UEFI Device Firmware
  DeviceId:             ea0fd...
  Guid:                 9e21f...
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              184.79.3722
  VersionLowest:        184.79.3722
  VersionFormat:        triplet
  Icon:                 audio-card
  Created:              2020-10-24
  UpdateState:          success

UEFI Device Firmware
  DeviceId:             c1958...
  Guid:                 b87a9...
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.20
  VersionLowest:        0.1.20
  VersionFormat:        triplet
  Icon:                 audio-card
  Created:              2020-10-24

UEFI Device Firmware
  DeviceId:             0a6ca...
  Guid:                 89aab...
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              2.0.33505
  VersionLowest:        0.0.1
  VersionFormat:        triplet
  Icon:                 audio-card
  Created:              2020-10-24

UEFI Device Firmware
  DeviceId:             ad87c...
  Guid:                 1a30f...
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              1.0.8
  VersionLowest:        0.0.1
  VersionFormat:        triplet
  Icon:                 audio-card
  Created:              2020-10-24

LENSE30512GMSP34MEAT3TA
  DeviceId:             f2759...
  Guid:                 70cc2...
  Guid:                 0a2bd...
  Summary:              NVM Express Solid State Drive
  Plugin:               nvme
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Vendor:               Lenovo
  VendorId:             NVME:0x17AA
  Version:              2.5.0412
  VersionFormat:        plain
  Icon:                 drive-harddisk
  Created:              2020-10-24
```

hughsie, 2020-10-26:
> Would you know why I have 3 audio cards showing up here?

It's the default icon for an unknown UEFI device.

ChauhanT, 2020-10-26:
@hughsie Ah dang, forgot to notice that the field was Icon. Thanks ^_^

kmauleon, 2020-10-26:
hi @ChauhanT sorry but can you please try this instead... 
**fwupdmgr report-history**

ChauhanT, 2020-10-27:
@kmauleon Hi, it just says 

```
$ fwupdmgr report-history
No reports require uploading

```

I remember that during the update process `fwupdmgr` ha doffered to upload some reports for improving the software and I had agreed. So the report from my laptop should be uploaded to the server sometime within the last week or so.

kmauleon, 2020-10-28:
hi @ChauhanT .. oh i don't know how to check that haha... but anyway, you already have the latest MEFW. But we plan to release another by end of this year.  Can we check the behavior again by that time and if no problem then maybe we can close this one?

@jvanschie you also have the latest MEFW already...

@smalltimer @galibozek can you please execute the below commands...
     **fwupdmgr refresh**
     **fwupdmgr update**
restart system manually and wait for updates to complete...system will reboot
     **fwupdmgr get-devices**

thank you very much

ChauhanT, 2020-10-28:
@kmauleon Hi, since I'm running some computations on my machine I'll get back to you when I can reboot (it might be a few days though...). I did run `fwupdmgr refresh` and `fwupdmgr update` and there were no upgrades to any firmware. So I don't think a reboot will install anything. 

As far as I am concerned, the issue can be closed. Thank you for looking into it!

galibozek, 2020-10-28:
> hi @ChauhanT .. oh i don't know how to check that haha... but anyway, you already have the latest MEFW. But we plan to release another by end of this year. Can we check the behavior again by that [...]

I have send report, "UpdateError" : "failed to run update on reboot",
after update is still the same:

```20L8S3DK1A
?
??Thunderbolt host controller:
?     Device ID:          ca1ad4f978a9dafc8ccd728854d262046d7e4779
?     Summary:            Unmatched performance for high-speed I/O
?     Current version:    20.00
?     Vendor:             Lenovo (TBT:0x0109)
?     GUIDs:              3849839e-d8b2-5941-9e69-5acc397d3ed3
?                         7e3190a2-d5a5-5a9e-b582-494b47f7288c
?                         061f2847-061b-5cfb-9c39-3dbfe2b49d69
?                         65a3d897-4a18-5d27-83d8-eb8a231b7aa4
?     Device Flags:       ? Internal device
?                         ? Updatable
?                         ? System requires external power source
?                         ? Supported on remote server
?                         ? Device stages updates
?   
??USB3.0 Hub:
?     Device ID:          da77984c82b59c6fc69516431f467fd9a8d39a7f
?     Summary:            USB 3.x Hub
?     Current version:    50.13
?     Vendor:             VIA Labs, Inc. (USB:0x17EF)
?     Install Duration:   15 seconds
?     GUIDs:              f14cd52d-9b50-58b5-a1e5-97ce1c6b149f
?                         83fae0fd-e028-5e9e-b4ed-e3563dc35c69
?                         fc0a4e01-6957-5925-b27f-d165bf20fffa
?                         622acf8b-e390-5cf7-8134-b7f7a7600ed8
?                         084aee58-f98e-5be6-a772-ff70f3bdc6db
?     Device Flags:       ? Updatable
?                         ? Cryptographic hash verification is available
?                         ? Device stages updates
?                         ? Device can recover flash failures
?   
??Embedded Controller:
?     Device ID:          323bda90831520867f8c02544efb9711dfca9fb6
?     Current version:    0.1.20
?     Minimum Version:    0.1.20
?     Vendor:             DMI:LENOVO
?     GUIDs:              4d254d6e-cd67-477b-97d5-bc3048af45c4
?                         cc9cf2fe-ba41-532b-bc58-17ca9cc3c63d
?     Device Flags:       ? Internal device
?                         ? Updatable
?                         ? System requires external power source
?                         ? Supported on remote server
?                         ? Needs a reboot after installation
?                         ? Device is usable for the duration of the update
?   
??Intel Management Engine:
?     Device ID:          118b3d8cfafa8781890c4d3e23a4deb9fcb18e5a
?     Current version:    184.77.3664
?     Minimum Version:    184.77.3664
?     Vendor:             DMI:LENOVO
?     Update State:       success
?     GUIDs:              39a39e45-38ce-4b05-809b-60351bf88ec4
?                         e6e2469e-013d-564c-917a-5938b9277c49
?     Device Flags:       ? Internal device
?                         ? Updatable
?                         ? System requires external power source
?                         ? Supported on remote server
?                         ? Needs a reboot after installation
?                         ? Device is usable for the duration of the update
?   
??LENSE30256GMSP34MEAT3TA:
?     Device ID:          c82980cd3fe66f7d0a8bb83b353cbc271177efd4
?     Summary:            NVM Express Solid State Drive
[... 35 lines cut ...]
?       Vendor:           UEFI:Linux Foundation
?       Install Duration: 1 second
?       GUIDs:            14503b3d-73ce-5d06-8137-77c68972a341
?                         5971a208-da00-5fce-b5f5-1234342f9cf7
?                         c6682ade-b5ec-57c4-b687-676351208742
?                         f8ba2887-9411-5c36-9cee-88995bb39731
?       Device Flags:     ? Internal device
?                         ? Updatable
?                         ? Needs a reboot after installation
?     
??USB2.0 Hub:
      Device ID:          983c3cffc6fd36d32b00b62928d30721eaeb93db
      Summary:            USB 2.x Hub
      Current version:    50.13
      Vendor:             VIA Labs, Inc. (USB:0x17EF)
      Install Duration:   15 seconds
      GUIDs:              767b34cc-9dd5-5a4f-b566-90a1cb244e48
                          f14cd52d-9b50-58b5-a1e5-97ce1c6b149f
                          2b6b5c49-69ab-5ff2-8650-28b5ecc4e7f5
                          33d8793e-68bb-56f1-a203-3c60fe81e51b
      Device Flags:       ? Updatable
                          ? Cryptographic hash verification is available
                          ? Device stages updates
                          ? Device can recover flash failures
```

kmauleon, 2020-10-28:
> @kmauleon Hi, since I'm running some computations on my machine I'll get back to you when I can reboot (it might be a few days though...). I did run `fwupdmgr refresh` and `fwupdmgr update` and [...]

Thank you very much @ChauhanT ! ^_^

kmauleon, 2020-10-29:
hi @galibozek thank you very much for the information... but i found it weird that your system has 2 GUIDs for MEFW as below
??Intel Management Engine:
?     Device ID:          118b3d8cfafa8781890c4d3e23a4deb9fcb18e5a
?     Current version:    184.77.3664
?     Minimum Version:    184.77.3664
?     Vendor:             DMI:LENOVO
?     Update State:       success
**?     GUIDs:              39a39e45-38ce-4b05-809b-60351bf88ec4
?                         e6e2469e-013d-564c-917a-5938b9277c49**
let me ping the owner with this behavior... will get back to you once i have update... thank you very much

jvanschie, 2020-10-29:
> hi @ChauhanT .. oh i don't know how to check that haha... but anyway, you already have the latest MEFW. But we plan to release another by end of this year. Can we check the behavior again by that [...]

@kmauleon 

I managed to install the update manually with fwupdmgr.

The install succeeded but afterwards I had troubles with the boot manager.
After a fresh install of ubuntu everything started to work fine.

kmauleon, 2020-10-29:
hi @jvanschie thank you very much for the information
_The install succeeded but afterwards I had troubles with the boot manager_ >> do you remember which updates resulted to this?
I am sorry that you have to reinstall ubuntu... @hughsie do we have any idea what's causing the boot manager to break?

hughsie, 2020-10-30:
> do we have any idea what's causing the boot manager to break

Some older system UEFI firmware did "helpful" things to the BootXXXX variables (which is why we use `Linux-Firmware-Updater` on Lenovo, not `Linux Firmware Updater` like everywhere else...), I think this has all stopped in newer Lenovo firmware.

kmauleon, 2020-10-31:
@hughsie so the only way to fix broken boot manager is to reinstall Linux?

hughsie, 2020-10-31:
> so the only way to fix broken boot manager is to reinstall Linux

I've not seen this myself; I'm guessing you could also boot a rescue USB disk and just reinstall grub.

kmauleon, 2020-11-03:
thanks for the information @hughsie ... is there a link that we could point customer to on how to **create a bootable rescue USB disk and reinstall grub**?

galibozek, 2020-11-06:
version 11.8.80.3746 is available on pcsupport.lenovo.com, can I have a version for fwupd?

mrhpearson, 2020-11-06:
Forwarded the request to the firmware team.

kmauleon, 2020-11-08:
hi @galibozek @mrhpearson this version of firmware is targeted to be released to LVFS by December.  Please wait.... Thank you very much.

smalltimer, 2020-11-18:
Hi, ran into this issue once again. This time the fix was simply uninstalling and reinstalling fwupdmgr from the debian repos. The firmware upgraded very nicely after that.

I have been running linux for more than 16 years now, and you have no idea how magical fwupdmgr feels. Never in the history of linux has upgrading/downgrading firmware been this easy. Awesome job guys <3

kmauleon, 2020-11-18:
thank you very much for the tips @smalltimer !!

galibozek, 2020-11-21:
@kmauleon i installed today version 11.8.80.3746 (ed0b54330fe171837aec61b037ec3d465e105fed99a14385b46174d410e62635-Lenovo-ThinkPad-T480s-ConsumerMEFirmware-11.8.80.3746.cab)

and i have same effect

fwupdmgr get-topology --show-all:
```
??Intel Management Engine:
?     Device ID:          118b3d8cfafa8781890c4d3e23a4deb9fcb18e5a
?     Current version:    184.77.3664
?     Minimum Version:    184.77.3664
?     Vendor:             DMI:LENOVO
?     Update State:       success
?     GUIDs:              39a39e45-38ce-4b05-809b-60351bf88ec4
?                         e6e2469e-013d-564c-917a-5938b9277c49
?     Device Flags:       ? Internal device
?                         ? Updatable
?                         ? System requires external power source
?                         ? Supported on remote server
?                         ? Needs a reboot after installation
?                         ? Device is usable for the duration of the update
 

? ??Intel AMT [unprovisioned]:
? ?     Device ID:        e2623122c99d58220498aacbfcfdb1baebbae3c5
? ?     Summary:          Hardware and firmware technology for remote out-of-band management
? ?     Current version:  11.8.80.3746
? ?     Bootloader Version:11.8.80.3746
? ?     Vendor:           Intel Corporation
? ?     GUID:             2800f812-b7b4-2d4b-aca8-46e0ff65814c
? ?     Device Flags:     ? Internal device
```

bios show version: 11.8.80.3746,

sudo fwupdmgr get-updates
```
20L8S3DK1A
?
??Intel Management Engine:
  ?   Device ID:          118b3d8cfafa8781890c4d3e23a4deb9fcb18e5a
  ?   Current version:    184.77.3664
  ?   Minimum Version:    184.77.3664
  ?   Vendor:             DMI:LENOVO
  ?   Update State:       success
  ?   GUIDs:              39a39e45-38ce-4b05-809b-60351bf88ec4
  ?                       e6e2469e-013d-564c-917a-5938b9277c49 ? UEFI\RES_{39A39E45-38CE-4B05-809B-60351BF88EC4}
  ?   Device Flags:       ? Internal device
  ?                       ? Updatable
  ?                       ? System requires external power source
  ?                       ? Supported on remote server
  ?                       ? Needs a reboot after installation
  ?                       ? Device is usable for the duration of the update
  ? 
  ??ThinkPad T480s Corporate ME Update:
        New version:      184.79.3722
        Remote ID:        lvfs
        Summary:          Lenovo ThinkPad T480s Corporate ME Firmware
        License:          Proprietary
        Size:             7.5?MB
        Created:          2016-07-08
        Urgency:          High
        Details:          https://pcsupport.lenovo.com/de/en/search?query=N22RM14W
        Vendor:           Lenovo Ltd.
        Flags:            is-upgrade
        Description:      
        ? 8 Intel Platform Update 2020.1
        
        Version 11.8.79.3722 (LVFS: 184.79.3722)
        
        Problem Fixes
        
        ? Mitigated the following security vulnerabilities:
        
        Security issues fixed:
        
        ? CVE-2020-8758
```

kmauleon, 2020-11-24:
hi @galibozek ... thank you very much for the information i'll forward this to owner...

galibozek, 2020-12-03:
hi, @kmauleon @mrhpearson is something new in the topic?

kmauleon, 2020-12-03:
hi @galibozek... i'll ping owner again... but BIOS is registering the new version 11.8.80.3746 already... so it seem ESRT table is not updated correctly thus incorrect version for fwupd... thank you very much...

galibozek, 2020-12-04:
@kmauleon that is, it can be understood that the update has been applied?

kmauleon, 2020-12-06:
hi @galibozek ... yes you already have the latest MEFW ^_^

lijunwang1990, 2020-12-14:
@galibozek if your still get fail result after excute "sudo fwupdmgr get-updates",could you please uninstalling and reinstalling fwupdmgr to try again? or use "$ sudo fwupdmgr downgrade" and "$ sudo fwupdmgr update", I see someone fixed this issue with this method. thank you.

kmauleon, 2021-01-04:
hi @lijunwang1990 @galibozek already updated MEFW. but fwupdmgr still sees it as not updated. so it seem ESRT table was not updated correctly as below, thus, your team may need to investigate this behavior further.  thank you very much.

**bios show version: 11.8.80.3746**

sudo fwupdmgr get-updates
20L8S3DK1A
?
??Intel Management Engine:
  ?   Device ID:          118b3d8cfafa8781890c4d3e23a4deb9fcb18e5a
  **?   Current version:    184.77.3664**

galibozek, 2021-02-01:
still the same in 11.8.82.3838... :/

kmauleon, 2021-02-02:
hi @lijunwang1990 **ESRT table** was not updated correctly and was used by fwupd as reference. it will appear every time that mefw is not updated even if it is actually updated as seen in BIOS settings. thus, your **BIOS team** may need to investigate this behavior further. thank you very much.

galibozek, 2021-02-02:
@kmauleon and what's next with this problem? someone diagnoses it?

kmauleon, 2021-02-03:
hi @galibozek BIOS dev is still checking on this but no feedback yet as of this moment... will update once i have some information... very sorry for the slow response...
**@lijunwang1990 can we escalate the investigation on this please**

kmauleon, 2021-02-04:
update from BIOS Dev... thank you very much....
**no clear progress yet regarding BIOS ESRT problem investigation**

nonsite, 2021-02-12:
as described, same here
ThinkPad T480 Ubuntu 20.04.2 LTS Kernel 5.4.0-65-generic BIOS N24ET61W

galibozek, 2021-03-17:
Version 11.8.83.3874 did not fix the problem, still...
Does anyone have any plans to solve this problem? @kmauleon

kmauleon, 2021-03-17:
hi @galibozek ... really very sorry for the very slow investigation but still no clear update from BIOS dev regarding the issue T_T

ChauhanT, 2021-07-23:
Hi guys. I'm back again. This time I've got the following error: 

```sh
Downloading 0.1.22 for UEFI Device Firmware...
Decompressing…           [    \                                  ]failed to compile 5b41632e511b0a4f42b665ea95f8dd8acb95669a:prefix=components: Error on line 46 char 2: Document must begin with an element (e.g. <book>)
``` 

Reboots will not fix this. Any idea what might be going wrong here? Thanks!

hughsie, 2021-07-23:
@ChauhanT I've just merged https://gitlab.com/fwupd/lvfs-website/-/commit/fab289d36cf34992a11e875dca5a11d82b21ab79 -- can you tell me the filename of the file you downloaded please and then I'll resign it on the LVFS with the workaround in place.

smalltimer, 2021-07-23:
Hi @hughsie . Thank you for getting in touch so quickly! 

I am using `fwupd` from the Debian Buster repository. I'm not sure how to get the filename. If you could tell me how, I will be happy to do it right away. 

Thanks again!

hughsie, 2021-07-23:
Ahh, I meant the 41476a3066b9b459bca87936f63235c52f13ac047306dfb06d4d4782aa400b64-Lenovo-ThinkPad-T480-ECFirmware-1.22.cab bit -- which I've guessed as that was the one I regenerated this morning.  Can you do `fwupdmgr refresh --force` and try again please?

smalltimer, 2021-07-23:
Hi, just tried it, and more of the same. Give me a couple of hours though. I think there might be something weird going on with my `grub` too. I have purged `grub` from the system (using `refind`), and it still shows up on boot. Such an epic PITA `grub` is :|

ChauhanT, 2021-07-23:
@hughsie Just purged `refind` and went back to a `grub2` -only boot. Now I get `Failed to update metadata for lvfs: '48A6D80E4538BAC2' is not a valid signature` when trying to `sudo fwupdmgr refresh --force`. Not sure what's going on there.

smalltimer, 2021-07-23:
@hughsie Just purged refind and went back to a grub2 -only boot. Now I get Failed to update metadata for lvfs: '48A6D80E4538BAC2' is not a valid signature when trying to sudo fwupdmgr refresh --force. Not sure what's going on there.

smalltimer, 2021-08-05:
Hi, I still get the error `Failed to update metadata for lvfs: '48A6D80E4538BAC2' is not a valid signature`. Any workarounds for this?

galibozek, 2021-08-10:
@kmauleon how is progress with my problem?

kmauleon, 2021-08-11:
hi @galibozek i am sorry but i can't get the BIOS dev to investigate ESRT issue further unless we recreate the issue.  i am not sure how to go from here.... @mrhpearson @hughsie any suggestions?

[13 more replies]

closed 2022-11-16 by kmauleon