fwupd/firmware-lenovo #74: No firmware for the AMD Versions of the T14 / T14s

aussetg, 2020-09-22:
It's available on the support page but

1. No .cab is made available
2. The AMD versions aren't even present in fwupd

hughsie, 2020-09-22:
Which support page? Thanks.

mrhpearson, 2020-09-23:
Hi @aussetg 

I assume you mean the Lenovo support site.

I'll chase this up with the firmware team to confirm but firmware for these platforms is supposed to be delivered to LVFS so if it's not up yet it should be coming soon.

As a note - we haven't actually completed our Linux enablement of the platform yet and I know the BIOS team were working through some BIOS+Linux related issues for the last couple of weeks so I suspect that might be part of the reason why. I believe we're very close to sign off so please wait just a bit longer.

Thanks
Mark

jvnknvlgl, 2020-10-12:
Just wondering - what is the status on this?

mrhpearson, 2020-10-15:
Hi - Still in progress, I've asked for an ETA.

mrhpearson, 2020-10-19:
Hi @jvnknvlgl - the firmware team believe they should have it at the beginning of November. Will let you know when it's posted.
Thanks
Mark

fkru, 2020-11-09:
> Hi @jvnknvlgl - the firmware team believe they should have it at the beginning of November. Will let you know when it's posted.
> Thanks
> Mark

There have been three BIOS updates so far (most recent version is 1.27), but unfortunately not through LVFS. Moreover, Lenovo's T14 AMD was certified by Ubuntu 20.04 LTS with BIOS 1.24 some time ago. So, what's up? Regards, Frank

mrhpearson, 2020-11-09:
I'll check the latest and report back.

fkru, 2020-11-10:
> I'll check the latest and report back.

It seems they have just released some updates at LVFS, including BIOS 1.25 (https://fwupd.org/lvfs/search?csrf_token=IjRiZGRmMDliMDVkNjQyNjkwODliMmY2YmMwODJiMDM3MTljZGYwNWIi.X6o0rg.XY8ysR6ZQZmjU0QyQ9-Sy68M9_U&value=T14) Thx. Regards, Frank

fkru, 2020-11-10:
Tried to install BIOS 1.25 through 'fwupd update' on a T14, ending with the message "sucessfully installed firmware". After a reboot, however, I am still with version 1.5. Anyone else experiencing this issue?

jpasquier, 2020-11-10:
I successfully updated the firmware (to 1.25) on my T14 amd gen1 through `sudo fwupdmgr update` (after `fwupdmgr refresh`). Regards

fkru, 2020-11-10:
JFY: Today's firmware update to 1.25 has been removed from LVFS.

mrhpearson, 2020-11-11:
Yes - there was a GUID clash with the E495 so it had to be pulled whilst that was fixed. Apologies!
Mark

jpasquier, 2020-11-11:
I successfully applied the update yesterday. Do I now have to do something more because of this GUID clash? Thank you for your help.

SoMuchForSubtlety, 2020-11-17:
Should the firmware update be on LVFS again? I don't see any updates when I run `fwupdmgr refresh`.

fkru, 2020-11-22:
@Mark: Do you know by chance whether a bios release for T14 is in sight? Thx. Regards, Frank

mrhpearson, 2020-11-24:
Afraid no ETA yet - I sent a nudge this morning to find out what is happening.
Mark

fkru, 2020-12-03:
While Lenovo announced publicly a couple of months ago to ship T14 AMD with Ubuntu LTS 20.04, it seems they are incapable of providing working firmware updates at LVFS. It's rather disappointing!

SoMuchForSubtlety, 2020-12-15:
@mrhpearson any news? I though this was already released and only pulled because of a GUID clash.

aussetg, 2020-12-18:
1.29 (latest) is on LVFS

Seems to fix every single oddity I had with sleep

j13m126, 2020-12-18:
Only for the T14s though. I think @SoMuchForSubtlety was talking about the T14 AMD model, where there is no update on LVFS yet.

SoMuchForSubtlety, 2020-12-18:
Yes, I have a T14 (20UD0013GE)

kmauleon, 2020-12-21:
hi all... really sorry for the delay for this firmware. as per owner target availability is December 24... thank you very much.

fkru, 2020-12-21:
Installed today's upate to system firmware version 1.25 for T14 AMD via LVFS ("sucessfully installed firmware"), but after a reboot I still have version 1.05. Any idea? Anyone else seeing this? Thx.

kmauleon, 2020-12-21:
hi @fkru can you please attach screen shot of your BIOS Setup Menu here... I'll forward to owner... thank you very much

jpasquier, 2020-12-22:
Hi, I had managed to do the update during the few hours it was first available (November 10th). Now that the update is available again, it is not recognized in the history. Looks like for my system there is no more device corresponding to this firmware. I'm afraid I won't be able to apply future updates. Is there a possibility to update or correct my history and my device list?

The command `sudo systemctl status fwupd.service` gives

```
● fwupd.service - Firmware update daemon
     Loaded: loaded (/lib/systemd/system/fwupd.service; static)
     Active: active (running) since Tue 2020-12-22 03:53:30 CET; 2s ago
       Docs: https://fwupd.org/
   Main PID: 1781 (fwupd)
      Tasks: 6 (limit: 37532)
     Memory: 15.8M
     CGroup: /system.slice/fwupd.service
             └─1781 /usr/libexec/fwupd/fwupd

Dec 22 03:53:29 t14 systemd[1]: Starting Firmware update daemon...
Dec 22 03:53:29 t14 fwupd[1781]: 02:53:29:0837 FuPluginDfu          04f2:b6d0 is missing download capability
Dec 22 03:53:30 t14 fwupd[1781]: 02:53:30:0243 FuEngine             failed to update history database: device ID 3ba7ea853f9613adbcbb9326dbe5f0decd677a67 was not found
Dec 22 03:53:30 t14 systemd[1]: Started Firmware update daemon.
```

and `sudo fwupdmgr get-history` gives

```
20UES1Y200
│
... [history of other devices]
│     
└─System Firmware:
  │   Device ID:          3ba7ea853f9613adbcbb9326dbe5f0decd677a67
  │   Previous version:   0.1.5
  │   Update State:       needs-reboot
  │   Last modified:      2020-11-10 07:28
  │   GUID:               9bfd8b34-080b-40a1-47b3-ca1efa289e6d
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Cryptographic hash verification is available
  │                       • Device is usable for the duration of the update
  │ 
  └─  New version:      0.1.25
        Remote ID:        lvfs
        License:          Unknown
```

In addition, the device does not seem to be listed by the command `sudo fwupdmgr get-devices`:

```
20UES1Y200
│
├─Integrated Camera:
│     Device ID:          54f0d9041b6c5438c7ff825f5139559c5ca1b222
│     Current version:    58.18
│     Vendor:             Chicony Electronics Co.,Ltd. (USB:0x04F2)
│     Serial Number:      0001
│     GUIDs:              cd900697-9e98-53bb-9f13-fac06bc26f77 ← USB\VID_04F2&PID_B6D0&REV_5818
│                         f8d483f9-9284-51b6-9363-7e6d59c40a97 ← USB\VID_04F2&PID_B6D0
│     Device Flags:       • Updatable
│   
├─Prometheus:
│ │   Device ID:          50efef912094e784a03628350441f6927c4920b0
│ │   Summary:            Fingerprint reader
│ │   Current version:    10.01.3158804
│ │   Vendor:             Synaptics (USB:0x06CB)
│ │   Install Duration:   2 seconds
│ │   Serial Number:      60902600455565
│ │   GUIDs:              09e8bf16-3e69-50f5-bb66-c7a040248352 ← USB\VID_06CB&PID_00BD&REV_0000
│ │                       8088f861-6318-5b1e-9ce4-fbddbedb09ac ← USB\VID_06CB&PID_00BD
[... 119 lines cut ...]
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
```

and the device is not listed as having the latest firmware version by the command `sudo fwupdmgr get-updates`:

```
Devices with no available firmware updates: 
 • Integrated Camera
 • SKHynix HFS001TD9TNI-L2B0B
 • System Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI dbx
Devices with the latest available firmware version:
 • Prometheus
 • Prometheus IOTA Config
No updates available for remaining devices
```

Thank you in advance for your help.

kmauleon, 2020-12-22:
hi @jpasquier can you please try the following commands... thank you very much.
**fwupdmgr downgrade** _(choose which firmware to downgrade)_

after successful downgrade, please try the following again...
**fwupdmgr refresh
fwupdmgr update**

jpasquier, 2020-12-22:
@kmauleon Thank you for your message. It does not work. The command `sudo fwupdmgr downgrade 3ba7ea853f9613adbcbb9326dbe5f0decd677a67` gives:

```
failed to find 3ba7ea853f9613adbcbb9326dbe5f0decd677a67
Choose a device:
0.	Cancel
1.	50efef912094e784a03628350441f6927c4920b0 (Prometheus)
2.	30469829b6adc759baf3b0c3cf19454985aa73a2 (Prometheus IOTA Config)```
```
I get the same result with the GUID

```
failed to find any device providing 9bfd8b34-080b-40a1-47b3-ca1efa289e6d
Choose a device:
0.	Cancel
1.	50efef912094e784a03628350441f6927c4920b0 (Prometheus)
2.	30469829b6adc759baf3b0c3cf19454985aa73a2 (Prometheus IOTA Config)
```

kmauleon, 2020-12-22:
how about this one... please download cab file >> https://fwupd.org/downloads/a66f0fe4894d6e00661e455ce1ef371e026310f2a85c934b60693a9c5af1a946-Lenovo-ThinkPad-T14-P14s-AMD-SystemFirmware-1.25.cab

then execute below command...
**fwupdmgr --allow-older --allow-reinstall install** _cab filename_

jpasquier, 2020-12-22:
Doesn't work either. I get:

```
Decompressing…           [***************************************]
No supported devices found
```

Thank you

kmauleon, 2020-12-22:
let me ping owner @jpasquier hopefully they reply asap... will update once  i get reply from them... thank you very much...

jvnknvlgl, 2020-12-22:
I'm having the same issue, I updated in the first hours on November 10th.

jvnknvlgl, 2020-12-22:
If it helps anything, here is the output  of `get-history`:

`[jvnknvlgl@thinkpad-t14 ~]$ fwupdmgr get-history 
20UDCTO1WW
│
└─System Firmware:
  │   Device ID:          3ba7ea853f9613adbcbb9326dbe5f0decd677a67
  │   Previous version:   0.1.5
  │   Update State:       needs-reboot
  │   Last modified:      2020-11-10 10:51
  │   GUID:               9bfd8b34-080b-40a1-47b3-ca1efa289e6d
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Cryptographic hash verification is available
  │                       • Device is usable for the duration of the update
  │ 
  └─  New version:      0.1.25
        Remote ID:        lvfs
        License:          Unknown
`

kmauleon, 2020-12-22:
hi @jvnknvlgl thank you for the information... already forwarded this to owner...

33Fraise33, 2020-12-23:
I actually seem to have the same issuel. Device is not listed and also not showing up onder get-history. I'm unsure if I did the update through fwupdmgr or not but I'm running 1.25 at the moment. 

```
$ fwupdmgr get-devices 
20UDCTO1WW
│
├─Integrated Camera:
│     Device ID:          54f0d9041b6c5438c7ff825f5139559c5ca1b222
│     Current version:    60.1
│     Vendor:             Azurewave (USB:0x13D3)
│     GUIDs:              c9b54303-2a5c-5da1-a441-bda12b11fff3
│                         db842a92-1141-541b-965a-2733a5726e17
│     Device Flags:       • Updatable
│   
├─SKHynix HFS512GD9TNI-L2B0B:
│     Device ID:          71b677ca0f1bc2c5b804fa1d59e52064ce589293
│     Summary:            NVM Express Solid State Drive
│     Current version:    11710C10
│     Vendor:             SK hynix (NVME:0x1C5C)
│     GUIDs:              424e8633-6c06-5c41-bd78-fbcb483dc0fb
│                         b7603cf7-873e-59e8-b7aa-9d8a9877b6e7
│                         af90743d-93b0-5a7f-b3aa-08db33ab1f82
│                         85ef24c5-fc0a-57a6-ab37-5711ac8acfbb
│                         930d3ce1-d73d-582e-84df-190dad19f650
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─System Firmware:
│ │   Device ID:          3f10cbb75c39627fc0d14eddd274374207a5c753
│ │   Current version:    65561
│ │   Vendor:             LENOVO (DMI:LENOVO)
│ │   GUIDs:              01453b71-da0c-4832-9f4f-e378245339c7
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a
│ │                       38ad8832-d322-5d73-aff7-77b8ae0f1a6f
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─UEFI dbx:
│       Device ID:        362301da643102b9f38477387e2193e57abaa590
│       Summary:          UEFI Revocation Database
│       Current version:  77
│       Minimum Version:  77
│       Vendor:           UEFI:Linux Foundation
│       Install Duration: 1 second
│       GUIDs:            14503b3d-73ce-5d06-8137-77c68972a341
│                         5971a208-da00-5fce-b5f5-1234342f9cf7
│                         c6682ade-b5ec-57c4-b687-676351208742
│                         f8ba2887-9411-5c36-9cee-88995bb39731
│       Device Flags:     • Internal device
│                         • Updatable
│                         • Needs a reboot after installation
│     
├─UEFI Device Firmware:
│     Device ID:          1c8d3b6c7cb08805789e84e05fc8f3b097237f1e
│     Current version:    1057227777
│     Vendor:             DMI:LENOVO
│     GUIDs:              3b7f92a6-36d0-49ae-a2d1-272f9a067274
│                         4108f09c-02bc-5381-b4ed-0bc8dbd56848
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
[... 48 lines cut ...]
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update

```
```
$ fwupdmgr get-history
20UDCTO1WW
│
└─Prometheus:
  │   Device ID:          50efef912094e784a03628350441f6927c4920b0
  │   Previous version:   10.01.3121519
  │   Update State:       success
  │   Last modified:      2020-10-23 13:27
  │   GUID:               09e8bf16-3e69-50f5-bb66-c7a040248352
  │   Device Flags:       • Updatable
  │                       • Supported on remote server
  │                       • Reported to remote server
  │                       • Cryptographic hash verification is available
  │ 
  └─  New version:      10.01.3158804
        Remote ID:        lvfs
        License:          Unknown

```

jggithub2017, 2020-12-26:
```
$ fwupdmgr get-history
20UDCTO1WW
│
└─System Firmware:
  │   Device ID:          3ba7ea853f9613adbcbb9326dbe5f0decd677a67
  │   Previous version:   0.1.5
  │   Update State:       needs-reboot
  │   Last modified:      2020-11-10 13:48
  │   GUID:               9bfd8b34-080b-40a1-47b3-ca1efa289e6d
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Needs a reboot after installation
  │                       • Cryptographic hash verification is available
  │                       • Device is usable for the duration of the update
  │ 
  └─  New version:      0.1.25
        License:          Unknown
      
```

Same problem as @jvnknvlgl.

fkru, 2020-12-27:
> hi @fkru can you please attach screen shot of your BIOS Setup Menu here... I'll forward to owner... thank you very much

The issue that I am experiencing seems to be different from the one reported recently. In fact, the update to version 1.25 is allegedly installed successfully, but after a reboot firmware is still at version 1.5. Shall I open a new bug report?

$ fwupdmgr get-history   
20UES00L00
│
└─System Firmware:
  │   Device ID:          3ba7ea853f9613adbcbb9326dbe5f0decd677a67
  │   Previous version:   0.1.5
  │   Update State:       failed
  │   Update Error:       failed to run update on reboot
  │   Last modified:      2020-12-27 16:14
  │   GUID:               9bfd8b34-080b-40a1-47b3-ca1efa289e6d
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Cryptographic hash verification is available
  │                       • Device is usable for the duration of the update
  │ 
  └─ThinkPad T14/P14s Gen 1 AMD System Update:
        New version:      0.1.25
        Remote ID:        lvfs
        Zusammenfassung:  Lenovo ThinkPad T14/P14s Gen 1 AMD System Firmware
        Licence:          Proprietary
        Size:             17,9 MB
        Created:          2016-07-08
        Urgency:          High
        Vendor:           Lenovo Ltd.
        Flags:            is-upgrade
        Beschreibung:     
        This stable release fixes the following issues:
        
        • Fixed an issue where BIOS has incorrect GUID in ACPI table.In case user cannot update the UEFI
        
        BIOS  later version,must update v1.25 first.
        
        • Truncated TCO logo at POST screen on both HD andFHD Display.
        • Update PI patch code.
        • Fixed an issue that TP&Trackpoint/Keyboard have no function after close lid during
        
        Restart/Full shutdown/S4.
        
        • Fixed an issue that the POST hang up with Lindy cable.

fkru, 2020-12-27:
JFYI: I have managed to update to version 1.25 (by disabling secure boot!), with 'fwupdmgr get-history': 

20UES00L00
│
└─System Firmware:
  │   Device ID:          3ba7ea853f9613adbcbb9326dbe5f0decd677a67
  │   Previous version:   0.1.5
  │   Update State:       needs-reboot
  │   Last modified:      2020-12-27 17:16
  │   GUID:               9bfd8b34-080b-40a1-47b3-ca1efa289e6d
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Cryptographic hash verification is available
  │                       • Device is usable for the duration of the update
  │ 
  └─  New version:      0.1.25
        Remote ID:        lvfs
        Licence:          Unbekannt

kmauleon, 2020-12-28:
hi @fkru thank you very much for the update... does this mean we can close the issue?  please advise... thank you very much.  happy holidays ^_^

jvnknvlgl, 2020-12-28:
> hi @fkru thank you very much for the update... does this mean we can close the issue? please advise... thank you very much. happy holidays ^_^

My issue still persists, so I don't think so.

kmauleon, 2020-12-28:
hi @fkru I meant for the original issue "No firmware for the AMD Versions of the T14 / T14s
#74" ... then please open a new ticket for the new issue for proper tracking... thank you very much.

jvnknvlgl, 2020-12-28:
Well, I'd say I still don't have a firmware available for my T14 through fwupdmgr. Which is where the original issue is about, and in this thread there are a lot of other people with the same issue, because they all installed the faulty update on the 10th of November. If you want to I can make a new issue, but there's already a lot of information in here already.

kmauleon, 2020-12-28:
hi @fkru I apologize for misunderstanding... I thought original issue was already fixed since [1.25](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadR1BET.firmware) is already available.  ok...let's just keep this open until everyone agrees. for the other issue, I already forwarded to owner.  sorry but response may be delayed due to holidays. will feedback once I received updates from them.  thank you very much....

fkru, 2020-12-28:
> hi @fkru I apologize for misunderstanding... I thought original issue was already fixed since [1.25](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadR1BET.firmware) is already available. [...]

Agreed. Since version 1.25 is a prerequisite for 1.29, which is available on Lenovo's website since 23 Dec, it is important to keep this bug open until the issue is solved. Thx.

jvnknvlgl, 2020-12-28:
I have found a way to manually upgrade the UEFI to 1.29, to hopefully make way for future updates via LVFS:

1. Download the correct Bootable CD firmware from the Lenovo Support website, for mine it's this page: https://pcsupport.lenovo.com/us/en/products/laptops-and-netbooks/thinkpad-t-series-laptops/thinkpad-t14-type-20ud-20ue/downloads/driver-list/component?name=BIOS%2FUEFI
2. Convert the ISO to a bootable image using the following command: `geteltorito -o bootable.img download.iso`. You might need to install `geteltorito`, it's available on Fedora.
3. Write the image file to a USB drive using your favorite way to do such things, I used Gnome Disks.
4. Disable secure boot, otherwise the USB drive will not boot.
5. Boot the USB drive and follow the steps, you will now be updated to 1.29 after a reboot.
6. R-enable secure boot if you desire.

```
[jvnknvlgl@thinkpad-t14 ~]$ sudo dmidecode -s bios-version
R1BET60W(1.29 )
```
The device still doesn't show up if I run `fwupdmgr`, but here's hoping it will show up for the next UEFI update.

jggithub2017, 2020-12-28:
> I have found a way to manually upgrade the UEFI to 1.29, to hopefully make way for future updates via LVFS:
> 
>     1. Download the correct Bootable CD firmware from the Lenovo Support website, for mine it's this page: https://pcsupport.lenovo.com/us/en/products/laptops-and-netbooks/thinkpad-t-series-laptops/thinkpad-t14-type-20ud-20ue/downloads/driver-list/component?name=BIOS%2FUEFI
> 
>     2. Convert the ISO to a bootable image using the following command: `geteltorito -o bootable.img download.iso`. You might need to install `geteltorito`, it's available on the standard Fedora repositories.
> 
>     3. Write the image file to a USB drive using your favorite way to do such things, I used Gnome Disks.
> 
>     4. Disable secure boot, otherwise the USB drive will not boot.
> 
>     5. Boot the USB drive and follow the steps, you will now be updated to 1.29 after a reboot.
> 
>     6. Re-enable secure boot if you desire.
> 
> 
> ```
> [jvnknvlgl@thinkpad-t14 ~]$ sudo dmidecode -s bios-version
> R1BET60W(1.29 )
> ```
> 
> The device still doesn't show up if I run `fwupdmgr`, but here's hoping it will show up for the next UEFI update.

Thank you! Worked perfectly!

fkru, 2020-12-29:
Thanks for the information. Incidentally, do you have problems with 1.29 and resume after hibernation, which I am experiencing after the upgrade to version 1.25?

fkru, 2020-12-29:
> Thanks for the information. Incidentally, do you have problems with 1.29 and resume after hibernation, which I am experiencing after the upgrade to version 1.25?

FYI: The hibernation issue was not due to the firmware update. Sorry for the noise.

fkru, 2020-12-30:
Version 1.29 is now available via LVFS, the installation went smoothly.

$ fwupdmgr get-history
20UES00L00
│
└─System Firmware:
  │   Device ID:          3f10cbb75c39627fc0d14eddd274374207a5c753
  │   Previous version:   0.1.29
  │   Update State:       success
  │   Last modified:      2020-12-30 10:20
  │   GUID:               01453b71-da0c-4832-9f4f-e378245339c7
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Cryptographic hash verification is available
  │                       • Device is usable for the duration of the update
  │ 
  └─ThinkPad T14/P14s Gen 1 AMD System Update:
        New version:      0.1.29
        Remote ID:        lvfs
        Zusammenfassung:  Lenovo ThinkPad T14/P14s Gen 1 AMD System Firmware
        License:          Proprietary
        Size:             17,9 MB
        Created:          2020-12-21
        Urgency:          High
        Vendor:           Lenovo Ltd.
        Beschreibung:     
        This stable release fixes the following issues:
        
        • Fixed System will auto wake up when putting system to S3 then close lid.
        • Fixed System auto resume from S3 after attach 45W/65W typec power adapter in S3 mode.
        • Fixed mouse test have no function after shutdown in LENOVO Diagnost.
        • Fixed GUI setup screen and small lenovo logo show overlap when click blank on BIOS setup.
        • Fixed the system can be awakened from S3 by any key under Linux mode.
        
        Some new functionality has also been added:
        
        • Added Enable 802.11ax for Ukraine shipment.
        • Added back flash prevention for Type11 Country code.
        • Added BOE low power panel on T14 AMD platform.
        • Supported PI_DLS mode on MP battery.
        • Added Wake up on Alarm BIOS feature.
        
        Note:If can't update to V1.29, need update V1.25 first.

morphy2k, 2020-12-30:
I currently have version 1.27 installed (done a week ago via the flash drive method) and am not being offered an update to 1.29.

```
❯ fwupdmgr update 
Devices with no available firmware updates: 
 • Integrated Camera
 • SKHynix HFS512GD9TNI-L2B0B
 • System Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI dbx
Devices with the latest available firmware version:
 • Prometheus
 • Prometheus IOTA Config
 ```

fkru, 2020-12-30:
> I currently have version 1.27 installed (done a week ago via the flash drive method) and am not being offered an update to 1.29.

Have you tried 'fwupdmgr --force refresh' followed by 'fwupdmgr update'?

morphy2k, 2020-12-30:
> > I currently have version 1.27 installed (done a week ago via the flash drive method) and am not being offered an update to 1.29.
> 
> Have you tried 'fwupdmgr --force refresh' followed by 'fwupdmgr update'?

Oh `--force` was missing. Everything went smoothly, thank you!

```
❯ fwupdmgr get-history 
20UD0013GE
│
└─System Firmware:
  │   Device ID:          3f10cbb75c39627fc0d14eddd274374207a5c753
  │   Previous version:   0.1.27
  │   Update State:       success
  │   Last modified:      2020-12-30 11:22
  │   GUID:               01453b71-da0c-4832-9f4f-e378245339c7
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Cryptographic hash verification is available
  │                       • Device is usable for the duration of the update
  │ 
  └─ThinkPad T14/P14s Gen 1 AMD System Update:
        New version:      0.1.29
        Remote ID:        lvfs
        Summary:          Lenovo ThinkPad T14/P14s Gen 1 AMD System Firmware
        License:          Proprietary
        Size:             17.9 MB
        Created:          2020-12-21
        Urgency:          High
        Vendor:           Lenovo Ltd.
        Description:      
        This stable release fixes the following issues:
        
        • Fixed System will auto wake up when putting system to S3 then close lid.
        • Fixed System auto resume from S3 after attach 45W/65W typec power adapter in S3 mode.
        • Fixed mouse test have no function after shutdown in LENOVO Diagnost.
        • Fixed GUI setup screen and small lenovo logo show overlap when click blank on BIOS setup.
        • Fixed the system can be awakened from S3 by any key under Linux mode.
        
        Some new functionality has also been added:
        
        • Added Enable 802.11ax for Ukraine shipment.
        • Added back flash prevention for Type11 Country code.
        • Added BOE low power panel on T14 AMD platform.
        • Supported PI_DLS mode on MP battery.
        • Added Wake up on Alarm BIOS feature.
        
        Note:If can't update to V1.29, need update V1.25 first.
      

```

syscll, 2020-12-30:
Running BIOS 1.27 via manual flash and also not seeing the 1.29 update via LVFS.
```
$ fwupdmgr update
Devices with no available firmware updates:
 • Integrated Camera
 • Samsung SSD 970 EVO Plus 1TB
 • System Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI dbx
No updatable devices
```

Have tried with the `--force` flag:
```
$ fwupdmgr refresh --force
Updating lvfs
Downloading…             [***************************************]
Successfully downloaded new metadata: 0 local devices supported

$ fwupdmgr get-updates
Devices with no available firmware updates:
 • Integrated Camera
 • Samsung SSD 970 EVO Plus 1TB
 • System Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI dbx
No updatable devices
```

fkru, 2020-12-30:
It should be 'fwupdmgr --force refresh'.

syscll, 2020-12-30:
Have tried both, but I don't think it matters which order the flag is in- as long as it's present.

Have been using `fwupdmgr refresh --force` for the past 12 months or so.

Asheboy, 2020-12-30:
I'm also running 1.25 and can't see 1.29.

```
❯ fwupdmgr get-history 
20UDCTO1WW
│
├─Prometheus:
│ │   Device ID:          50efef912094e784a03628350441f6927c4920b0
│ │   Previous version:   10.01.3121519
│ │   Update State:       success
│ │   Last modified:      2020-10-18 07:58
│ │   GUID:               09e8bf16-3e69-50f5-bb66-c7a040248352
│ │   Device Flags:       • Updatable
│ │                       • Supported on remote server
│ │                       • Reported to remote server
│ │                       • Cryptographic hash verification is available
│ │ 
│ └─Prometheus Fingerprint Reader Device Update:
│       New version:      10.01.3158804
│       Remote ID:        lvfs
│       Summary:          Firmware for the Synaptics Prometheus Fingerprint Reader device
│       Licence:          Proprietary
│       Size:             400.3 kB
│       Created:          2020-05-22
│       Urgency:          Medium
│       Vendor:           Synaptics Inc.
│       Duration:         2 seconds
│       Description:      
│       New features and enhancements:
│       
│       • Support Linux system
│     
├─WDC PC SN520 SDAPMUW-128G-1001:
│ │   Device ID:          310f45f1f223064b5c16bf6dff31146755a64480
│ │   Previous version:   20260001
│ │   Update State:       success
│ │   Last modified:      2020-10-18 07:59
│ │   GUID:               c0ff0f18-855d-5415-af06-a36723074fc9
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Reported to remote server
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─PCSN520 Device Update:
│       New version:      20370001
│       Remote ID:        lvfs
│       Summary:          WD PCSN520 Firmware Update for Lenovo PC
│       Licence:          Proprietary
│       Size:             2.1 MB
│       Created:          2020-08-31
│       Urgency:          High
│       Vendor:           Western Digital
│       Description:      
│       Do NOT turn off your computer or remove the AC adapter while update is in progress.
│       
│       The computer shall be restarted after updating firmware completely. The device may not properly function until you shut down or reboot PC.
│       
│       Firmware update may fail when HDD password is set in BIOS. In that case, remove HDD password then try update again.
│       
│       Supported Product Scope
│       
│       Lenovo ThinkPad, ThinkCentre , ThinkStation , IdeaCentre
│     
└─System Firmware:
  │   Device ID:          3ba7ea853f9613adbcbb9326dbe5f0decd677a67
  │   Previous version:   0.1.9
  │   Update State:       needs-reboot
  │   Last modified:      2020-12-30 14:53
  │   GUID:               9bfd8b34-080b-40a1-47b3-ca1efa289e6d
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Cryptographic hash verification is available
  │                       • Device is usable for the duration of the update
  │ 
  └─  New version:      0.1.25
        Remote ID:        lvfs
        Licence:          Unknown
```

```
❯ fwupdmgr --force refresh
Updating lvfs
Downloading…             [***************************************]
Successfully downloaded new metadata: 3 local devices supported

~
❯ fwupdmgr get-updates 
Devices with no available firmware updates: 
 • Integrated Camera
 • Samsung SSD 970 EVO 500GB
 • System Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI dbx
Devices with the latest available firmware version:
 • Prometheus
 • Prometheus IOTA Config
 • WDC PC SN520 SDAPMUW-128G-1001
No updates available for remaining devices
```

Asheboy, 2020-12-31:
Update available for me this morning and upgraded from 1.25 to 1.29 successfully. Thank you!

33Fraise33, 2020-12-31:
I can also confirm this is now working, I applied the initial update in
november when it came out and can now update through fwupdmgr like it
should be.

syscll, 2020-12-31:
Also managed to successfully upgrade from 1.27 -> 1.29 this morning.

FYI: for anyone having issues with the firmware not being applied on reboot, I had to disable `Boot Order Lock` in the BIOS for it to work.

kmauleon, 2021-01-04:
hi @jvnknvlgl @fkru  and all thanks for your help and update
hi @aussetg ... with this new update 1.29 in LVFS, can we close this ticket now?  please advise.  thank you very much.

applesun123, 2021-01-04:
> Thank you! Worked perfectly!

as we set the v1.29 to stable on 12/31/2021, it should work available now. could you please use "fwupdmgr force --refresh" command instead of "fwupdmgr" command if you want to upgrade BIOS firmware next update.

[78 more replies]

closed 2022-03-30 by mrhpearson