fwupd/firmware-lenovo #75: Lenovo Basic/Pro/Ultra dock FW is missing

joakim-tjernlund, 2020-09-29:
Adding this to track/inform Lenovo that this is highly missed!

hughsie, 2020-09-29:
FYI: https://github.com/fwupd/fwupd/blob/master/plugins/vli/vli-usbhub-lenovo.quirk#L1 -- just waiting for the Lenovo ODM to finish testing.

joakim-tjernlund, 2020-09-29:
In particular:
```
─USB3.1 Hub:
│     Device ID:          d68b5d74453efa336214e42d39ec96008a3fed64
│     Summary:            USB 3.x Hub
│     Current version:    10.74
│     Vendor:             VIA Labs, Inc. (USB:0x17EF)
│     Install Duration:   15 seconds
│     GUIDs:              e2c96871-8be3-530e-a796-5652cee64eeb
│                         351195ca-bd3e-5a03-911c-0be6686d91dd ← USB\VID_17EF&PID_3070&REV_0A74
│                         2c08e9d9-1f25-5564-b4db-49867dd2c593 ← USB\VID_17EF&PID_3070
│                         816fc838-78d7-5deb-85ef-a95935318198 ← USB\VID_17EF&PID_3070&HUB_0010
│                         7705d2da-41a0-57db-9fb3-c831b76638ad ← USB\VID_17EF&PID_3070&DEV_VL820Q7
│     Device Flags:       • Updatable
│                         • Cryptographic hash verification is available
│                         • Device stages updates
│                         • Device can recover flash failures

─USB2.0 Hub:
│     Device ID:          70835e1192ddb447c964de775916a4cd505ec3b3
│     Summary:            USB 2.x Hub
│     Current version:    10.74
│     Vendor:             VIA Labs, Inc. (USB:0x17EF)
│     Install Duration:   15 seconds
│     GUIDs:              112b3acf-d42a-53cc-91c8-458c3b58f359 ← USB\VID_17EF&PID_3071&REV_0A74
│                         e2c96871-8be3-530e-a796-5652cee64eeb ← USB\VID_17EF&PID_3071
│                         3545c1f3-4d81-59f7-b9d9-d25d5811e4be ← USB\VID_17EF&PID_3071&HUB_10
│                         9af0af2d-99ff-551b-9134-e94809cc8016 ← USB\VID_17EF&PID_3071&DEV_VL820Q7
│     Device Flags:       • Updatable
│                         • Cryptographic hash verification is available
│                         • Device stages updates
│                         • Device can recover flash failures

joakim-tjernlund, 2020-09-29:
> FYI: https://github.com/fwupd/fwupd/blob/master/plugins/vli/vli-usbhub-lenovo.quirk#L1 -- just waiting for the Lenovo ODM to finish testing.

I can see my dock in there :) Thanks for letting me know and congrats to 20 million downloads :)
A truly fantastic achievement!

hughsie, 2020-09-29:
I have all three docks ready to test. :)

joakim-tjernlund, 2020-09-29:
Any idea when that is going to happen? Next week, next month or ... ?

joakim-tjernlund, 2020-09-29:
Cold we a a symbolic name in there too, like Lenovo Dock or similar?
Now it just says USB device which is a bit generic.

mrhpearson, 2020-09-30:
I'm looking into this issue as I'm not sure why it's taking so long to release. As a note it's tracked internally with LO-737 (for when you want to do a "hey Mark whats the update on this one" it's a helpful reference for me)

joakim-tjernlund, 2020-09-30:
> I'm looking into this issue as I'm not sure why it's taking so long to release. As a note it's tracked internally with LO-737 (for when you want to do a "hey Mark whats the update on this one" it's a helpful reference for me)

Thanks, I will drop a reminder in a few days :)

joakim-tjernlund, 2020-10-05:
> I'm looking into this issue as I'm not sure why it's taking so long to release. As a note it's tracked internally with LO-737 (for when you want to do a "hey Mark whats the update on this one" it's a helpful reference for me)

A gentle "hey Mark whats the update on this one" ping for LO-737 :)

mrhpearson, 2020-10-05:
Hi,
Afraid the person I need to talk to for this one is based in China and it's a national holiday there this week (golden week) so I'm waiting for a response. Hopefully will have an update for you next week.
Mark

joakim-tjernlund, 2020-10-09:
Meanwhile, could we get symbolic names in fwupd for the Dock ?

hughsie, 2020-10-09:
> could we get symbolic names in fwupd for the Dock

We certainly could; although equally we can "inherit" the names from the metadata once it's been made public.

joakim-tjernlund, 2020-10-09:
> > could we get symbolic names in fwupd for the Dock
> 
> We certainly could; although equally we can "inherit" the names from the metadata once it's been made public.

You mean that this name will be visible/inherited when FW is released automatically?
No need for a new fwupd release ?

hughsie, 2020-10-09:
> You mean that this name will be visible/inherited when FW is released automatically?

It's not set up that way right now, but it certainly could be. I know the descriptions are not great at the moment when there are layered composite devices.

joakim-tjernlund, 2020-10-20:
Yet a ping for LO-737

mrhpearson, 2020-10-21:
Hi,
I did hear back - they are still working on it, apparently all components except the audio component are completed. They are still working on the audio component firmware as they are having some issues there.
I've asked for an ETA and I will keep an eye on it.
Thanks
Mark

joakim-tjernlund, 2020-10-23:
OK, might that problem be: https://github.com/fwupd/fwupd/commit/b814af80ef7bc3c8842c57ca3e0ed7f104434d17 ?

hughsie, 2020-10-23:
@joakim-tjernlund it might be indeed. Monday is the 1.5.0 release of fwupd, so that should unlock various things.

joakim-tjernlund, 2020-10-26:
.. and now is 1.5 released :) 
Will we see Dock FW in the next few days perhaps ?

hughsie, 2020-10-26:
@joakim-tjernlund you made it into the blog post: https://blogs.gnome.org/hughsie/2020/10/26/new-fwupd-1-5-0-release/

joakim-tjernlund, 2020-10-26:
"You know who you are."  I sure do :)
Keeping an eye om LVFS now

mrhpearson, 2020-10-26:
Awesome :) 

I'll go and send another nudge to the dock team and see if this releases them - it will be really nice to get this one up on LVFS.

I need to add, because I noticed the title of the thread: We're working on the **Thunderbolt3 workstation Gen2** dock firmware. I don't think the ultra dock is covered as we haven't Linux certified that. Just wanted to set expectations. I will confirm though as I'm just guessing

Mark

joakim-tjernlund, 2020-10-26:
I really hoping for Ultra as the a lot of embedded team here is using it and running Linux on them.

hughsie, 2020-10-26:
> I don't think the ultra dock is covered

If it helps, I have all thee kinds of dock here (Ultra, Basic, the other one) and the VLI updates apply fine as VLI were testing this a few months ago. It's what we used to validate the VLI plugin all those moons ago.

joakim-tjernlund, 2020-10-26:
Somewhat off but I noticed there is an Windows release of fwupd-1.5.0 too. Is that capable of
updating BIOS/Dock FW too? 
If so we really need the Ultra FW update too ...

hughsie, 2020-10-26:
>  Is that capable of updating BIOS/Dock FW too

All the USB hub firmware: yes (including PD and audio), the synaptics MST controller and TBT: no.

joakim-tjernlund, 2020-10-28:
@mrhpearson , still no sign of Dock FW in LVFS or an ETA.
Can you provide an ETA for dock FW in general and Ultra Dock FW in particular ?

mrhpearson, 2020-10-29:
It turns out fwupd-1.5 is not the solution we were waiting for (sorry, that's a bit Jedi...)

I'm working on tracking down what we're still missing - there are a fair few confusing threads going on.

Mark

joakim-tjernlund, 2020-10-30:
OK, lets hope the upcoming 1.5.1 will close some of those threads.

hughsie, 2020-11-02:
1.5.1 out, I'll poke ODMs in the morning.

joakim-tjernlund, 2020-11-02:
> 1.5.1 out, I'll poke ODMs in the morning.

Thanks alot!

joakim-tjernlund, 2020-11-06:
@mrhpearson ?

mrhpearson, 2020-11-06:
Progress - but not done yet. There are still some failure conditions that they are working through. It's being actively worked on

joakim-tjernlund, 2020-11-06:
Why not ask @hughsie for help ?

mrhpearson, 2020-11-06:
He's already involved. I was just updating the status here

joakim-tjernlund, 2020-11-16:
I can see a Dock FW on LVFS now :)

joakim-tjernlund, 2020-11-19:
It is raining Dock FW now :)
Will we see  Basic/Pro/Ultra Dock FW now?

joakim-tjernlund, 2020-11-25:
@mrhpearson ?

joakim-tjernlund, 2020-12-07:
@mrhpearson , any news?

mrhpearson, 2020-12-07:
I tested the update manually myself last week on the TBT3 WS Gen2 dock and it works. The Docking team are still doing their testing and I saw some emails about getting it up on LVFS but it's not there yet. I don't have an ETA yet I'm afraid but it is getting there.

FYI - fwupd 1.5.2 or later will be required. It's available on Fedora in updates-testing, or as a snap for other distro's.

mrhpearson, 2020-12-07:
Oh, and I assume you're interested in the Ultra dock scanning the above. My understanding is that a lot of the components are the same so it will work there too...but I don't know for certain what will happen.
I don't have one of those to test with myself. Once the TBT3 dock is done hopefully it will become more obvious...

joakim-tjernlund, 2020-12-17:
1.5.4 is out, any news w.r.t Lenovo Dock FW ?

joakim-tjernlund, 2021-01-12:
1.5.5 is out. @mrhpearson any news ? Particularly about Ultra

mrhpearson, 2021-01-12:
Hmm - I asked on this last week and didn't get a reply. I'll go chase
Mark

joakim-tjernlund, 2021-02-08:
@mrhpearson , please chase harder :)

mrhpearson, 2021-02-10:
Yeah - I'm frustrated too...honestly it's getting ridiculous.
They have completed their testing - my understanding is they are just putting it up on LVFS and I don't have any good reasons why its' not there yet. Sorry :( I believe that team are all off for Chinese New Year so expect at least one more week - I have requested an update for when they get back.

mrhpearson, 2021-02-17:
They found an issue in testing with the DP component not being detected (and hence updated). I'm trying to understand better what is going on (and honestly, how the heck this wasn't caught 4 months ago)

LordChaos73, 2021-02-25:
1.5.7 seems to be released; any good news for the TB3 Gen2 dock owners out there?

joakim-tjernlund, 2021-03-04:
Hi @mrhpearson , been a while now and I am too curious to just wait :)
Any news?

mrhpearson, 2021-03-08:
It's on LVFS...but under embargo still. Will keep pushing until it's moved to enabled - we're getting there.

joakim-tjernlund, 2021-03-08:
Nice:) I guess it is waiting for @hughsie to give it a go before moving to enabled.

mrhpearson, 2021-03-08:
Nope - they test it (again) and then there are some folk who can move it to enabled status - at least that's my understanding.

Richard has been great (as always!) helping behind the scenes but he's off the hook for anything right now :)

sclel016, 2021-03-10:
Really appreciate all the effort that's being put into the issue, hopefully it's getting close to the finish line!

joakim-tjernlund, 2021-03-15:
@mrhpearson , I see a Dock FW on LVFS now but for Thunderbolt 3 dock.
Is there a Ultra Dock FW on the way as well?

mrhpearson, 2021-03-15:
Hi,

As noteed - the docking firmware is up under testing state. I just tried it on my dock (P1G2 /w Fedora) and it updated cleanly and succesfully. There is a known issue around the DisplayPort component where it has to be connected and used for it to be updated, if you're using it then we believe it will be fine, and if you're not using it....then you likely don't care (and if you start using it updates should work after) - I'm still messing around with it myself to understand the permutations but we made the call not to delay getting the firmware up based on that.

I think the FW will be moved to available soon (I sent my results into the team to help speed that along). If anybody does enable testing and tries it and either has issues or is succesful let me know. I think we're all slightly nervous about this one :)

For Ultra dock - it's not explicitely covered but my understanding is where the components are the same then you'll get the update. So it should work....I don't have one myself so I'd be interested to hear how it goes (I'll see if we can get one in our team too). This does come with the warning that it's not in the docking teams remit to deliver - we had to get a dock done first and (as we've all seen) that has been a challenge in itself. Once the TBT2G3 dock is out and stable I'll go and test the waters for extending the program.

Mark

joakim-tjernlund, 2021-03-15:
> For Ultra dock - it's not explicitely covered but my understanding is where the components are the same then you'll get the update. So it should work....I don't have one myself so I'd be interested [...]

FYI, what is in LVFS testing does not match my Dock. Details about my Dock is in the beginning of this issue.

joakim-tjernlund, 2021-03-16:
I think @hughsie has this Dock and tested it LONG time ago. A bit strange that there isn't a Dock FW on LVFS by now.

hughsie, 2021-03-16:
Yes, I have the ultra/pro/etc docs too -- all are mostly the same components, although with different VID/PIDs. I did all the VIA validation testing on them all those moons ago.

joakim-tjernlund, 2021-03-16:
Try requesting Ultra Dock FW explicitly and see what happens.

LordChaos73, 2021-03-17:
Has anyone tested the firmware on their TB3G2 dock by any chance?

[5 more replies]

closed 2023-02-15 by hughsie