fwupd/firmware-lenovo #76: T480s Embedded Controller Firmware 1.21 still missing

der-eismann, 2020-10-01:
Hey everyone,

On 12th July Lenovo published the package with [BIOS 1.42 & EC FW 1.21](https://pcsupport.lenovo.com/de/de/products/laptops-and-netbooks/thinkpad-t-series-laptops/thinkpad-t480s-type-20l7-20l8/downloads/DS502226) on their support website. While the 1.42 BIOS updates are available via LVFS ([with](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN22ETXXW.firmware) & [without](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN22ETXXP.firmware) Computrace) the package for the [EC firmware](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN22HT.firmware) is still on version 1.20. 
This is especially frustrating since the BIOS explicitly states: 
> Support Lid Sensor Enable/Disable in ThinkPad Setup - Config - Power.
> (Note) ECP version 1.21 or later is required for this support.

It was possible for me to download the *.zip on their support page and upgrade my EC FW with the *.cab file, but I would prefer the LVFS in the future.

mrhpearson, 2020-10-05:
Thanks for the report.
LO-747 created to track internally. 
@kmauleon - adding you here for awareness.
Mark

der-eismann, 2021-08-04:
Closing because it's been available for a while.

closed 2021-08-04 by der-eismann