fwupd/firmware-lenovo #77: UEFI Firmware for T490s is missing / EC & ME FW outdated

der-eismann, 2020-10-07:
Hey everyone,

It's me again. While showing my colleagues the capabilities of fwupd we noticed that the BIOS files for their T490s are completely missing on the LVFS website, others are outdated.

**UEFI Update**

Here's the `fwupd get-devices` output for the UEFI part:
```
├─System Firmware:
│     Device ID:           ca8e899e46f1da1776d16410bc92f2cf0d963433
│     Current version:     0.1.55
│     Vendor:              LENOVO (DMI:LENOVO)
│     GUID:                6300ba52-d9ea-4ca0-88f9-f89ef73cc07b
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
```

As mentioned before this is completely missing on the LVFS website at the moment, but the current firmware 1.70 is available as a *.cab file via the [T490s download page](https://pcsupport.lenovo.com/de/en/products/laptops-and-netbooks/thinkpad-t-series-laptops/thinkpad-t490s-type-20nx-20ny/downloads/DS539062).
What's interesting is that the output of `fwupd get-updates` says `• System Firmware has the latest available firmware version` instead of `has no available firmware updates`, so maybe it's hidden on the LVFS?

**Embedded Controller Firmware**

This one is just outdated. On the [T490s download page](https://pcsupport.lenovo.com/de/en/products/laptops-and-netbooks/thinkpad-t-series-laptops/thinkpad-t490s-type-20nx-20ny/downloads/DS539062) there's version 1.20 available, on the LVFS it's [still 1.19](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN2JHT.firmware). I'm also not sure what the `ThinkPad T490s/Fx390` in the title should mean. Is there a ThinkPad Fx390? The X390 has a different EC FW.

**ME Firmware**

Severely outdated! In this case there is version 12.0.68.1606 available on the [download page](https://pcsupport.lenovo.com/de/en/products/laptops-and-netbooks/thinkpad-t-series-laptops/thinkpad-t490s-type-20nx-20ny/downloads/DS539850) while the LVFS is still on 12.0.47.1524 ([consumer](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN2JRN.firmware) & [corporate](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN2JRM.firmware)). The LVFS version was released in October 2019 and there were four new versions published since then.

kmauleon, 2020-10-07:
hi @der-eismann thank you very much for the information... calling the attention of the owner....

kmauleon, 2020-10-22:
target to release new versions by november...please wait...thank you very much

der-eismann, 2020-10-22:
Thanks for the update, I'll keep an eye on it!

joakim-tjernlund, 2020-11-16:
Great, I am also looking forward to T490s updates

kmauleon, 2021-01-04:
hi latest MEFW already available please check >> https://fwupd.org/lvfs/search?value=T490s
As for BIOS/EC, will follow-up owner.  thank you very much....

der-eismann, 2021-03-09:
It's getting better, on the LVFS we have BIOS 1.70 while on the Lenovo page BIOS 1.71 is offered since 02 Feb 2021. EC and ME FW are the latest version. I don't really understand the delay or release cycles of this, but it's improving.

kmauleon, 2021-03-10:
hi all... latest BIOS uploaded here >> https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN2JETXXP.firmware
closing this ticket... thank you very much.

closed 2021-03-10 by kmauleon