fwupd/firmware-lenovo #78: UEFI firmware missing for P1 Gen3

bkchr, 2020-10-12:
According to the [site](https://pcsupport.lenovo.com/de/en/products/laptops-and-netbooks/thinkpad-p-series-laptops/thinkpad-p1-gen-3-type-20th-20tj/downloads/driver-list/component?name=BIOS%2FUEFI) there is a newer UEFI version, however `fwupdmgr update` says that everything is up-to-date.

`get-devices` output:
```
└─UEFI dbx:
      Device ID:          362301da643102b9f38477387e2193e57abaa590
      Summary:            UEFI Revocation Database
      Current version:    83
      Minimum Version:    83
      Vendor:             UEFI:Linux Foundation
      Install Duration:   1 second
      GUIDs:              14503b3d-73ce-5d06-8137-77c68972a341 ← UEFI\CRT_A9087D1044AD18F7A94916D284CBC01827CF23CD8F60B79072C9CAA1FEF4D649
                          5971a208-da00-5fce-b5f5-1234342f9cf7 ← UEFI\CRT_A9087D1044AD18F7A94916D284CBC01827CF23CD8F60B79072C9CAA1FEF4D649&ARCH_X64
                          c6682ade-b5ec-57c4-b687-676351208742 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503
                          f8ba2887-9411-5c36-9cee-88995bb39731 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503&ARCH_X64
```

hughsie, 2020-10-12:
Is that literally the output of `fwupdmgr get-devices`? You've got a lot of devices missing...

bkchr, 2020-10-12:
Sorry, I thought that only UEFI would be relevant. Full output:

```
20THCTO1WW
│
├─Thunderbolt Controller:
│     Device ID:          864bdf85b8041be89abd1891dc3c11c45827df11
│     Summary:            Unmatched performance for high-speed I/O
│     Current version:    59.00
│     Vendor:             Lenovo (TBT:0x0109)
│     GUIDs:              e5126b67-ecd3-52e5-b420-549c7283d56e
│                         531e674f-ffb5-5496-be61-b7d97bf97d6a
│                         825f6432-e785-54ad-b655-60e79e55b24a
│                         56460c46-4e6f-50eb-abe9-abf43677702e
│     Device Flags:       • Internal device
│                         • Updatable
│                         • Requires AC power
│                         • Device stages updates
│   
├─Integrated Camera:
│     Device ID:          9037d31b0ec04c23441d60a55c29f07b6869eb59
│     Current version:    60.1
│     Vendor:             Azurewave (USB:0x13D3)
│     GUIDs:              4dc71255-84a9-5adc-8eab-c5982db48621
│                         9284c551-0b4c-51ee-905a-168b8787290c
│     Device Flags:       • Updatable
│   
├─SAMSUNG MZVLB1T0HBLR-000L7:
│     Device ID:          08e1798bf5d9cb56a0290b552cab6c1a371b5089
│     Summary:            NVM Express Solid State Drive
│     Current version:    4M2QEXF7
│     Vendor:             Samsung Electronics Co Ltd (NVME:0x144D)
│     GUIDs:              6e54c992-d302-59ab-b454-2d26ddd63e6d
│                         47335265-a509-51f7-841e-1c94911af66b
│                         b77af583-f45c-537c-8584-fc5d1c80fa20
│     Device Flags:       • Internal device
│                         • Updatable
│                         • Requires AC power
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
└─UEFI dbx:
      Device ID:          362301da643102b9f38477387e2193e57abaa590
      Summary:            UEFI Revocation Database
      Current version:    83
      Minimum Version:    83
      Vendor:             UEFI:Linux Foundation
      Install Duration:   1 second
      GUIDs:              14503b3d-73ce-5d06-8137-77c68972a341
                          5971a208-da00-5fce-b5f5-1234342f9cf7
                          c6682ade-b5ec-57c4-b687-676351208742
                          f8ba2887-9411-5c36-9cee-88995bb39731
      Device Flags:       • Internal device
                          • Updatable
                          • Needs a reboot after installation
```

mrhpearson, 2020-10-13:
Curious - yours isn't showing the system firmware or EC firmware. Which version of fwupdmgr are you running? They are showing up on mine with version 1.4.6

That being said - I don't think the firmware is available on LVFS yet. It often takes them a little while to get them but I'll go and ask. As an aside - I believe these updates are useful, I was having some suspend issues without them

Mark

bkchr, 2020-10-13:
Hmm, I also got 1.4.6. Not sure why they are not showing. So, there should be a new entry in this list?

Yeah, I updated the BIOS now manually with `fwupdate`. I didn't had problems with suspend, but I had thermal issues and my laptop was shutting down already multiple times.

hughsie, 2020-10-13:
> yours isn't showing the system firmware or EC firmware

Are capsule updates switched off in the BIOS perhaps?

mrhpearson, 2020-10-15:
As a note - the firmware for the P1G3 itself isn't up on LVFS yet - it's under testing still. I don't have an ETA yet.
Mark

kmauleon, 2021-02-01:
Latest System, EC, and ME firmware are already uploaded to [LVFS](https://fwupd.org/lvfs/search?value=P1+Gen+3).  please check.... thank you very much

closed 2021-02-01 by kmauleon