fwupd/firmware-lenovo #79: Thinkpad P52 updates no longer listed

rugubara, 2020-10-13:
```
PF16W6Y2 ~ # fwupdmgr get-devices
20M90019RT
│
└─Samsung SSD 970 EVO 2TB:
      Device ID:          03281da317dccd2b18de2bd1cc70a782df40ed7e
      Summary:            NVM Express Solid State Drive
      Current version:    1B2QEXE7
      Vendor:             Samsung Electronics Co Ltd (NVME:0x144D)
      Serial Number:      S464NB0K704432V
      GUIDs:              6e54c992-d302-59ab-b454-2d26ddd63e6d ← NVME\VEN_144D&DEV_A808&REV_00
                          47335265-a509-51f7-841e-1c94911af66b ← NVME\VEN_144D&DEV_A808
                          4494b540-d91c-5ae9-9963-3b0f03e7fe2b ← Samsung SSD 970 EVO 2TB
      Device Flags:       • Internal device
                          • Updatable
                          • Requires AC power
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
```
I can see only my aftermarket EVO 970. The main system UEFI and intel ME no longer listed. I used to update them ok before:

```
PF16W6Y2 ~ # fwupdmgr get-history
20M90019RT
│
├─UEFI Device Firmware:
│ │   Device ID:          0f3c64df11304fc5711ef77a93e3e042e2738907
│ │   Previous version:   0.1.15
│ │   Update State:       success
│ │   Last modified:      2019-12-21 13:21
│ │   GUID:               88b8ba7e-296b-4f9e-929f-dcc19318dbcf
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • Requires AC power
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Reported to remote server
│ │                       • Device is usable for the duration of the update
│ │
│ └─  New version:      0.1.16
│       Remote ID:        lvfs
│       License:          Unknown
│
├─SAMSUNG MZVLB512HAJQ-000L7:
│ │   Device ID:          c6a0cfba7c7d81e253fce571e1d1e9f6003ae1c7
│ │   Previous version:   4L2QEXA7
│ │   Update State:       success
│ │   Last modified:      2020-03-14 19:30
│ │   GUID:               6e54c992-d302-59ab-b454-2d26ddd63e6d
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • Requires AC power
│ │                       • Needs a reboot after installation
│ │                       • Reported to remote server
│ │                       • Device is usable for the duration of the update
│ │
│ └─  New version:      5L2QEXA7
│       License:          Unknown
│
├─Intel Management Engine:
│ │   Device ID:          619057f1d0da67b3fa4e9f74e09fc9218ce32643
│ │   Previous version:   192.47.1524
│ │   Update State:       needs-reboot
│ │   Last modified:      2020-08-30 16:35
│ │   GUID:               971682af-21a0-46e3-91c0-825702c2ba70
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • Requires AC power
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Device is usable for the duration of the update
│ │
│ └─  New version:      192.64.1551
│       Remote ID:        lvfs
│       License:          Unknown
│
└─System Firmware:
  │   Device ID:          65b6a9dc7b7df18bdff003584b51bf21373e3aa6
  │   Previous version:   0.1.33
  │   Update State:       needs-reboot
  │   Last modified:      2020-08-30 16:35
  │   GUID:               1e1fe415-74e8-49e1-9508-106b3d13d50d
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • Requires AC power
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Cryptographic hash verification is available
  │                       • Device is usable for the duration of the update
  │
  └─  New version:      0.1.37
        Remote ID:        lvfs
        License:          Unknown
```
What did I do wrong?

kmauleon, 2020-10-22:
hi @rugubara have you tried the following commands....thanks
    fwupdmgr refresh
    fwupdmgr get-devices

rugubara, 2020-10-25:
Just tried.
```
PF16W6Y2 ~ # fwupdmgr refresh
Updating lvfs
Downloading…             [***************************************]
Downloading…             [***************************************]
Successfully downloaded new metadata: 0 local devices supported
PF16W6Y2 ~ # fwupdmgr get-devices
20M90019RT
│
└─Samsung SSD 970 EVO 2TB:
      Device ID:          03281da317dccd2b18de2bd1cc70a782df40ed7e
      Summary:            NVM Express Solid State Drive
      Current version:    1B2QEXE7
      Vendor:             Samsung Electronics Co Ltd (NVME:0x144D)
      Serial Number:      S464NB0K704432V
      GUIDs:              6e54c992-d302-59ab-b454-2d26ddd63e6d ← NVME\VEN_144D&DEV_A808&REV_00
                          47335265-a509-51f7-841e-1c94911af66b ← NVME\VEN_144D&DEV_A808
                          4494b540-d91c-5ae9-9963-3b0f03e7fe2b ← Samsung SSD 970 EVO 2TB
      Device Flags:       • Internal device
                          • Updatable
                          • Requires AC power
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update

PF16W6Y2 #
```

Still no system firmware supported :-(

kmauleon, 2020-10-26:
forwarded your concern to owner... please wait... thank you very much

kmauleon, 2021-01-04:
hi @rugubara can you please try this...
**fwupdmgr --force refresh
fwupdmgr update**