fwupd/firmware-lenovo #81: Lenovo X1 Carbon 6th Gen fails to update system firmware, embedded controller

dead10ck, 2020-11-05:
**Describe the bug**
I have these two updates to perform on a Lenovo X1 Carbon 6th Gen:

```
20KGS3Y900
│
├─Embedded Controller:
│ │   Device ID:           9698faabddf0d7b18925cfbbda95f8b0d0dacc53
│ │   Current version:     0.1.21
│ │   Minimum Version:     0.1.21
│ │   Vendor:              DMI:LENOVO
│ │   Update State:        success
│ │   GUID:                3babca5f-b2bf-4f4b-a72e-2bdc84eb4019
│ │   Device Flags:        • Internal device
│ │                        • Updatable
│ │                        • Requires AC power
│ │                        • Supported on remote server
│ │                        • Needs a reboot after installation
│ │                        • Device is usable for the duration of the update
│ │ 
│ └─ThinkPad X1 Carbon 6th Embedded Controller Update:
│       New version:       0.1.22
│       Remote ID:         lvfs
│       Summary:           Lenovo ThinkPad X1 Carbon 6th Embedded Controller Firmware
│       License:           Proprietary
│       Size:              767.1 kB
│       Created:           2016-07-08
│       Urgency:           High
│       Vendor:            Lenovo Ltd.
│       Flags:             is-upgrade
│       Description:       Lenovo ThinkPad X1 Carbon 6th Embedded Controller Firmware
│       
│       Fixed an issue where ThinkVision T24m-10 monitor might not connected properly.
│     
└─System Firmware:
  │   Device ID:           1c53551e7da69d896138fac1ae131c83ad46d923
  │   Current version:     0.1.49
  │   Minimum Version:     0.0.1
  │   Vendor:              LENOVO (DMI:LENOVO)
  │   Update State:        success
  │   GUID:                a4b51dca-8f97-4310-8821-3330f83c9135
  │   Device Flags:        • Internal device
  │                        • Updatable
  │                        • Requires AC power
  │                        • Supported on remote server
  │                        • Needs a reboot after installation
  │                        • Cryptographic hash verification is available
  │                        • Device is usable for the duration of the update
  │ 
  └─ThinkPad X1 Carbon 6th System Update:
        New version:       0.1.50
        Remote ID:         lvfs
        Summary:           Lenovo ThinkPad X1 Carbon 6th System Firmware
        License:           Proprietary
        Size:              9.5 MB
        Created:           2016-07-08
        Urgency:           High
        Vendor:            Lenovo Ltd.
        Flags:             is-upgrade
        Description:       Lenovo ThinkPad X1 Carbon 6th System Firmware
        
         • Update includes a security fix.
         • Updated the CPU microcode.
         • Updated the Diagnostics module to version 04.15.000.
        
        Security issues fixed:
        
         • CVE-2020-0593
         • CVE-2020-0592
         • CVE-2020-0591
[... 296 lines cut ...]
├── initramfs-0-rescue-43e843afee574658881078e37cbd57c4.img
├── initramfs-5.8.15-201.fc32.x86_64.img
├── initramfs-5.8.16-200.fc32.x86_64.img
├── initramfs-5.8.17-200.fc32.x86_64.img
├── loader
│   └── entries [error opening dir]
├── lost+found [error opening dir]
├── memtest86+-5.01
├── System.map-5.8.15-201.fc32.x86_64
├── System.map-5.8.16-200.fc32.x86_64
├── System.map-5.8.17-200.fc32.x86_64
├── vmlinuz-0-rescue-43e843afee574658881078e37cbd57c4
├── vmlinuz-5.8.15-201.fc32.x86_64
├── vmlinuz-5.8.16-200.fc32.x86_64
└── vmlinuz-5.8.17-200.fc32.x86_64

6 directories, 77 files
```

**Additional questions**
- Operating system and version: Fedora 32
- Have you tried rebooting? Yes
- Is this a regression? Yes: https://github.com/fwupd/fwupd/issues/1538
- Are you using an NVMe disk? yes
- Is secure boot enabled? no

lijunwang1990, 2020-11-06:
Hi @dead10ck could you please use this commond have a try?
$ sudo fwupdmgr downgrade
This should give you a list of your recent updates. Just downgrade the looping updates. Don't worry though. Once you've downgraded, just go ahead and run
$ sudo fwupdmgr update
some users can fix similar issue by this commond. thank you.

dead10ck, 2020-11-06:
May I ask why and how you think downgrading my firmware will fix the issue? Isn't this a fwupd bug?

dead10ck, 2020-11-10:
It looks like after upgrading fwupd to 1.5.1, the same two firmware updates succeeded.

closed 2020-11-10 by dead10ck